Hamlib/Hamlib #505: Yaesu retries not working

mdblack98, 2021-01-11:
Mikael -- need to narrow down the question_mark_response_means_rejected flag usage

If it's only needed by mode for certain rigs than it should be coded by mode for certain rigs.
The comment doesn't say what rigs this applies to.
Right now it's disallowing IF; command repeats for all Yaesu rigs and appears as though the flag doesn't get reset in all paths.
The rig_get_mode

https://github.com/Hamlib/Hamlib/commit/956be046e748066577a34508cbfe68718954edcf

On Monday, January 11, 2021, 04:38:33 PM CST, Bill Somerville <g4wjs@classdesign.com> wrote:

Rather more concerning is that while testing with the master HEAD commit I see that retries when a Yaesu rig returns a rig busy status '?;' are no longer being done. I see several examples, why is '?;' now being treated as a fatal command rejection error ratther than a rig busy condition that can be retried?

warning: [22:08:25.820593][RIGCTRL][debug] () : rig_get_mode called
warning: [22:08:25.820593][RIGCTRL][trace] () : elapsed_ms: start = 0,0
warning: [22:08:25.822594][RIGCTRL][trace] () : rig_get_mode: cache check age=1000000ms
warning: [22:08:25.822594][RIGCTRL][trace] () : rig_get_mode: cache miss age=1000000ms
warning: [22:08:25.823594][RIGCTRL][debug] () : newcat_get_mode called
warning: [22:08:25.823594][RIGCTRL][debug] () : newcat_valid_command called
warning: [22:08:25.824595][RIGCTRL][trace] () : newcat_valid_command MD
warning: [22:08:25.824595][RIGCTRL][trace] () : newcat_set_vfo_from_alias: alias vfo = currVFO
warning: [22:08:25.825597][RIGCTRL][trace] () : newcat_get_mode: cmd_str = MD0;
warning: [22:08:25.826598][RIGCTRL][trace] () : rig_flush: called for serial device
warning: [22:08:25.826598][RIGCTRL][debug] () : serial_flush called
warning: [22:08:25.827598][RIGCTRL][debug] () : tcflush
warning: [22:08:25.988704][RIGCTRL][trace] () : cmd_str = MD0;
warning: [22:08:25.988704][RIGCTRL][debug] () : write_block called
warning: [22:08:25.995710][RIGCTRL][trace] () : write_block(): TX 4 bytes
warning: [22:08:25.995710][RIGCTRL][trace] () : 0000    4d 44 30 3b                                         MD0;
warning: [22:08:25.996711][RIGCTRL][trace] () : read_string called, rxmax=129
warning: [22:08:26.040742][RIGCTRL][trace] () : read_string(): RX 5 characters
warning: [22:08:26.040742][RIGCTRL][trace] () : 0000    4d 44 30 32 3b                                      MD02;
warning: [22:08:26.041741][RIGCTRL][trace] () : newcat_get_cmd: read count = 5, ret_data = MD02;
warning: [22:08:26.041741][RIGCTRL][debug] () : newcat_rmode_width called
warning: [22:08:26.041741][RIGCTRL][debug] () : rig_passband_normal called
warning: [22:08:26.042744][RIGCTRL][debug] () : rig_passband_normal: return filter#6, width=2400
warning: [22:08:26.042744][RIGCTRL][trace] () : newcat_get_mode: returning newcat_get_rx_bandwidth
warning: [22:08:26.043744][RIGCTRL][debug] () : newcat_get_rx_bandwidth called, vfo=VFOA, mode=USB
warning: [22:08:26.043744][RIGCTRL][debug] () : newcat_valid_command called
warning: [22:08:26.043744][RIGCTRL][trace] () : newcat_valid_command SH
warning: [22:08:26.043744][RIGCTRL][trace] () : newcat_set_vfo_from_alias: alias vfo = VFOA
warning: [22:08:26.044742][RIGCTRL][trace] () : newcat_get_cmd: cache invalidated
warning: [22:08:26.044742][RIGCTRL][trace] () : rig_flush: called for serial device
warning: [22:08:26.044742][RIGCTRL][debug] () : serial_flush called
warning: [22:08:26.044742][RIGCTRL][debug] () : tcflush
warning: [22:08:26.045743][RIGCTRL][trace] () : cmd_str = SH0;
warning: [22:08:26.045743][RIGCTRL][debug] () : write_block called
warning: [22:08:26.051747][RIGCTRL][trace] () : write_block(): TX 4 bytes
warning: [22:08:26.051747][RIGCTRL][trace] () : 0000    53 48 30 3b                                         SH0;
warning: [22:08:26.051747][RIGCTRL][trace] () : read_string called, rxmax=129
warning: [22:08:26.096778][RIGCTRL][trace] () : read_string(): RX 6 characters
warning: [22:08:26.096778][RIGCTRL][trace] () : 0000    53 48 30 33 30 3b                                   SH030;
warning: [22:08:26.096778][RIGCTRL][trace] () : newcat_get_cmd: read count = 6, ret_data = SH030;
warning: [22:08:26.097778][RIGCTRL][trace] () : newcat_get_rx_bandwidth: w=30
warning: [22:08:26.097778][RIGCTRL][trace] () : rig_flush: called for serial device
warning: [22:08:26.097778][RIGCTRL][debug] () : serial_flush called
warning: [22:08:26.097778][RIGCTRL][debug] () : tcflush
warning: [22:08:26.097778][RIGCTRL][trace] () : cmd_str = NA0;
warning: [22:08:26.098779][RIGCTRL][debug] () : write_block called
warning: [22:08:26.105784][RIGCTRL][trace] () : write_block(): TX 4 bytes
warning: [22:08:26.106789][RIGCTRL][trace] () : 0000    4e 41 30 3b                                         NA0;
warning: [22:08:26.107786][RIGCTRL][trace] () : read_string called, rxmax=129
warning: [22:08:26.173832][RIGCTRL][trace] () : read_string(): RX 2 characters
warning: [22:08:26.173832][RIGCTRL][trace] () : 0000    3f 3b                                               ?;
warning: [22:08:26.174831][RIGCTRL][trace] () : newcat_get_cmd: read count = 2, ret_data = ?;
warning: [22:08:26.174831][RIGCTRL][error] () : newcat_get_cmd: Command rejected by the rig: 'NA0;'
warning: [22:08:26.175833][RIGCTRL][error] () : get_narrow: unable to parse width from '?;'
warning: [22:08:26.175833][RIGCTRL][trace] () : rig_get_mode: retcode after get_mode=-8
warning: [22:08:26.175833][RIGCTRL][trace] () : elapsed_ms: start = 0,0
warning: [22:08:26.176833][RIGCTRL][trace] () : elapsed_ms: after gettime, start = 1610402906,176833200
[... 171 lines cut ...]
warning: [22:08:33.729911][RIGCTRL][trace] () : newcat_valid_command IF
warning: [22:08:33.730912][RIGCTRL][trace] () : rig_flush: called for serial device
warning: [22:08:33.730912][RIGCTRL][debug] () : serial_flush called
warning: [22:08:33.730912][RIGCTRL][debug] () : tcflush
warning: [22:08:33.730912][RIGCTRL][trace] () : cmd_str = IF;
warning: [22:08:33.730912][RIGCTRL][debug] () : write_block called
warning: [22:08:33.737917][RIGCTRL][trace] () : write_block(): TX 3 bytes
warning: [22:08:33.737917][RIGCTRL][trace] () : 0000    49 46 3b                                            IF;
warning: [22:08:33.737917][RIGCTRL][trace] () : read_string called, rxmax=129
warning: [22:08:33.792954][RIGCTRL][trace] () : read_string(): RX 2 characters
warning: [22:08:33.792954][RIGCTRL][trace] () : 0000    3f 3b                                               ?;
warning: [22:08:33.793956][RIGCTRL][trace] () : newcat_get_cmd: read count = 2, ret_data = ?;
warning: [22:08:33.793956][RIGCTRL][error] () : newcat_get_cmd: Command rejected by the rig: 'IF;'
warning: [22:08:33.794956][RIGCTRL][trace] () : elapsed_ms: start = 0,0
warning: [22:08:33.794956][RIGCTRL][trace] () : elapsed_ms: after gettime, start = 1610402913,794956900
warning: [22:08:33.794956][RIGCTRL][error] () : newcat_get_vfo_mode: incorrect length of IF response, expected 27 or 28, got 2
warning: [22:08:33.795957][RIGCTRL][trace] () : elapsed_ms: start = 1610402912,789281100
warning: [22:08:33.795957][RIGCTRL][trace] () : elapsed_ms: elapsed_msecs=10000
warning: [22:08:33.795957][RIGCTRL][error] () : rig_get_vfo: returning -8(Protocol error)
More complete log attached.

73
Bill
G4WJS.
On 11/01/2021 10:42, Bill Somerville wrote:

mikaelnousiainen, 2021-01-12:
@mdblack98 @g4wjs Ok. Let's take a step back and look at what's happening here.

According to the debug log above, it's the `newcat_get_cmd` that is failing to perform the retries. Assuming the code running there is the latest master (before your commenting of `question_mark_response_means_rejected` lines), `newcat_get_cmd` already had the `return RIG_ERJCTED` commented out and should do the retries, so there is something else going on. See: https://github.com/Hamlib/Hamlib/blame/614f08b6720e3d8d44821bf47405866529f94920/rigs/yaesu/newcat.c#L9165

That said, `newcat_get_cmd` should have the condition ` if (question_mark_response_means_rejected)` for the rejected response. I looked at the code paths for that variable and it should get reset. So it's not about that variable being set incorrectly at all now.

What is the number of retries here and which rig is this?

Maybe Bill was running a version of Hamlib where the `return RIG_ERJCTED` wasn't yet commented out?

I can create a PR which should have the correct logic here. The commands affected by the `?` response being invalid (and rig not being busy) are very specific cases only and do not affect the "IF" command for example.

mikaelnousiainen, 2021-01-12:
@mdblack98 @g4wjs Right, I noticed what the issue is. `newcat_get_cmd` did not set `rc = -RIG_BUSBUSY;` in case of the `?` response, so the loop exits immediately. I'll push out a PR in a moment for you to test.

mikaelnousiainen, 2021-01-12:
@mdblack98 @g4wjs Please test fix in #506

mdblack98, 2021-01-12:
Thinking about this it really is an exception to the valid_commands.
So perhaps when we switch modes for those specific rigs we turn those valid_commands "FALSE" if RTTY and TRUE if not RTTY.But it has to rig and mode specific.
Then it will get an "invalid command" response which is appropriate.
Mike W9MDB

g4wjs, 2021-01-12:
Hi Mikael,

I tried your PR merged with the master at 935b520. I am running with an 
FT-2000 connected by a remote serial port over TCP/IP, this should work 
but seems to invoke many busy responses from the rig. If the retries 
were working reliably the communications would be fine but there are 
still some issues. Firstly even though the retry parameter is 3 (which 
is too low IMHO) it is only doing 2 retries at most. Here are some 
exchanges that are problematic:

Although this one works after a retry, I'm not sure it should be 
invoking a timed out read_string call as the initial FA; command gives 
an immediate rig busy response. The FA; command could probably be 
retried without waiting, particularly if there were a higher number of 
retries. Anyway this is minor as the result is eventually correct.

warning: [20:34:36.710545][RIGCTRL][trace] () : rig_get_cache_timeout_ms: called selection=0
warning: [20:34:36.710545][RIGCTRL][trace] () : rig_set_cache_timeout_ms: called selection=0, ms=0
warning: [20:34:36.710545][RIGCTRL][debug] () : rig_get_freq called vfo=currVFO
warning: [20:34:36.710545][RIGCTRL][trace] () : vfo_fixup: vfo=currVFO
warning: [20:34:36.710545][RIGCTRL][trace] () : vfo_fixup: Leaving currVFO alone
warning: [20:34:36.711546][RIGCTRL][trace] () : get_cache_freq: vfo=currVFO, freq=0
warning: [20:34:36.711546][RIGCTRL][trace] () : elapsed_ms: start = 0,0
warning: [20:34:36.711546][RIGCTRL][trace] () : rig_get_freq: cache check age=1000000ms
warning: [20:34:36.821620][RIGCTRL][trace] () : rig_get_freq: cache miss age=1000000ms, cached_vfo=None, asked_vfo=currVFO
warning: [20:34:36.821620][RIGCTRL][debug] () : newcat_get_freq called
warning: [20:34:36.821620][RIGCTRL][trace] () : newcat_get_freq: passed vfo = currVFO
warning: [20:34:36.822621][RIGCTRL][debug] () : newcat_valid_command called
warning: [20:34:36.822621][RIGCTRL][trace] () : newcat_valid_command FA
warning: [20:34:36.822621][RIGCTRL][debug] () : newcat_valid_command called
warning: [20:34:36.822621][RIGCTRL][trace] () : newcat_valid_command FB
warning: [20:34:36.823619][RIGCTRL][trace] () : newcat_set_vfo_from_alias: alias vfo = currVFO
warning: [20:34:36.823619][RIGCTRL][debug] () : newcat_valid_command called
warning: [20:34:36.823619][RIGCTRL][trace] () : newcat_valid_command FA
warning: [20:34:36.823619][RIGCTRL][trace] () : cmd_str = FA;
warning: [20:34:36.824620][RIGCTRL][trace] () : rig_flush: called for serial device
warning: [20:34:36.824620][RIGCTRL][debug] () : serial_flush called
warning: [20:34:36.824620][RIGCTRL][debug] () : tcflush
warning: [20:34:36.824620][RIGCTRL][trace] () : cmd_str = FA;
warning: [20:34:36.824620][RIGCTRL][debug] () : write_block called
warning: [20:34:36.831625][RIGCTRL][trace] () : write_block(): TX 3 bytes
warning: [20:34:36.831625][RIGCTRL][trace] () : 0000    46 41 3b                                            FA;
warning: [20:34:36.831625][RIGCTRL][trace] () : read_string called, rxmax=129
warning: [20:34:36.886662][RIGCTRL][trace] () : read_string(): RX 2 characters
warning: [20:34:36.886662][RIGCTRL][trace] () : 0000    3f 3b                                               ?;
warning: [20:34:36.886662][RIGCTRL][trace] () : newcat_get_cmd: read count = 2, ret_data = ?;
warning: [20:34:36.887663][RIGCTRL][warning] () : newcat_get_cmd: Rig busy - retrying: 'FA;'
warning: [20:34:36.887663][RIGCTRL][trace] () : read_string called, rxmax=129
warning: [20:34:39.089144][RIGCTRL][warning] () : read_string(): Timed out 2.201481 seconds after 0 chars
warning: [20:34:39.089144][RIGCTRL][trace] () : rig_flush: called for serial device
warning: [20:34:39.089144][RIGCTRL][debug] () : serial_flush called
warning: [20:34:39.089144][RIGCTRL][debug] () : tcflush
warning: [20:34:39.090144][RIGCTRL][trace] () : cmd_str = FA;
warning: [20:34:39.090144][RIGCTRL][debug] () : write_block called
warning: [20:34:39.096149][RIGCTRL][trace] () : write_block(): TX 3 bytes
warning: [20:34:39.096149][RIGCTRL][trace] () : 0000    46 41 3b                                            FA;
warning: [20:34:39.096149][RIGCTRL][trace] () : read_string called, rxmax=129
warning: [20:34:39.151185][RIGCTRL][trace] () : read_string(): RX 11 characters
warning: [20:34:39.151185][RIGCTRL][trace] () : 0000    46 41 32 31 30 37 34 30 30 30 3b                    FA21074000;
warning: [20:34:39.152186][RIGCTRL][trace] () : newcat_get_cmd: read count = 11, ret_data = FA21074000;
warning: [20:34:39.152186][RIGCTRL][trace] () : newcat_get_freq: freq = 21074000.000000 Hz for vfo VFOA
warning: [20:34:39.152186][RIGCTRL][trace] () : elapsed_ms: start = 0,0
warning: [20:34:39.152186][RIGCTRL][trace] () : elapsed_ms: after gettime, start = 1610483679,152186100
warning: [20:34:39.152186][RIGCTRL][trace] () : elapsed_ms: start = 0,0
warning: [20:34:39.153188][RIGCTRL][trace] () : elapsed_ms: after gettime, start = 1610483679,153188100
warning: [20:34:39.153188][RIGCTRL][trace] () : rig_get_freq: cache reset age=999000ms, vfo=currVFO, freq=21074000

This one is completely incorrect. The rig replies to the FA set 
frequency command with a rig busy response, the trace says it is 
retrying the command but it doesn't actually retry the command. The 
[... 754 lines cut ...]
warning: [20:34:27.806572][RIGCTRL][debug] () : rig_register: rig_register (32001)
warning: [20:34:27.807573][RIGCTRL][debug] () : rig_register called
warning: [20:34:27.807573][RIGCTRL][debug] () : rig_register: rig_register (32002)
[... cut ...]
warning: [20:34:34.780251][RIGCTRL][debug] () : rig_init: rig has VFO_MEM
warning: [20:34:34.780251][RIGCTRL][debug] () : newcat_init called
warning: [20:34:34.781251][RIGCTRL][debug] () : rig_token_lookup called
warning: [20:34:34.781251][RIGCTRL][debug] () : rig_confparam_lookup called
warning: [20:34:34.782252][RIGCTRL][debug] () : rig_set_conf called
warning: [20:34:34.782252][RIGCTRL][debug] () : rig_confparam_lookup called
warning: [20:34:34.782252][RIGCTRL][debug] () : rig_set_conf: rig_pathname='COM23'
warning: [20:34:34.782252][

mikaelnousiainen, 2021-01-12:
@g4wjs The retry mechanism was broken because of other reasons, I've fixed both that and made the use of the question_mark flag specific to rig model/mode in #507 -- please test with the code from the new PR (I've tested it on FTDX101)

mikaelnousiainen, 2021-01-12:
@g4wjs BTW I've been testing FT-2000 earlier too, but with a direct serial port connection with no "busy" responses at all -- so it may be the remote TCP/IP serial port implementation and its latencies that cause additional challenges with Hamlib!

g4wjs, 2021-01-12:
On 12/01/2021 22:25, Mikael Nousiainen wrote:
> @g4wjs <https://github.com/g4wjs> BTW I've been testing FT-2000 
> earlier too, but with a direct serial port connection with no "busy" 
> responses at all -- so it may be the remote TCP/IP serial port 
> implementation and its latencies that cause additional challenges with 
> Hamlib!

Hi Mikael,

no doubt but the transactions are valid and should work. It seems a good 
opportunity to sort out the messed up retry mechanisms.

73
Bill
G4WJS.

mdblack98, 2021-01-12:
Are you testing with WSJT-X?
What about band changes?  If you do consistent read frequency does it get the ? response during band change?I've seen that on other rigs.
Mike

mikaelnousiainen, 2021-01-12:
I haven't tested FT-2000 specifically with WSJT-X, but only with my own remote control software, which does poll the rig actively. Band changes haven't been an issue so far.

mdblack98, 2021-01-12:
You can use rig->state.current_mode
Instead of rig_get_mode

Mike W9MDB

g4wjs, 2021-01-13:
On 12/01/2021 22:21, Mikael Nousiainen wrote:
> @g4wjs <https://github.com/g4wjs> The retry mechanism was broken
> because of other reasons, I've fixed both that and made the use of the
> question_mark flag specific to rig model/mode in #507 [...]

Hi Mikael,

I ran a test with the master branch at commit a9a806f8 which has PR #507 
merged. It is much better and I have reasonable control of the rig now. 
But I am still seeing set commands that get a '?;' response not being 
retried. Several examples below and a full log of the session attached.

Here the BS05; command gets a busy response and is not retried even 
though the trace says it will retry:

warning: [12:47:11.055133][RIGCTRL][trace] () : rig_set_freq: TARGETABLE_FREQ vfo=currVFO
warning: [12:47:11.055133][RIGCTRL][debug] () : newcat_set_freq called
warning: [12:47:11.056134][RIGCTRL][debug] () : newcat_valid_command called
warning: [12:47:11.056134][RIGCTRL][trace] () : newcat_valid_command FA
warning: [12:47:11.056134][RIGCTRL][debug] () : newcat_valid_command called
warning: [12:47:11.056134][RIGCTRL][trace] () : newcat_valid_command FB
warning: [12:47:11.056134][RIGCTRL][trace] () : newcat_set_freq: passed vfo = currVFO
warning: [12:47:11.057135][RIGCTRL][trace] () : newcat_set_freq: passed freq = 14074000.000000 Hz
warning: [12:47:11.057135][RIGCTRL][trace] () : newcat_set_vfo_from_alias: alias vfo = currVFO
warning: [12:47:11.057135][RIGCTRL][debug] () : newcat_valid_command called
warning: [12:47:11.057135][RIGCTRL][trace] () : newcat_valid_command BS
warning: [12:47:11.057135][RIGCTRL][trace] () : newcat_band_index: freq=1.4074e+07, band=5
warning: [12:47:11.057135][RIGCTRL][trace] () : newcat_band_index: freq=7.0475e+06, band=3
warning: [12:47:11.058135][RIGCTRL][trace] () : newcat_band_index: freq=1.4074e+07, band=5
warning: [12:47:11.058135][RIGCTRL][trace] () : rig_flush: called for serial device
warning: [12:47:11.058135][RIGCTRL][debug] () : serial_flush called
warning: [12:47:11.058135][RIGCTRL][debug] () : tcflush
warning: [12:47:11.058135][RIGCTRL][trace] () : cmd_str = BS05;
warning: [12:47:11.058135][RIGCTRL][debug] () : write_block called
warning: [12:47:11.081151][RIGCTRL][trace] () : write_block(): TX 5 bytes
warning: [12:47:11.081151][RIGCTRL][trace] () : 0000    42 53 30 35 3b                                      BS05;
warning: [12:47:11.081151][RIGCTRL][trace] () : cmd_str = ID;
warning: [12:47:11.081151][RIGCTRL][debug] () : write_block called
warning: [12:47:11.087159][RIGCTRL][trace] () : write_block(): TX 3 bytes
warning: [12:47:11.087159][RIGCTRL][trace] () : 0000    49 44 3b                                            ID;
warning: [12:47:11.087159][RIGCTRL][trace] () : read_string called, rxmax=129
warning: [12:47:11.119474][RIGCTRL][trace] () : read_string(): RX 2 characters
warning: [12:47:11.119474][RIGCTRL][trace] () : 0000    3f 3b                                               ?;
warning: [12:47:11.119474][RIGCTRL][trace] () : newcat_set_cmd: read count = 2, ret_data = ?;
warning: [12:47:11.119474][RIGCTRL][warning] () : newcat_set_cmd: Rig busy - retrying: 'BS05;'
warning: [12:47:11.120475][RIGCTRL][trace] () : read_string called, rxmax=129
warning: [12:47:11.153497][RIGCTRL][trace] () : read_string(): RX 7 characters
warning: [12:47:11.153497][RIGCTRL][trace] () : 0000    49 44 30 32 35 31 3b                                ID0251;
warning: [12:47:11.153497][RIGCTRL][trace] () : newcat_set_cmd: read count = 7, ret_data = ID0251;
warning: [12:47:11.153497][RIGCTRL][trace] () : newcat_set_freq:804 cmd_str = FA14074000;
warning: [12:47:11.154497][RIGCTRL][trace] () : rig_flush: called for serial device
warning: [12:47:11.154497][RIGCTRL][debug] () : serial_flush called
warning: [12:47:11.154497][RIGCTRL][debug] () : tcflush
warning: [12:47:11.154497][RIGCTRL][trace] () : cmd_str = FA14074000;
warning: [12:47:11.154497][RIGCTRL][debug] () : write_block called
warning: [12:47:11.160502][RIGCTRL][trace] () : write_block(): TX 11 bytes
warning: [12:47:11.160502][RIGCTRL][trace] () : 0000    46 41 31 34 30 37 34 30 30 30 3b                    FA14074000;
warning: [12:47:11.160502][RIGCTRL][trace] () : cmd_str = ID;
warning: [12:47:11.160502][RIGCTRL][debug] () : write_block called
warning: [12:47:11.176511][RIGCTRL][trace] () : write_block(): TX 3 bytes
warning: [12:47:11.176511][RIGCTRL][trace] () : 0000    49 44 3b                                            ID;
warning: [12:47:11.176511][RIGCTRL][trace] () : read_string called, rxmax=129
warning: [12:47:11.230130][RIGCTRL][trace] () : read_string(): RX 7 characters
warning: [12:47:11.230130][RIGCTRL][trace] () : 0000    49 44 30 32 35 31 3b                                ID0251;
warning: [12:47:11.230130][RIGCTRL][trace] () : newcat_set_cmd: read count = 7, ret_data = ID0251;
warning: [12:47:11.230130][RIGCTRL][trace] () : newcat_band_index: freq=7.0475e+06, band=3
warning: [12:47:11.231131][RIGCTRL][trace] () : newcat_band_index: freq=1.4074e+07, band=5
warning: [12:47:11.231131][RIGCTRL][trace] () : newcat_band_index: freq=7.0475e+06, band=3
warning: [12:47:11.231131][RIGCTRL][trace] () : newcat_band_index: freq=1.4074e+07, band=5
warning: [12:47:11.231131][RIGCTRL][trace] () : newcat_set_freq: band changing? old=5, new=3
[... 753 lines cut ...]
warning: [12:46:50.557079][RIGCTRL][debug] () : rig_register called
warning: [12:46:50.557079][RIGCTRL][debug] () : rig_register: rig_register (6007)
[... cut ...]
warning: [12:46:50.559080][RIGCTRL][debug] () : rig_register: rig_register (8010)
warning: [12:46:50.559080][RIGCTRL][debug] () : rig_register called
warning: [12:46:50.559080][RIGCTRL][debug] () : rig_register: rig_register (8011)
warning: [12:46:50.559080][RIGCTRL][debug] () : initrigs4_drake: _init called
warning: [12:46:50.560082][RIGCTRL][debug] () : rig_register called
warning: [12:46:50.560082][RIGCTRL][debug] () : rig_register: rig_register (9002)
warning: [12:46:50.560082][RIGCTRL][debug] () : rig_register calle

mdblack98, 2021-01-13:
I'd like to see what's going on before this BS05 sequence.....there's some timing problem where it's too early to send the BS05 command or the serial data is corrupted.
I'm going to post a patch shortly to fix the retry on this type of sequence.

 
 

Hi Mikael,

I ran a test with the master branch at commit a9a806f8 which has PR #507 
merged. It is much better and I have reasonable control of the rig now. 
But I am still seeing set commands that get a '?;' response not being 
retried. Several examples below and a full log of the session attached.

Here the BS05; command gets a busy response and is not retried even 
though the trace says it will retry:

warning: [12:47:11.055133][RIGCTRL][trace] () : rig_set_freq: TARGETABLE_FREQ vfo=currVFO
warning: [12:47:11.055133][RIGCTRL][debug] () : newcat_set_freq called
warning: [12:47:11.056134][RIGCTRL][debug] () : newcat_valid_command called
warning: [12:47:11.056134][RIGCTRL][trace] () : newcat_valid_command FA
warning: [12:47:11.056134][RIGCTRL][debug] () : newcat_valid_command called
warning: [12:47:11.056134][RIGCTRL][trace] () : newcat_valid_command FB
warning: [12:47:11.056134][RIGCTRL][trace] () : newcat_set_freq: passed vfo = currVFO
warning: [12:47:11.057135][RIGCTRL][trace] () : newcat_set_freq: passed freq = 14074000.000000 Hz
warning: [12:47:11.057135][RIGCTRL][trace] () : newcat_set_vfo_from_alias: alias vfo = currVFO
warning: [12:47:11.057135][RIGCTRL][debug] () : newcat_valid_command called
warning: [12:47:11.057135][RIGCTRL][trace] () : newcat_valid_command BS
warning: [12:47:11.057135][RIGCTRL][trace] () : newcat_band_index: freq=1.4074e+07, band=5
warning: [12:47:11.057135][RIGCTRL][trace] () : newcat_band_index: freq=7.0475e+06, band=3
warning: [12:47:11.058135][RIGCTRL][trace] () : newcat_band_index: freq=1.4074e+07, band=5
warning: [12:47:11.058135][RIGCTRL][trace] () : rig_flush: called for serial device
warning: [12:47:11.058135][RIGCTRL][debug] () : serial_flush called
warning: [12:47:11.058135][RIGCTRL][debug] () : tcflush
warning: [12:47:11.058135][RIGCTRL][trace] () : cmd_str = BS05;
warning: [12:47:11.058135][RIGCTRL][debug] () : write_block called
warning: [12:47:11.081151][RIGCTRL][trace] () : write_block(): TX 5 bytes
warning: [12:47:11.081151][RIGCTRL][trace] () : 0000 42 53 30 35 3b BS05;
warning: [12:47:11.081151][RIGCTRL][trace] () : cmd_str = ID;
warning: [12:47:11.081151][RIGCTRL][debug] () : write_block called
warning: [12:47:11.087159][RIGCTRL][trace] () : write_block(): TX 3 bytes
warning: [12:47:11.087159][RIGCTRL][trace] () : 0000 49 44 3b ID;
warning: [12:47:11.087159][RIGCTRL][trace] () : read_string called, rxmax=129
warning: [12:47:11.119474][RIGCTRL][trace] () : read_string(): RX 2 characters
warning: [12:47:11.119474][RIGCTRL][trace] () : 0000 3f 3b ?;
warning: [12:47:11.119474][RIGCTRL][trace] () : newcat_set_cmd: read count = 2, ret_data = ?;
warning: [12:47:11.119474][RIGCTRL][warning] () : newcat_set_cmd: Rig busy - retrying: 'BS05;'
warning: [12:47:11.120475][RIGCTRL][trace] () : read_string called, rxmax=129
warning: [12:47:11.153497][RIGCTRL][trace] () : read_string(): RX 7 characters
warning: [12:47:11.153497][RIGCTRL][trace] () : 0000 49 44 30 32 35 31 3b ID0251;
warning: [12:47:11.153497][RIGCTRL][trace] () : newcat_set_cmd: read count = 7, ret_data = ID0251;
warning: [12:47:11.153497][RIGCTRL][trace] () : newcat_set_freq:804 cmd_str = FA14074000;
warning: [12:47:11.154497][RIGCTRL][trace] () : rig_flush: called for serial device
warning: [12:47:11.154497][RIGCTRL][debug] () : serial_flush called
warning: [12:47:11.154497][RIGCTRL][debug] () : tcflush
warning: [12:47:11.154497][RIGCTRL][trace] () : cmd_str = FA14074000;
warning: [12:47:11.154497][RIGCTRL][debug] () : write_block called
warning: [12:47:11.160502][RIGCTRL][trace] () : write_block(): TX 11 bytes
warning: [12:47:11.160502][RIGCTRL][trace] () : 0000 46 41 31 34 30 37 34 30 30 30 3b FA14074000;
warning: [12:47:11.160502][RIGCTRL][trace] () : cmd_str = ID;
warning: [12:47:11.160502][RIGCTRL][debug] () : write_block called
warning: [12:47:11.176511][RIGCTRL][trace] () : write_block(): TX 3 bytes
warning: [12:47:11.176511][RIGCTRL][trace] () : 0000 49 44 3b ID;
warning: [12:47:11.176511][RIGCTRL][trace] () : read_string called, rxmax=129
warning: [12:47:11.230130][RIGCTRL][trace] () : read_string(): RX 7 characters
warning: [12:47:11.230130][RIGCTRL][trace] () : 0000 49 44 30 32 35 31 3b ID0251;
warning: [12:47:11.230130][RIGCTRL][trace] () : newcat_set_cmd: read count = 7, ret_data = ID0251;
warning: [12:47:11.230130][RIGCTRL][trace] () : newcat_band_index: freq=7.0475e+06, band=3
warning: [12:47:11.231131][RIGCTRL][trace] () : newcat_band_index: freq=1.4074e+07, band=5
warning: [12:47:11.231131][RIGCTRL][trace] () : newcat_band_index: freq=7.0475e+06, band=3
warning: [12:47:11.231131][RIGCTRL][trace] () : newcat_band_index: freq=1.4074e+07, band=5
[... 754 lines cut ...]
warning: [12:46:50.557079][RIGCTRL][debug] () : rig_register called
warning: [12:46:50.557079][RIGCTRL][debug] () : rig_register: rig_register (6007)
warning: [12:46:50.557079][RIGCTRL][debug] () : rig_register called
warning: [12:46:50.557079][RIGCTRL][debug] () : rig_register: rig_register (6005)
warning: [12:46:50.557079][RIGCTRL][debug] () : initrigs4_uniden: _init called
warning: [12:46:50.557079][RIGCTRL][debug] () : rig_register called
warning: [12:46:50.558079][RIGCTRL][debug] () : rig_register: rig_register (8003)
[... cut ...]
warning: [12:46:50.559080][RIGCTRL][debug] () : rig_register: rig_register (8010)
warning: [12:46:50.559080][RIGCTRL][debug] () : rig_register called
warning: [12:46:50.559080][RIGCTRL][debug] () : rig_register: rig_register (8011)
warning: [12:46:50.559080][RIGCTRL][debug] () : initrigs4_drake: _init called
warning: [12:46:50.560082][RIGCTRL][debug] () : rig_register called
warning: [12:46:50.560082][RIGCTRL][debug] () : rig_register: rig_register (9002)
warning: [12:46:50.560082][RIGCTRL][debug] () : rig_register calle

g4wjs, 2021-01-13:
On 13/01/2021 13:43, Michael Black wrote:
> I'd like to see what's going on before this BS05 sequence.....there's 
> some timing problem where it's too early to send the BS05 command or 
> the serial data is corrupted.
> I'm going to post a patch shortly to fix the retry on this type of 
> sequence.

Mike,

the whole log of the session was attached.

The reasons for the busy responses are irrelevant at the moment, I 
suspect a handshaking issue somewhere, what is important is that the 
retry mechanism is broken. Until that is fixed there no point in 
worrying about how the busy responses are occurring, the transactions 
are valid, the responses are from the rig and the reaction of Hamlib to 
those responses is wrong.

73
Bill
G4WJS.

mdblack98, 2021-01-13:
I added a 200ms delay before doing the BS command -- perhaps that fixes the ?; response.And I fixed the retry on this case too.
Mike W9MDB

 
 

Mike,

the whole log of the session was attached.

The reasons for the busy responses are irrelevant at the moment, I 
suspect a handshaking issue somewhere, what is important is that the 
retry mechanism is broken. Until that is fixed there no point in 
worrying about how the busy responses are occurring, the transactions 
are valid, the responses are from the rig and the reaction of Hamlib to 
those responses is wrong.

73
Bill
G4WJS.

g4wjs, 2021-01-13:
Hi Mike,

that is somewhat better, much more responsive as many of the unnecessary 
timed out reads have gone. Full session log attached. But there are 
still examples like this where a read_string after a '?;' response 
should not be attempted, just flush and retry is what is needed.

warning: [15:03:33.749982][RIGCTRL][debug] () : rig_set_split_mode called
warning: [15:03:33.750981][RIGCTRL][debug] () : newcat_set_mode called
warning: [15:03:33.750981][RIGCTRL][debug] () : newcat_valid_command called
warning: [15:03:33.750981][RIGCTRL][trace] () : newcat_valid_command MD
warning: [15:03:33.750981][RIGCTRL][trace] () : newcat_set_vfo_from_alias: alias vfo = VFOB
warning: [15:03:33.750981][RIGCTRL][debug] () : newcat_set_mode: generic mode = PKTUSB
warning: [15:03:33.750981][RIGCTRL][debug] () : newcat_modechar called
warning: [15:03:33.751982][RIGCTRL][trace] () : newcat_modechar: return C for PKTUSB
warning: [15:03:33.751982][RIGCTRL][trace] () : rig_flush: called for serial device
warning: [15:03:33.751982][RIGCTRL][debug] () : serial_flush called
warning: [15:03:33.751982][RIGCTRL][debug] () : tcflush
warning: [15:03:33.751982][RIGCTRL][trace] () : cmd_str = MD1C;
warning: [15:03:33.751982][RIGCTRL][debug] () : write_block called
warning: [15:03:33.758987][RIGCTRL][trace] () : write_block(): TX 5 bytes
warning: [15:03:33.758987][RIGCTRL][trace] () : 0000    4d 44 31 43 3b                                      MD1C;
warning: [15:03:33.758987][RIGCTRL][trace] () : cmd_str = ID;
warning: [15:03:33.758987][RIGCTRL][debug] () : write_block called
warning: [15:03:33.764994][RIGCTRL][trace] () : write_block(): TX 3 bytes
warning: [15:03:33.764994][RIGCTRL][trace] () : 0000    49 44 3b                                            ID;
warning: [15:03:33.765992][RIGCTRL][trace] () : read_string called, rxmax=129
warning: [15:03:33.810025][RIGCTRL][trace] () : read_string(): RX 2 characters
warning: [15:03:33.810025][RIGCTRL][trace] () : 0000    3f 3b                                               ?;
warning: [15:03:33.810025][RIGCTRL][trace] () : newcat_set_cmd: read count = 2, ret_data = ?;
warning: [15:03:33.810025][RIGCTRL][warning] () : newcat_set_cmd: Rig busy - retrying: 'MD1C;'
warning: [15:03:33.811027][RIGCTRL][trace] () : read_string called, rxmax=129
warning: [15:03:36.007333][RIGCTRL][warning] () : read_string(): Timed out 2.196306 seconds after 0 chars
warning: [15:03:36.007333][RIGCTRL][trace] () : rig_flush: called for serial device
warning: [15:03:36.007333][RIGCTRL][debug] () : serial_flush called
warning: [15:03:36.007333][RIGCTRL][debug] () : tcflush
warning: [15:03:36.008331][RIGCTRL][trace] () : cmd_str = MD1C;
warning: [15:03:36.008331][RIGCTRL][debug] () : write_block called
warning: [15:03:36.014336][RIGCTRL][trace] () : write_block(): TX 5 bytes
warning: [15:03:36.014336][RIGCTRL][trace] () : 0000    4d 44 31 43 3b                                      MD1C;
warning: [15:03:36.014336][RIGCTRL][trace] () : cmd_str = ID;
warning: [15:03:36.014336][RIGCTRL][debug] () : write_block called
warning: [15:03:36.021359][RIGCTRL][trace] () : write_block(): TX 3 bytes
warning: [15:03:36.021359][RIGCTRL][trace] () : 0000    49 44 3b                                            ID;
warning: [15:03:36.021359][RIGCTRL][trace] () : read_string called, rxmax=129
warning: [15:03:36.087408][RIGCTRL][trace] () : read_string(): RX 7 characters
warning: [15:03:36.087408][RIGCTRL][trace] () : 0000    49 44 30 32 35 31 3b                                ID0251;
warning: [15:03:36.087408][RIGCTRL][trace] () : newcat_set_cmd: read count = 7, ret_data = ID0251;

Here's a case where retries are exhausted, it tries 4 times which is not 
really consistent with the caps setting of 3 retries but better to do 
too many than not enough. I know this is a special case but I don't see 
a problem with a relatively high number of retries, especially if 
unnecessary timeouts are avoided. I am not sure 3 retries is enough for 
Yaesu rigs like this one.

warning: [15:04:03.719144][RIGCTRL][debug] () : rig_set_freq called vfo=currVFO, freq=7.074e+06
warning: [15:04:03.719144][RIGCTRL][trace] () : vfo_fixup: vfo=currVFO
warning: [15:04:03.720144][RIGCTRL][trace] () : vfo_fixup: Leaving currVFO alone
warning: [15:04:03.720144][RIGCTRL][trace] () : rig_set_freq: TARGETABLE_FREQ vfo=currVFO
warning: [15:04:03.720144][RIGCTRL][debug] () : newcat_set_freq called
warning: [15:04:03.720144][RIGCTRL][debug] () : newcat_valid_command called
warning: [15:04:03.720144][RIGCTRL][trace] () : newcat_valid_command FA
warning: [15:04:03.720144][RIGCTRL][debug] () : newcat_valid_command called
warning: [15:04:03.721145][RIGCTRL][trace] () : newcat_valid_command FB
warning: [15:04:03.721145][RIGCTRL][trace] () : newcat_set_freq: passed vfo = currVFO
warning: [15:04:03.721145][RIGCTRL][trace] () : newcat_set_freq: passed freq = 7074000.000000 Hz
warning: [15:04:03.721145][RIGCTRL][trace] () : newcat_set_vfo_from_alias: alias vfo = currVFO
warning: [15:04:03.721145][RIGCTRL][debug] () : newcat_valid_command called
warning: [15:04:03.721145][RIGCTRL][trace] () : newcat_valid_command BS
[... 755 lines cut ...]
warning: [15:03:08.093110][RIGCTRL][trace] () : newcat_get_cmd: read count = 4, ret_data = VS0;
warning: [15:03:08.093110][RIGCTRL][debug] () : newcat_get_vfo_mode called
warning: [15:03:08.094111][RIGCTRL][debug] () : newcat_valid_command called
warning: [15:03:08.094111][RIGCTRL][trace] () : newcat_valid_command IF
[... cut ...]
warning: [15:03:08.159152][RIGCTRL][debug] () : serial_flush called
warning: [15:03:08.159152][RIGCTRL][debug] () : tcflush
warning: [15:03:08.159152][RIGCTRL][trace] () : cmd_str = IF;
warning: [15:03:08.159152][RIGCTRL][debug] () : write_block called
warning: [15:03:08.165156][RIGCTRL][trace] () : write_block(): TX 3 bytes
warning: [15:03:08.165156][RIGCTRL][trace] () : 0000    49 46 3b                                            IF;
warning: [15:03:08.165156][RIGCTRL][trace] () : read_string called, rxmax=129
warning: [15:0

mdblack98, 2021-01-13:
tries = 1 + retries
we can't flush since the ID; command response needs to have time to read and the flush would happen too soon.
I'm wondering if sending the ID command too quickly can cause this?
So I put an 11ms wait between the command and the ID; request.
Maybe that's enough to trigger the 10ms CAT TOT timeout.
Mike

 
 

r
`c:\test-install\wsjtx-dev\Debug\bin\wsjtx.exe' has changed; re-reading symbols.
Starting program: c:\test-install\wsjtx-dev\Debug\bin\wsjtx.exe -r test 
[New Thread 7172.0x36e8]
[New Thread 7172.0x2788]
[New Thread 7172.0xb4]
warning: [15:02:56.430321][SYSLOG][info] C:\Users\bill\src\k1jt\wsjtx\WSJTXLogging.cpp(161) WSJTXLogging: Read logging configuration file: C:/Users/bill/AppData/Local/WSJT-X - test/wsjtx_log_config.ini
warning: [15:02:56.431322][SYSLOG][info] C:\Users\bill\src\k1jt\wsjtx\WSJTXLogging.cpp(230) WSJTXLogging: Log Start
warning: [15:02:56.431322][SYSLOG][info] C:\Users\bill\src\k1jt\wsjtx\main.cpp(215) main: WSJT-X - test v2.4.0-devel ef4787 by K1JT, G4WJS, K9AN, and IV3NWV - Program startup
warning: [15:02:56.441328][SYSLOG][info] C:\Users\bill\src\k1jt\wsjtx\L10nLoader.cpp(70) L10nLoader: locale: language: English script: Latin country: United Kingdom ui-languages: en-GB, en-US
warning: [15:02:56.442329][SYSLOG][info] C:\Users\bill\src\k1jt\wsjtx\L10nLoader.cpp(87) L10nLoader: Loaded Qt translations for current locale from resources
warning: [15:02:56.443333][SYSLOG][info] C:\Users\bill\src\k1jt\wsjtx\L10nLoader.cpp(113) L10nLoader: Loaded WSJT-X base translation file from :/Translations based on language en
warning: [15:02:56.443333][SYSLOG][info] C:\Users\bill\src\k1jt\wsjtx\L10nLoader.cpp(123) L10nLoader: Loaded WSJT-X translations for current locale from resources
warning: [15:02:57.762214][SYSLOG][info] C:\Users\bill\src\k1jt\wsjtx\main.cpp(400) main: shmem size:48275456
[New Thread 7172.0x408]
warning: [15:02:58.172490][RIGCTRL][debug] () : initrigs4_dummy: _init called
warning: [15:02:58.172490][RIGCTRL][debug] () : rig_register called
warning: [15:02:58.172490][RIGCTRL][debug] () : rig_register: rig_register (1)
warning: [15:02:58.172490][RIGCTRL][debug] () : rig_register called
warning: [15:02:58.173491][RIGCTRL][debug] () : rig_register: rig_register (2)
warning: [15:02:58.173491][RIGCTRL][debug] () : rig_register called
warning: [15:02:58.173491][RIGCTRL][debug] () : rig_register: rig_register (4)
warning: [15:02:58.173491][RIGCTRL][debug] () : rig_register called
warning: [15:02:58.173491][RIGCTRL][debug] () : rig_register: rig_register (5)
warning: [15:02:58.174492][RIGCTRL][debug] () : rig_register called
warning: [15:02:58.174492][RIGCTRL][debug] () : rig_register: rig_register (6)
warning: [15:02:58.174492][RIGCTRL][debug] () : yaesu: initrigs4_yaesu called
warning: [15:02:58.174492][RIGCTRL][debug] () : rig_register called
warning: [15:02:58.174492][RIGCTRL][debug] () : rig_register: rig_register (1021)
warning: [15:02:58.174492][RIGCTRL][debug] () : rig_register called
warning: [15:02:58.175493][RIGCTRL][debug] () : rig_register: rig_register (1027)
warning: [15:02:58.175493][RIGCTRL][debug] () : rig_register called
warning: [15:02:58.175493][RIGCTRL][debug] () : rig_register: rig_register (1010)
warning: [15:02:58.175493][RIGCTRL][debug] () : rig_register called
warning: [15:02:58.175493][RIGCTRL][debug] () : rig_register: rig_register (1005)
warning: [15:02:58.175493][RIGCTRL][debug] () : rig_register called
warning: [15:02:58.176496][RIGCTRL][debug] () : rig_register: rig_register (1006)
warning: [15:02:58.176496][RIGCTRL][debug] () : rig_register called
warning: [15:02:58.176496][RIGCTRL][debug] () : rig_register: rig_register (1007)
warning: [15:02:58.176496][RIGCTRL][debug] () : rig_register called
warning: [15:02:58.176496][RIGCTRL][debug] () : rig_register: rig_register (1039)
warning: [15:02:58.177495][RIGCTRL][debug] () : rig_register called
warning: [15:02:58.177495][RIGCTRL][debug] () : rig_register: rig_register (1009)
warning: [15:02:58.177495][RIGCTRL][debug] () : rig_register called
warning: [15:02:58.177495][RIGCTRL][debug] () : rig_register: rig_register (1020)
warning: [15:02:58.177495][RIGCTRL][debug] () : rig_register called
warning: [15:02:58.177495][RIGCTRL][debug] () : rig_register: rig_register (1001)
warning: [15:02:58.178494][RIGCTRL][debug] () : rig_register called
warning: [15:02:58.178494][RIGCTRL][debug] () : rig_register: rig_register (1022)
warning: [15:02:58.178494][RIGCTRL][debug] () : rig_register called
warning: [15:02:58.178494][RIGCTRL][debug] () : rig_register: rig_register (1023)
warning: [15:02:58.178494][RIGCTRL][debug] () : rig_register called
warning: [15:02:58.179494][RIGCTRL][debug] () : rig_register: rig_register (1011)
warning: [15:02:58.179494][RIGCTRL][debug] () : rig_register called
warning: [15:02:58.179494][RIGCTRL][debug] () : rig_register: rig_register (1015)
warning: [15:02:58.179494][RIGCTRL][debug] () : rig_register called
warning: [15:02:58.179494][RIGCTRL][debug] () : rig_register: rig_register (1013)
warning: [15:02:58.179494][RIGCTRL][debug] () : rig_register called
warning: [15:02:58.179494][RIGCTRL][debug] () : rig_register: rig_register (1014)
warning: [15:02:58.180497][RIGCTRL][debug] () : rig_register called
[... 593 lines cut ...]
warning: [15:03:08.093110][RIGCTRL][trace] () : newcat_get_cmd: read count = 4, ret_data = VS0;
warning: [15:03:08.093110][RIGCTRL][debug] () : newcat_get_vfo_mode called
[... cut ...]
warning: [15:03:08.158153][RIGCTRL][trace] () : rig_flush: called for serial device
warning: [15:03:08.159152][RIGCTRL][debug] () : serial_flush called
warning: [15:03:08.159152][RIGCTRL][debug] () : tcflush
warning: [15:03:08.159152][RIGCTRL][trace] () : cmd_str = IF;
warning: [15:03:08.159152][RIGCTRL][debug] () : write_block called
warning: [15:03:08.165156][RIGCTRL][trace] () : write_block(): TX 3 bytes
warning: [15:03:08.165156][RIGCTRL][trace] () : 0000 49 46 3b IF;
warning: [15:03:08.165156][RIGCTRL][trace] () : read_string called, rxmax=129
warning: [15:0

g4wjs, 2021-01-13:
On 13/01/2021 15:42, Michael Black wrote:
> tries = 1 + retries
> we can't flush since the ID; command response needs to have time to
> read and the flush would happen too soon.
> I'm wondering if sending the ID command too quickly can cause this? [...]

Hi Mike,

attached is another session log using commit b5de94da(I set retries to 
10 just in case although that was not needed in this case). I don't see 
any instances where a flush fails to remove unwanted responses, but I 
see several instances where there are still unnecessary reads that time 
out, like this:

warning: [20:02:15.901206][RIGCTRL][debug] () : rig_set_freq called vfo=currVFO, freq=1.4074e+07
warning: [20:02:15.901206][RIGCTRL][trace] () : vfo_fixup: vfo=currVFO
warning: [20:02:15.902206][RIGCTRL][trace] () : vfo_fixup: Leaving currVFO alone
warning: [20:02:15.902206][RIGCTRL][trace] () : rig_set_freq: TARGETABLE_FREQ vfo=currVFO
warning: [20:02:15.902206][RIGCTRL][debug] () : newcat_set_freq called
warning: [20:02:15.902206][RIGCTRL][debug] () : newcat_valid_command called
warning: [20:02:15.907210][RIGCTRL][trace] () : newcat_valid_command FA
warning: [20:02:15.907210][RIGCTRL][debug] () : newcat_valid_command called
warning: [20:02:15.907210][RIGCTRL][trace] () : newcat_valid_command FB
warning: [20:02:15.908212][RIGCTRL][trace] () : newcat_set_freq: passed vfo = currVFO
warning: [20:02:15.908212][RIGCTRL][trace] () : newcat_set_freq: passed freq = 14074000.000000 Hz
warning: [20:02:15.908212][RIGCTRL][trace] () : newcat_set_vfo_from_alias: alias vfo = currVFO
warning: [20:02:15.908212][RIGCTRL][debug] () : newcat_valid_command called
warning: [20:02:15.909215][RIGCTRL][trace] () : newcat_valid_command BS
warning: [20:02:15.909215][RIGCTRL][trace] () : newcat_band_index: freq=1.4074e+07, band=5
warning: [20:02:15.909215][RIGCTRL][trace] () : newcat_band_index: freq=7.0475e+06, band=3
warning: [20:02:16.110347][RIGCTRL][trace] () : newcat_band_index: freq=1.4074e+07, band=5
warning: [20:02:16.110347][RIGCTRL][trace] () : rig_flush: called for serial device
warning: [20:02:16.110347][RIGCTRL][debug] () : serial_flush called
warning: [20:02:16.110347][RIGCTRL][debug] () : tcflush
warning: [20:02:16.111346][RIGCTRL][trace] () : cmd_str = BS05;
warning: [20:02:16.111346][RIGCTRL][debug] () : write_block called
warning: [20:02:16.117352][RIGCTRL][trace] () : write_block(): TX 5 bytes
warning: [20:02:16.117352][RIGCTRL][trace] () : 0000    42 53 30 35 3b                                      BS05;
warning: [20:02:16.129360][RIGCTRL][trace] () : cmd_str = ID;
warning: [20:02:16.129360][RIGCTRL][debug] () : write_block called
warning: [20:02:16.151375][RIGCTRL][trace] () : write_block(): TX 3 bytes
warning: [20:02:16.151375][RIGCTRL][trace] () : 0000    49 44 3b                                            ID;
warning: [20:02:16.151375][RIGCTRL][trace] () : read_string called, rxmax=129
warning: [20:02:16.184397][RIGCTRL][trace] () : read_string(): RX 2 characters
warning: [20:02:16.184397][RIGCTRL][trace] () : 0000    3f 3b                                               ?;
warning: [20:02:16.184397][RIGCTRL][trace] () : newcat_set_cmd: read count = 2, ret_data = ?;
warning: [20:02:16.184397][RIGCTRL][warning] () : newcat_set_cmd: Rig busy - retrying: 'BS05;'
warning: [20:02:16.185398][RIGCTRL][trace] () : read_string called, rxmax=129
warning: [20:02:18.381300][RIGCTRL][warning] () : read_string(): Timed out 2.195902 seconds after 0 chars
warning: [20:02:18.381300][RIGCTRL][trace] () : rig_flush: called for serial device
warning: [20:02:18.381300][RIGCTRL][debug] () : serial_flush called
warning: [20:02:18.382299][RIGCTRL][debug] () : tcflush
warning: [20:02:18.382299][RIGCTRL][trace] () : cmd_str = BS05;
warning: [20:02:18.382299][RIGCTRL][debug] () : write_block called
warning: [20:02:18.388304][RIGCTRL][trace] () : write_block(): TX 5 bytes
warning: [20:02:18.388304][RIGCTRL][trace] () : 0000    42 53 30 35 3b                                      BS05;
warning: [20:02:18.400315][RIGCTRL][trace] () : cmd_str = ID;
warning: [20:02:18.400315][RIGCTRL][debug] () : write_block called
warning: [20:02:18.406488][RIGCTRL][trace] () : write_block(): TX 3 bytes
warning: [20:02:18.406488][RIGCTRL][trace] () : 0000    49 44 3b                                            ID;
warning: [20:02:18.406488][RIGCTRL][trace] () : read_string called, rxmax=129
warning: [20:02:18.461524][RIGCTRL][trace] () : read_string(): RX 7 characters
warning: [20:02:18.461524][RIGCTRL][trace] () : 0000    49 44 30 32 35 31 3b                                ID0251;
warning: [20:02:18.461524][RIGCTRL][trace] () : newcat_set_cmd: read count = 7, ret_data = ID0251;
warning: [20:02:18.462524][RIGCTRL][trace] () : newcat_set_freq:828 cmd_str = FA14074000;
warning: [20:02:18.462524][RIGCTRL][trace] () : rig_flush: called for serial device
warning: [20:02:18.462524][RIGCTRL][debug] () : serial_flush called
warning: [20:02:18.463525][RIGCTRL][debug] () : tcflush
warning: [20:02:18.463525][RIGCTRL][trace] () : cmd_str = FA14074000;
warning: [20:02:18.463525][RIGCTRL][debug] () : write_block called
[... 739 lines cut ...]
warning: [20:01:48.846742][RIGCTRL][debug] () : rig_register called
warning: [20:01:48.846742][RIGCTRL][debug] () : rig_register: rig_register (16008)
warning: [20:01:48.847744][RIGCTRL][debug] () : rig_register called
warning: [20:01:48.847744][RIGCTRL][debug] () : rig_register: rig_register (16002)
[... cut ...]
warning: [20:01:48.849742][RIGCTRL][debug] () : rig_register: rig_register (16005)
warning: [20:01:48.849742][RIGCTRL][debug] () : rig_register called
warning: [20:01:48.849742][RIGCTRL][debug] () : rig_register: rig_register (17001)
warning: [20:01:48.850742][RIGCTRL][debug] () : rig_register called
warning: [20:01:48.850742][RIGCTRL][debug] () : rig_register: rig_register (17002)
warning: [20:01:48.850742][RIGCTRL][debug] () : initrigs4_kachina: _init called
warning: [20:01:48.850742][RIGCTRL][debug] () : rig_register called
warning: [20:01:48.850742][RIGCTRL][debug

mdblack98, 2021-01-13:
Try again....
My goal is to get rid of this ?; return on BS;  -- I added a 200ms wait after the BS; command assuming it needs some time and now it will requery freq/mode and won't do anything if the band is already set.
Mike W9MDB

 
 

Hi Mike,

attached is another session log using commit b5de94da(I set retries to 
10 just in case although that was not needed in this case). I don't see 
any instances where a flush fails to remove unwanted responses, but I 
see several instances where there are still unnecessary reads that time 
out, like this:

warning: [20:02:15.901206][RIGCTRL][debug] () : rig_set_freq called vfo=currVFO, freq=1.4074e+07
warning: [20:02:15.901206][RIGCTRL][trace] () : vfo_fixup: vfo=currVFO
warning: [20:02:15.902206][RIGCTRL][trace] () : vfo_fixup: Leaving currVFO alone
warning: [20:02:15.902206][RIGCTRL][trace] () : rig_set_freq: TARGETABLE_FREQ vfo=currVFO
warning: [20:02:15.902206][RIGCTRL][debug] () : newcat_set_freq called
warning: [20:02:15.902206][RIGCTRL][debug] () : newcat_valid_command called
warning: [20:02:15.907210][RIGCTRL][trace] () : newcat_valid_command FA
warning: [20:02:15.907210][RIGCTRL][debug] () : newcat_valid_command called
warning: [20:02:15.907210][RIGCTRL][trace] () : newcat_valid_command FB
warning: [20:02:15.908212][RIGCTRL][trace] () : newcat_set_freq: passed vfo = currVFO
warning: [20:02:15.908212][RIGCTRL][trace] () : newcat_set_freq: passed freq = 14074000.000000 Hz
warning: [20:02:15.908212][RIGCTRL][trace] () : newcat_set_vfo_from_alias: alias vfo = currVFO
warning: [20:02:15.908212][RIGCTRL][debug] () : newcat_valid_command called
warning: [20:02:15.909215][RIGCTRL][trace] () : newcat_valid_command BS
warning: [20:02:15.909215][RIGCTRL][trace] () : newcat_band_index: freq=1.4074e+07, band=5
warning: [20:02:15.909215][RIGCTRL][trace] () : newcat_band_index: freq=7.0475e+06, band=3
warning: [20:02:16.110347][RIGCTRL][trace] () : newcat_band_index: freq=1.4074e+07, band=5
warning: [20:02:16.110347][RIGCTRL][trace] () : rig_flush: called for serial device
warning: [20:02:16.110347][RIGCTRL][debug] () : serial_flush called
warning: [20:02:16.110347][RIGCTRL][debug] () : tcflush
warning: [20:02:16.111346][RIGCTRL][trace] () : cmd_str = BS05;
warning: [20:02:16.111346][RIGCTRL][debug] () : write_block called
warning: [20:02:16.117352][RIGCTRL][trace] () : write_block(): TX 5 bytes
warning: [20:02:16.117352][RIGCTRL][trace] () : 0000 42 53 30 35 3b BS05;
warning: [20:02:16.129360][RIGCTRL][trace] () : cmd_str = ID;
warning: [20:02:16.129360][RIGCTRL][debug] () : write_block called
warning: [20:02:16.151375][RIGCTRL][trace] () : write_block(): TX 3 bytes
warning: [20:02:16.151375][RIGCTRL][trace] () : 0000 49 44 3b ID;
warning: [20:02:16.151375][RIGCTRL][trace] () : read_string called, rxmax=129
warning: [20:02:16.184397][RIGCTRL][trace] () : read_string(): RX 2 characters
warning: [20:02:16.184397][RIGCTRL][trace] () : 0000 3f 3b ?;
warning: [20:02:16.184397][RIGCTRL][trace] () : newcat_set_cmd: read count = 2, ret_data = ?;
warning: [20:02:16.184397][RIGCTRL][warning] () : newcat_set_cmd: Rig busy - retrying: 'BS05;'
warning: [20:02:16.185398][RIGCTRL][trace] () : read_string called, rxmax=129
warning: [20:02:18.381300][RIGCTRL][warning] () : read_string(): Timed out 2.195902 seconds after 0 chars
warning: [20:02:18.381300][RIGCTRL][trace] () : rig_flush: called for serial device
warning: [20:02:18.381300][RIGCTRL][debug] () : serial_flush called
warning: [20:02:18.382299][RIGCTRL][debug] () : tcflush
warning: [20:02:18.382299][RIGCTRL][trace] () : cmd_str = BS05;
warning: [20:02:18.382299][RIGCTRL][debug] () : write_block called
warning: [20:02:18.388304][RIGCTRL][trace] () : write_block(): TX 5 bytes
warning: [20:02:18.388304][RIGCTRL][trace] () : 0000 42 53 30 35 3b BS05;
warning: [20:02:18.400315][RIGCTRL][trace] () : cmd_str = ID;
warning: [20:02:18.400315][RIGCTRL][debug] () : write_block called
warning: [20:02:18.406488][RIGCTRL][trace] () : write_block(): TX 3 bytes
warning: [20:02:18.406488][RIGCTRL][trace] () : 0000 49 44 3b ID;
warning: [20:02:18.406488][RIGCTRL][trace] () : read_string called, rxmax=129
warning: [20:02:18.461524][RIGCTRL][trace] () : read_string(): RX 7 characters
warning: [20:02:18.461524][RIGCTRL][trace] () : 0000 49 44 30 32 35 31 3b ID0251;
warning: [20:02:18.461524][RIGCTRL][trace] () : newcat_set_cmd: read count = 7, ret_data = ID0251;
warning: [20:02:18.462524][RIGCTRL][trace] () : newcat_set_freq:828 cmd_str = FA14074000;
warning: [20:02:18.462524][RIGCTRL][trace] () : rig_flush: called for serial device
warning: [20:02:18.462524][RIGCTRL][debug] () : serial_flush called
warning: [20:02:18.463525][RIGCTRL][debug] () : tcflush
warning: [20:02:18.463525][RIGCTRL][trace] () : cmd_str = FA14074000;
[... 740 lines cut ...]
warning: [20:01:48.846742][RIGCTRL][debug] () : rig_register called
warning: [20:01:48.846742][RIGCTRL][debug] () : rig_register: rig_register (16008)
warning: [20:01:48.847744][RIGCTRL][debug] () : rig_register called
warning: [20:01:48.847744][RIGCTRL][debug] () : rig_register: rig_register (16002)
warning: [20:01:48.847744][RIGCTRL][debug] () : rig_register called
warning: [20:01:48.847744][RIGCTRL][debug] () : rig_register: rig_register (16009)
warning: [20:01:48.847744][RIGCTRL][debug] () : rig_register called
warning: [20:01:48.848741][RIGCTRL][debug] () : rig_register: rig_register (16011)
warning: [20:01:48.848741][RIGCTRL][debug] () : rig_register called
[... cut ...]
warning: [20:01:48.849742][RIGCTRL][debug] () : rig_register: rig_register (16005)
warning: [20:01:48.849742][RIGCTRL][debug] () : rig_register called
warning: [20:01:48.849742][RIGCTRL][debug] () : rig_register: rig_register (17001)
warning: [20:01:48.850742][RIGCTRL][debug] () : rig_register called
warning: [20:01:48.850742][RIGCTRL][debug] () : rig_register: rig_register (17002)
warning: [20:01:48.850742][RIGCTRL][debug] () : initrigs4_kachina: _init called
warning: [20:01:48.850742][RIGCTRL][debug] () : rig_register called
warning: [20:01:48.850742][RIGCTRL][debug

g4wjs, 2021-01-13:
On 13/01/2021 21:57, Michael Black wrote:
> Try again....
> My goal is to get rid of this ?; return on BS;  -- I added a 200ms 
> wait after the BS; command assuming it needs some time and now it will 
> requery freq/mode and won't do anything if the band is already set.
> Mike W9MDB
>
Hi Mike,

I not sure you should be chasing down the reason for the '?;' responses 
I get with the remote serial port software, I'd rather see the retry 
mechanism solid. Once that happens I can make a normal build and send it 
to the guy who has the FT-2000, then we can get a log with a normal 
local serial port connection.

I'll see if I can get one more test run this evening, it may be too late 
and have to wait until tomorrow.

73
Bill
G4WJS.

mdblack98, 2021-01-13:
I don't think we can reduce the timeout after we get a ?; response -- in this case I think that's indicating the rig is busy.  The only way I can think of is to recognize commands that should be 100% reliable and do a busy wait loop which could be faster.
Something likeif cmd=="FA" and reply = "?" then 10ms sleep and retry 30 times.Or something like that.

So I'm hoping the last change I put gives the BS command enough time to do the band change.
Mike

 
 

Hi Mike,

I not sure you should be chasing down the reason for the '?;' responses 
I get with the remote serial port software, I'd rather see the retry 
mechanism solid. Once that happens I can make a normal build and send it 
to the guy who has the FT-2000, then we can get a log with a normal 
local serial port connection.

I'll see if I can get one more test run this evening, it may be too late 
and have to wait until tomorrow.

73
Bill
G4WJS.

g4wjs, 2021-01-14:
Hi Mike,

attached is another session using commit 32d5f2b4. The BS commands are 
now a bit better as they are being sent when necessary for VFO A, but no 
attempt to do the required band changes for VFO B. I think this will be 
confusing for owners of Yasesu rigs with dual Rx like the FT-2000 since 
it is not treating the two receivers equally. Surely the logic when a 
frequency change is requested on rigs with a VS command should be:

set VFO X to frequency f:
    use FX; to get the current frequency
    if VFO X current frequency is on a new band:
        use VS; to get the selected VFO
        if selected VFO is not X:
            use VS to select VFO X
        end-if
        send BS command to select new band
        if previously selected VFO was not X:
            use VS to select previous VFO
        end-if
        use FX to set the target frequency on VFO X (could read first to 
avoid redundant change)
    end-if

Other than that the remaining read_string() timeouts are annoying and 
there was a case where 3 retries was not enough, otherwise it seems better.

73
Bill
G4WJS.

On 13/01/2021 22:15, Michael Black wrote:
> I don't think we can reduce the timeout after we get a ?; response --
> in this case I think that's indicating the rig is busy.  The only way
> I can think of is to recognize commands that should be [...]

r
Starting program: c:\test-install\wsjtx-dev\Debug\bin\wsjtx.exe -r test 
[New Thread 12888.0x2ad8]
[New Thread 12888.0x24c]
[New Thread 12888.0x1e90]
warning: [16:17:41.619661][SYSLOG][info] C:\Users\bill\src\k1jt\wsjtx\WSJTXLogging.cpp(161) WSJTXLogging: Read logging configuration file: C:/Users/bill/AppData/Local/WSJT-X - test/wsjtx_log_config.ini
warning: [16:17:41.619661][SYSLOG][info] C:\Users\bill\src\k1jt\wsjtx\WSJTXLogging.cpp(230) WSJTXLogging: Log Start
warning: [16:17:41.620662][SYSLOG][info] C:\Users\bill\src\k1jt\wsjtx\main.cpp(215) main: WSJT-X - test   v2.3.0-rc4 b741eb  by K1JT, G4WJS, and K9AN - Program startup
warning: [16:17:41.633671][SYSLOG][info] C:\Users\bill\src\k1jt\wsjtx\L10nLoader.cpp(70) L10nLoader: locale: language: English script: Latin country: United Kingdom ui-languages: en-GB, en-US
warning: [16:17:41.634672][SYSLOG][info] C:\Users\bill\src\k1jt\wsjtx\L10nLoader.cpp(87) L10nLoader: Loaded Qt translations for current locale from resources
warning: [16:17:41.634672][SYSLOG][info] C:\Users\bill\src\k1jt\wsjtx\L10nLoader.cpp(113) L10nLoader: Loaded WSJT-X base translation file from :/Translations based on language en
warning: [16:17:41.634672][SYSLOG][info] C:\Users\bill\src\k1jt\wsjtx\L10nLoader.cpp(123) L10nLoader: Loaded WSJT-X translations for current locale from resources
warning: [16:17:43.380898][SYSLOG][info] C:\Users\bill\src\k1jt\wsjtx\main.cpp(400) main: shmem size:48275456
[New Thread 12888.0x369c]
warning: [16:17:43.935268][RIGCTRL][debug] () : initrigs4_dummy: _init called
warning: [16:17:43.936268][RIGCTRL][debug] () : rig_register called
warning: [16:17:43.936268][RIGCTRL][debug] () : rig_register: rig_register (1)
warning: [16:17:43.936268][RIGCTRL][debug] () : rig_register called
warning: [16:17:43.936268][RIGCTRL][debug] () : rig_register: rig_register (2)
warning: [16:17:43.936268][RIGCTRL][debug] () : rig_register called
warning: [16:17:43.937268][RIGCTRL][debug] () : rig_register: rig_register (4)
warning: [16:17:43.937268][RIGCTRL][debug] () : rig_register called
warning: [16:17:43.937268][RIGCTRL][debug] () : rig_register: rig_register (5)
warning: [16:17:43.937268][RIGCTRL][debug] () : rig_register called
warning: [16:17:43.938270][RIGCTRL][debug] () : rig_register: rig_register (6)
warning: [16:17:43.938270][RIGCTRL][debug] () : yaesu: initrigs4_yaesu called
warning: [16:17:43.938270][RIGCTRL][debug] () : rig_register called
warning: [16:17:43.938270][RIGCTRL][debug] () : rig_register: rig_register (1021)
warning: [16:17:43.938270][RIGCTRL][debug] () : rig_register called
warning: [16:17:43.939270][RIGCTRL][debug] () : rig_register: rig_register (1027)
warning: [16:17:43.939270][RIGCTRL][debug] () : rig_register called
warning: [16:17:43.939270][RIGCTRL][debug] () : rig_register: rig_register (1010)
warning: [16:17:43.939270][RIGCTRL][debug] () : rig_register called
warning: [16:17:43.940270][RIGCTRL][debug] () : rig_register: rig_register (1005)
[... 747 lines cut ...]
warning: [16:17:54.876142][RIGCTRL][trace] () : newcat_valid_command FA
warning: [16:17:54.876142][RIGCTRL][trace] () : cmd_str = FA;
warning: [16:17:54.876142][RIGCTRL][trace] () : rig_flush: called for serial device
warning: [16:17:54.876142][RIGCTRL][debug] () : serial_flush called
warning: [16:17:54.876142][RIGCTRL][debug] () : tcflush
warning: [16:17:54.877141][RIGCTRL][trace] () : cmd_str = FA;
warning: [16:17:54.877141][RIGCTRL][debug] () : write_block called
warning: [16:17:54.883146][RIGCTRL][trace] () : write_block(): TX 3 bytes
warning: [16:17:54.973397][RIGCTRL][trace] () : 0000    46 41 3b                                            FA;
warning: [16:17:54.973397][RIGCTRL][trace] () : read_string called, rxmax=129
warning: [16:17:54.974398][RIGCTRL][trace] () : read_string(): RX 11 characters
warning: [16:17:54.974398][RIGCTRL][trace] () : 0000    46 41 31 30 31 33 36 30 35 35 3b                    FA10136055;
warning: [16:17:54.974398][RIGCTRL][trace] () : newcat_get_cmd: read count = 11, ret_data = FA10136055;
warning: [16:17:54.974398][RIGCTRL][trace] () : newcat_get_freq: freq = 10136055.000000 Hz for vfo VFOA
[... cut ...]
warning: [16:17:54.976401][RIGCTRL][trace] () : rig_get_freq: cache reset age=999000ms, vfo=currVFO, freq=10136055
warning: [16:17:54.976401][RIGCTRL][trace] () : elapsed_ms: start = 0,0
warning: [16:17:54.976401][RIGCTRL][trace] () : elapsed_ms: after gettime, start = 1610641074,976401300
warning: [16:17:54.976401][RIGCTRL][trace] () : elapsed_ms: start = 0,0
warning: [16:17:54.976401][RIGCTRL][trace] () : elapsed_ms: after gettime, start = 1610641074,976401300
warning: [16:17:54.977401][RIGCTRL][trace] () : elapsed_ms: start = 0,0
warning: [16:17:54.977401][RIGCTRL][trace]

mdblack98, 2021-01-14:
If there's an attachment to that email it's not showing up.Can you email me the attachment separately?
I'm thinking of changing the Yaesu and Kenwood retry logic completely.  We're using "ID;" as the verification command.
What would be better would be value verifcation.For example;
Rather thanFA14074000;ID;
We do thisFA14074000;FA;If we don't get the same frequency back in the correct "FA14074000" we repeat after a short timeout....maybe 50ms or so and do 10 triies for around 500ms of effort total.  Higher level retries would still do their thing...this would be it's own internal retry mechanism.  So you wouild still get 3 retries on top of this but highly unlikely to need it.
So all set commands would ask for verification if possible
This would do three things#1 Verifies the command actually worked#2 Eliminates any need for example for ?; response#3 Eliminates long read timeouts as long as the rig is talking at all.

 
 

r
Starting program: c:\test-install\wsjtx-dev\Debug\bin\wsjtx.exe -r test 
[New Thread 12888.0x2ad8]
[New Thread 12888.0x24c]
[New Thread 12888.0x1e90]
warning: [16:17:41.619661][SYSLOG][info] C:\Users\bill\src\k1jt\wsjtx\WSJTXLogging.cpp(161) WSJTXLogging: Read logging configuration file: C:/Users/bill/AppData/Local/WSJT-X - test/wsjtx_log_config.ini
warning: [16:17:41.619661][SYSLOG][info] C:\Users\bill\src\k1jt\wsjtx\WSJTXLogging.cpp(230) WSJTXLogging: Log Start
warning: [16:17:41.620662][SYSLOG][info] C:\Users\bill\src\k1jt\wsjtx\main.cpp(215) main: WSJT-X - test v2.3.0-rc4 b741eb by K1JT, G4WJS, and K9AN - Program startup
warning: [16:17:41.633671][SYSLOG][info] C:\Users\bill\src\k1jt\wsjtx\L10nLoader.cpp(70) L10nLoader: locale: language: English script: Latin country: United Kingdom ui-languages: en-GB, en-US
warning: [16:17:41.634672][SYSLOG][info] C:\Users\bill\src\k1jt\wsjtx\L10nLoader.cpp(87) L10nLoader: Loaded Qt translations for current locale from resources
warning: [16:17:41.634672][SYSLOG][info] C:\Users\bill\src\k1jt\wsjtx\L10nLoader.cpp(113) L10nLoader: Loaded WSJT-X base translation file from :/Translations based on language en
warning: [16:17:41.634672][SYSLOG][info] C:\Users\bill\src\k1jt\wsjtx\L10nLoader.cpp(123) L10nLoader: Loaded WSJT-X translations for current locale from resources
warning: [16:17:43.380898][SYSLOG][info] C:\Users\bill\src\k1jt\wsjtx\main.cpp(400) main: shmem size:48275456
[New Thread 12888.0x369c]
warning: [16:17:43.935268][RIGCTRL][debug] () : initrigs4_dummy: _init called
warning: [16:17:43.936268][RIGCTRL][debug] () : rig_register called
warning: [16:17:43.936268][RIGCTRL][debug] () : rig_register: rig_register (1)
warning: [16:17:43.936268][RIGCTRL][debug] () : rig_register called
warning: [16:17:43.936268][RIGCTRL][debug] () : rig_register: rig_register (2)
warning: [16:17:43.936268][RIGCTRL][debug] () : rig_register called
warning: [16:17:43.937268][RIGCTRL][debug] () : rig_register: rig_register (4)
warning: [16:17:43.937268][RIGCTRL][debug] () : rig_register called
warning: [16:17:43.937268][RIGCTRL][debug] () : rig_register: rig_register (5)
warning: [16:17:43.937268][RIGCTRL][debug] () : rig_register called
warning: [16:17:43.938270][RIGCTRL][debug] () : rig_register: rig_register (6)
warning: [16:17:43.938270][RIGCTRL][debug] () : yaesu: initrigs4_yaesu called
warning: [16:17:43.938270][RIGCTRL][debug] () : rig_register called
warning: [16:17:43.938270][RIGCTRL][debug] () : rig_register: rig_register (1021)
warning: [16:17:43.938270][RIGCTRL][debug] () : rig_register called
warning: [16:17:43.939270][RIGCTRL][debug] () : rig_register: rig_register (1027)
warning: [16:17:43.939270][RIGCTRL][debug] () : rig_register called
warning: [16:17:43.939270][RIGCTRL][debug] () : rig_register: rig_register (1010)
warning: [16:17:43.939270][RIGCTRL][debug] () : rig_register called
warning: [16:17:43.940270][RIGCTRL][debug] () : rig_register: rig_register (1005)
warning: [16:17:43.940270][RIGCTRL][debug] () : rig_register called
warning: [16:17:43.940270][RIGCTRL][debug] () : rig_register: rig_register (1006)
warning: [16:17:43.940270][RIGCTRL][debug] () : rig_register called
warning: [16:17:43.940270][RIGCTRL][debug] () : rig_register: rig_register (1007)
warning: [16:17:43.941272][RIGCTRL][debug] () : rig_register called
warning: [16:17:43.941272][RIGCTRL][debug] () : rig_register: rig_register (1039)
warning: [16:17:43.941272][RIGCTRL][debug] () : rig_register called
warning: [16:17:43.941272][RIGCTRL][debug] () : rig_register: rig_register (1009)
warning: [16:17:43.941272][RIGCTRL][debug] () : rig_register called
warning: [16:17:43.942273][RIGCTRL][debug] () : rig_register: rig_register (1020)
warning: [16:17:43.942273][RIGCTRL][debug] () : rig_register called
warning: [16:17:43.942273][RIGCTRL][debug] () : rig_register: rig_register (1001)
warning: [16:17:43.942273][RIGCTRL][debug] () : rig_register called
warning: [16:17:43.943273][RIGCTRL][debug] () : rig_register: rig_register (1022)
warning: [16:17:43.943273][RIGCTRL][debug] () : rig_register called
warning: [16:17:43.943273][RIGCTRL][debug] () : rig_register: rig_register (1023)
warning: [16:17:43.943273][RIGCTRL][debug] () : rig_register called
warning: [16:17:43.943273][RIGCTRL][debug] () : rig_register: rig_register (1011)
warning: [16:17:43.944273][RIGCTRL][debug] () : rig_register called
warning: [16:17:43.944273][RIGCTRL][debug] () : rig_register: rig_register (1015)
warning: [16:17:43.944273][RIGCTRL][debug] () : rig_register called
[... cut ...]
warning: [16:17:54.976401][RIGCTRL][trace] () : rig_get_freq: cache reset age=999000ms, vfo=currVFO, freq=10136055
warning: [16:17:54.976401][RIGCTRL][trace] () : elapsed_ms: start = 0,0
warning: [16:17:54.976401][RIGCTRL][trace] () : elapsed_ms: after gettime, start = 1610641074,976401300
warning: [16:17:54.976401][RIGCTRL][trace] () : elapsed_ms: start = 0,0
warning: [16:17:54.976401][RIGCTRL][trace] () : elapsed_ms: after gettime, start = 1610641074,976401300
warning: [16:17:54.977401][RIGCTRL][trace] () : elapsed_ms: start = 0,0
warning: [16:17:54.977401][RIGCTRL][trace]

mikaelnousiainen, 2021-01-14:
@g4wjs It has been a bit difficult to follow the conversation between the debug logs, so I'm not sure if this was covered while you were talking about VFO / band changes.

The FT-2000 rig I used (via remote access) for improving its Hamlib backend has one weird issue: Whenever I try to switch to VFOB, the rig stays on VFOB only for a couple of seconds and the switches back to VFOA. I was using Hamlib on command line with debug, so I'm sure nothing else was sending commands the radio between the commands I sent from Hamlib.

The operator also said that indicator "ANT1" was blinking on the radio display before it switches back to VFOA -- I'm not sure what is going on here. Would you happen to know?

g4wjs, 2021-01-14:
On 14/01/2021 22:20, Mikael Nousiainen wrote:
> @g4wjs <https://github.com/g4wjs> It has been a bit difficult to
> follow the conversation between the debug logs, so I'm not sure if
> this was covered while you were talking about VFO / band [...]
Hi Mikael,

I don't have direct access to the FT-2000 but I believe it works like this.

Press the 'B' button and it goes into VFO B setting mode for 5 seconds. 
While that happens one of the LEDs flashes orange. I would guess the 
VS1; command is doing the same thing and CAT commands for VFO B have to 
be sent within that 5 second period.

I believe the flashing ANT 1 on the display is a reminder that it is the 
VFO B aerial, assumedly for Tx, as the selected Rx aerial is common to 
VFO A and VFO B since the FT-2000 has a single common front end as far 
as the first mixer.

The FTdx500, and FTdx101 as well probably, have genuine independent main 
and sub receivers so may have a different mechanism for the VS command.

73
Bill
G4WJS.

mdblack98, 2021-01-14:
I implemented a new retry mechanism.  It tries to validate set commands with the get command to verify that the set actually worked.So we don' really care if what answer we get as we expect to see that the set cmd worked.  Any other answer is just ignored.50ms sleep between each attempt.  May need to be longer like 100ms.  5 tries are done for each command so along with 3 overhead retries give 15 tries taking 750ms+overhead or so worse case.
Let's see how it works.
Mike W9MDB

mdblack98, 2021-01-17:
Retries are working so closing this one.

closed 2021-01-17 by mdblack98