Hamlib/Hamlib #511: Yaesu stop bits change to 2

mdblack98, 2021-01-16:
It appears all Yaesu rigs should be using 2 stop bits.
So will be changing to default to 2 stop bits in 4.2

closed 2021-01-17 by mdblack98

mikaelnousiainen, 2021-01-17:
@mdblack98 I wonder how did the CAT commands still were functioning properly with 1 stop bit? Is there official documentation for this?

mdblack98, 2021-01-17:
It's a crap shoot....will frequently work with 1 stop bit.
Bill was testing an FT2000 over a remote serial port and getting a lot of retries until he changed to 2 stop bits on the whole serial chain.
Mike

mikaelnousiainen, 2021-01-17:
@mdblack98 That's interesting, great that you were able to figure this out. I'll have to test this with an FT-991, as all commands have been very slow to execute on that rig.

mdblack98, 2021-01-17:
If it's still slow please send a debug log.
The retries might explain the slowness...
Mike