Hamlib/Hamlib #523: compilation on MinGW fails with undefined reference to 'debugmsgsave'

N0NB, 2021-01-30:
MinGW 32 bit build fails as follows:

```
/usr/bin/i686-w64-mingw32-ld: rigctl-rigctl.o: in function `snprintf':
/usr/share/mingw-w64/include/stdio.h:790: undefined reference to `debugmsgsave'
/usr/bin/i686-w64-mingw32-ld: rigctl-rigctl.o: in function `main':
/home/nate/builds/hamlib-4.1~git/tests/rigctl.c:436: undefined reference to `debugmsgsave'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.1~git/tests/rigctl.c:437: undefined reference to `debugmsgsave'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.1~git/tests/rigctl.c:554: undefined reference to `debugmsgsave'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.1~git/tests/rigctl.c:620: undefined reference to `debugmsgsave'
/usr/bin/i686-w64-mingw32-ld: rigctl-rigctl.o:/home/nate/builds/hamlib-4.1~git/tests/rigctl.c:626: more undefined references to `debugmsgsave' follow
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:991: rigctl.exe] Error 1
make[2]: *** Waiting for unfinished jobs....
/usr/bin/i686-w64-mingw32-ld: rigctld-rigctld.o: in function `snprintf':
/usr/share/mingw-w64/include/stdio.h:790: undefined reference to `debugmsgsave'
/usr/bin/i686-w64-mingw32-ld: rigctld-rigctld.o: in function `sync_callback':
/home/nate/builds/hamlib-4.1~git/tests/rigctld.c:157: undefined reference to `debugmsgsave'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.1~git/tests/rigctld.c:161: undefined reference to `debugmsgsave'
/usr/bin/i686-w64-mingw32-ld: rigctld-rigctld.o: in function `handle_error':
/home/nate/builds/hamlib-4.1~git/tests/rigctld.c:218: undefined reference to `debugmsgsave'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.1~git/tests/rigctld.c:223: undefined reference to `debugmsgsave'
/usr/bin/i686-w64-mingw32-ld: rigctld-rigctld.o:/home/nate/builds/hamlib-4.1~git/tests/rigctld.c:171: more undefined references to `debugmsgsave' follow
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:999: rigctld.exe] Error 1
/usr/bin/i686-w64-mingw32-ld: rigmem-rigmem.o: in function `snprintf':
/usr/share/mingw-w64/include/stdio.h:790: undefined reference to `debugmsgsave'
/usr/bin/i686-w64-mingw32-ld: rigmem-rigmem.o: in function `main':
/home/nate/builds/hamlib-4.1~git/tests/rigmem.c:236: undefined reference to `debugmsgsave'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.1~git/tests/rigmem.c:237: undefined reference to `debugmsgsave'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.1~git/tests/rigmem.c:320: undefined reference to `debugmsgsave'
/usr/bin/i686-w64-mingw32-ld: rigmem-sprintflst.o: in function `snprintf':
/usr/share/mingw-w64/include/stdio.h:790: undefined reference to `debugmsgsave'
/usr/bin/i686-w64-mingw32-ld: rigmem-sprintflst.o:/home/nate/builds/hamlib-4.1~git/tests/sprintflst.c:44: more undefined references to `debugmsgsave' follow
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:1007: rigmem.exe] Error 1
/usr/bin/i686-w64-mingw32-ld: rigsmtr.o: in function `snprintf':
/usr/share/mingw-w64/include/stdio.h:790: undefined reference to `debugmsgsave'
/usr/bin/i686-w64-mingw32-ld: rigsmtr.o: in function `main':
/home/nate/builds/hamlib-4.1~git/tests/rigsmtr.c:245: undefined reference to `debugmsgsave'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.1~git/tests/rigsmtr.c:246: undefined reference to `debugmsgsave'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:1011: rigsmtr.exe] Error 1
make[2]: Leaving directory '/home/nate/builds/hamlib-4.1~git/tests'
make[1]: *** [Makefile:1783: install] Error 2
make[1]: Leaving directory '/home/nate/builds/hamlib-4.1~git/tests'
make: *** [Makefile:602: install-recursive] Error 1
```

MinGW 64 bit is less verbose:

```
/usr/bin/x86_64-w64-mingw32-ld: rigctld-rigctld.o:rigctld.c:(.rdata$.refptr.debugmsgsave[.refptr.debugmsgsave]+0x0): undefined reference to `debugmsgsave'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:999: rigctld.exe] Error 1
make[2]: *** Waiting for unfinished jobs....
/usr/bin/x86_64-w64-mingw32-ld: rigctl-rigctl.o:rigctl.c:(.rdata$.refptr.debugmsgsave[.refptr.debugmsgsave]+0x0): undefined reference to `debugmsgsave'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:991: rigctl.exe] Error 1
/usr/bin/x86_64-w64-mingw32-ld: rigmem-rigmem.o:rigmem.c:(.rdata$.refptr.debugmsgsave[.refptr.debugmsgsave]+0x0): undefined reference to `debugmsgsave'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:1007: rigmem.exe] Error 1
make[2]: Leaving directory '/home/nate/builds/hamlib-4.1~git/tests'
make[1]: *** [Makefile:1783: install] Error 2
make[1]: Leaving directory '/home/nate/builds/hamlib-4.1~git/tests'
make: *** [Makefile:602: install-recursive] Error 1
```

mdblack98, 2021-01-30:
Please do a "make clean" first and then if that doesn't work a fresh clone.Sounds like yours is missing something.Works here...
Sounds like include/hamlib/rig.h did not get updated in your copyIt should have#define DEBUGMSGSAVE_SIZE 16384extern char debugmsgsave[DEBUGMSGSAVE_SIZE];  // last debug msg
And src/rig.c should havechar debugmsgsave[DEBUGMSGSAVE_SIZE];

N0NB, 2021-01-30:
* On 2021 30 Jan 10:05 -0600, Michael Black wrote:
> Please do a "make clean" first and then if that doesn't work a fresh
> clone.  Sounds like yours is missing something.

> Works here...  Sounds like include/hamlib/rig.h did not get updated in
> your copy

This was from the tarball that was newly cloned from the master branch
this morning.  If you'd like to look at it, it is at:

http://n0nb.users.sourceforge.net/hamlib-4.1~git-cd7811f3-20210130.tar.gz

> It should have:
> #define DEBUGMSGSAVE_SIZE 16384
> extern char debugmsgsave[DEBUGMSGSAVE_SIZE];  // last debug msg

It does, lines 2858 and 2859.

> And src/rig.c should have:
> char debugmsgsave[DEBUGMSGSAVE_SIZE];

It does, line 293.

My script saves the generated tarball locally that is uploaded each
morning and then the next scripts do the MinGW 32/64 bit builds from
that tarball and between runs the build directory is deleted and the
tarball extracted anew for the 64 bit build so each one builds clean and
independent of the other.

I just did a pull again and the HEAD matches the tarball SHA1 of
cd7811f3.

73, Nate

-- 

"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."

Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

mdblack98, 2021-01-30:
That tarball compiled just fine on MINGW64 here.
nm rig.o | grep debugmsgsave0000000000000000 B debugmsgsave
 nm rigctld-rigctld.o|grep debugmsgsave0000000000000000 r .rdata$.refptr.debugmsgsave0000000000000000 R .refptr.debugmsgsave                 U debugmsgsave

.Mike

 
 

> Please do a "make clean" first and then if that doesn't work a fresh
> clone. Sounds like yours is missing something.

> Works here... Sounds like include/hamlib/rig.h did not get updated in
> your copy

This was from the tarball that was newly cloned from the master branch
this morning. If you'd like to look at it, it is at:

http://n0nb.users.sourceforge.net/hamlib-4.1~git-cd7811f3-20210130.tar.gz

> It should have:
> #define DEBUGMSGSAVE_SIZE 16384
> extern char debugmsgsave[DEBUGMSGSAVE_SIZE];  // last debug msg

It does, lines 2858 and 2859.

> And src/rig.c should have:
> char debugmsgsave[DEBUGMSGSAVE_SIZE];

It does, line 293.

My script saves the generated tarball locally that is uploaded each
morning and then the next scripts do the MinGW 32/64 bit builds from
that tarball and between runs the build directory is deleted and the
tarball extracted anew for the 64 bit build so each one builds clean and
independent of the other.

I just did a pull again and the HEAD matches the tarball SHA1 of
cd7811f3.

73, Nate

N0NB, 2021-01-30:
It looks like my MinGW 32 bit (Debian Buster) is getting confused, note the second line of output where it is referencing `/usr/share/mingw-w64/include/stdio.h`:

```
/usr/bin/i686-w64-mingw32-ld: rigmem-rigmem.o: in function `snprintf':
/usr/share/mingw-w64/include/stdio.h:790: undefined reference to `debugmsgsave'
/usr/bin/i686-w64-mingw32-ld: rigmem-rigmem.o: in function `main':
/home/nate/builds/hamlib-4.1~git/tests/rigmem.c:236: undefined reference to `debugmsgsave'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.1~git/tests/rigmem.c:237: undefined reference to `debugmsgsave'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.1~git/tests/rigmem.c:320: undefined reference to `debugmsgsave'
/usr/bin/i686-w64-mingw32-ld: rigmem-sprintflst.o: in function `snprintf':
/usr/share/mingw-w64/include/stdio.h:790: undefined reference to `debugmsgsave'
/usr/bin/i686-w64-mingw32-ld: rigmem-sprintflst.o:/home/nate/builds/hamlib-4.1~git/tests/sprintflst.c:44: more undefined references to `debugmsgsave' follow
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:1007: rigmem.exe] Error 1
```

The version of MinGW 32 bit on Debian Buster:

```
$ i686-w64-mingw32-gcc --version
i686-w64-mingw32-gcc (GCC) 8.3-win32 20190406
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

```
$ /usr/bin/i686-w64-mingw32-ld --version
GNU ld (GNU Binutils) 2.31.1
Copyright (C) 2018 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.
```

mdblack98, 2021-01-30:
It sounds like it's not linking the new library.  The library should contain that defintion.
Mike W9MDB

N0NB, 2021-01-31:
Mike fixed this with commit 9d302051

closed 2021-01-31 by N0NB