Hamlib/Hamlib #526: Stray `dummy_get_channel called` in rigerror

df7cb, 2021-02-04:
Seen in 4.1 and git head, looks like a leftover from debugging:

```
perl /usr/share/doc/libhamlib-perl/examples/perltest.pl
get_conf:		path = "/dev/Rig", retry = 50
freq:			14266000
get_mode:		FM
width:			15000
get_vfo:		VFOA
Backend copyright:	LGPL
Model:			Dummy
Manufacturer:		Hamlib
Backend version:	20200606.0
get_info:		Nothing much (dummy)
VOX delay:		0
VOX delay:		5
get_channel status:	0 = Command completed successfully
dummy_get_channel called                             <--------------- there
VFO:			VFOA, 14266000
Attenuators:		10 20 30 0 0 0 0 0
```

(On 4.1, there's a huge amount of bogus whitespace around it, but that's fixed in head.)

mdblack98, 2021-02-04:
Not a leftover -- new rig_error return contains the last debug message now -- may soon contain the last two messages.
So I'll modify perltest.pl to allow for the new format.

closed 2021-02-04 by mdblack98

df7cb, 2021-02-04:
Thanks for the feedback!
But, how is that supposed to be useful? If some program like wsjtx gets a rig error and tells me in the popup window "whatever_get_channel called", does that help me with fixing the problem?

mdblack98, 2021-02-04:
It doesn't help you -- it helps those of us who can fix the problem.
I'm still working on gettting more useful messages -- it's a work in progress.