Hamlib/Hamlib #541: Support for Yaesu FT-7800/7900

TuxedoHam, 2021-02-10:
I tried to reprogram my Yaesu FT-7800 with Chirp to find out under hamlib 4.0 does not support it anymore.

$ rigctl --version
rigctl(d), Hamlib 4.0

Copyright (C) 2000-2012 Stephane Fillod
Copyright (C) 2000-2003 Frank Singleton
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

mdblack98, 2021-02-10:
What do you mean it doesn't support it?The rig numbers did change if that's what you missed.
I don't see where anybody has tested the 7800.  Hopefully the 7900 is the same.  Which one do you have?
rigctl -l | grep 7800  3056  Icom                   IC-7800                 20210203.0      Untested 
Mike W9MDB

TuxedoHam, 2021-02-10:
I have a Yaesu FT-7800 and it no longer in hamlib under Debian, so I can't use rigctl etc as of 4.0

N0NB, 2021-02-10:
* On 2021 10 Feb 00:29 -0600, TuxedoHam wrote:
> I have a Yaesu FT-7800 and it no longer in hamlib under Debian, so I
> can't use rigctl etc as of 4.0

To my knowledge mobile radios like that do not support CAT control that
Hamlib supports.  Are you sure you're not confusing Chirp with Hamlib?

Also, Hamlib does not drop support for a radio.  There is no reason to
do so.  In fact, support for the old FT-980 was just added/enhanced!
Again, VHF/UHF FM radios typically lack the kind of CAT protocol that
HF/multi-mode VHF/UHF radios have so the only option in the Free
Software world is Chirp which does memory writes into the EEPROM.

73, Nate

-- 

"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."

Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

TuxedoHam, 2021-02-10:
N0NB, oh thanks, I checked my desktop that not been upgraded for a bit so I checked to rigctl on there and it does not have the FT-7800.

sorry for the mix up, it has to be a problem with "Chirp" rather than "Hamlib", I thinking that "Chirp" relied on "Hamlib" for programming radios

Cheers & 73 VK5ZSH

closed 2021-02-10 by TuxedoHam

N0NB, 2021-02-10:
* On 2021 10 Feb 04:55 -0600, TuxedoHam wrote:
> N0NB, oh thanks, I checked my desktop that not been upgraded for a bit
> so I checked to rigctl on there and it does not have the FT-7800.
> 
> sorry for the mix up, it has to be a problem with "Chirp" rather than
> "Hamlib", I thinking that "Chirp" relied on "Hamlib" for programming
> radios

That's quite all right.  Good luck on getting Chirp sorted.  I've used
it with a couple of radios so I'm a user of it too.

73, Nate

-- 

"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."

Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819