Hamlib/Hamlib #546: Android version fails to build

sh123, 2021-02-14:
Trying to build with NDK version 22.0.6917172
```
$ android/hamlib-compile
build-binary.mk:654: Android NDK: Module hamlib depends on undefined modules: adat alinco amsat aor ars celestron drake dummy easycomm ether6 flexradio fodtrack gnuradio gs232a heathkit icom jrc kachina kenwood kit kylix lowe m2 microtune minivna pcr prm80 racal rft rotorez rs sartek skanti spid tapr tentec ts7400 tuner uniden winradio wj yaesu
```

mdblack98, 2021-02-16:
Try a pull and compile again.  Looks like the paths needed to be updated in the Android.mk file.

mdblack98, 2021-02-16:
Here's some info on Android USB.  Looks like Android 5.0 can do audio.
https://source.android.com/devices/audio/usb

sh123, 2021-02-17:
Now it gives this error:
```
build-binary.mk:654: Android NDK: Module hamlib depends on undefined modules: amsat ars celestron easycomm ether6 fodtrack gnuradio gs232a heathkit kylix m2 microtune minivna rotorez sartek spid ts7400   
```

also some warnings for adat, alinco, aor, drake, dummy, flexradio:
```
LOCAL_LDLIBS is always ignored for static libraries
non-system libraries in linker flags: -lhamlib    
Android NDK:     This is likely to result in incorrect builds. Try using LOCAL_STATIC_LIBRARIES    
Android NDK:     or LOCAL_SHARED_LIBRARIES instead to list the library dependencies of the    
Android NDK:     current module    
```

sh123, 2021-03-01:
@mdblack98 , it compiles fine now (haven't tested at run time), thank you.

closed 2021-03-01 by mdblack98