Hamlib/Hamlib #560: rigctl and rigctld need to try to reopen rig when an error occurs

mdblack98, 2021-02-22:
Right now rigctl/rigctld does not try to reopen the rig when a rig error occurs.
Commented out the reopen attempt as it was the incorrect way to do it and was causing problems with PiWeb.
So now running "rigctld -m 4" for example will no longer reconnect if the network connection to FLRig aborts for any reason.
rigctl and rigctld should both attempt to reconnect to network or serial devices when a hard device error occurs but should return an error and simply try to reconnect on each command.
e.g.
rigctl -r com3 
f (error occurs because serial port reset -- should attempt a re-open a couple of times)
if reopen does not succed return error
f (again tries to reopen rig)
f (again tried to reopen rig)

So hamlib will do 3 reopen attempts with a 1 second wait between each and hopefully succeed.  If not will return error and on next command will do 3 open attempts again....ad nauseum.

mikaelnousiainen, 2021-02-23:
@mdblack98 This is a welcome change. However, I'm thinking if the retries here are unnecessary? They will after all cause a 3-second delay after a command that fails with an I/O error. Wouldn't a single close/open combination be sufficient? After all, there are retries in the backends too and client using rigctl is able to perform retries accordingly if needed (as it knows the command failed anyway). What I'm after is giving more control of the logic and retries to the client.

mdblack98, 2021-02-23:
It's only on a hard error which should hardly ever occur -- but does -- seen lots of report of com port burps, network burps (I see those).
The internal retries are when there is no hard error.   So it will combine with those retries and then put a 3 second attempt to try and reconnect to the rig too.
Needs some testing...but this may solve the com ports that drop out due to RFI and devices that go to sleep and such.
Mike W9MDB

mdblack98, 2021-03-07:
rigctl reconnects properly now
rigctld reconnects properly now

closed 2021-03-07 by mdblack98