Hamlib/Hamlib #565: rigctl M PKTLSB does wierd things on KX3.

n9seo, 2021-02-26:
Hi,

I have been trying to nail down a wierd issue in hamlib/rigctl on KX3.  I have the latest kx3 firmware 02.95 and dsp 01.52 (taken from the KX3 menu readout).   So in short I can probably provide whatever logs you might need but lacked finding it so let me know.   

So if one sets the rig to 'data rev' mode from the rigs panel rigctl is fine this way:

```
$ rigctl -m 2045 -r /dev/ttyUSB0 m
PKTLSB
2000
```

Now if we try to set it from rigctl like :

```
$ rigctl -m 2045 -r /dev/ttyUSB0 m
PKTUSB
2000
$ rigctl -m 2045 -r /dev/ttyUSB0 M PKTLSB 2000
$ rigctl -m 2045 -r /dev/ttyUSB0 m
PKTLSB
500
```

And instead of going to data rev mode it throws the rig into Data 'AFSK 45 bps' mode instead of 'DATA A' mode set to LSB.  This is commonly used for modes like RTTY or  legacy HF Packet so it'd be nice to fix this if its possible and not something maybe lost in the translation from hamlib to CAT.  It seems to work just fine in apps like win4k3suite or nap3.  Im running linux but can test those just fine over serial from a VM or WINE.

I noticed this when I was trying to write a hardware file for quisk in python but traced it back to something wierd hamlib does.  Im not sure but I thought it was fine some time ago maybe in hamlib 2 or 3.   

This is built using :
```
$ rigctl -V
rigctl Hamlib 4.2~git
Last commit was Sun Feb 14 22:46:41 2021 +0000 SHA=003bc9

```

Not sure if it helps or not but posting this for you since Im using Manjaro and needed to test this against the latest code I modified the 'hamlib-git' AUR package to match dependancy for 'hamlib' with my PKGBUILD:

```
pkgname=hamlib
pkgver=4.2
pkgrel=1
pkgdesc="Ham radio equipment control libraries"
arch=('i686' 'x86_64' 'armv5h' 'armv6h' 'armv7h')
url="http://hamlib.org"
license=('GPL' 'LGPL')
depends=('perl' 'python' 'tcl' 'libxml2' 'libusb')
makedepends=('libtool' 'swig')
#source=('hamlib-git::git+git://git.code.sf.net/p/hamlib/code')
# testing code
source=('hamlib-git::git+https://github.com/n9seo/Hamlib.git')
options=('!emptydirs')
conflicts=('hamlib')
provides=('hamlib')
md5sums=('SKIP')
sha256sums=('SKIP')

build() {
	export PYTHON=/usr/bin/python2

	cd ${srcdir}/${pkgname}-git
	./bootstrap
	./configure \
		--prefix=/usr \
		--sbindir=/usr/bin \
		--with-perl-binding \
		--with-python-binding \
		--with-xml-support \
		--with-tcl-binding
	make
}

package() {
	cd ${srcdir}/${pkgname}-git

	make DESTDIR=$pkgdir install

	/usr/bin/find $pkgdir -name '.packlist' -delete
	/usr/bin/find $pkgdir -name '*.pod' -delete
}

```

So let me know if i can help, i can test.

n9seo, 2021-02-26:
What might make this interesting that 'data' has submodes on kx3.   But for whatever reasons PKTUSB defaults to data mode and PKTLSB seems to always send me to afsk 45bps.

mdblack98, 2021-02-26:
Please add "-vvvvv -Z" and send the debug log.
Mike W9MDB

mdblack98, 2021-02-26:
Also make sure you have the latest pull from the master repo.
I just made some significant changes today.
Mike W9MDB

n9seo, 2021-02-26:
Hi,
Just this morning I rebuilt it using commit id f6fd0e887b4f4ff9fd283972c583bb8241076dd8 of master.   

```
$ rigctl -vvvvv -Z -m 2045 -r /dev/ttyUSB0 m
2021-02-26:12:07:30.035943: rigctl Hamlib 4.2~git
Last commit was Fri Feb 26 05:48:32 2021 +0000 SHA=f6fd0e
2021-02-26:12:07:30.035996: Report bugs to <hamlib-developer@lists.sourceforge.net>

2021-02-26:12:07:30.036010: rig.c(365):rig_init entered
2021-02-26:12:07:30.036022: rig_check_rig_caps: p1=0x7f5a2c02f320, p2=0x7f5a2c035690, rig_model=0x7f5a2c02f320, macro_name=0x7f5a2c035690
2021-02-26:12:07:30.036080: initrigs4_kenwood called
2021-02-26:12:07:30.036099: rig_register called
2021-02-26:12:07:30.036112: rig_register: rig_register (2012)
2021-02-26:12:07:30.036131: rig_register called
2021-02-26:12:07:30.036146: rig_register: rig_register (2013)
2021-02-26:12:07:30.036161: rig_register called
2021-02-26:12:07:30.036241: rig_register: rig_register (2001)
2021-02-26:12:07:30.036258: rig_register called
2021-02-26:12:07:30.036276: rig_register: rig_register (2025)
2021-02-26:12:07:30.036287: rig_register called
2021-02-26:12:07:30.036306: rig_register: rig_register (2003)
2021-02-26:12:07:30.036321: rig_register called
2021-02-26:12:07:30.036333: rig_register: rig_register (2004)
2021-02-26:12:07:30.036350: rig_register called
2021-02-26:12:07:30.036363: rig_register: rig_register (2016)
2021-02-26:12:07:30.036378: rig_register called
2021-02-26:12:07:30.036398: rig_register: rig_register (2024)
2021-02-26:12:07:30.036410: rig_register called
2021-02-26:12:07:30.036424: rig_register: rig_register (2005)
2021-02-26:12:07:30.036434: rig_register called
2021-02-26:12:07:30.036443: rig_register: rig_register (2007)
2021-02-26:12:07:30.036452: rig_register called
2021-02-26:12:07:30.036461: rig_register: rig_register (2009)
2021-02-26:12:07:30.036471: rig_register called
2021-02-26:12:07:30.036480: rig_register: rig_register (2010)
2021-02-26:12:07:30.036496: rig_register called
2021-02-26:12:07:30.036507: rig_register: rig_register (2022)
2021-02-26:12:07:30.036518: rig_register called
2021-02-26:12:07:30.036533: rig_register: rig_register (2014)
2021-02-26:12:07:30.036544: rig_register called
2021-02-26:12:07:30.036557: rig_register: rig_register (2030)
2021-02-26:12:07:30.036572: rig_register called
2021-02-26:12:07:30.036585: rig_register: rig_register (2021)
2021-02-26:12:07:30.036598: rig_register called
2021-02-26:12:07:30.036610: rig_register: rig_register (2029)
2021-02-26:12:07:30.036623: rig_register called
2021-02-26:12:07:30.036636: rig_register: rig_register (2043)
2021-02-26:12:07:30.036649: rig_register called
2021-02-26:12:07:30.036667: rig_register: rig_register (2044)
2021-02-26:12:07:30.036681: rig_register called
2021-02-26:12:07:30.036692: rig_register: rig_register (2045)
2021-02-26:12:07:30.036701: rig_register called
2021-02-26:12:07:30.036713: rig_register: rig_register (2047)
2021-02-26:12:07:30.036730: rig_register called
2021-02-26:12:07:30.036746: rig_register: rig_register (2038)
2021-02-26:12:07:30.036760: rig_register called
2021-02-26:12:07:30.036775: rig_register: rig_register (2002)
2021-02-26:12:07:30.036790: rig_register called
2021-02-26:12:07:30.036803: rig_register: rig_register (2011)
2021-02-26:12:07:30.036816: rig_register called
2021-02-26:12:07:30.036829: rig_register: rig_register (2006)
2021-02-26:12:07:30.036844: rig_register called
2021-02-26:12:07:30.036855: rig_register: rig_register (2008)
2021-02-26:12:07:30.036866: rig_register called
2021-02-26:12:07:30.036878: rig_register: rig_register (2015)
2021-02-26:12:07:30.036890: rig_register called
2021-02-26:12:07:30.036906: rig_register: rig_register (2026)
2021-02-26:12:07:30.036922: rig_register called
2021-02-26:12:07:30.036937: rig_register: rig_register (2017)
[... 1443 lines cut ...]
2021-02-26:12:28:27.971165: 0000    49 44 3b                                            ID;             
2021-02-26:12:28:27.971176: read_string called, rxmax=35
2021-02-26:12:28:27.986660: read_string(): RX 6 characters
2021-02-26:12:28:27.986694: 0000    49 44 30 31 37 3b                                   ID017;          
2021-02-26:12:28:27.986706: kenwood_transaction: read_string(len=6)='ID017;'
2021-02-26:12:28:27.986718: kenwood_transaction: No data expected, checking ID; in ID017;
2021-02-26:12:28:27.986729: kenwood_transaction: returning RIG_OK, retval=0
2021-02-26:12:28:27.986756: kenwood_transaction: returning retval=0
2021-02-26:12:28:27.986769: kenwood.c(556):kenwood_transaction return
2021-02-26:12:28:27.986784: kenwood.c(967):kenwood_close return
2021-02-26:12:28:27.986797: port_close called
2021-02-26:12:28:27.986811: ser_close called
2021-02-26:12:28:27.986825: ser_close: restoring options
2021-02-26:12:28:27.987003: serial.c(817):ser_close return
2021-02-26:12:28:27.987030: rig.c(236):remove_opened_rig return
2021-02-26:12:28:27.987042: rig.c(1196):rig_close return
2021-02-26:12:28:27.987053: rig.c(1215):rig_cleanup entered
2021-02-26:12:28:27.987064: kenwood_cleanup called
2021-02-26:12:28:27.987074: kenwood.c(727):kenwood_cleanup return
2021-02-26:12:28:27.987086: rig.c(1240):rig_cleanup return
```

I hope we got expected details for you.   Thanks for reading.

-k

n9seo, 2021-02-26:
Hi My guess reading this is that pktlsb wants to use data RTTY submode in hamlib so maybe my expectations are off.  Not sure if it was meant to handle the submodes?

n9seo, 2021-02-26:
Hi,

I think this may be helpful I dug this out the KX3 programmer reference: https://ftp.elecraft.com/KX3/Manuals%20Downloads/K3S&K3&KX3&KX2%20Pgmrs%20Ref,%20G4.pdf

Look at "MD $"  where it states Data submode id is '6' and data-rev = '9' in the raw CAT command.

mdblack98, 2021-02-26:
PKTLSB does AFSK A mode for the K3 series.
    case RIG_MODE_PKTLSB:        mode = RIG_MODE_RTTY;        snprintf(cmd_m, sizeof(cmd_m),                 "DT1"); /* AFSK A mode - AFSK on LSB optimised for RTTY, VFO dial is MARK */        break;
    case RIG_MODE_PKTUSB:        mode = RIG_MODE_RTTY;        snprintf(cmd_m, sizeof(cmd_m),                 "DT0"); /* DATA A mode - AFSK on USB general, VFO dial is suppressed carrier QRG */

Mike W9MDB

mdblack98, 2021-02-26:
Do another pull and try again please.
Made some more changes to try and get the VFO handling correct.
Mike W9MDB

 

    On Friday, February 26, 2021, 06:30:34 AM CST, n9seo <notifications@github.com> wrote:  
 
 

Hi,
Just this morning I rebuilt it using commit id f6fd0e8 of master.
$ rigctl -vvvvv -Z -m 2045 -r /dev/ttyUSB0 m
2021-02-26:12:07:30.035943: rigctl Hamlib 4.2~git
Last commit was Fri Feb 26 05:48:32 2021 +0000 SHA=f6fd0e
2021-02-26:12:07:30.035996: Report bugs to <hamlib-developer@lists.sourceforge.net>

2021-02-26:12:07:30.036010: rig.c(365):rig_init entered
2021-02-26:12:07:30.036022: rig_check_rig_caps: p1=0x7f5a2c02f320, p2=0x7f5a2c035690, rig_model=0x7f5a2c02f320, macro_name=0x7f5a2c035690
2021-02-26:12:07:30.036080: initrigs4_kenwood called
2021-02-26:12:07:30.036099: rig_register called
2021-02-26:12:07:30.036112: rig_register: rig_register (2012)
2021-02-26:12:07:30.036131: rig_register called
2021-02-26:12:07:30.036146: rig_register: rig_register (2013)
2021-02-26:12:07:30.036161: rig_register called
2021-02-26:12:07:30.036241: rig_register: rig_register (2001)
2021-02-26:12:07:30.036258: rig_register called
2021-02-26:12:07:30.036276: rig_register: rig_register (2025)
2021-02-26:12:07:30.036287: rig_register called
2021-02-26:12:07:30.036306: rig_register: rig_register (2003)
2021-02-26:12:07:30.036321: rig_register called
2021-02-26:12:07:30.036333: rig_register: rig_register (2004)
2021-02-26:12:07:30.036350: rig_register called
2021-02-26:12:07:30.036363: rig_register: rig_register (2016)
2021-02-26:12:07:30.036378: rig_register called
2021-02-26:12:07:30.036398: rig_register: rig_register (2024)
2021-02-26:12:07:30.036410: rig_register called
2021-02-26:12:07:30.036424: rig_register: rig_register (2005)
2021-02-26:12:07:30.036434: rig_register called
2021-02-26:12:07:30.036443: rig_register: rig_register (2007)
2021-02-26:12:07:30.036452: rig_register called
2021-02-26:12:07:30.036461: rig_register: rig_register (2009)
2021-02-26:12:07:30.036471: rig_register called
2021-02-26:12:07:30.036480: rig_register: rig_register (2010)
2021-02-26:12:07:30.036496: rig_register called
2021-02-26:12:07:30.036507: rig_register: rig_register (2022)
2021-02-26:12:07:30.036518: rig_register called
2021-02-26:12:07:30.036533: rig_register: rig_register (2014)
2021-02-26:12:07:30.036544: rig_register called
2021-02-26:12:07:30.036557: rig_register: rig_register (2030)
2021-02-26:12:07:30.036572: rig_register called
2021-02-26:12:07:30.036585: rig_register: rig_register (2021)
2021-02-26:12:07:30.036598: rig_register called
2021-02-26:12:07:30.036610: rig_register: rig_register (2029)
2021-02-26:12:07:30.036623: rig_register called
2021-02-26:12:07:30.036636: rig_register: rig_register (2043)
2021-02-26:12:07:30.036649: rig_register called
2021-02-26:12:07:30.036667: rig_register: rig_register (2044)
2021-02-26:12:07:30.036681: rig_register called
2021-02-26:12:07:30.036692: rig_register: rig_register (2045)
2021-02-26:12:07:30.036701: rig_register called
2021-02-26:12:07:30.036713: rig_register: rig_register (2047)
2021-02-26:12:07:30.036730: rig_register called
2021-02-26:12:07:30.036746: rig_register: rig_register (2038)
2021-02-26:12:07:30.036760: rig_register called
2021-02-26:12:07:30.036775: rig_register: rig_register (2002)
2021-02-26:12:07:30.036790: rig_register called
2021-02-26:12:07:30.036803: rig_register: rig_register (2011)
2021-02-26:12:07:30.036816: rig_register called
2021-02-26:12:07:30.036829: rig_register: rig_register (2006)
[... 959 lines cut ...]
2021-02-26:12:28:26.525284: initrigs4_kenwood called
2021-02-26:12:28:26.525298: rig_register called
2021-02-26:12:28:26.525308: rig_register: rig_register (2012)
2021-02-26:12:28:26.525327: rig_register called
2021-02-26:12:28:26.525336: rig_register: rig_register (2013)
2021-02-26:12:28:26.525349: rig_register called
2021-02-26:12:28:26.525358: rig_register: rig_register (2001)
2021-02-26:12:28:26.525371: rig_register called
2021-02-26:12:28:26.525384: rig_register: rig_register (2025)
2021-02-26:12:28:26.525397: rig_register called
2021-02-26:12:28:26.525414: rig_register: rig_register (2003)
2021-02-26:12:28:26.525430: rig_register called
2021-02-26:12:28:26.525441: rig_register: rig_register (2004)
2021-02-26:12:28:26.525454: rig_register called
2021-02-26:12:28:26.525467: rig_register: rig_register (2016)
2021-02-26:12:28:26.525478: rig_register called
2021-02-26:12:28:26.525491: rig_register: rig_register (2024)
2021-02-26:12:28:26.525504: rig_register called
2021-02-26:12:28:26.525517: rig_register: rig_register (2005)
2021-02-26:12:28:26.525529: rig_register called
2021-02-26:12:28:26.525541: rig_register: rig_register (2007)
2021-02-26:12:28:26.525556: rig_register called
2021-02-26:12:28:26.525568: rig_register: rig_register (2009)
2021-02-26:12:28:26.525584: rig_register called
2021-02-26:12:28

n9seo, 2021-02-26:
Hi Mike,

I refreshed the git repo and reran the build and got this instead:

``` CCLD     libhamlib-elecraft.la
make[1]: Leaving directory '/source/aur/hamlib-git/src/hamlib-git/amplifiers/elecraft'
Making all in src
make[1]: Entering directory '/source/aur/hamlib-git/src/hamlib-git/src'
  CC       rig.lo
  CC       serial.lo
  CC       register.lo
  CC       misc.lo
  CC       event.lo
misc.c:2149:11: error: conflicting types for ‘rig_get_caps_int’
 2149 | long long rig_get_caps_int(rig_model_t rig_model, long long rig_caps)
      |           ^~~~~~~~~~~~~~~~
In file included from misc.c:53:
../include/hamlib/rig.h:1974:18: note: previous declaration of ‘rig_get_caps_int’ was here
 1974 | extern long long rig_get_caps_int(rig_model_t rig_model, enum rig_caps_int_e rig_caps);
      |                  ^~~~~~~~~~~~~~~~
make[1]: *** [Makefile:597: misc.lo] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/source/aur/hamlib-git/src/hamlib-git/src'
make: *** [Makefile:610: all-recursive] Error 1
```

mdblack98, 2021-02-26:
Yup...missed a commit...please do another pull....Mike

n9seo, 2021-02-26:
Hello
Perhaps another missed one for ftdx101mp.c?  

```
make[1]: *** No rule to make target 'ftdx101mp.c', needed by 'ftdx101mp.lo'.  Stop.
make[1]: *** Waiting for unfinished jobs....
```

mdblack98, 2021-02-26:
oops...try again...
Mike

n9seo, 2021-02-26:
Hi,

No worries, is the nature of this stuff.   That fixed compile so I will do some more poking around in this new commit and report back later.

n9seo, 2021-02-27:
Hi,
VFO errors are gone at least.   Here are some more outputs without those errors if that helps.  

Here is rigctl -vvvvv -Z -m 2045 -r /dev/ttyUSB0 m from rig being in normal data mode 
[get_mode_data.txt](https://github.com/Hamlib/Hamlib/files/6053407/get_mode_data.txt)

Here is rigctl -vvvvv -Z -m 2045 -r /dev/ttyUSB0 m from rig being set to data reverse mode 
[get_mode_data_rev.txt](https://github.com/Hamlib/Hamlib/files/6053408/get_mode_data_rev.txt)

This is rigctl -vvvvv -Z -m 2045 -r /dev/ttyUSB0 M PKTLSB 0
[set_mode_pktlsb_0.txt](https://github.com/Hamlib/Hamlib/files/6053410/set_mode_pktlsb_0.txt)

This is rigctl -vvvvv -Z -m 2045 -r /dev/ttyUSB0 m after the last cmd.
[get_mode_after_rigctl_set_mode_pktlsb_0.txt](https://github.com/Hamlib/Hamlib/files/6053412/get_mode_after_rigctl_set_mode_pktlsb_0.txt)

mdblack98, 2021-02-27:
Are you saying there's something wrong with any of those?
Mike W9MDB

n9seo, 2021-02-27:
Hi,

VFO is fine. Its just that from what I can see that set_mode doesnt do PKTLSB ala DATA REV in LSB mode with the sub mode set to DATA A.  Despite feeding it the same parameters it 'reads' state out of the radio from a manual set, it wont put it back the same way.   How does one send submodes with rigctl manually?   Surely we arent to live in a jail of 1 submode but be able to select?

n9seo, 2021-02-27:
I am really trying to get DM 9 and DT0 together I guess if that makes sense.  Not sure how I can send this.

n9seo, 2021-02-27:
ER sorry MD9.

n9seo, 2021-02-27:
Hi,

First sorry if I am communicating poorly.  Bear with me.  I usually am better at fixing things myself but not well versed in c/c++.   Hack and small changes is about my level.

Maybe this makes more sense to say this way.  I think this code here https://github.com/Hamlib/Hamlib/blob/26cfe0bb3f9d5a1f5cd0f2ae1290fb72bde18eb6/rigs/kenwood/k3.c#L1067-L1089

I do not believe that RIG_MODE_PKTLSB and RIG_MODE_RTTY are one in the same here and this should be more granular.  
Perhaps the present logic for RIG_MODE_PKTLSB should DT0 and RIG_MODE_RTTY  be DT1.  I may hack it into a test fork but not really too comfortable with making c changes.

MD
```
ET/RSP format:  MDn;or  MD$n;where nis 1 (LSB), 2 (USB), 3 (CW), 4 (FM), 5 (AM), 6 (DATA), 7 (CW-REV), or 9 (DATA-REV). Notes:(1)  K3 only: In Diversity Mode(accessed by holding SUB), sending MDn;sets both main and sub mode to n. (2) DATA and DATA-REV select the data sub-mode that was last in effect on the present band. (To read/set data sub-mode, use DT.) The norm/rev conditions for the K3’s data sub-modes are handled in two pairs at present: DATA A/PSK D, and AFSK A/FSK D. E.g., if the radio is set up for DATA A mode, alternating between MD6and MD9will cause both DATA A and  PSK D to be set to the same normal/reverse condition.
```

DT
```
11DT (DATA Sub-Mode; GET/SET)SET/RSP format:  DTn;where nis the data sub-mode last used with VFO A, whether or not DATA mode is in effect: 0 (DATA A), 1 (AFSK A), 2 (FSKD), or 3 (PSK D).  See MDfor setting data normal/reverse. In Diversity Mode(K3 only, accessed by sending  DV1or via a hold of SUB), sending DTnmatches the sub receiver’s mode to the main receiver’s. Notes:(1) Use DT only when the transceiver is in DATA mode; otherwise, the returned value may not be valid.(2) In AI2/3 modes, changing the data sub-mode results in both FWand ISresponses. (3) The present data sub-mode is also reported as part of the IFcommand, although this requires that K31be in effect. Refer to the IFcommand for details.
```

n9seo, 2021-02-27:
Hi Mike,

I got my confidence up a bit and figured it out I think.  Maybe the code can say what I failed to.

This PR i just made seems to work for me, but probably would want K3 folks to try it too.

I hope it makes sense what I did here https://github.com/Hamlib/Hamlib/pull/574

mdblack98, 2021-02-27:
There are modes available that may make more sense.
Seems these should map to DT3RIG_MODE_PSK -- MD6RIG_MODE_PSKR -- MD9
It would seem these would be DT2RIG_MODE_RTTY -- MD9RIG_MODE_RTTYR -- MD6
And these would be DT1RIG_MODE_PKTUSB -- MD6RIG_MODE_PKBLSB -- MD9
RTTYR is actually upper side band and RTTY is lower side band.  It's conceptually backwards from everything else.

mdblack98, 2021-02-27:
Trying this in plain text.
There are modes available that may make more sense.

Seems these should map to DT3
RIG_MODE_PSK -- MD6
RIG_MODE_PSKR -- MD9

It would seem these would be DT2
RIG_MODE_RTTY -- MD9
RIG_MODE_RTTYR -- MD6

And these would be DT1
RIG_MODE_PKTUSB -- MD6
RIG_MODE_PKBLSB -- MD9

RTTYR is actually upper side band and RTTY is lower side band.  It's conceptually backwards from everything else.

Mike W9MDB

n9seo, 2021-02-27:
Hi,
I think i follow.   This is backwards to what you'd think.  RTTYR (reverse) would usually be USB but it sets the rig to LSB here.  I must be missing something here but this is doing the trick for me.   I am messing around with the other missing mode types you mentioned so maybe in the coming days I will have more time to look.

```
    case RIG_MODE_PKTLSB:
        mode = RIG_MODE_RTTYR;
        snprintf(cmd_m, sizeof(cmd_m),
                 "DT0"); /* DATA A mode - DATA (REV) on LSB optimized for HF Packet, VFO dial is suppressed carrier QRG */
        break;

    case RIG_MODE_PKTUSB:
        mode = RIG_MODE_RTTY;
        snprintf(cmd_m, sizeof(cmd_m),
                 "DT0"); /* DATA A mode - DATA on USB general, VFO dial is suppressed carrier QRG */
        break;
```

n9seo, 2021-03-04:
Hi,

Since https://github.com/Hamlib/Hamlib/pull/582 was merged I found that 41bfd57eb389ef6e884f119ccc770af6acb17642 broke get_mode reporting for PKTLSB now reports falsely as PKTUSB.   Reverting to this commit fixes the reporting but the RTTY/RTTYR modes get flipped again.

mdblack98, 2021-03-04:
I assume you're fixing that?
Mike W9MDB

n9seo, 2021-03-05:
Hi Mike,

Well, I had meant to refactor it again after realizing you changed alot more that broke my stuff.  Hence why I closed that PR, 20 minutes later you decided to reopen and merge it.    Not sure why you'd reopen a closed PR and merge it.  *I* must have closed it for a reason.  Now not sure what/where to go from here.

mdblack98, 2021-03-05:
I thought you submitted a new patch that fixed the RTTY problem...that's what the one I merged looked like.
Sounds like you need to fork again and try another patch/pull request.
Mike W9MDB

n9seo, 2021-03-05:
OK,

I will set the kenwood.c back as well I think thats needed does that sound good?

mdblack98, 2021-03-05:
Set kenwood.c back to what?  Reversing the RTTY/RTTYR?  Nope...that was wrong.
What exactly is not working now?
Mike W9MDB

n9seo, 2021-03-05:
Hi the biggest thing is that there is a mismatch now between set_mode and get_mode.

E.g. set to PKTUSB and get_mode reports inverse.  

So based upon the findings that outline the rigs panel and in the docs the table looks like this.

 * DT0 defaults MD6 to USB
 * DT1 defaults MD6 to LSB
 * DT2 defaults MD6 to LSB
 * DT3 defaults MD6 to USB

So MD6 sends "DATA" mode indicator to the rig and MD9 inverts the sideband mode and makes the mode indicator say DATA REV in that case.

Attached set mode and get mode for pktlsb.   

[get_mode_pktusb.txt](https://github.com/Hamlib/Hamlib/files/6091166/get_mode_pktusb.txt)
[set_mode_pktusb.txt](https://github.com/Hamlib/Hamlib/files/6091167/set_mode_pktusb.txt)

n9seo, 2021-03-05:
> Attached set mode and get mode for pktlsb.

Sorry, meant PKTUSB.

mdblack98, 2021-03-05:
We're missing some debug....make sure you have 5 v's.
And which one is wrong?
Mike W9MDB

n9seo, 2021-03-05:
> We're missing some debug....make sure you have 5 v's. 
Resubmitted.
[get_mode_pktusb.txt](https://github.com/Hamlib/Hamlib/files/6091254/get_mode_pktusb.txt)
[set_mode_pktusb.txt](https://github.com/Hamlib/Hamlib/files/6091255/set_mode_pktusb.txt)

>And which one is wrong?

Immidiately after set_mode get_mode reports PKTLSB instead of PKTUSB.

n9seo, 2021-03-05:
Also I should add that the rig is set properly but get_mode does not report properly.

mdblack98, 2021-03-05:
Try again....I reversed the logic
Mike W9DMB

n9seo, 2021-03-05:
Hi,

I commented on that commit but also think that resolves it for now.   Ill put some miles on it and see how this goes.

closed 2021-03-05 by mdblack98