Hamlib/Hamlib #570: Too many clang-cl warnings

gvanem, 2021-02-26:
I've managed to build Hamlib and all program on Windows using:
 * MSVC-2019; 32 and 64-bit.
 * MinGW-w64; 32 and 64-bit.
 * clang-cl; 32 and 64-bit.
 
Compare the Appveyor job-results from: 
* `gcc -Wall` and 6 disabled warnings.  Result: **0** warnings; [MinGW job](https://ci.appveyor.com/project/gvanem/hamlib/build/job/84u66b4d8f4x82go).
 
with:
 * `clang-cl -Wall` with 47 disabled warnings. Result: **25** warnings; [clang-cl job](https://ci.appveyor.com/project/gvanem/hamlib/build/job/c4r0pml9tvtwtq6p).

`clang-cl` gives some  C++ warnings I don't  understand, but the `-Wconditional-uninitialized` warnings looks
important too me. I've not studied them all. 

I'm not sure which clang version they have on AppVeyor. Locally I use clang-cl v11. and it gives pretty much the same warnings.
Maybe some are false positives? But it could be improved.

mdblack98, 2021-02-26:
The C warnings should be cleared up now....some of those are false positives but still easy to suppress.  clang isn't recognizing loops that set vars.
So this logicint i;int n;for(i=0;i<2;++i) n = i;printf("%d\n",n);
Seems to produce a warning that n may not be set which is wrong.
Mike W9MDB

gvanem, 2021-02-26:
> So this logicint i;int n;for(i=0;i<2;++i) n = i;printf("%d\n",n);

I would *really* help if you stated where this is.

mdblack98, 2021-02-26:
Just an example....
I guess you want to report a bug to clang?
One false positive was after this loop that definitely sets ttfilter
       for (ttfilter = 0; tt550_tx_filters[ttfilter] != 0; ttfilter++)        {            if (tt550_tx_filters[ttfilter] == width)            {                break;            }        }
rigs/tentec/tt550.c(917,29): warning: variable 'ttfilter' may be uninitialized when used here [-Wconditional-uninitialized]                            ttfilter,                            ^~~~~~~~rigs/tentec/tt550.c(815,28): note: initialize the variable 'ttfilter' to silence this warning    int mdbuf_len, ttfilter, retval;                           ^                            = 02 warnings generated.

gvanem, 2021-02-27:
> I guess you want to report a bug to clang?

In that case clang-cl was wrong since the code was quite contorted. But why not just initialise some locals at top?
And building with `-DHAVE_USRP`, the situation is even worse. Not to speak of 'extra/gnuradio/*.c*` which doesn't even compile.

mdblack98, 2021-02-27:
All the warning you showed earlier for the C code should be gone now....I did do all the init's to suppress those warnings.
Please post a new log showing any more warnings....
Mike W9MDB

gvanem, 2021-02-27:
> I did do all the init's to suppress those warnings.

Yes I saw it and tried it. But why not use a consistent code-style?

mdblack98, 2021-02-27:
What's inconsistent?

Mike W9MDB

Yes I saw it and tried it. But why not use a consistent code-style?

gvanem, 2021-02-28:
> What's inconsistent?

Space around assignments for example; `x = -1;` vs `x=-1;`. The latter is hard to read.

gvanem, 2021-02-28:
And there is this [warning](https://ci.appveyor.com/project/gvanem/hamlib/build/job/oafsy72icm2f52si):
```
src/rig.c(5738,5): warning: expression which evaluates to zero treated as a null pointer constant of type 'const char *'
[-Wnon-literal-null-conversion]
    RETURNFUNC(RIG_OK);
    ^~~~~~~~~~~~~~~~~~
```
in a function that's never used AFAICS.

mdblack98, 2021-02-28:
I haven't done an astyle recently so that will be cleaned up.We get lots of style differences from many contributors and legacy code.
Mike W9MDB

closed 2021-03-09 by mdblack98