Hamlib/Hamlib #575: FTdx101D - delayed F Main response

Ian-G3VPX, 2021-02-27:
Using rigctld in my piWebCAT project.
All well on a Hamlib-master download from 20th Feb

Updated today to latest Hamlib-master

- All well except for tuning -     \freq set    - very delayed repsonses
- Ran telnet with log (attached)
- f Main
3600000     = ok

f  Sub
3600000     = ok

F Main 3700000
RPRT0   - frequency set ok - but approx 1.3 sec delay before RPRT0

F Sub 3700000
RTRT -8    - approx 1.3 sec dela before RPRT -8

Thus remote tuning of rig impossible
Restored the 20thFeb Hamlib-master   
>  all well again - very fast  Ajax response time
     ( 18ms response time : web client > RPI server > rig > RPi server > web client
Ian
G3VPX
[Fdx101D_log.txt](https://github.com/Hamlib/Hamlib/files/6055198/Fdx101D_log.txt)

mdblack98, 2021-02-27:
Try again....I think I fixed it...
Mike

Ian-G3VPX, 2021-02-28:
New version installed
- F Main 3600000 returns RPRT0 without a delay ... but not effect on rig
- F SUb 3700000 returns RTRT0 without a delay ... but no effect on rig
[FTdx101D_2_log.txt](https://github.com/Hamlib/Hamlib/files/6056522/FTdx101D_2_log.txt)

Ian-G3VPX, 2021-03-01:
FTdx101D F Main and F Sub now working correctly

Ian 
G3VPX

closed 2021-03-01 by mdblack98