Hamlib/Hamlib #576: RTTY/RTTYR inconsistent

mdblack98, 2021-02-27:
RTTY/RTTYR handling across rigs is inconsistent
RTTYR is upper side band
RTTY is lower side band

icom.c does not know about RTTY modes
newcat.c has them backwards
kenwood.c calls it FSK (mode 6) and FSKR (mode 9) -- do we assume FSK=RTTY and FSKR=RTTYR?

N0NB, 2021-02-27:
* On 2021 27 Feb 14:28 -0600, Michael Black wrote:
> RTTY/RTTYR handling across rigs is inconsistent
> RTTYR is upper side band
> RTTY is lower side band
> 
> icom.c does not know about RTTY modes
> newcat.c has them backwards
> kenwood.c calls it FSK (mode 6) and FSKR (mode 9) -- do we assume FSK=RTTY and FSKR=RTTYR?

Mike, I seem to recall some radio (my K3?) being able to set the
preferred RTTY sideband via the radio menu system but it would not be
reflected in the CAT query, etc.  In other words, the actual sideband
was a local setting where RTTY/RTTYR could be one or the other on the
radio but from a CAT standpoint, all that could be known was RTTY/RTTYR.

73, Nate

-- 

"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."

Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

mdblack98, 2021-02-27:
Are you thinking of these sections?
AFSK A also uses Audio-shift transmit, but isoptimized for RTTY. The VFO displays theRTTY mark frequency, and LSB is “normal"

Mike W9MDB 

 
 

Mike, I seem to recall some radio (my K3?) being able to set the
preferred RTTY sideband via the radio menu system but it would not be
reflected in the CAT query, etc. In other words, the actual sideband
was a local setting where RTTY/RTTYR could be one or the other on the
radio but from a CAT standpoint, all that could be known was RTTY/RTTYR.

73, Nate

mdblack98, 2021-03-03:
Yes some rigs do let you choose.  Probably part of the confusion where people though RTTYR meant lower sideband.
Turns out newcat.c has it right
TS-450 manual for example shows MD6 (RTTY-LSB aka RTTY) and MD9 (RTTY-USB aka RTTYR)
So only the kenwood.c was wrong.

closed 2021-03-28 by mdblack98