Hamlib/Hamlib #584: Commit 82126a4 breaks MinGW builds on Debian Buster (32 and 64 bit)

N0NB, 2021-03-04:
Git bisect shows commit 82126a4 to be the suspected culprit.

I am seeing the following errors from the build-w32.sh script on my Debian Buster VM for MinGW builds:

```
make[1]: Leaving directory '/home/nate/builds/hamlib-4.2~git/src'
Making install in tests
make[1]: Entering directory '/home/nate/builds/hamlib-4.2~git/tests'
make  install-am
make[2]: Entering directory '/home/nate/builds/hamlib-4.2~git/tests'
  CC       rigctl-rigctl.o
  CC       rigctl-rigctl_parse.o
  CC       rigctl-dumpcaps.o
  CCLD     rigctl.exe
/usr/bin/i686-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_get_vfo_list':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:2233: undefined reference to `rig_sprintf_vfo'
/usr/bin/i686-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_get_vfo_info':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:2199: undefined reference to `rig_sprintf_vfo'
/usr/bin/i686-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_get_ant':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:4244: undefined reference to `rig_sprintf_ant'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:4260: undefined reference to `rig_sprintf_ant'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:4269: undefined reference to `rig_sprintf_ant'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:4244: undefined reference to `rig_sprintf_ant'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:4260: undefined reference to `rig_sprintf_ant'
/usr/bin/i686-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_get_level':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:2952: undefined reference to `rig_sprintf_level'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:2957: undefined reference to `sprintf_level_ext'
/usr/bin/i686-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_set_vfo':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:2135: undefined reference to `rig_sprintf_vfo'
/usr/bin/i686-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_scan':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:3403: undefined reference to `rig_sprintf_scan'
/usr/bin/i686-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_vfo_op':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:3378: undefined reference to `rig_sprintf_vfop'
/usr/bin/i686-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_get_parm':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:3223: undefined reference to `rig_sprintf_parm'
/usr/bin/i686-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_set_parm':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:3150: undefined reference to `rig_sprintf_parm'
/usr/bin/i686-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_get_func':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:3088: undefined reference to `rig_sprintf_func'
/usr/bin/i686-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_set_func':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:3050: undefined reference to `rig_sprintf_func'
/usr/bin/i686-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_set_level':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:2871: undefined reference to `rig_sprintf_level'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:2876: undefined reference to `sprintf_level_ext'
/usr/bin/i686-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_set_split_vfo':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:2710: undefined reference to `rig_sprintf_vfo'
/usr/bin/i686-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_set_split_freq_mode':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:2646: undefined reference to `rig_sprintf_mode'
/usr/bin/i686-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_set_split_mode':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:2591: undefined reference to `rig_sprintf_mode'
/usr/bin/i686-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_set_mode':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:2084: undefined reference to `rig_sprintf_mode'
/usr/bin/i686-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `dump_chan':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:3967: undefined reference to `rig_sprintf_func'
/usr/bin/i686-w64-mingw32-ld: rigctl-dumpcaps.o: in function `range_print':
/home/nate/builds/hamlib-4.2~git/tests/dumpcaps.c:854: undefined reference to `rig_sprintf_vfo'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.2~git/tests/dumpcaps.c:859: undefined reference to `rig_sprintf_mode'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.2~git/tests/dumpcaps.c:864: undefined reference to `rig_sprintf_ant'
/usr/bin/i686-w64-mingw32-ld: rigctl-dumpcaps.o: in function `dumpcaps':
/home/nate/builds/hamlib-4.2~git/tests/dumpcaps.c:334: undefined reference to `rig_sprintf_func'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.2~git/tests/dumpcaps.c:337: undefined reference to `rig_sprintf_func'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.2~git/tests/dumpcaps.c:343: undefined reference to `rig_sprintf_level_gran'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.2~git/tests/dumpcaps.c:362: undefined reference to `rig_sprintf_level_gran'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.2~git/tests/dumpcaps.c:374: undefined reference to `rig_sprintf_parm_gran'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.2~git/tests/dumpcaps.c:377: undefined reference to `rig_sprintf_parm_gran'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.2~git/tests/dumpcaps.c:392: undefined reference to `rig_sprintf_mode'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.2~git/tests/dumpcaps.c:404: undefined reference to `rig_sprintf_vfo'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.2~git/tests/dumpcaps.c:414: undefined reference to `rig_sprintf_vfop'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.2~git/tests/dumpcaps.c:417: undefined reference to `rig_sprintf_scan'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.2~git/tests/dumpcaps.c:616: undefined reference to `rig_sprintf_mode'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.2~git/tests/dumpcaps.c:648: undefined reference to `rig_sprintf_mode'
/usr/bin/i686-w64-mingw32-ld: rigctl-dumpcaps.o: in function `print_ext':
/home/nate/builds/hamlib-4.2~git/tests/dumpcaps.c:43: undefined reference to `print_ext_param'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:981: rigctl.exe] Error 1
make[2]: Leaving directory '/home/nate/builds/hamlib-4.2~git/tests'
make[1]: *** [Makefile:1682: install] Error 2
make[1]: Leaving directory '/home/nate/builds/hamlib-4.2~git/tests'
make: *** [Makefile:602: install-recursive] Error 1
```

mdblack98, 2021-03-04:
You may need to do a "configure" again.  I had to move the sprintf routine
Mike W9MDB

N0NB, 2021-03-04:
Did a fresh clone of the master branch (latest commit), `./boostrap`, `./configure`, `make dist`, then unpacked the tarball into `~/builds` and ran `build-w64.sh hamlib-4.2~git` and still see:

```
make[1]: Leaving directory '/home/nate/builds/hamlib-4.2~git/src'
Making install in tests
make[1]: Entering directory '/home/nate/builds/hamlib-4.2~git/tests'
make  install-am
make[2]: Entering directory '/home/nate/builds/hamlib-4.2~git/tests'
  CC       rigctl-rigctl.o
  CC       rigctl-rigctl_parse.o
  CC       rigctl-dumpcaps.o
  CCLD     rigctl.exe
/usr/bin/x86_64-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_get_vfo_list':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:2238: undefined reference to `rig_sprintf_vfo'
/usr/bin/x86_64-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_get_vfo_info':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:2204: undefined reference to `rig_sprintf_vfo'
/usr/bin/x86_64-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_get_ant':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:4249: undefined reference to `rig_sprintf_ant'
/usr/bin/x86_64-w64-mingw32-ld: /home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:4265: undefined reference to `rig_sprintf_ant'
/usr/bin/x86_64-w64-mingw32-ld: /home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:4274: undefined reference to `rig_sprintf_ant'
/usr/bin/x86_64-w64-mingw32-ld: /home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:4249: undefined reference to `rig_sprintf_ant'
/usr/bin/x86_64-w64-mingw32-ld: /home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:4265: undefined reference to `rig_sprintf_ant'
/usr/bin/x86_64-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_set_vfo':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:2140: undefined reference to `rig_sprintf_vfo'
/usr/bin/x86_64-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_scan':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:3408: undefined reference to `rig_sprintf_scan'
/usr/bin/x86_64-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_vfo_op':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:3383: undefined reference to `rig_sprintf_vfop'
/usr/bin/x86_64-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_get_parm':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:3228: undefined reference to `rig_sprintf_parm'
/usr/bin/x86_64-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_set_parm':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:3155: undefined reference to `rig_sprintf_parm'
/usr/bin/x86_64-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_get_func':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:3093: undefined reference to `rig_sprintf_func'
/usr/bin/x86_64-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_set_func':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:3055: undefined reference to `rig_sprintf_func'
/usr/bin/x86_64-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_get_level':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:2957: undefined reference to `rig_sprintf_level'
/usr/bin/x86_64-w64-mingw32-ld: /home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:2962: undefined reference to `sprintf_level_ext'
/usr/bin/x86_64-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_set_level':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:2876: undefined reference to `rig_sprintf_level'
/usr/bin/x86_64-w64-mingw32-ld: /home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:2881: undefined reference to `sprintf_level_ext'
/usr/bin/x86_64-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_set_split_vfo':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:2715: undefined reference to `rig_sprintf_vfo'
/usr/bin/x86_64-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_set_split_freq_mode':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:2651: undefined reference to `rig_sprintf_mode'
/usr/bin/x86_64-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_set_split_mode':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:2596: undefined reference to `rig_sprintf_mode'
/usr/bin/x86_64-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_set_mode':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:2089: undefined reference to `rig_sprintf_mode'
/usr/bin/x86_64-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `dump_chan':
/home/nate/builds/hamlib-4.2~git/tests/rigctl_parse.c:3972: undefined reference to `rig_sprintf_func'
/usr/bin/x86_64-w64-mingw32-ld: rigctl-dumpcaps.o: in function `range_print':
/home/nate/builds/hamlib-4.2~git/tests/dumpcaps.c:854: undefined reference to `rig_sprintf_vfo'
/usr/bin/x86_64-w64-mingw32-ld: /home/nate/builds/hamlib-4.2~git/tests/dumpcaps.c:859: undefined reference to `rig_sprintf_mode'
/usr/bin/x86_64-w64-mingw32-ld: /home/nate/builds/hamlib-4.2~git/tests/dumpcaps.c:864: undefined reference to `rig_sprintf_ant'
/usr/bin/x86_64-w64-mingw32-ld: rigctl-dumpcaps.o: in function `dumpcaps':
/home/nate/builds/hamlib-4.2~git/tests/dumpcaps.c:334: undefined reference to `rig_sprintf_func'
/usr/bin/x86_64-w64-mingw32-ld: /home/nate/builds/hamlib-4.2~git/tests/dumpcaps.c:337: undefined reference to `rig_sprintf_func'
/usr/bin/x86_64-w64-mingw32-ld: /home/nate/builds/hamlib-4.2~git/tests/dumpcaps.c:343: undefined reference to `rig_sprintf_level_gran'
[... cut ...]
/usr/bin/x86_64-w64-mingw32-ld: /home/nate/builds/hamlib-4.2~git/tests/dumpcaps.c:648: undefined reference to `rig_sprintf_mode'
/usr/bin/x86_64-w64-mingw32-ld: rigctl-dumpcaps.o: in function `print_ext':
/home/nate/builds/hamlib-4.2~git/tests/dumpcaps.c:43: undefined reference to `print_ext_param'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:981: rigctl.exe] Error 1
make[2]: Leaving directory '/home/nate/builds/hamlib-4.2~git/tests'
make[1]: *** [Makefile:1682: install] Error 2
make[1]: Leaving directory '/home/nate/builds/hamlib-4.2~git/tests'
```

mdblack98, 2021-03-04:
Try again

N0NB, 2021-03-05:
Everything is working now.  I manually updated the daily snapshots and they'll run again in the morning.

closed 2021-03-05 by N0NB