Hamlib/Hamlib #590: FTdx101D - recurrence of RTRP -1 on button reads

Ian-G3VPX, 2021-03-08:
6thMarch version working fine in piWebCAT
Todays version - multiple errors with recurrence of RPRT -1 problem on \get_func

Log attached - from piWebCAT startup
Seems to start ok - many controls correctly set from rig ... and two frequencies
Then quickly goes wrong with errors on read

[FTdx101D_08Feb2021_piWebCAT_start_log.txt](https://github.com/Hamlib/Hamlib/files/6101501/FTdx101D_08Feb2021_piWebCAT_start_log.txt)

mdblack98, 2021-03-09:
Make sure you end every command with "q" to quit your connection.
And if that doesn't fix it then try commenting out this block of code in rigctld.c and see if this fixes your problem.

        // if we get a hard error we try to reopen the rig again        // this should cover short dropouts that can occur        if (retcode == -RIG_EIO || retcode == 2)        {            int retry = 3;            rig_debug(RIG_DEBUG_ERR, "%s: i/o error\n", __func__)
            do            {                retcode = rig_close(my_rig);                hl_usleep(1000 * 1000);                rig_debug(RIG_DEBUG_ERR, "%s: rig_close retcode=%d\n", __func__, retcode);                retcode = rig_open(my_rig);                rig_debug(RIG_DEBUG_ERR, "%s: rig_open retcode=%d\n", __func__, retcode);            }            while (retry-- > 0 && retcode != RIG_OK);
        }

mdblack98, 2021-03-09:
Sending again in text mode...

Mike

On Monday, March 8, 2021, 10:48:30 PM CST, Black Michael <mdblack98@yahoo.com> wrote: 

Make sure you end every command with "q" to quit your connection.

And if that doesn't fix it then try commenting out this block of code in rigctld.c and see if this fixes your problem.

        // if we get a hard error we try to reopen the rig again
        // this should cover short dropouts that can occur
        if (retcode == -RIG_EIO || retcode == 2)
        {
            int retry = 3;
            rig_debug(RIG_DEBUG_ERR, "%s: i/o error\n", __func__)

            do
            {
                retcode = rig_close(my_rig);
                hl_usleep(1000 * 1000);
                rig_debug(RIG_DEBUG_ERR, "%s: rig_close retcode=%d\n", __func__, retcode);
                retcode = rig_open(my_rig);
                rig_debug(RIG_DEBUG_ERR, "%s: rig_open retcode=%d\n", __func__, retcode);
            }
            while (retry-- > 0 && retcode != RIG_OK);

        }

Ian-G3VPX, 2021-03-09:
Fixed...
This morning, after no changes, using yesterday's Hamib-master the problem didn't happen.
( - baffled -- yesterday I did two download  / builds - to avoid erroneous reports)
However   - q  added as follows   = Thans for that 
		socket_write($socket, $catString);
		$ans = socket_read($socket, 1000);
		socket_write($socket, "q\n");		
		socket_close($socket);
Downloaded this mornings Hamlib-master (9th Feb)   - problems:

Using telnet:   cannot set freq      l Main STRENGTH always reports full scale (60)
(Rpeated download on another RPI   -same)
- raised as an issue   - with telnet driven log
Ian
GeVPX