Hamlib/Hamlib #592: FTDX101MP power level incorrect in JTDX

mdblack98, 2021-03-08:
geohartmut@yahoo.de
PWR  JTDX
10   17
20   35
30   46
40   59
50   71
60   80
70   88
80   98
90   112
100  134
120  198
140  198
160  198
180  198

Ian-G3VPX, 2021-03-11:
l

Looks ok on FTdx101D   ... but only does 100w max

l Main RFPOWER_METER_WATTS   100
100.000000

l Main RFPOWER_METER_WATTS    80
100.000000

l Main RFPOWER_METER_WATTS    80
82.758621

l Main RFPOWER_METER_WATTS    60 
63.793102

l Main RFPOWER_METER_WATTS    50    
52.586205

l Main RFPOWER_METER_WATTS    40
42.452831

l Main RFPOWER_METER_WATTS    30
30.188679

l Main RFPOWER_METER_WATTS    20     
21.785715

l Main RFPOWER_METER_WATTS    10 
12.142858

l Main RFPOWER_METER_WATTS     5
4.605263

Ian G3VPX

mdblack98, 2021-03-11:
You got some debug info?
And I need to know how it matches with the power level too.
If you could do it in 20W intervals....
Mike W9MDB

closed 2021-03-15 by mdblack98