Hamlib/Hamlib #594: FTdx101D 9th Feb  - cannot set freq. Smeter full scale

Ian-G3VPX, 2021-03-09:
Using rigctld driven by telnet:

f Main    - returns correct frequency

F Main 3600000   retruns RPRT 0    but no freq change on rig

l Main STRENGTH  -    always returns 60 (fsd) irrespective of S meter reading on rig

Log generated for these three commands

Ian
G3VPX
[FTdx101D_09Feb21_log.txt](https://github.com/Hamlib/Hamlib/files/6108163/FTdx101D_09Feb21_log.txt)

Ian-G3VPX, 2021-03-09:
Fixed   - thanks
Ian
G3VPX

mdblack98, 2021-03-10:
FYI...hamlib is having problems doing set_ptt -- if you see delays it's because it's unnecessarily repeating set_ptt...Could be causing other problems too but at least this one is repeatable.
I'm working on it...
Mike

mdblack98, 2021-03-10:
I need another test on this....it might be fixed now.
Mike W9MDB