Hamlib/Hamlib #600: Icom backends broken

mikaelnousiainen, 2021-03-11:
@mdblack98 I checked out the current master (on March 11th) and noticed Icom backends are somehow broken. None of the commands that read data are working indicating "collision on the bus". Tested on IC-7300 and IC-7600.

Here's the initialization:

```
serial.c(278):serial_setup entered
serial_setup: tcgetattr
serial_setup: cfmakeraw
serial_setup: cfsetispeed=115200,0x1002
serial_setup: cfsetospeed=115200,0x1002
serial_setup: data_bits=8
serial_setup: parity=0
serial_setup: tcsetattr TCSANOW
serial.c(618):serial_setup return(0)
serial.c(629):serial_flush entered
tcflush
serial.c(661):serial_flush return(0)
serial.c(253):serial_open return(0)
rig.c(214):add_opened_rig return(0)
icom.c(758):icom_rig_open entered
icom_rig_open: IC-7300 v20210310.3
icom_get_usb_echo_off called
icom_get_usb_echo_off: retry temp set to 0
frame.c(315):icom_transaction entered
icom_transaction: cmd=0x03, subcmd=0xffffffff, payload_len=0
frame.c(119):icom_one_transaction entered
frame.c(56):make_cmd_frame entered
frame.c(90):make_cmd_frame return(6)
rig_flush: called for serial device
serial.c(629):serial_flush entered
tcflush
serial.c(661):serial_flush return(0)
write_block called
write_block(): TX 6 bytes
0000    fe fe 94 e0 03 fd                                   ......
frame.c(365):read_icom_frame entered
read_string called, rxmax=80
read_string(): RX 11 characters
0000    fe fe e0 94 03 00 80 71 03 00 fd                    .......q...
frame.c(399):read_icom_frame return(11)
frame.c(282):icom_one_transaction return(-14)
icom_transaction: retry=0: Communication bus collision
frame.c(399):read_icom_frame return(11)
frame.c(282):icom_one_transaction return(-14)
icom_transaction: retry=0: Communication bus collision
frame.c(399):read_icom_frame return(11)
frame.c(282):icom_one_transaction return(-14)
frame.c(282):icom_one_transaction return(-14)
icom_transaction: failed: Communication bus collision
frame.c(282):icom_one_transaction return(-14)
icom_transaction: retry=0: Communication bus collision
frame.c(399):read_icom_frame return(11)
frame.c(282):icom_one_transaction return(-14)
frame.c(282):icom_one_transaction return(-14)icom_transaction: failed: Communication bus collision
frame.c(282):icom_one_transaction return(-14)
icom_transaction: retry=0: Communication bus collision
frame.c(399):read_icom_frame return(11)
```

Here's `get_freq` command:

```
Rig command: f
rigctl_parse: input_line: f
rigctl_parse: vfo_opt=0
rigctl_parse.c(1984):rigctl_get_freq entered
rig_get_freq called vfo=currVFO
vfo_fixup: vfo=currVFO
vfo_fixup: Leaving currVFO alone
get_cache_freq:  vfo=currVFO, current_vfo=currVFO
[... 205 lines cut ...]
frame.c(282):icom_one_transaction return(-14)icom_transaction: failed: Communication bus collision
frame.c(282):icom_one_transaction return(-14)
icom_transaction: retry=0: Communication bus collision
frame.c(399):read_icom_frame return(11)
frame.c(282):icom_one_transaction return(-14)
frame.c(282):icom_one_transaction return(-14)icom_transaction: retry=0: Communication bus collision
frame.c(399):read_icom_frame return(11)
frame.c(282):icom_one_transaction return(-14)
frame.c(282):icom_one_transaction return(-14)
frame.c(345):icom_transaction return(-14)
icom_get_usb_echo_off: ack_len=0
icom_get_usb_echo_off: USB echo off detected
icom.c(742):icom_get_usb_echo_off return(1)
icom.c(827):icom_rig_open return(0)
rig.c(2439):rig_get_vfo entered
rig_get_vfo: no get_vfo
rig.c(2453):rig_get_vfo return(-11)
rig_open: Icom rig so default vfo = currVFO
rig.c(1044):rig_open return(0)
main: rig_open retcode=0
rigctl_parse.c(642):rigctl_parse entered
rigctl_parse: called, interactive=1
```

The same setup was working 100% fine before updating.

mikaelnousiainen, 2021-03-11:
@mdblack98 Looking at this response: `fe fe e0 94 03 00 80 71 03 00 fd` it seems like my IC-7300 is not including the OK/NG acknowledgement byte. I recently upgraded to firmware version 1.40, so I'm not sure if that is the reason. I'll have to check with some other rigs.

mikaelnousiainen, 2021-03-11:
@mdblack98 Right, so I tested this quickly on the IC-7600 too: It looks like command that **set** something return the OK/NG byte without the actual command parameters in the responses, but commands that **get** data do not return the OK/NG byte in the responses, so we would need to handle this in the transactions -> The code that reads the `frm_len - 2` byte should be executed only for **set**-type commands (see changes made #596 and the check above for `NAK`)

mdblack98, 2021-03-11:
That should be fixed now...please test the latest.

closed 2021-03-11 by mdblack98