Hamlib/Hamlib #601: ICOM Transceive packet recognition

mdblack98, 2021-03-11:
Add ability to recognize Icom transceive packets and update internal cache along with appropriate callbacks
e.g. IC275 freq transceive packet where 0x00 is the broadcast address.
fe fe 00 10 00 40 45 30 44 01 fd

mdblack98, 2021-06-22:
As part of this enhancement will have to disable rig_flush and be able to process all async or unexpected messages that may be received.
K3 is a really good example of bad interaction with CAT commands being processed late.
Here's a log showing an ID command timing out and subsequenct FA commands out of sequence.
[2021-06-22 15:21:16.356939][00:07:11.947498][RIGCTRL:trace] 0000    46 54 31 3b                                         FT1;
[2021-06-22 15:21:16.357220][00:07:11.947766][RIGCTRL:trace] write_block(): TX 3 bytes
[2021-06-22 15:21:16.357220][00:07:11.947778][RIGCTRL:trace] 0000    49 44 3b                                            ID;
[2021-06-22 15:21:16.357220][00:07:11.947785][RIGCTRL:trace] read_string called, rxmax=35
[2021-06-22 15:21:16.377854][00:07:11.968538][RIGCTRL:trace] read_string(): RX 2 characters
[2021-06-22 15:21:16.377854][00:07:11.968554][RIGCTRL:trace] 0000    3f 3b                                               ?;
[2021-06-22 15:21:16.377854][00:07:11.968561][RIGCTRL:trace] kenwood_transaction: read_string(len=2)='?;'
[2021-06-22 15:21:16.377854][00:07:11.968574][RIGCTRL:error] kenwood_transaction: Unknown command or rig busy 'FT1'
[2021-06-22 15:21:16.377854][00:07:11.968582][RIGCTRL:error] kenwood_transaction: Retrying shortly
[2021-06-22 15:21:17.378801][00:07:12.969406][RIGCTRL:trace] kenwood_transaction: cmdstr = FT1
[2021-06-22 15:21:17.378801][00:07:12.969693][RIGCTRL:trace] write_block(): TX 4 bytes
[2021-06-22 15:21:17.378801][00:07:12.969711][RIGCTRL:trace] 0000    46 54 31 3b                                         FT1;
[2021-06-22 15:21:17.378801][00:07:12.969880][RIGCTRL:trace] write_block(): TX 3 bytes
[2021-06-22 15:21:17.378801][00:07:12.969895][RIGCTRL:trace] 0000    49 44 3b                                            ID;
[2021-06-22 15:21:17.378801][00:07:12.969904][RIGCTRL:trace] read_string called, rxmax=35
[2021-06-22 15:21:17.379799][00:07:12.970712][RIGCTRL:trace] read_string(): RX 6 characters
[2021-06-22 15:21:17.379799][00:07:12.970726][RIGCTRL:trace] 0000    49 44 30 31 37 3b                                   ID017;
[2021-06-22 15:21:17.379799][00:07:12.970734][RIGCTRL:trace] kenwood_transaction: read_string(len=6)='ID017;'
[2021-06-22 15:21:17.379799][00:07:12.970745][RIGCTRL:trace] kenwood_transaction: No data expected, checking ID; in ID017;
[2021-06-22 15:21:17.379799][00:07:12.970752][RIGCTRL:trace] kenwood_transaction: returning RIG_OK, retval=0
[2021-06-22 15:21:17.379799][00:07:12.970759][RIGCTRL:debug] kenwood.c(576):kenwood_transaction return(0)
[2021-06-22 15:21:17.379799][00:07:12.970765][RIGCTRL:debug] kenwood.c(1299):kenwood_set_split_vfo return(0)
[2021-06-22 15:21:17.379799][00:07:12.970772][RIGCTRL:debug] rig.c(4433):rig_set_split_vfo return(0)
[2021-06-22 15:21:17.379799][00:07:12.970824][RIGCTRL:debug] rig.c(2702):rig_get_vfo entered
[2021-06-22 15:21:17.379799][00:07:12.970832][RIGCTRL:trace] rig_get_vfo: cache check age=1172ms
[2021-06-22 15:21:17.379799][00:07:12.970839][RIGCTRL:trace] rig_get_vfo: cache miss age=1172ms
[2021-06-22 15:21:17.379799][00:07:12.970852][RIGCTRL:trace] ../../src/src/rig.c(2738) trace
[2021-06-22 15:21:17.379799][00:07:12.970858][RIGCTRL:debug] k3.c(1287):k3_get_vfo entered
[2021-06-22 15:21:17.379799][00:07:12.970865][RIGCTRL:debug] k3.c(1291):k3_get_vfo return(0)
[2021-06-22 15:21:17.379799][00:07:12.970871][RIGCTRL:debug] rig.c(2758):rig_get_vfo return(0)
[2021-06-22 15:21:17.379799][00:07:12.970874][RIGCTRL:trace] VFO=VFOB
[2021-06-22 15:21:17.379799][00:07:12.970884][RIGCTRL:debug] rig.c(4505):rig_get_split_vfo entered
[2021-06-22 15:21:17.379799][00:07:12.970890][RIGCTRL:trace] rig_get_split_vfo: cache check age=0ms
[2021-06-22 15:21:17.379799][00:07:12.970896][RIGCTRL:trace] rig_get_split_vfo: cache hit age=0ms
[2021-06-22 15:21:17.379799][00:07:12.970902][RIGCTRL:debug] rig.c(4540):rig_get_split_vfo return(0)
[2021-06-22 15:21:17.379799][00:07:12.970905][RIGCTRL:trace] rig_get_split_vfo split=1 VFO=VFOB
[2021-06-22 15:21:17.379799][00:07:12.970914][RIGCTRL:debug] rig_get_freq(1976) called vfo=VFOA
[2021-06-22 15:21:17.379799][00:07:12.970931][RIGCTRL:trace] vfo_fixup: vfo=VFOA, vfo_curr=VFOB
[2021-06-22 15:21:17.379799][00:07:12.970939][RIGCTRL:debug] ../../src/src/rig.c(1984) vfo=VFOA, curr_vfo=VFOB
[2021-06-22 15:21:17.379799][00:07:12.970966][RIGCTRL:trace] rig_get_freq: cache miss age=1107ms, cached_vfo=VFOA, asked_vfo=VFOA
[2021-06-22 15:21:17.379799][00:07:12.970972][RIGCTRL:debug] kenwood_get_freq called
[2021-06-22 15:21:17.379799][00:07:12.970977][RIGCTRL:debug] kenwood_safe_transaction called
[2021-06-22 15:21:17.379799][00:07:12.970993][RIGCTRL:debug] kenwood_transaction called
[2021-06-22 15:21:17.379799][00:07:12.970998][RIGCTRL:trace] kenwood_transaction: cmdstr = FA
[2021-06-22 15:21:17.380796][00:07:12.971390][RIGCTRL:trace] write_block(): TX 3 bytes
[2021-06-22 15:21:17.380796][00:07:12.971403][RIGCTRL:trace] 0000    46 41 3b                                            FA;
[2021-06-22 15:21:17.380796][00:07:12.971410][RIGCTRL:trace] read_string called, rxmax=51
[2021-06-22 15:21:17.402080][00:07:12.992633][RIGCTRL:trace] read_string(): RX 2 characters
[2021-06-22 15:21:17.402080][00:07:12.992650][RIGCTRL:trace] 0000    3f 3b                                               ?;
[2021-06-22 15:21:17.402080][00:07:12.992657][RIGCTRL:trace] kenwood_transaction: read_string(len=2)='?;'
[... cut ...]
[2021-06-22 15:21:18.406487][00:07:13.997010][RIGCTRL:trace] 0000    49 46 3b                                            IF;
[2021-06-22 15:21:18.406487][00:07:13.997017][RIGCTRL:trace] read_string called, rxmax=128
[2021-06-22 15:21:18.460512][00:07:14.051059][RIGCTRL:trace] read_string(): RX 14 characters
[2021-06-22 15:21:18.460512][00:07:14.051081][RIGCTRL:trace] 0000    46 41 30 30 30 31 34 30 37 34 30 30 31 3b           FA00014074001;
[2021-06-22 15:21:18.460512][00:07:14.051089][RIGCTRL:trace] kenwood_transaction: read_string(len=14)='FA00014074001;'

mikaelnousiainen, 2023-11-10:
@mdblack98 This is already implemented in the Icom backend (requires async mode to be enabled) and works -- as long as Hamlib can track the current VFO (-> user doesn't change VFO from the rig buttons). Not sure if there is anything else one can do to make it better, because of the CI-V limitations.