Hamlib/Hamlib #609: Hamlib error: IO error while opening connection to rig

kb3mkd, 2021-03-15:
I have been using my IC-7410 wtih hamlib for digital modes for some time. Yesterday I used my rig on ssb, switching the data and data off mode to mic,acc. Today i switched it back to USB and tried to use the system as normal, but get the error above. 
The only changes I made were the 2 changes in the rig, and changing them back should have gone back the way it was, I would think.
Using Linux Mint.

mdblack98, 2021-03-15:
Can you provide some debug info?  Using -vvvvv?
Aer you running rigctl or some app with hamlib built in?

kb3mkd, 2021-03-15:
wsjtx

mdblack98, 2021-03-15:
What operating system?
Mike W9MDB

mdblack98, 2021-03-15:
What version of WSTJ-X too please?

kb3mkd, 2021-03-15:
v2.1.0

kb3mkd, 2021-03-15:
I mentioned in my original post Linux Mint

mdblack98, 2021-03-15:
I just noticed the Linux Mint.
Please provide 
rigctl -m 3067 -r /dev/ttyUSB0 -s 19200 -vvvvv -Z f >&log.txt
And send the log.txt file 
Mike W9MDB

kb3mkd, 2021-03-15:
rigctl: invalid option -- 'Z'
And then a bunch of usage info The version is the wsjtx version 2.1.0
Hamlib version is coming up as 3.1-7build1

mdblack98, 2021-03-15:
Please upgrade wsjt-x and hamlib -- both of those versions are no longer supported.
Mike W9MDB

kb3mkd, 2021-03-15:
Is there a repository I can get these from or must I recompile?

mdblack98, 2021-03-15:
It would appear the .deb package is what you want
https://www.physics.princeton.edu/pulsar/k1jt/wsjtx.html

Mike W9MDB

kb3mkd, 2021-03-15:
Thanks Mike!

closed 2021-03-15 by mdblack98