Hamlib/Hamlib #610: FTdx101D - majority of \set_freq Main failing

Ian-G3VPX, 2021-03-15:
Version frrom 15th March - Mode problem fixed
- but has new problem.
- piWebCAT - noticed frequency control of rig slow and jerky (was very fast)
- Observed web clinet > server traffic on browser Inspect element > network - Timing
- A majority of freq writes taking 600ms (usually about 20ms)
- Unclear whether these are failing or just slowing everything
- Log attached  shows start up, running for few seconds, then started rapid freq writes
from piWebCAT mousewheel tuner
- (Can't break it with telnet)
- Ian
- G3VPX
- 
[FTdx101D_freq_write_log.txt](https://github.com/Hamlib/Hamlib/files/6145089/FTdx101D_freq_write_log.txt)
-

mdblack98, 2021-03-16:
2021-03-15:23:29:00.500488: Connection closed from localhost:37508
2021-03-15:23:29:00.528558: Connection opened from localhost:37510
2021-03-15:23:29:00.528788: sync_callback: client lock engaged
2021-03-15:23:29:00.528846: sync_callback: client lock disengaged
2021-03-15:23:29:00.528893: handle_socket: doing rigctl_parse vfo_mode=1
2021-03-15:23:29:00.528949: rigctl_parse.c(642):rigctl_parse entered
2021-03-15:23:29:00.528994: rigctl_parse: called, interactive=1
2021-03-15:23:29:00.529079: rigctl_parse: cmd=\(5c)
2021-03-15:23:29:00.529131: rigctl_parse: cmd=set_freq
2021-03-15:23:29:00.529182: rigctl_parse: vfo_opt=1
2021-03-15:23:29:00.529230: rig_parse_vfo called
2021-03-15:23:29:00.529274: rigctl_parse: debug1
2021-03-15:23:29:00.529320: rigctl_parse: debug3
2021-03-15:23:29:00.529370: rigctl_parse: debug4 arg1= 
2021-03-15:23:29:00.529422: rigctl_parse: debug5
2021-03-15:23:29:00.529466: rigctl_parse: debug10
2021-03-15:23:29:00.529511: sync_callback: client lock engaged
2021-03-15:23:29:00.529556: rigctl(d): F 'Main' '3694179' '' ''
2021-03-15:23:29:00.529599: rigctl_parse: vfo_opt=1
2021-03-15:23:29:00.529641: rigctl_parse.c(1960):rigctl_set_freq entered
2021-03-15:23:29:00.529717: rig_set_freq called vfo=Main, freq=3694179
2021-03-15:23:29:00.529766: vfo_fixup: vfo=Main
2021-03-15:23:29:00.529809: vfo_fixup: final vfo=Main
2021-03-15:23:29:00.529851: rig_set_freq: TARGETABLE_FREQ vfo=Main
2021-03-15:23:29:00.529933: newcat.c(722):newcat_set_freq entered
2021-03-15:23:29:00.529978: newcat_valid_command FA
2021-03-15:23:29:00.530022: newcat.c(6699):newcat_valid_command return(1)
2021-03-15:23:29:00.530065: newcat_valid_command FB
2021-03-15:23:29:00.530106: newcat.c(6699):newcat_valid_command return(1)
2021-03-15:23:29:00.530146: newcat_set_freq: passed vfo = Main
2021-03-15:23:29:00.530196: newcat_set_freq: passed freq = 3694179.000000 Hz
2021-03-15:23:29:00.530239: newcat.c(6902):newcat_set_vfo_from_alias entered
2021-03-15:23:29:00.530279: newcat_set_vfo_from_alias: alias vfo = Main
2021-03-15:23:29:00.530321: newcat.c(6945):newcat_set_vfo_from_alias return(0)
2021-03-15:23:29:00.530363: newcat_set_freq: special_60m=0, 60m freq=0, is_ftdx3000=0
2021-03-15:23:29:00.530408: newcat.c(866)newcat_set_freq: rig->state.current_vfo=Main
2021-03-15:23:29:00.530467: rig_get_freq called vfo=VFOA
2021-03-15:23:29:00.530508: vfo_fixup: vfo=VFOA
2021-03-15:23:29:00.530552: vfo_fixup: final vfo=VFOA
2021-03-15:23:29:00.530593: rig.c(1464):rig_get_cache entered
2021-03-15:23:29:00.530633: rig_get_cache:  vfo=VFOA, current_vfo=Main
2021-03-15:23:29:00.530683: rig_get_cache: vfo=VFOA, freq=3694079
2021-03-15:23:29:00.530728: rig.c(1550):rig_get_cache return(0)
2021-03-15:23:29:00.530769: rig_get_freq: cache check1 age=10000ms
2021-03-15:23:29:00.530808: rig_get_freq: cache miss age=10000ms, cached_vfo=VFOA, asked_vfo=VFOA
2021-03-15:23:29:00.530853: newcat.c(1105):newcat_get_freq entered
2021-03-15:23:29:00.530893: newcat_get_freq: passed vfo = VFOA
2021-03-15:23:29:00.530930: newcat_valid_command FA
2021-03-15:23:29:00.530971: newcat.c(6699):newcat_valid_command return(1)
2021-03-15:23:29:00.531008: newcat_valid_command FB
2021-03-15:23:29:00.531046: newcat.c(6699):newcat_valid_command return(1)
2021-03-15:23:29:00.531086: newcat.c(6902):newcat_set_vfo_from_alias entered
2021-03-15:23:29:00.531124: newcat_set_vfo_from_alias: alias vfo = VFOA
2021-03-15:23:29:00.531163: newcat.c(6945):newcat_set_vfo_from_alias return(0)
2021-03-15:23:29:00.531202: newcat_valid_command FA
2021-03-15:23:29:00.531240: newcat.c(6699):newcat_valid_command return(1)
2021-03-15:23:29:00.531279: cmd_str = FA;
2021-03-15:23:29:00.531321: newcat.c(9452):newcat_get_cmd entered
2021-03-15:23:29:00.531370: rig_flush: called for serial device
2021-03-15:23:29:00.531417: serial.c(629):serial_flush entered
2021-03-15:23:29:00.531463: tcflush
2021-03-15:23:29:00.531514: serial.c(661):serial_flush return(0)
2021-03-15:23:29:00.531557: cmd_str = FA;
2021-03-15:23:29:00.531602: write_block called
2021-03-15:23:29:00.536804: write_block(): TX 3 bytes
2021-03-15:23:29:00.536937: 0000    46 41 3b                                            FA;             
2021-03-15:23:29:00.537026: read_string called, rxmax=129
2021-03-15:23:29:00.537218: read_string(): RX 12 characters
2021-03-15:23:29:00.537300: 0000    46 41 30 30 33 36 39 34 30 37 39 3b                 FA003694079;    
2021-03-15:23:29:00.537376: newcat_get_cmd: read count = 12, ret_data = FA003694079;
[... 44 lines cut ...]
2021-03-15:23:29:00.545486: newcat.c(428):newcat_band_index return(1)
2021-03-15:23:29:00.545544: newcat_band_index: freq=3.69408e+06, band=1
2021-03-15:23:29:00.545601: newcat.c(428):newcat_band_index return(1)
2021-03-15:23:29:00.545615: newcat_set_freq: band changing? old=1, new=1
2021-03-15:23:29:00.545677: newcat.c(1086):newcat_set_freq return(0)
2021-03-15:23:29:00.545746: rig.c(1377):set_cache_freq entered
2021-03-15:23:29:00.545797: set_cache_freq:  vfo=Main, current_vfo=Main
2021-03-15:23:29:00.545850: set_cache_freq: set vfo=Main to freq=0
2021-03-15:23:29:00.545901: rig.c(1459):set_cache_freq return(0)
2021-03-15:23:29:00.545948: rig_get_freq called vfo=Main
2021-03-15:23:29:00.545991: vfo_fixup: vfo=Main
2021-03-15:23:29:00.546034: vfo_fixup: final vfo=Main
2021-03-15:23:29:00.546077: rig.c(1464):rig_get_cache entered
2021-03-15:23:29:00.546120: rig_get_cache:  vfo=Main, current_vfo=Main
2021-03-15:23:29:00.546166: rig_get_cache: vfo=Main, freq=0
[... cut ...]
2021-03-15:23:29:00.546297: rig_get_freq: Main cache hit age=0ms, freq=0
2021-03-15:23:29:00.546343: rig.c(1872):rig_get_freq return(0)
2021-03-15:23:29:00.596491: rig_set_freq: freq not set correctly?? got 0 asked for 3694179, retry=3
2021-03-15:23:29:00.596568: newcat.c(722):newcat_set_freq entered
2021-03-15:23:29:00.596613: newcat_valid_command FA
2021-03-15:23:29:00.596661: newcat.c(6699):newcat_valid_command return(1)
2021-03-15:23:29:00.596704: newcat_valid_command FB
2021-03-15:23:29:00.596747: newcat.c(6699):newcat_valid_command return(1)

mdblack98, 2021-03-16:
Please try again...hopefully fixed now.Mike W9MDB

Ian-G3VPX, 2021-03-16:
Problem fixed  - fast tuning,  no dropouts
Thanks
Ian
G3VPX

closed 2021-03-16 by mdblack98