Hamlib/Hamlib #612: WSJT-X commands failing randomly on latest master

mikaelnousiainen, 2021-03-16:
I pulled the latest changes (commit 5dfb3b85671e8542a0016aae4d63f5ef76bab023) and now WSJT-X rigctld commands (control over network via Hamlib NET rigctl) fail very randomly. The commands and responses seem to vary, I've been getting "invalid parameter", "I/O error" and "command rejected by the rig". It all seems like it's got something to do with timings -- maybe the latest changes with caching affect this, so could #610 be related?

I can usually make the commands fail when clicking the test CAT/PTT buttons in WSJT-X settings. During actual use of WSJT-X, it may take a while until it fails. None of this has happened for me before. I've been running the same WSJT-X v2.2.2 for a long time and it's been stable.

Here's an example of IC-7600 failing to set PTT off in WSJT-X settings when testing PTT (apologies for the extra mess, it's the application logging):

```
64,"process":"/usr/local/bin/rigctld","line":"rigctl(d): T 'currVFO' '0' '' ''\n"}
64,"process":"/usr/local/bin/rigctld","line":"rigctl_parse.c(642):rigctl_parse entered\n"}
64,"process":"/usr/local/bin/rigctld","line":"rigctl_parse: vfo_opt=0\n"}
64,"process":"/usr/local/bin/rigctld","line":"rigctl_parse: called, interactive=1\n"}
64,"process":"/usr/local/bin/rigctld","line":"rigctl_parse.c(2279):rigctl_set_ptt entered\n"}
64,"process":"/usr/local/bin/rigctld","line":"rigctl_parse: cmd==0x0a\n"}
64,"process":"/usr/local/bin/rigctld","line":"rig.c(2602):rig_set_ptt entered\n"}
64,"process":"/usr/local/bin/rigctld","line":"icom_set_ptt called\n"}
64,"process":"/usr/local/bin/rigctld","line":"frame.c(325):icom_transaction entered\n"}
64,"process":"/usr/local/bin/rigctld","line":"icom_transaction: cmd=0x1c, subcmd=0x00, payload_len=1\n"}
64,"process":"/usr/local/bin/rigctld","line":"frame.c(119):icom_one_transaction entered\n"}
64,"process":"/usr/local/bin/rigctld","line":"frame.c(56):make_cmd_frame entered\n"}
64,"process":"/usr/local/bin/rigctld","line":"frame.c(90):make_cmd_frame return(8)\n"}
64,"process":"/usr/local/bin/rigctld","line":"rig_flush: called for serial device\n"}
64,"process":"/usr/local/bin/rigctld","line":"serial.c(629):serial_flush entered\n"}
64,"process":"/usr/local/bin/rigctld","line":"tcflush\n"}
64,"process":"/usr/local/bin/rigctld","line":"serial.c(661):serial_flush return(0)\n"}
64,"process":"/usr/local/bin/rigctld","line":"write_block called\n"}
64,"process":"/usr/local/bin/rigctld","line":"write_block(): TX 8 bytes\n"}
64,"process":"/usr/local/bin/rigctld","line":"0000    fe fe 7a e0 1c 00 00 fd                             ..z.....        \n"}
64,"process":"/usr/local/bin/rigctld","line":"frame.c(375):read_icom_frame entered\n"}
64,"process":"/usr/local/bin/rigctld","line":"read_string called, rxmax=80\n"}
64,"process":"/usr/local/bin/rigctld","line":"rigctl_parse: cmd=+(2b)\n"}
64,"process":"/usr/local/bin/rigctld","line":"rigctl_parse: cmd=get_level\n"}
64,"process":"/usr/local/bin/rigctld","line":"rigctl_parse: vfo_opt=0\n"}
64,"process":"/usr/local/bin/rigctld","line":"rigctl_parse: debug1\n"}
64,"process":"/usr/local/bin/rigctld","line":"rigctl_parse: debug3\n"}
64,"process":"/usr/local/bin/rigctld","line":"rigctl_parse: debug4 arg1= \n"}
64,"process":"/usr/local/bin/rigctld","line":"rigctl_parse: debug5\n"}
64,"process":"/usr/local/bin/rigctld","line":"rigctl_parse: debug10\n"}
64,"process":"/usr/local/bin/rigctld","line":"read_string(): RX 8 characters\n"}
64,"process":"/usr/local/bin/rigctld","line":"0000    fe fe 7a e0 1c 00 00 fd                             ..z.....        \n"}
64,"process":"/usr/local/bin/rigctld","line":"frame.c(409):read_icom_frame return(8)\n"}
64,"process":"/usr/local/bin/rigctld","line":"frame.c(375):read_icom_frame entered\n"}
64,"process":"/usr/local/bin/rigctld","line":"read_string called, rxmax=80\n"}
64,"process":"/usr/local/bin/rigctld","line":"read_string(): RX 6 characters\n"}
64,"process":"/usr/local/bin/rigctld","line":"0000    fe fe e0 7a fa fd                                   ...z..          \n"}
64,"process":"/usr/local/bin/rigctld","line":"frame.c(409):read_icom_frame return(6)\n"}
64,"process":"/usr/local/bin/rigctld","line":"frame.c(285):icom_one_transaction return(-9)\n"}
64,"process":"/usr/local/bin/rigctld","line":"icom_transaction: failed: Command rejected by the rig\n"}
64,"process":"/usr/local/bin/rigctld","line":"frame.c(409):read_icom_frame return(6)\n"}
64,"process":"/usr/local/bin/rigctld","line":"frame.c(285):icom_one_transaction return(-9)\n"}
64,"process":"/usr/local/bin/rigctld","line":"icom_transaction: failed: Command rejected by the rig\n"}
64,"process":"/usr/local/bin/rigctld","line":"frame.c(409):read_icom_frame return(6)\n"}
```

mikaelnousiainen, 2021-03-16:
@mdblack98 One thing worth checking here: the scenario where I get these errors is using two connections to rigctld, which send commands at the same time. Maybe something is broken with locking/mutexes?

I keep seeing weird errors like:

```
806,"process":"/usr/local/bin/rigctld","line":"icom_rig_open: IC-7600 v20210314.0\n"}
806,"process":"/usr/local/bin/rigctld","line":"icom_get_usb_echo_off called\n"}
806,"process":"/usr/local/bin/rigctld","line":"icom_get_usb_echo_off: retry temp set to 0\n"}
806,"process":"/usr/local/bin/rigctld","line":"frame.c(325):icom_transaction entered\n"}
806,"process":"/usr/local/bin/rigctld","line":"icom_transaction: cmd=0x03, subcmd=0xffffffff, payload_len=0\n"}
806,"process":"/usr/local/bin/rigctld","line":"frame.c(119):icom_one_transaction entered\n"}
806,"process":"/usr/local/bin/rigctld","line":"frame.c(56):make_cmd_frame entered\n"}
806,"process":"/usr/local/bin/rigctld","line":"frame.c(90):make_cmd_frame return(6)\n"}
806,"process":"/usr/local/bin/rigctld","line":"rig_flush: called for serial device\n"}
806,"process":"/usr/local/bin/rigctld","line":"serial.c(629):serial_flush entered\n"}
806,"process":"/usr/local/bin/rigctld","line":"tcflush\n"}
806,"process":"/usr/local/bin/rigctld","line":"serial.c(661):serial_flush return(0)\n"}
806,"process":"/usr/local/bin/rigctld","line":"write_block called\n"}
806,"process":"/usr/local/bin/rigctld","line":"write_block(): TX 6 bytes\n"}
806,"process":"/usr/local/bin/rigctld","line":"0000    fe fe 7a e0 03 fd                                   ..z...          \n"}
806,"process":"/usr/local/bin/rigctld","line":"frame.c(375):read_icom_frame entered\n"}
806,"process":"/usr/local/bin/rigctld","line":"read_string called, rxmax=80\n"}
806,"process":"/usr/local/bin/rigctld","line":"read_string(): RX 6 characters\n"}
806,"process":"/usr/local/bin/rigctld","line":"read_string(): read() failed - Resource temporarily unavailable\n"}
806,"process":"/usr/local/bin/rigctld","line":"frame.c(391):read_icom_frame return(-6)\n"}
806,"process":"/usr/local/bin/rigctld","line":"frame.c(254):icom_one_transaction return(-6)\n"}
806,"process":"/usr/local/bin/rigctld","line":"icom_transaction: retry=3: IO error\n"}
806,"process":"/usr/local/bin/rigctld","line":"frame.c(391):read_icom_frame return(-6)\n"}
806,"process":"/usr/local/bin/rigctld","line":"frame.c(254):icom_one_transaction return(-6)\n"}
806,"process":"/usr/local/bin/rigctld","line":"icom_transaction: retry=3: IO error\n"}
806,"process":"/usr/local/bin/rigctld","line":"frame.c(391):read_icom_frame return(-6)\n"}
806,"process":"/usr/local/bin/rigctld","line":"frame.c(254):icom_one_transaction return(-6)\n"}
806,"process":"/usr/local/bin/rigctld","line":"frame.c(254):icom_one_transaction return(-6)\n"}
806,"process":"/usr/local/bin/rigctld","line":"0000    fe fe 7a e0 03 fd                                   ..z...          \n"}
806,"process":"/usr/local/bin/rigctld","line":"frame.c(409):read_icom_frame return(6)\n"}
806,"process":"/usr/local/bin/rigctld","line":"icom_one_transaction: frm_len=6, frm_len-1=fd, frm_len-2=03\n"}
806,"process":"/usr/local/bin/rigctld","line":"frame.c(303):icom_one_transaction return(0)\n"}
806,"process":"/usr/local/bin/rigctld","line":"frame.c(355):icom_transaction return(0)\n"}
806,"process":"/usr/local/bin/rigctld","line":"icom_get_usb_echo_off: ack_len=1\n"}
806,"process":"/usr/local/bin/rigctld","line":"icom_get_usb_echo_off: USB echo on detected\n"}
```

mdblack98, 2021-03-16:
In this sequence all that it's doing is asking for frequency....please add the -Z switch wo we can see timing.I suspect this is some behavior of your serial port needing a pause after opening it.  Does your cable need DTR or RTS set for powering it?What kind of serial adapter do you have?
Mike W9MDB

mikaelnousiainen, 2021-03-16:
@mdblack98 The IC-7600 here is connected directly via the native rig USB port (no special RTS/DTR stuff needed). And this setup has been working fine for a long time, but I started getting these errors yesterday, after upgrading Hamlib to latest master.

There are two rigctld connections here:
1) Remote control app polling the rig (e.g. reading freq/mode etc)
2) WSJT-X doing its things

When I'm running 1) only, I don't get any errors.

mdblack98, 2021-03-16:
Need more debug before and this and the -Z switch to see timing.
This is a valid PTT=OFF command so it would appear the rig is rejecting it because of whatever happened just before this.The NAK wasn't being detected correctly before the latest commits so we may be discovering some timing behavior in the 7600.
Mike W9MDB

mdblack98, 2021-03-16:
Please use -Z and provide more debug info before the NAK error occurs.
As for the serial error it does appear to be recovering from it so not sure there's anything to fix there...but you didn't provide enough debug to see where it recovered.
Mike W9MDB

mikaelnousiainen, 2021-03-16:
@mdblack98 Ok, now I've got more logs with the timestamps.

Some questions first though:

1. Would the current Hamlib master be somehow incompatible with WSJT-X v2.2.2?

2. I'm seeing rig close/open calls in the logs and I'm not sure why would that need to happen...

This is the command-line output from WSJT-X -- why does WSJT-X complain about unknown settings?

```
Hamlib: netrigctl_open: unknown setting='targetable_vfo=0x0'
Hamlib: netrigctl_open: unknown setting='has_set_vfo=1'
Hamlib: netrigctl_open: unknown setting='has_get_vfo=0'
Hamlib: netrigctl_open: unknown setting='has_set_freq=1'
Hamlib: netrigctl_open: unknown setting='has_get_freq=1'
Hamlib: rig_set_ptt: return code=-5
Hamlib: netrigctl_close: done status=Command completed successfully
```

And here's a log containing WSJT-X Hamlib commands and responses, which I proxy through a WebSocket connection. The commands start failing near the end, as you'll see:

```
{"level":"debug","ts":"2021-03-16T17:12:17.075+0200",,"msg":"Command from Hamlib client","command":"\\chk_vfo\n"}
{"level":"debug","ts":"2021-03-16T17:12:17.127+0200",,"msg":"Response from Hamlib server","response":"0\n"}
{"level":"debug","ts":"2021-03-16T17:12:17.128+0200",,"msg":"Command from Hamlib client","command":"\\dump_state\n"}
{"level":"debug","ts":"2021-03-16T17:12:17.184+0200",,"msg":"Response from Hamlib server","response":"1\n"}
{"level":"debug","ts":"2021-03-16T17:12:17.194+0200",,"msg":"Response from Hamlib server","response":"3063\n"}
{"level":"debug","ts":"2021-03-16T17:12:17.195+0200",,"msg":"Response from Hamlib server","response":"0\n"}
{"level":"debug","ts":"2021-03-16T17:12:17.195+0200",,"msg":"Response from Hamlib server","response":"30000.000000 60000000.000000 0xc0401dbf -1 -1 0x16000000 0x3\n"}
{"level":"debug","ts":"2021-03-16T17:12:17.195+0200",,"msg":"Response from Hamlib server","response":"0 0 0 0 0 0 0\n"}
{"level":"debug","ts":"2021-03-16T17:12:17.195+0200",,"msg":"Response from Hamlib server","response":"1810000.000000 2000000.000000 0xc0001dbf 2000 100000 0x16000000 0x3\n"}
{"level":"debug","ts":"2021-03-16T17:12:17.195+0200",,"msg":"Response from Hamlib server","response":"3500000.000000 3800000.000000 0xc0001dbf 2000 100000 0x16000000 0x3\n"}
{"level":"debug","ts":"2021-03-16T17:12:17.195+0200",,"msg":"Response from Hamlib server","response":"7000000.000000 7100000.000000 0xc0001dbf 2000 100000 0x16000000 0x3\n"}
{"level":"debug","ts":"2021-03-16T17:12:17.195+0200",,"msg":"Response from Hamlib server","response":"10100000.000000 10150000.000000 0xc0001dbf 2000 100000 0x16000000 0x3\n"}
{"level":"debug","ts":"2021-03-16T17:12:17.195+0200",,"msg":"Response from Hamlib server","response":"14000000.000000 14350000.000000 0xc0001dbf 2000 100000 0x16000000 0x3\n"}
{"level":"debug","ts":"2021-03-16T17:12:17.252+0200",,"msg":"Response from Hamlib server","response":"18068000.000000 18168000.000000 0xc0001dbf 2000 100000 0x16000000 0x3\n"}
{"level":"debug","ts":"2021-03-16T17:12:17.252+0200",,"msg":"Response from Hamlib server","response":"21000000.000000 21450000.000000 0xc0001dbf 2000 100000 0x16000000 0x3\n"}
{"level":"debug","ts":"2021-03-16T17:12:17.252+0200",,"msg":"Response from Hamlib server","response":"24890000.000000 24990000.000000 0xc0001dbf 2000 100000 0x16000000 0x3\n"}
{"level":"debug","ts":"2021-03-16T17:12:17.252+0200",,"msg":"Response from Hamlib server","response":"28000000.000000 29700000.000000 0xc0001dbf 2000 100000 0x16000000 0x3\n"}
{"level":"debug","ts":"2021-03-16T17:12:17.252+0200",,"msg":"Response from Hamlib server","response":"50000000.000000 54000000.000000 0xc0001dbf 2000 100000 0x16000000 0x3\n"}
{"level":"debug","ts":"2021-03-16T17:12:17.252+0200",,"msg":"Response from Hamlib server","response":"1810000.000000 2000000.000000 0x400001 1000 30000 0x16000000 0x3\n"}
{"level":"debug","ts":"2021-03-16T17:12:17.252+0200",,"msg":"Response from Hamlib server","response":"3500000.000000 3800000.000000 0x400001 1000 30000 0x16000000 0x3\n"}
{"level":"debug","ts":"2021-03-16T17:12:17.252+0200",,"msg":"Response from Hamlib server","response":"7000000.000000 7100000.000000 0x400001 1000 30000 0x16000000 0x3\n"}
{"level":"debug","ts":"2021-03-16T17:12:17.252+0200",,"msg":"Response from Hamlib server","response":"10100000.000000 10150000.000000 0x400001 1000 30000 0x16000000 0x3\n"}
{"level":"debug","ts":"2021-03-16T17:12:17.252+0200",,"msg":"Response from Hamlib server","response":"14000000.000000 14350000.000000 0x400001 1000 30000 0x16000000 0x3\n"}
{"level":"debug","ts":"2021-03-16T17:12:17.252+0200",,"msg":"Response from Hamlib server","response":"18068000.000000 18168000.000000 0x400001 1000 30000 0x16000000 0x3\n"}
{"level":"debug","ts":"2021-03-16T17:12:17.253+0200",,"msg":"Response from Hamlib server","response":"21000000.000000 21450000.000000 0x400001 1000 30000 0x16000000 0x3\n"}
{"level":"debug","ts":"2021-03-16T17:12:17.253+0200",,"msg":"Response from Hamlib server","response":"24890000.000000 24990000.000000 0x400001 1000 30000 0x16000000 0x3\n"}
{"level":"debug","ts":"2021-03-16T17:12:17.253+0200",,"msg":"Response from Hamlib server","response":"28000000.000000 29700000.000000 0x400001 1000 30000 0x16000000 0x3\n"}
{"level":"debug","ts":"2021-03-16T17:12:17.253+0200",,"msg":"Response from Hamlib server","response":"50000000.000000 54000000.000000 0x400001 1000 30000 0x16000000 0x3\n"}
{"level":"debug","ts":"2021-03-16T17:12:17.253+0200",,"msg":"Response from Hamlib server","response":"0 0 0 0 0 0 0\n"}
[... cut ...]
{"level":"debug","ts":"2021-03-16T17:12:20.449+0200",,"msg":"Command from Hamlib client","command":"q\n"}
{"level":"debug","ts":"2021-03-16T17:12:21.817+0200",,"msg":"Response from Hamlib server","response":"RPRT 0\n"}
```

I've attached the full rigctld log around the failing commands. Search for `set_ptt` around 15:12:20, timestamps are in UTC in the rigctld log file:

[wrc-hamlib-rigctld-error-1-snippet.log.gz](https://github.com/Hamlib/Hamlib/files/6150204/wrc-hamlib-rigctld-error-1-snippet.log.gz)

mikaelnousiainen, 2021-03-16:
@mdblack98 This is the first error that I see in the logs:

```
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.888218: sync_callback: client lock engaged\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.888291: rigctl(d): l 'currVFO' 'VOXDELAY' '' ''\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.888365: rigctl_parse: vfo_opt=0\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.888431: rigctl_parse.c(3040):rigctl_get_level entered\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.888506: rig_parse_level called\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.888580: icom_get_level called\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.888641: icom_get_level: i=0\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.888700: icom_get_level: i=1\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.888755: icom_get_level: i=2\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.888807: icom_get_level: i=3\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.888867: icom_get_cmd called\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.888933: icom_get_raw_buf called\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.888998: frame.c(325):icom_transaction entered\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.889072: icom_transaction: cmd=0x1a, subcmd=0x05, payload_len=2\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.889152: frame.c(119):icom_one_transaction entered\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.889228: frame.c(56):make_cmd_frame entered\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.889292: frame.c(90):make_cmd_frame return(9)\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.889356: rig_flush: called for serial device\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.889422: serial.c(629):serial_flush entered\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.889475: tcflush\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.889563: serial.c(661):serial_flush return(0)\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.889635: write_block called\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.889857: write_block(): TX 9 bytes\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.893926: 0000    fe fe 7a e0 1a 05 01 67 fd                          ..z....g.       \n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.894020: frame.c(375):read_icom_frame entered\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.894100: read_string called, rxmax=80\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.896952: read_string(): RX 9 characters\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.897489: 0000    fe fe 7a e0 1a 05 01 67 fd                          ..z....g.       \n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.897568: frame.c(409):read_icom_frame return(9)\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.897654: frame.c(375):read_icom_frame entered\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.897795: read_string called, rxmax=80\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.911840: read_string(): RX 10 characters\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.912004: 0000    fe fe e0 7a 1a 05 01 67 02 fd                       ...z...g..      \n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.912082: frame.c(409):read_icom_frame return(10)\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.912158: icom_one_transaction: frm_len=10, frm_len-1=fd, frm_len-2=02\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.912229: frame.c(303):icom_one_transaction return(0)\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.912297: frame.c(355):icom_transaction return(0)\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.912359: icom.c(7399):icom_get_raw_buf return(-5)\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.912496: icom.c(2403):icom_get_cmd return(-5)\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.912565: icom.c(2861):icom_get_level return(-5)\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.912634: rigctl_parse.c(3114):rigctl_get_level return(-5)\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.912698: rigctl_parse: vfo_opt=0\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.912754: rigctl_parse: return#1 RPRT -5\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.913085: rigctl_parse: retcode=-5\n"}
02,"process":"/usr/local/bin/rigctld","line":"2021-03-16:15:12:14.913170: sync_callback: client lock disengaged\n"}
```

The command succeeds, but `icom_get_raw_buf` has some new code (commit ef2e8a4d94d854c179e0ca16beef8c4147ee6eea) that does a check for `ACK` + `NAK` bytes. I'm not sure how that could ever work with get commands that don't return the ACK/NAK bytes?

mikaelnousiainen, 2021-03-16:
@mdblack98 Additionally, both of these checks in `icom_get_raw_buf` may read outside `ack_len` (I'm aware that the same checks are present in the code in many other places):

1. `if (ack_len >= 1 && ackbuf[0] != ACK && ackbuf[1] != NAK)` <- `ack_len` may be 1, but `ackbuf[1]` still accessed
2. `if (ack_len != 1 || ackbuf[0] != ACK)` <- `ack_len` may be 0, but `ackbuf[0]` still accessed

mikaelnousiainen, 2021-03-16:
@mdblack98 Thanks for the changes, but we still need to remove both of these if-conditions from these functions `icom_get_raw_buf` and `icom_get_powerstat`. The error I'm getting is specifically matching the first condition (-RIG_ETIMEOUT == -5) and the second condition won't work with get responses either.

```
        if ((ack_len >= 1 && ackbuf[0] != ACK) && (ack_len >= 2 && ackbuf[1] != NAK))
        {
            //  if we don't get ACK/NAK some serial corruption occurred
            // so we'll call it a timeout for retry purposes
            RETURNFUNC(-RIG_ETIMEOUT);
        }

        if (ack_len != 1 || (ack_len >= 1 && ackbuf[0] != ACK))
        {
            rig_debug(RIG_DEBUG_ERR, "%s: ack NG (%#.2x), len=%d\n", __func__,
                      ackbuf[0], ack_len);
            RETURNFUNC(-RIG_ERJCTED);
        }
```

mikaelnousiainen, 2021-03-16:
@mdblack98 Digging deeper. I tested locally by removing the code mentioned in my previous comment and it certainly gets rid of errors with levels. However, the underlying reason for these errors seems to be the following:

1) My remote control app is running commands that result in "soft" errors (e.g. get_level for a level that is not implemented by a backend) -- which should be totally OK and not cause any side-effects

2) However, `rigctl_parse` returns a special value `2` in case of *ANY* error, see: https://github.com/Hamlib/Hamlib/blob/27192d0c813aa79c6eb09d85077c7a0fae5489e7/tests/rigctl_parse.c#L1735

3) This will cause the rigctld (and plain rigctl too!) to close and re-open the rig, see: https://github.com/Hamlib/Hamlib/blob/27192d0c813aa79c6eb09d85077c7a0fae5489e7/tests/rigctld.c#L1092 and https://github.com/Hamlib/Hamlib/blob/27192d0c813aa79c6eb09d85077c7a0fae5489e7/tests/rigctl.c#L621

4) The rigctld network client calling the soft-failing command will be fine, but `rig_open` and `rig_close` will cause issues for any other rigctld clients -- leading to errors during the time the rig is closed.

What I think should be changed (at least) is the return value logic here:

https://github.com/Hamlib/Hamlib/blob/27192d0c813aa79c6eb09d85077c7a0fae5489e7/tests/rigctl_parse.c#L1735

Could we simply do `RETURNFUNC(retcode);` to make the full return code information available for rigctl loops?

mikaelnousiainen, 2021-03-16:
@mdblack98 Of course returning the error code breaks the rigctl loops, as they expect certain subset of values only.

I can try to push a fix for this issue the following way:

1) Remove the "magic" return value 2 (if it's not used elsewhere)

2) Add a concept of a soft error, which would include at least error codes "invalid parameter", "function not available" and "function not implemented" -- in these cases the rigctl loop would continue running and NOT try to re-open the rig.

Sounds good?

mikaelnousiainen, 2021-03-16:
@mdblack98 I've got an initial fix in this branch: https://github.com/mikaelnousiainen/Hamlib/tree/fix-unnecessary-rig-reopen -- will push out a PR once I've tested it at least briefly.

mikaelnousiainen, 2021-03-17:
Initial tests look good, I'll have a couple of rounds of testing today before creating a PR

mikaelnousiainen, 2021-03-17:
PR #614 out, closing this, so we can continue discussion / iterate there if more changes are needed.

closed 2021-03-17 by mikaelnousiainen