Hamlib/Hamlib #613: newly upgraded system still won't connect.

kb3mkd, 2021-03-16:
WSJT-X v2.3.0
Linux Mint 20
Hamlib 3.3-10build1
Getting a rig failure. IO error
Settings serial port /dev/tty/USB0
8-N-1
PTT method CAT
Mode USB
Split rig

mdblack98, 2021-03-16:
Please upgrade hamlib to at least 4.1Seems
Mike W9MDB

kb3mkd, 2021-03-16:
Is there a debian / ubuntu / mint repository or do I have to compile it?

kb3mkd, 2021-03-17:
I tried compiling it, I get 
Hamlib error:                                                                         IO error
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                 [... cut ...]                                                                                                                                                                                                                                                                                                                                                                                                                                                                                           rig.c(745):rig_open return

mdblack98, 2021-03-18:
Are you sure you have access to the port?  Try as root....

closed 2021-03-21 by mdblack98