Hamlib/Hamlib #634: LIBUSB problem

mdblack98, 2021-03-27:
it looks like the above commit causes critical libusb code to be #ifdef
excluded, I am seeing this on MS Windows MinGW builds. The configuration
macro HAVE_LIBUSB_H is being used and that is not defined, perhaps
HAVE_LIBUSB_1_0_LIBUSB_H should have been used? Offending code is at
src/iofunc.c lines 149 and 206. Also src/rotator.c lines 64, 421, and 533.

73
Bill
G4WJS.

mdblack98, 2021-03-27:
I take you don't see any LIBUSB defined in include/config.h ?
Mike W9MDB

g4wjs, 2021-03-27:
> I take you don't see any LIBUSB defined in include/config.h ? Mike W9MDB

/* Define if libusb-1.0 is available */
#define HAVE_LIBUSB 1

/* Define to 1 if you have the <libusb-1.0/libusb.h> header file. */
#define HAVE_LIBUSB_1_0_LIBUSB_H 1

/* Define to 1 if you have the <libusb.h> header file. */
/* #undef HAVE_LIBUSB_H */

73
Bill
G4WJS.

mdblack98, 2021-03-27:
Hopefully fixed in my latest patch.
Mike

closed 2021-03-28 by mdblack98