Hamlib/Hamlib #645: Malachite SDR TS-480 not working

mdblack98, 2021-04-04:
Malachite SDR uses TS-480 emulation but does not work with set_freq in hamlib.
Seems SDR does not like having the ID command sent immediately after a set command (at least FA/FB)
FA0000185000;ID;

So we'll add new parm NO_ID that can be set true that will not send the ID; with every set command

mdblack98, 2021-04-06:
Malachite firmware has a bug when changing across HF/VHF/UHF/GHz where it doesn't change frequency and you have to select a different frequency and then your target to get it to change.  I'm not inclined to fix a firmware bug like this in hamlib.

timo12357, 2021-07-01:
Hi, I volunteer to test this wonderful fix.
I have the following equipment
- Malachite with version 1.10c Firmware
- Ubuntu 21.04 with libhamlib-utils 4.0 installed with APT.

How to test?

mdblack98, 2021-07-01:
Since it doesn't appear there are any Ubuntu packages > 4.0 you would have to compile yourself.
Remove your current hamlib package

You can download 4.2 from here:
https://github.com/Hamlib/Hamlib/releases/download/4.2/hamlib-4.2.tar.gz

tar xzf hamlib-4.2.tar.gzcd hamlib-4.2./bootstrap./configure --prefix=/usrmake install

Mike W9MDB

timo12357, 2021-07-02:
Hello,
1st test report:

Did the following

1. wget https://github.com/Hamlib/Hamlib/releases/download/4.2/hamlib-4.2.tar.gz: OK
2. sudo apt purge libhamlib-utils: OK
3. tar xzf hamlib-4.2.tar.gz: OK
4. cd hamlib-4.2/: OK
5. ./bootstrap: No such file
6. ./configure --prefix=/usr: OK
7. make: OK
8. sudo make install: OK
9. rigctl -l: rigctl: symbol lookup error: rigctl: undefined symbol: debugmsgsave

73, Timo

mdblack98, 2021-07-02:
You've got an old library sitting around.
What does "ldd rigctl" show for the path to libhamlib.so?
Mike

 
 

   
   - wget https://github.com/Hamlib/Hamlib/releases/download/4.2/hamlib-4.2.tar.gz: OK
   - cd hamlib-4.2/: OK
   - ./bootstrap: No such file
   - make: OK
   - sudo make install: OK

timo12357, 2021-07-02:
> What does "ldd rigctl" show for the path to libhamlib.so?

sudo ldd rigctl: ldd: ./rigctl: No such file or directory
sudo ldd libhamlib.so: ldd: ./libhamlib.so: No such file or directory

Timo

mdblack98, 2021-07-02:
Well you apparently aren't in the right place for rigctl
which rigctl/usr/bin/rigctl
ldd /usr/bin/rigctl
Mike W9MDB

timo12357, 2021-07-05:
> Well you apparently aren't in the right place for rigctl 

which rigctl:
/usr/bin/rigctl

ldd /usr/bin/rigctl
	linux-vdso.so.1 (0x00007ffc3512f000)
	libhamlib.so.4 => /lib/x86_64-linux-gnu/libhamlib.so.4 (0x00007f1b9af26000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f1b9af04000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f1b9ad18000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f1b9ad11000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f1b9abc3000)
	libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007f1b9aba6000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f1b9b8d2000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007f1b9ab7b000)

73

mdblack98, 2021-07-05:
I'll bet you also have a libhamlib in /usr/libThat is where it is usually installed when building.
You need to remove your system hamlib package.
Or rename /lib/x86_64-linux-gnu/libhamlib.so.4
Mike W9MDB

timo12357, 2021-07-06:
> I'll bet you also have a libhamlib in /usr/lib
Yes, that is right.

>You need to remove your system hamlib package. 
Removing it removes also all my ham software, so not an option at this stage. Tried it and had to reinstall.

>Or rename /lib/x86_64-linux-gnu/libhamlib.so.4
$ sudo mv /usr/lib/x86_64-linux-gnu/libhamlib.so.4 /usr/lib/x86_64-linux-gnu/libhamlib.so.4.old

$ ./configure --prefix=/usr: OK
$ make: OK
$ sudo make install: OK
$ rigctl -l now lists available rigs OK

Is there a way I can see what version of Hamlib I am using?

Tests
$ grig -m 2028 -r /dev/ttyACM1 -s 19200 -d 3
- Shows frequency of the Malahit SDR, but can not change anything on the SDR

flrig
- Selected TS480SAT
- Same functionality as with grig

 73, Timo

mdblack98, 2021-07-07:
rigctl --version
Should show the shared library version.
Do this and send me the log.txt
rigctl -m 2028 -r /dev/ttyACM1 -s 19200 -vvvvvv F 14074123
Mike W9MDB

timo12357, 2021-07-07:
>

>  rigctl --version Should show the shared library version. 

$ rigctl --version
rigctl Hamlib 4.2 Mon May 17 02:40:24 2021 +0000 SHA=cdad07

> Do this and send me the log.txt 

$ rigctl -m 2028 -r /dev/ttyACM1 -s 19200 -vvvvvv F 14074123 
rigctl Hamlib 4.2 Mon May 17 02:40:24 2021 +0000 SHA=cdad07
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_check_rig_caps: p1=0x7f6307a79a80, p2=0x7f6307a7fdf8, rig_model=0x7f6307a79a80, macro_name=0x7f6307a7fdf8
initrigs4_kenwood called
rig_register called
rig_register: rig_register (2012)
rig_register called
rig_register: rig_register (2013)
rig_register called
rig_register: rig_register (2001)
rig_register called
rig_register: rig_register (2025)
rig_register called
rig_register: rig_register (2003)
rig_register called
rig_register: rig_register (2004)
rig_register called
rig_register: rig_register (2016)
rig_register called
rig_register: rig_register (2024)
rig_register called
rig_register: rig_register (2005)
rig_register called
rig_register: rig_register (2007)
rig_register called
rig_register: rig_register (2009)
rig_register called
rig_register: rig_register (2010)
rig_register called
rig_register: rig_register (2022)
rig_register called
rig_register: rig_register (2014)
rig_register called
rig_register: rig_register (2030)
rig_register called
rig_register: rig_register (2021)
rig_register called
rig_register: rig_register (2029)
rig_register called
rig_register: rig_register (2043)
rig_register called
rig_register: rig_register (2044)
rig_register called
rig_register: rig_register (2045)
rig_register called
rig_register: rig_register (2047)
rig_register called
rig_register: rig_register (2038)
rig_register called
rig_register: rig_register (2002)
rig_register called
rig_register: rig_register (2011)
rig_register called
rig_register: rig_register (2006)
rig_register called
rig_register: rig_register (2008)
rig_register called
rig_register: rig_register (2015)
rig_register called
rig_register: rig_register (2026)
rig_register called
rig_register: rig_register (2017)
rig_register called
rig_register: rig_register (2033)
rig_register called
rig_register: rig_register (2042)
rig_register called
rig_register: rig_register (2020)
rig_register called
rig_register: rig_register (2023)
rig_register called
rig_register: rig_register (2027)
rig_register called
rig_register: rig_register (2034)
rig_register called
rig_register: rig_register (2031)
rig_register called
rig_register: rig_register (2039)
rig_register called
rig_register: rig_register (2037)
rig_register called
rig_register: rig_register (2028)
rig_register called
rig_register: rig_register (2019)
rig_register called
rig_register: rig_register (2032)
rig_register called
rig_register: rig_register (2036)
rig_register called
rig_register: rig_register (2048)
rig_register called
rig_register: rig_register (2040)
rig_register called
rig_register: rig_register (2041)
rig_register called
rig_register: rig_register (2046)
rig_register called
rig_register: rig_register (2049)
kenwood.c(5351):initrigs4_kenwood return(0)
rig_init: rig has VFO_A
rig_init: rig has VFO_B
ts480_init called
kenwood_init called, version 20210427/20210427.1
kenwood_init: if_len = 37
kenwood.c(738):kenwood_init return(0)
ts480.c(1247):ts480_init return(0)
rigctl_parse.c(1916):set_conf entered
rigctl_parse.c(1950):set_conf return(0)
rig.c(669):rig_open entered
iofunc.c(68):port_open entered
serial.c(144):serial_open entered
serial_open: /dev/ttyACM1
serial.c(278):serial_setup entered
serial_setup: tcgetattr
serial_setup: cfmakeraw
serial_setup: cfsetispeed=19200,0x000e
serial_setup: cfsetospeed=19200,0x000e
serial_setup: data_bits=8
serial_setup: parity=0
serial_setup: tcsetattr TCSANOW
serial.c(618):serial_setup return(0)
serial.c(629):serial_flush entered
tcflush
serial.c(661):serial_flush return(0)
serial.c(253):serial_open return(0)
iofunc.c(185):port_open return(0)
rig.c(217):add_opened_rig return(0)
kenwood_open called
kenwood_get_id called
kenwood_transaction called
kenwood_transaction: cmdstr = ID
rig_flush: called for serial device
serial.c(629):serial_flush entered
tcflush
serial.c(661):serial_flush return(0)
write_block called
write_block(): TX 3 bytes
0000    49 44 3b                                            ID;             
read_string called, rxmax=128
read_string(): Timed out 0.200 seconds after 0 chars
kenwood_transaction: read_string(len=0)=''
kenwood_transaction: read_string retval < 0, retval = -5, retry_read=0, rs->rigport.retry=0
kenwood_transaction: returning retval=-5
kenwood.c(577):kenwood_transaction return(-5)
kenwood.c(1014):kenwood_get_id return(-5)
kenwood_open: no response to get_id from rig...continuing anyways.
kenwood_open: rig has_rit2
kenwood_transaction called
kenwood_transaction: cmdstr = FA
rig_flush: called for serial device
serial.c(629):serial_flush entered
tcflush
serial.c(661):serial_flush return(0)
write_block called
write_block(): TX 3 bytes
0000    46 41 3b                                            FA;             
read_string called, rxmax=128
read_string(): Timed out 0.200 seconds after 0 chars
kenwood_transaction: read_string(len=0)=''
kenwood_transaction: read_string retval < 0, retval = -5, retry_read=0, rs->rigport.retry=0
kenwood_transaction: returning retval=-5
kenwood.c(577):kenwood_transaction return(-5)
kenwood_open: no response from rig
kenwood.c(856):kenwood_open return(-5)
rig.c(1000):rig_open return(-5)
rig_open: error = Communication timed out
kenwood_open: no response from rig
kenwood.c(856):kenwood_open return(-5)
rig.c(1000):rig_open return(-5) 

73, Timo OH2LVZ

timo12357, 2021-07-07:
Here is another listing using /dev/ttyACM0
$ rigctl -m 2028 -r /dev/ttyACM0 -s 19200 -vvvvvv F 14074123 
rigctl Hamlib 4.2 Mon May 17 02:40:24 2021 +0000 SHA=cdad07
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_check_rig_caps: p1=0x7f8682a6da80, p2=0x7f8682a73df8, rig_model=0x7f8682a6da80, macro_name=0x7f8682a73df8
initrigs4_kenwood called
rig_register called
rig_register: rig_register (2012)
rig_register called
rig_register: rig_register (2013)
rig_register called
rig_register: rig_register (2001)
rig_register called
rig_register: rig_register (2025)
rig_register called
rig_register: rig_register (2003)
rig_register called
rig_register: rig_register (2004)
rig_register called
rig_register: rig_register (2016)
rig_register called
rig_register: rig_register (2024)
rig_register called
rig_register: rig_register (2005)
rig_register called
rig_register: rig_register (2007)
rig_register called
rig_register: rig_register (2009)
rig_register called
rig_register: rig_register (2010)
rig_register called
rig_register: rig_register (2022)
rig_register called
rig_register: rig_register (2014)
rig_register called
rig_register: rig_register (2030)
rig_register called
rig_register: rig_register (2021)
rig_register called
rig_register: rig_register (2029)
rig_register called
rig_register: rig_register (2043)
rig_register called
rig_register: rig_register (2044)
rig_register called
rig_register: rig_register (2045)
rig_register called
rig_register: rig_register (2047)
rig_register called
rig_register: rig_register (2038)
rig_register called
rig_register: rig_register (2002)
rig_register called
rig_register: rig_register (2011)
rig_register called
rig_register: rig_register (2006)
rig_register called
rig_register: rig_register (2008)
rig_register called
rig_register: rig_register (2015)
rig_register called
rig_register: rig_register (2026)
rig_register called
rig_register: rig_register (2017)
rig_register called
rig_register: rig_register (2033)
rig_register called
rig_register: rig_register (2042)
rig_register called
[... 325 lines cut ...]
set_cache_freq(1427): freqMainB=0, modeMainB=, widthMainB=0
set_cache_freq: set vfo=VFOA to freq=103400000
set_cache_freq(1513): freqMainA=103400000, modeMainA=, widthMainA=0
set_cache_freq(1513): freqMainB=0, modeMainB=, widthMainB=0
rig.c(1514):set_cache_freq return(0)
rig.c(1878):rig_set_freq return(0)
rigctl_parse.c(1980):rigctl_set_freq return(0)
rigctl_parse: vfo_opt=0
rigctl_parse: retcode=0
rigctl_parse.c(1743):rigctl_parse return(0)
rigctl_parse.c(648):rigctl_parse entered
rigctl_parse: called, interactive=0
rigctl_parse.c(816):rigctl_parse return(1)
rig.c(1092):rig_close entered
iofunc.c(199):port_close entered
ser_close called
ser_close: restoring options
serial.c(818):ser_close return(0)
iofunc.c(233):port_close return(0)
rig.c(240):remove_opened_rig return(0)
rig.c(1226):rig_close return(0)
rig.c(1245):rig_cleanup entered
kenwood_cleanup called
kenwood.c(748):kenwood_cleanup return(0)
rig.c(1270):rig_cleanup return(0)

timo12357, 2021-07-14:
@mdblack98 any use of these reports?

mdblack98, 2021-07-15:
The master repo has model# 2049 now for the Malahit DSP.
Please give it a try.
Seems to be working but you can't set mode on the thing..just read mode.  I've left the set_mode enabled as perhaps they'll fix that one day.
Mike W9MDB

timo12357, 2021-07-17:
I did the following

$ sudo apt purge libhamlib-utils
$ gh repo clone Hamlib/Hamlib
$ cd Hamlib
$ ./bootstrap 
$ ./configure
$ make check
============================================================================
Testsuite summary for Hamlib 4.3~git
============================================================================
# TOTAL: 7
# PASS:  7
# SKIP:  0
# XFAIL: 0
# FAIL:  0
# XPASS: 0
# ERROR: 0

$ make
$ sudo make install
$ rigctl
rigctl: symbol lookup error: rigctl: undefined symbol: cookie_use

73, Timo OH2LVZ

mdblack98, 2021-07-17:
That indicates you have a old version of hamlib installed.
Remove any hamlib package you have.
You can also install over the top of any installed package
./configure --prefix=/usr
Mike W9MDB

closed 2021-07-20 by mdblack98

timo12357, 2021-07-20:
Hi
did 
$ make clean
$ ./configure --prefix=/usr
$ make
$ sudo make install
Now rigctl starts without errors

flrig works with TS-480SAT. Malachite can not be found. 
Frequency freely until you choose a frequency in the FM BC band. There frequency can not be set

mdblack98, 2021-07-20:
Hamlib has the Malachite -- try using that and ensure it works.

I'm able to set all the FM broadcast frequencies with that.
Mike W9MDB