Hamlib/Hamlib #652: Debug tracing

mdblack98, 2021-04-07:
See if we can provide a "level" indication in the debug messages to show the function call depth.  Would make reviewing the debug info easier to trace.

So level 1 at top and each level down would increment by 1 on entry and decrement by 1 on exit.
This can probably be done via the ENTERFUNC and RETURNFUNC.  But RETURNFUNC may be called without an associated ENTERFUNC and vice versa so a state machine will be needed to determine when to increment/decrement.
ENTERFUNC would be if (state==RETURNFUNC) ++depth
RETURNFUNC would be if (state == ENTERRFUNC) ++ depth
ENTERFUNC would set ENTERFUNC state
RETURNFUNC would set RETURNFUNC state
get_depth function/macro would return depth when state==ENTERFUNC and depth+1 when state==RETURNFUNC
ENTERFUNC would only increment depth if state==RETURNFUNC
RETURNFUNC would only decrement depth if state==ENTERFUNC

gvanem, 2022-06-06:
And all instances of `rig_debug(RIG_DEBUG_VERBOSE, "%s: _init called\n", __func__);`
should IMHO be replaced with `ENTERFUNC2` (or better yet  `ENTERFUNC2()`).

I find that there is a huge amount of code being run when debug is disabled! E.g. in ` rig_send_raw()`, the preprocessor
on `clang-cl` spits out these lined (edited and Astyled):
```c
  do // for ENTERFUNC
  {
    ++rig->state.depth;

    do
    {
      strcpy (debugmsgsave2, debugmsgsave);
      snprintf (debugmsgsave, sizeof (debugmsgsave), "%.*s%d:%s(%d):%s entered\n", rig->state.depth, spaces(), rig->state.depth, (strrchr ("src/rig.c",
                '/') ? strrchrfile, '/') + 1 : "src/rig.c"), 7285, __func__);
      (rig_debug) (RIG_DEBUG_VERBOSE, "%.*s%d:%s(%d):%s entered\n", rig->state.depth, spaces(), rig->state.depth, (strrchr ("src/rig.c", '/') ? strrchrfile,
                   '/') + 1 : "src/rig.c"), 7285, __func__);
    }
    while (0);
  }
  while (0);

  int retval = write_block_sync (&rs->rigport, send, send_len);

  if (retval < 0)
  {
    do  // For rig_debug():
    {
      strcpy (debugmsgsave2, debugmsgsave);
      snprintf (debugmsgsave, sizeof (debugmsgsave), "%s: write_block_sync() failed, result=%d\n", __func__, retval);
      (rig_debug) (RIG_DEBUG_ERR, "%s: write_block_sync() failed, result=%d\n", __func__, retval);
    }
    while (0);
  }
```

mdblack98, 2022-06-06:
This provides intelligent debug information for apps like WSJTX/JTDX and others when errors occur.  The strerror() function returns  the debug history even if it's not turned on.
At this point I don't intend to eliminate this.

gvanem, 2022-06-06:
> ... function returns the debug history even if it's not turned on.

Into `debugmsgsave[]`? But what are `debugmsgsave2[]` and `debugmsgsave3[]` used for?
But I think `strerror()` is wrongly used on Windows in many places.

mdblack98, 2022-06-06:
save2 and save3 are deprecated.  Will be removed eventually.
Where do think strerror is used incorrectly?  Quite possible...

mdblack98, 2022-10-11:
I found one and fixed it.
Which two did you find?
Mike

 
 

2 places in src/network.c for example.