Hamlib/Hamlib #656: Hamlib 4.1 -m fails with Kenwood radios

kc9xg, 2021-04-10:
I am having an issue with HamLib 4.1 installed on RPi4B with a current version of Raspberry Pi OS.

RigCtl  rigctl -m 2026 -r /dev/ttyUSB0 -s 9600 -t 4532 
Fails with:
 rig_open: error = Protocol error
     rig.c(972):rig_open 

  The -m ID parameter fails with the rig_open error when using assigned id's within the range of 2000-2999, but appears to succeed with assigned values values outside that range, such as 1020 and 3011.  

   If an unassigned ID number in the 2000-2999 range is used the follow error is noted:
Unknown rig num 2099, or initialization error.

TIA,
Bill

mdblack98, 2021-04-11:
Well rig# 2099 is unknown.  Numbers in the 2000 range only run from 2000-2049
#2026 is a TMD-700 which I doubt you have.
This lists the available rigs and their model numbersrigctl -l

Mike W9MDB

kc9xg, 2021-04-11:
Mike
    As I noted (with emphasis added):
   "The -m ID parameter fails with the rig_open error when using 
*assigned id's* *within the range of 2000-2999*, but appears to succeed 
with *assigned* *values outside that range*, such as 1020 and 3011."

   I referred to *assigned* id's as radio model ID's listed using rigctl 
-l. Sorry for any confusion.

    IOW, HamLib 4.1 does not allow me to select radios/hardware/software 
in my shack using *assigned* ID's in the 2000-2999 numeric range, 
including 2026 Kenwood D700, 2038 Elecraft XG3, 2048 FlexRadio/ANAN 
PowerSDR/Thetis but I can select my Yaesu FT817 using 1020.

Bill KC9XG

On 4/10/2021 10:22 PM, Michael Black wrote:
> Well rig# 2099 is unknown.  Numbers in the 2000 range only run from 
> 2000-2049
> #2026 is a TMD-700 which I doubt you have.
> This lists the available rigs and their model numbersrigctl -l
>
> Mike W9MDB
>
>
>
>

mdblack98, 2021-04-11:
Please provide some debug

rigctl -m 2048 -r COM1 -s 115200 -vvvvv -ZAdjust com port and baud rate to suit and send the debug info.
Mike W9MDB

kc9xg, 2021-04-11:
Mike,
"Please provide some debug

rigctl -m 2048 -r COM1 -s 115200 -vvvvv -ZAdjust com port and baud rate to suit and send the debug info.
Mike W9MDB"

RPi4:
pi@raspberrypi:~ $ rigctl -m 2048 -r /dev/ttyUSB0 -s 9600 -vvvvv -Z
2021-04-11:13:47:19.999622: rigctl Hamlib 4.1
Last commit was Mon Feb 01 01:56:33 2021 +0000 SHA=645717
2021-04-11:13:47:19.999728: Report bugs to <hamlib-developer@lists.sourceforge.net>

2021-04-11:13:47:19.999810: rig.c(361):rig_init entered
2021-04-11:13:47:19.999906: rig_check_rig_caps: p1=0xb6f91938, p2=0xb6f96efc, rig_model=0xb6f91938, macro_name=0xb6f96efc
2021-04-11:13:47:20.000149: initrigs4_kenwood called
2021-04-11:13:47:20.000267: rig_register called
2021-04-11:13:47:20.000346: rig_register: rig_register (2012)
2021-04-11:13:47:20.000458: rig_register called
2021-04-11:13:47:20.000538: rig_register: rig_register (2013)
2021-04-11:13:47:20.000655: rig_register called
2021-04-11:13:47:20.000779: rig_register: rig_register (2001)
2021-04-11:13:47:20.000883: rig_register called
2021-04-11:13:47:20.000984: rig_register: rig_register (2025)
2021-04-11:13:47:20.001086: rig_register called
2021-04-11:13:47:20.001187: rig_register: rig_register (2003)
2021-04-11:13:47:20.001289: rig_register called
2021-04-11:13:47:20.001390: rig_register: rig_register (2004)
2021-04-11:13:47:20.001498: rig_register called
2021-04-11:13:47:20.001599: rig_register: rig_register (2016)
2021-04-11:13:47:20.001701: rig_register called
2021-04-11:13:47:20.001803: rig_register: rig_register (2024)
2021-04-11:13:47:20.001906: rig_register called
2021-04-11:13:47:20.002044: rig_register: rig_register (2005)
2021-04-11:13:47:20.002149: rig_register called
2021-04-11:13:47:20.002252: rig_register: rig_register (2007)
2021-04-11:13:47:20.002356: rig_register called
2021-04-11:13:47:20.002457: rig_register: rig_register (2009)
2021-04-11:13:47:20.002559: rig_register called
2021-04-11:13:47:20.002660: rig_register: rig_register (2010)
2021-04-11:13:47:20.002761: rig_register called
2021-04-11:13:47:20.002861: rig_register: rig_register (2022)
2021-04-11:13:47:20.002962: rig_register called
2021-04-11:13:47:20.003063: rig_register: rig_register (2014)
2021-04-11:13:47:20.003164: rig_register called
2021-04-11:13:47:20.003266: rig_register: rig_register (2030)
2021-04-11:13:47:20.003368: rig_register called
2021-04-11:13:47:20.003468: rig_register: rig_register (2021)
2021-04-11:13:47:20.003570: rig_register called
2021-04-11:13:47:20.003671: rig_register: rig_register (2029)
2021-04-11:13:47:20.003772: rig_register called
2021-04-11:13:47:20.003873: rig_register: rig_register (2043)
2021-04-11:13:47:20.003978: rig_register called
2021-04-11:13:47:20.004079: rig_register: rig_register (2044)
2021-04-11:13:47:20.004181: rig_register called
2021-04-11:13:47:20.004281: rig_register: rig_register (2045)
2021-04-11:13:47:20.004383: rig_register called
2021-04-11:13:47:20.004484: rig_register: rig_register (2047)
2021-04-11:13:47:20.004586: rig_register called
2021-04-11:13:47:20.004687: rig_register: rig_register (2038)
2021-04-11:13:47:20.004788: rig_register called
2021-04-11:13:47:20.004889: rig_register: rig_register (2002)
2021-04-11:13:47:20.004991: rig_register called
2021-04-11:13:47:20.005093: rig_register: rig_register (2011)
2021-04-11:13:47:20.005196: rig_register called
2021-04-11:13:47:20.005297: rig_register: rig_register (2006)
2021-04-11:13:47:20.005400: rig_register called
2021-04-11:13:47:20.005502: rig_register: rig_register (2008)
2021-04-11:13:47:20.005604: rig_register called
2021-04-11:13:47:20.005706: rig_register: rig_register (2015)
2021-04-11:13:47:20.005808: rig_register called
[... 87 lines cut ...]
2021-04-11:13:47:21.737454: write_block(): TX 3 bytes
2021-04-11:13:47:21.737545: 0000    49 44 3b                                            ID;
2021-04-11:13:47:21.737601: read_string called, rxmax=128
2021-04-11:13:47:22.538522: read_string(): Timed out 0.800 seconds after 0 chars
2021-04-11:13:47:22.538588: kenwood_transaction: read_string(len=0)=''
2021-04-11:13:47:22.538638: kenwood_transaction: read_string retval < 0, retval = -5, retry_read=2, rs->rigport.retry=3
2021-04-11:13:47:22.538683: kenwood_transaction: cmdstr = ID
2021-04-11:13:47:22.538726: rig_flush: called for serial device
2021-04-11:13:47:22.538766: serial_flush called
2021-04-11:13:47:22.538806: tcflush
2021-04-11:13:47:22.538853: write_block called
2021-04-11:13:47:22.559024: write_block(): TX 3 bytes
2021-04-11:13:47:22.559091: 0000    49 44 3b                                            ID;
2021-04-11:13:47:22.559139: read_string called, rxmax=128
2021-04-11:13:47:23.360075: read_string(): Timed out 0.800 seconds after 0 chars
2021-04-11:13:47:23.360162: kenwood_transaction: read_string(len=0)=''
2021-04-11:13:47:23.360219: kenwood_transaction: read_string retval < 0, retval = -5, retry_read=3, rs->rigport.retry=3
2021-04-11:13:47:23.360279: kenwood_transaction: returning retval=-5
2021-04-11:13:47:23.360335: verify_flexradio_id: cannot get identification
2021-04-11:13:47:23.360392: rig.c(972):rig_open return
rig_open: error = Communication timed out

 rig.c(972):rig_open return

Bill KC9XG

mdblack98, 2021-04-11:
It's timing out which means one of 
#1 Rig is turned off#2 Com port is wrong#3 Baud rate is wrong#4 Some other serial port setting is wrong
I'm not familiar with setting up PowerSDR on Linux but on Windows you have to use a serial port emulator to talk to CAT.
Mike W9MDB

kc9xg, 2021-04-12:
Mike,

  I am not running PowerSDR/Thetis and Windows with hamlib, I have been 
trying to connect to my D700 using the RPi4, with rigctl -m 2026 -r 
/dev/ttyUSB0 -s 9600 -t 4532 without success.

     For whatever reason, rigctl on my RPi4 fails to open any ID number 
associated with the Kenwood CAT Protocol.  This includes Kenwood radios, 
PowerSDR/Thetis, PiHpSDR, Elecraft XG3 and more.

   I used CuteCom on the RPi4 to connect to the D700 serial port using a 
FTDI USB-RS232 adapter to determine if 2 way communications was 
possible.  Connected to the D700, powered the rig off, then back on. 
Issued DISP A command to display ASYNC/Serial port parameters.  See 
result below:

  As shown above, the RPi4/FTDI adapter can send data to and receive 
data from the D700 radio serial port. Does rigctl require the radio to 
echo commands? ECHO ON?  What about Flow control? What type?  Hardware 
or Software?  Presently the D700 is using hardware flow control.

     While examining the debug output I sent you, rigctl appears to be 
sending the ID; command to the D700 and is not receiving a response, 
causing a timeout.

2021-04-11:13:47:22.559024: write_block(): TX 3 bytes
2021-04-11:13:47:22.559091: 0000 *49 44 3b ID;*
2021-04-11:13:47:22.559139: read_string called, rxmax=128
2021-04-11:13:47:23.360075: *read_string(): Timed out 0.800 seconds 
after 0 chars*
2021-04-11:13:47:23.360162: kenwood_transaction: read_string(len=0)=''

Notes:  D700 TNC Command Mode commands such as ID, only work if sent 
with a carriage return or carriage return and Line feed characters. D700 
Command Mode commands should not include semi-colons.

When in D700 TNC Converse Mode, all commands should be terminated with a 
semi colon.  Example Converse Mode command : FA;

Sending ID; in Command Mode will fail to return the ID, even if a 
carriage return and Line feed is included.  It appears rigctl may not be 
appending a carriage return with the ID command.

If the D700 TNC is in APRS mode, or turned off, the ID +CR +LF command 
returns TM-D700.

If the D700 TNC is in Packet mode the ID +CR +LF command returns a ? 
indicating an error.  An unexpected response, certainly.

   I would like to request a change in Rigctl Kenwood CAT commands, to 
send only ID with carriage return and line feed.  No semicolon included. 
Not sure if this will resolve the issue, but may be worth a try.

TIA,

Bill KC9XG

mdblack98, 2021-04-12:
The TMD-700 rig has been in beta status for years.....so would seem somebody had some success before.
It would appear to be written for command mode based on your description as the terminating char is CR.By default it uses no flow control.  If you want hardware flow add this
--set-conf=serial_handshake=Hardware
I guess you have to turn on flow control?  It's an option?
If you find things are working OK we can promote this thing to stable.
Mike W9MDB

mdblack98, 2021-05-25:
D710 has two control ports....the port on the main head is for CAT and the one on the head is for TNC.

closed 2021-05-25 by mdblack98