Hamlib/Hamlib #672: add rigctlpassthru utility

mdblack98, 2021-04-22:
Add rigctlpassthru utility.
This will, for example, allow this type of setup
SATPC32 > rigctlpassthru > FLRig/rigctld > rig
SATPC32 will be able to send Doppler VFO updates in the rig's language and FLRig will be able to poll and set the rig.  rigctld will also be able to poll and command the rig.

mikaelnousiainen, 2021-04-23:
@mdblack98 I'm curious if this would allow use of waterfall data available in recent models like the 7300?

On the other hand, I think there could also be a way to provide streaming of this type of data via the regular Hamlib API and rigctld TCP connection. Maybe identified with some rig-specific custom data identifiers. Though there isn't really a way to provide push-type data in the Hamlib protocol right now, is there?

Hamlib could still take care of parsing the rig response frames to identify and extract the actual payload data in these cases, so a secondary pass-through connection could be avoided.

mdblack98, 2021-04-23:
There are the async callback events
I've not really tested them as nobody has ever talked/complained about them.
The passthru would only work with rigctld and flrig for example and is just means to talk directly to the rig from another application through the "master" rig controller.

In 4.3 I plan on implementing the transceive frame processing so will be examining that.  Purportedly async callbacks don't work on Windows -- have to confirm that and see if we can get that work...otherwise the transceive processing would be implemented a different way so it could be portable.
/**
 * \brief Callback functions and args for rig event.
 *
 * Some rigs are able to notify the host computer the operator changed
 * the freq/mode from the front panel, depressed a button, etc.
 *
 * Events from the rig are received through async io,
 * so callback functions will be called from the SIGIO sighandler context.
 *
 * Don't set these fields directly, use rig_set_freq_callback et. al. instead.
 *
 * Callbacks suit event based programming very well,
 * really appropriate in a GUI.
 *
 * \sa rig_set_freq_callback(), rig_set_mode_callback(), rig_set_vfo_callback(),
 *     rig_set_ptt_callback(), rig_set_dcd_callback()

mikaelnousiainen, 2021-04-23:
Right, I never really paid attention to this code and I'm seeing `icom_decode_event` that could be extended for this purpose. We will need a new callback though. It could be something more generic for custom rig data. Anyway, this deserves a separate issue later for some more detailed discussion.

mdblack98, 2021-04-23:
I thinking this could be done on another port.
So you'd send a command to rigctld to tell it what port to put the audio stream on.  Do we need TCP or UDP?
rigctld would spawn a thread to service audio to that port.

mikaelnousiainen, 2021-04-23:
AFAIK the stream that Icom provides is not audio, it's the FFT "buckets" delivered as small, fragmented CI-V packets. The format looks rather generic, but I'm thinking if Hamlib should have its own abstraction over it just to standardize the format rigctld provides. Other manufacturers may provide something similar, I'd need to have a look.

Regarding whether to use TCP or UDP... I'm not sure yet. A TCP server would probably be most straightforward to implement, but UDP would allow higher data rates (for future purposes). However, I think a TCP server might still be the correct choices as the rigctld server should not really care about where it sends the data IMO. With UDP it should be configured to send the data to a particular address, which gets more complicated.

mdblack98, 2021-04-23:
I'm thinking multicast UDP.
Also planning on doing another UDP multicast port of rig information....freq, mode, ptt, width.  So other apps can just listen for updates rather than polling.

mikaelnousiainen, 2021-04-23:
That could work. I guess the ports for these could be preconfigured on command line just like the TCP server port. Would there be any need for dynamic port definition? After all, this is all for a specific radio.

mikaelnousiainen, 2021-08-14:
@mdblack98 This kind of passthrough functionality is something I'd really like to use and to work on. In addition to satellite tracking, this would allow a variety of logging applications to read / control rigs while simultaneously using the same rig via rigctld.

IIRC Hamlib (rigctld / rotctld) does only have a simplified way of sending raw commands to rigs and it expects a specified response from a rig, so it doesn't fit well as a generic solution.

Do you have any ideas regarding how to approach this? I could write some initial code already for testing this and get it out for 4.4. How would we separate rig/rotator replies from commands made via rigctld and commands supplied via this pass-through?

mdblack98, 2021-08-14:
It would be a rigctlcom rewrite into a new utility.
One program that doesn't work with Hamlib is Ham Radio Deluxe as it wants a ton of information from rigs and expects complete control.  So that is the target to get working with this facility.  
It will be a little tricky....
I envisioned a standalone program like rigctlcom, perhaps called rigctlthru or rigctlpipe?  It would be just a serial passthrough via a network connection or a serial port pair with a control channel opened to rigctld.
It would take each byte and just send it directly to the connected port.  A separate thread would be monitoring the port to look for any bytes from the rig and forward each byte received back to the client...no waiting for end markers or any such thing either way.  But...once a stream starts to process for either write or ready there needs to be a lock until a timeout occurs...the problem there is some commands can take up to 5 seconds so we have to avoid echo responses and transceive responses from Icom rigs.  This also means we need locks in all the rig functions to avoid writing when the passthru is happening.So logic is like this:wait for byte    lock
    while byte -- would be nice to detect end-of-command to avoid having to time out here.  Would require recognizing the protocol.
        send
    unlock

This would use a new rigctld command to request and release the lock.
I also envision opening up rigctld automatically in WSJTX/JTDX alongside the regular rig control -- so no need to start rigctld separately.  The locks are important there too as that would be two threads controlling the rig.  This would eliminate the need to run rigctld seperately and route WSJTX/JTDX through it instead.
So#1 Put locks in all rig functions and write a lock test as part of "make check" to stress test this.#2 Add lock request/release function to rigctld#3 Create rigctlthru or rigctlpipe or whatever name you can come up with that might make sense,.

Mike W9MDB