Hamlib/Hamlib #677: Kenwood rit not working

mdblack98, 2021-04-24:
Mike

Me again, and again RIT. A friend asked me to look at hamlib and RIT on
the TS2000, as it appeared to be a bit odd.

I agree with him - Hamlib follows what the spec APPEARS to say, but I
think the spec is wrong.

For RD and RU, P1 says 00000-99999 (The offset frequency in Hz)

What it appears to actually DO is to apply that offset to the RIT as an
increment.

Attached output from a rigctl run.

I'd fix it and send a pull request, but I suspect the problem applies
to ALL Kenwood rigs, and the TS2000 is the only one I have access to.

Mike G0GJV

mikaelnousiainen, 2021-04-24:
Btw I'm just working on extending the TS-480 backend significantly and noticed the same. I ended up implementing RIT/XIT for it specifically because it cannot execute RC command if RIT/XIT is not active (so it toggles RIT during RC if it's not active). Anyway this is otherwise good.

mdblack98, 2021-04-24:
I think they are all like that.
So it should be implemented in kenwood.c

mdblack98, 2021-04-27:
May need a short sleep after setting rit this way...

kenwood_transaction called
kenwood_transaction: cache invalidated
kenwood_transaction: cmdstr = RU01200
rig_flush: called for serial device
serial.c(629):serial_flush entered
tcflush
serial.c(661):serial_flush return(0)
write_block called
write_block(): TX 8 bytes
0000    52 55 30 31 32 30 30 3b                            RU01200;
write_block called
write_block():493 failed -1 - Resource temporarily unavailable
kenwood_transaction: returning retval=-6
kenwood.c(563):kenwood_transaction return(-6)
kenwood.c(1814):kenwood_set_rit return(-6)
rig.c(4463):rig_set_rit return(-6)
rigctl_parse.c(2024):rigctl_set_rit return(-6)
rigctl_parse: vfo_opt=0
rigctl_parse: RIG_EIO?
rigctl_parse.c(1684):rigctl_parse return(-6)
main: i/o error
rig.c(1086):rig_close entered
kenwood_close called

closed 2021-04-28 by mdblack98