Hamlib/Hamlib #687: SDRAngel back end

mdblack98, 2021-05-02:
I did fork Hamlib to:

https://github.com/N5BRG/Hamlib.git

I have modified the repository with my code which includes a folder called rigs\sdrangel

This package will build on Ubuntu 20.04 and works with the SDRangel software and a Pluto, HackRF and RTLSDR. It does not support WSJTX due to issues which I believe are mostly related to timing.
I wanted to post this code in hopes you would take a look at what I have done so far and give me commits. I also want SDRangel developers to get the code and have a look as well and I will let them know about the repository.

The code has a dependency on the CURL library at https://curl.se/libcurl/. These functions are being used to do HTTP REST communication between my code and SDRangel when running.

I have no affiliation with any company or organization pushing this effort. All of my code is intended to be public domain. I have mostly left the notices at the top of the code alone. I made need to add more to my code as well.
I would like to get have this added to the main Hamlib package if it is worthy and delete my fork as soon as possible. If I need to maintain my fork as a way to implement changes to the sdrangel rig we can.
If this has enough interest I will be adding more capability to the package, more SDR devices and initialization for example.

Let me know if you need more info from me to take action on this.  

I did make some changes to rig.h and rig.c to add a couple of parameters. Otherwise I worked to have the sdrangel code stand alone.

I plan to add a windows version later. I will do this using Visual Studio.

N0NB, 2025-07-30:
@N5BRG

Are you still working on this?  If so, can you rebase against the current 4.7~git master branch and issue a pull request?

I'll also advise that files we distribute in the source tarball need to be licensed under the LGPL v2.1.  The `.c` and `.h` files need to have the license header while others, including `Makefile.am` generally omit it.

I think this would be useful to merge into Hamlib.

N5BRG, 2025-07-31:
Hi Nate,

I had stopped working on this when I ran into the issue of turn around timing. I did notice SDRAngle is getting by the timing issue now days so may need to see what’s up there.

As time permits I will try to catch things up and I am fine with adding what ever license info I need to add. Not being familiar with the details I took a look and found this reference. I guess all this has been reviewed and the community wants to go with the referenced license or must because it was already used at some point.

<https://www.gnu.org/licenses/why-not-lgpl.html>
Why you shouldn't use the Lesser GPL for your next library - GNU Project - Free Software Foundation<https://www.gnu.org/licenses/why-not-lgpl.html>
gnu.org<https://www.gnu.org/licenses/why-not-lgpl.html>
[gnu-head-mini.png]<https://www.gnu.org/licenses/why-not-lgpl.html>

Let me put this on my todo list. I am currently writing code for an analog magnetic sensor to be used as a position encoder on an antenna pointing system. Need to keep my brain wrapped around this for a bit longer.

Thanks for your email and the efforts you are putting into HamLib.

Bob N5BRG