Hamlib/Hamlib #695: Rig control via mulicast/broadcast

mdblack98, 2021-05-09:
Add ability to multicast rig info and control rig via common API via UDP network inteface.
This will allow multiple programs to listen instead of polling and also control the rig.

mikaelnousiainen, 2021-05-09:
@mdblack98 Some questions:

1. Do you have any concrete specs/plans on this?
2. Will this be a two-way interface also allowing control?
3. Should the same mechanism (even the same port / protocol) allow transfer of data like Icom/Kenwood waterfall data, audio and/or raw I/Q?

Just thinking how generic this feature would be regarding supporting SDR features.

mikaelnousiainen, 2021-05-09:
That said, I'm primarily interested in adding support for the Icom waterfall data. But for future extensibility, we could allow other types of data too.

Maybe some frame header/metadata in the UDP packets could allow extensions.

mdblack98, 2021-05-09:
#1 Just what's in the doc right now
#2 Yes two-way -- will include built-in HELP a client can query too
#3 I'd say waterfall data would need to be on another port.  Do you have any info on the serial bandwidth that needs?

mdblack98, 2021-05-11:
I'll probably default it to JSON but you'll be able to ask for TEXT by sending a packet.
I would prefer to keep all data in ASCII and not binary.  So is the data from these rigs 8-bit or 16-bit or what?

mikaelnousiainen, 2021-05-11:
@mdblack98 Right ok. IIRC the waterfall data was 8-bit, but some rigs (both Icom and Kenwood) already provide I/Q data too, where the data rate is significantly higher. I'm a bit concerned about the overhead of ASCII data over binary.

mikaelnousiainen, 2021-05-11:
@mdblack98 They do I/Q over Ethernet / WLAN. Serial port is obviously limited to waterfall FFT data only.

Just looked at the specs, Icoms output binary waterfall data (like CI-V protocol is) with values between 0-160. However, Kenwoods output waterfall data converted to hexadecimal ASCII (as the control protocol is ASCII-based). Values between 00h - 8Ch.

mikaelnousiainen, 2021-05-11:
Diving a bit deeper into the features.

The following rigs are able to send waterfall data over serial port (or Ethernet/WLAN if the rig has the connection):

- Icom IC-7300 -> Serial only
- Icom IC-7610 -> Serial + Ethernet
- Icom IC-7850/IC-7851 -> Serial + Ethernet
- Icom IC-9700 -> Serial + Ethernet
- Icom IC-R8600 -> Serial + Ethernet
- Icom IC-705 -> Serial + WLAN
- Kenwood TS-890S -> Serial + Ethernet
- Kenwood TS-990S -> Serial + Ethernet

The Ethernet/WLAN connectivity seems to allow faster update rates on the rigs where its available.

The following rigs are able to send wide-band (> 1 MHz bandwidth) I/Q data over Ethernet:

- Icom IC-7610
- Icom IC-R8600

All of the Icom rigs (at least) also seem to offer "IF output" via the USB audio codec instead of the normal audio signal. However, I'm not sure if the data is actually raw I/Q data or something else.

mikaelnousiainen, 2021-05-12:
Also, some ideas regarding the broadcast packet contents:

1. It's probably worth introducing a packet/data type field right from day 1, so that it's possible to broadcast other types of (or more extended) data than just get_rig_info. I think this multicast infrastructure could eventually support all transceive type features from rigs, so thag any parameter/setting change (RIT/XIT/levels/etc) could be delivered automatically to clients.
2. Regarding the current proposal in JSON format: the VFOs could be represented in a JSON array. Having indexes in JSON key strings is a bit suboptimal IMO.
3. For rigs that do not support transceive, Hamlib could potentially be set up to do internal polling to avoid the current model where the clients have to poll rigs.
4. I see this could be extended to rotctl and ampctl too. At least many rotator controllers do already support reporting their status automatically without queries.

mikaelnousiainen, 2021-05-14:
@mdblack98 Not sure what your schedule for implementing the UDP multicast code is, but I can definitely work on the waterfall integration. That's something lots of apps could take advantage of.

I think all types of UDP broadcast data can share most of the UDP broadcast layer code, so there needs to be some common abstaction on it similar to the rig ports -- or what do you think? We could probably use a similar key/value-format for the waterfall data too: the actual data could be encoded in HEX/ASCII or potentially using Base64 or similar encoding.

mdblack98, 2021-05-14:
Here's my thinking...
The 1st thing to implement would be polling for audio.
If we abstract this to rig_get_data(RIG *rig, void *buf, int *datasize, int *maxelements, enum RIG_DATA_TYPE datatype)buf would be whatever size is needed to support the data.  If NULL would be dynamically allocated internally along with datasize and maxelements being filled in by the backend.  If not NULL would be preallocated by the caller and datasize and maxelements must be correct.   So it could be called once witth buf=NULL and then re-use buf after that on subsequent calls.
RIG_DATA_TYPE would be a new enumeration of the different data types for audio, spectrum, etc. and could be added to extcmds (e.g. ic7300_extcmds) for appropriate rigs.
RIG_DATA_AUDIORIG_DATA_SPECTRUM
After than is done we can work on adding the multicast to send out the polled audio data.  Then we can implement the new rigctl server that will do all the polling itself
 Mike W9MDB

mikaelnousiainen, 2021-05-14:
@mdblack98 Ok, I get the idea, but with both Icom/Kenwood the waterfall data is being *pushed* by the rig once enabled. There is no way to poll it.

Also, I'm not sure what you meant with "polling for audio". Do some rigs output digital audio via control port as well -- have I missed something essential? Or are you planning to connect ALSA/PulseAudio devices to Hamlib?

mdblack98, 2021-05-14:
Is the audio polled or pushed?
Any pushed data is going to have to wait for the multicast server since it will have to take over polling and eliminate all the flushing going on and be able to receive data immediately.
Mike W9MDB

mdblack98, 2021-05-14:
And I have no intention of ever hooking up any audio stuff to hamlib....

mikaelnousiainen, 2021-05-15:
@mdblack98 Again, I'm not sure what you mean with audio data here. I'm only aware of the ability to stream waterfall (FFT) and I/Q data via rig control connections. Neither of those can be directly considered as audio.

Based on Icom and Kenwood specs, I have the impression they both *push* these types of data once enabled. Though with Icom CI-V specs it's not really 100% clear. There's the command `27 00`, which looks like it could be used to poll waveform data. On the other hand, there's the command `27 11`, which can be used to switch waveform data output ON/OFF -- could this be then the rig actively pushing the data? I'll really have to test this.

Regardless of the way the communication works (poll/push), I think this extra data should be exposed via a single interface/protocol by Hamlib -- or what do you think? I think the UDP multicast server would really be optimal for this use case.

I'm wondering how we should integrate all of this in rigctl / rigctld... Should there be commands in rigctl to poll/listen for this data and corresponding API functions in the library -> only rigctld would expose the data via network? It gets quite complicated quickly this way.

mdblack98, 2021-05-15:
I'm thinking of the audio as the I/Q data.  Eventually ends up as audio but I'll correct my term.
I guess we need the multicast server first so I'll start working on that.
It will simply be in a polling loop on the rig where commands can also be inserted into a queue.
I'm going to also make rig control a token-based approach. So a client will request control, be granted, send commands, and release.
Mike W9MDB

mikaelnousiainen, 2021-05-17:
@mdblack98 Ok, so I did some testing on my IC-7300 during the weekend and found out how the Icom waterfall data output works.

With the serial port connection, the following settings must be present in the radio settings:
* baud rate has to be the maximum of 115 200 bps
* `CI-V USB Port` must be set to `Unlink from [REMOTE]`

There are two commands that need to be executed to switch on waterfall output:

`27 10 01` - Switch scope ON
`27 11 01` - Switch ON waterfall data output

The latter command will cause the rig to start *pushing* waterfall data to the controller continuously.

The data is split into small ~50-byte-parts with the serial connection (probably to allow interleaved commands and responses - I guess), but with rigs that have Ethernet/WLAN the data is pushed in a single frame.

For IC-7300 the data is split in 11 parts with the first part containing only metadata, e.g. frequency / span info, and the total number of waterfall data bytes is 475. The number of FFT bins (data points) will vary between rigs and it seems really to depend on the screen size: e.g. 7300 and 705 have the same data length (475 bytes = 475 pixels), but 7610 has more (689 bytes = 689 pixels).

The incoming data is identified by the response command `27 00` so that command is used only as a response, not as a way to poll the waterfall data.

It's also worth noting that the data is *not* sent to the CI-V broadcast address (it's not transceive data).

An example snippet of the data, first frame starts at `*`:

```
                         *
000008c0: 0000 0000 00fd fefe e094 2700 0001 1100  ..........'.....
000008d0: 0090 6903 0000 5002 0000 00fd fefe e094  ..i...P.........
000008e0: 2700 0002 1100 0000 0000 0000 0000 0000  '...............
000008f0: 0000 0000 0000 0000 0000 0000 0700 0000  ................
00000900: 0000 0000 0000 0300 0000 0000 0000 0005  ................
00000910: 0000 0000 0000 00fd fefe e094 2700 0003  ............'...
```

This is all to say that we need to be able to handle push-type data to implement this. The multicast server and the `decode_event` type of transceive functionality need to be present here. It looks like the `set_trn` functionality that uses `decode_event` functions, such as `icom_decode_event`, is already in place -- and hopefully already works.

We will simply need to extend `icom_decode_event` to handle the command `27 00` as a special case of transceive type data -- in addition to the regular broadcast CI-V frames. We'll probably need another type of rig callback for the waterfall data, which would the allow the data to be piped to the UDP multicast server eventually. I'm thinking we can create a simple abstraction of the waterfall frames based on Icom's specs. I'll have a look at what kind of waterfall data Kenwood rigs provide, but IIRC it's very similar. The waterfall frames would include the frequency / span info in addition to some normalized, 8-bit waterfall data = FFT buckets that are evenly spaced.

Thoughts?

g4wjs, 2021-05-18:
How is this different from the existing rigctld server? That already allows multiple clients to access a rig.

Why are you using the term "broadcast" for Multicast UDP, they are two distinct and different things?

What is the purpose of the checksum field in the JSON example? It seems like unnecessary and confusing duplication as UDP datagrams already contain a checksum which is generated and verified at the UDP protocol layer.

Where is a description of the topology of counterparties in this protocol? I am looking for how a rig server knows which address to send to, how a client of that data replies to a specific rig server.

The presence of a sequence number in the example JSON message smells of a problem with message ordering and delivery guarantees. IMHO if a sequence number is needed then TCP/IP should be used with point to point connections and a traditional unicast IP connection model, like rigctld.

73
Bill
G4WJS.

mdblack98, 2021-05-18:
Nobody owns the broadcast term --  I'm not trying to teach users old network terminology but using something most users understand...like "radio broadcast".
This will allow one control program -- like rigctld or FLRig (I'm working with Dave on that one too).
All clients can just listen for updates instead of polling...especially good for passive clients that cause problems when wanting VFOB info for example or when wanting something different from what the rig is currently doing.
Sequence numbers are just for debugging.
The CRC is OSI layer 7 where UDP packet checksum is OSI Layer 4
We're also going to be implement cookies for requesting control of the rig.
Mike W9MDB

mdblack98, 2021-05-18:
We could just add a WATERFALL (or should it be FFT?) item and put the plain hex bytes (not JSON formatted) in the value.Client can then decode the string rather than JSON doing it.  Otherwise we use 150% more space for the data size with "0x00 0x01...."

So we just mash all the bytes together like this...not hard to decode if the client wants it.  WATERFALL will have to be requested by at least one client.
  "Waterfall": "XXXXXXXXXXXXX...."
I've also got an enhancement planned for transceive data so will be looking at those functions.
Mike

mikaelnousiainen, 2021-05-18:
@mdblack98 Could we call it SPECTRUM, as that's what the data represents, right? We'll need a couple of other metadata fields too, I'll try to document things here soon. Hex ASCII data should be fine, it's easy to parse after all.

mdblack98, 2021-05-18:
Sure...sounds good.

mikaelnousiainen, 2021-05-18:
@mdblack98 This is an initial spec for spectrum data I wrote down quickly based on both Icom and Kenwood rig manuals:

# Spectrum data content

```
SpectrumLength=475
SpectrumData=00AAFF75BD2AAA ...
SpectrumDataMinLevel=0
SpectrumDataMaxLevel=140
SpectrumDataMinStrength=-100
SpectrumDataMaxStrength=0
```

`SpectrumLength`: Number of bytes in `SpectrumData`. Not strictly required, but helps verifying the data is complete.

`SpectrumData`: 8-bit data in Hex ASCII format.

`SpectrumDataMinLevel`: The value that represents the minimum signal level. This may change depending on the settings of the rig -- at least on Kenwood the subscope provides different kind of data comparted to the main scope.

`SpectrumDataMaxLevel`: The value that represents the maximum signal level. This may change depending on the settings of the rig -- at least on Kenwood the subscope provides different kind of data comparted to the main scope.

`SpectrumDataMinStrength`: The strength of the minimun signal level in dB. This may change depending on the settings of the rig -- at least on Kenwood the subscope provides different kind of data comparted to the main scope. The level values are assumed to represent the dB strength scale in a linear way.

`SpectrumDataMinStrength`: The strength of the maximum signal level in dB. This may change depending on the settings of the rig -- at least on Kenwood the subscope provides different kind of data comparted to the main scope. The level values are assumed to represent the dB strength scale in a linear way.

# Spectrum types

`SpectrumType`: Icom has additional modes SCROLL-F and SCROLL-C to allow scrolling past the boundaries of the screen and have the spectrum flip over to the corresponding area/"page", but the SCROLL modes can be mapped to these. We can have extlevel to set the Icom-specific modes. Kenwood rigs seem to have the same options, excluding SCROLL-C, but it looks like fixed mode edges for Kenwood are more freely specifiable.
```
SpectrumType=FIXED|CENTER 
```

## Spectrum type: CENTER

If `SpectrumType=CENTER`, the following fields will be present:

```
SpectrumCenterFreq=14267000
SpectrumSpan=25000
```

## Spectrum type: FIXED

If `SpectrumType=FIXED`, the following fields will be present:

```
SpectrumLowFreq=14000000
SpectrumHighFreq=14250000
```

In JSON format this could be:

```
{
  "Spectrum": {
    "Length": 475,
    "Data": "00AAFF75BD2AAA ...",
    "Type": "FIXED|CENTER",
    "MinLevel": 0,
    "MaxLevel": 140,
    "MinStrength": -100,
    "MaxStrength": 0,

    // If Type=CENTER, the following fields will be present:
    "CenterFreq": 14267000,
    "Span": 25000,
    
    // If SpectrumType=FIXED, the following fields will be present:
    "LowFreq": 14000000,
    "HighFreq": 14250000
  }
}
```

mdblack98, 2021-05-18:
How is the frequency bin computed...is the example you provided based on real data?
A span of 25000 doesn't divide evenly by 475 so is the freq-per-bin actually 25000/475?So frequency is (center_freq-(center_freq/2))+(i*(25000/475)) ?Where is the index into the SpectrumData.
Mike

mikaelnousiainen, 2021-05-19:
@mdblack98 Yes, the example is mostly based on real values. The indices (bins) in the spectrum data will not map exactly to any integer frequencies. The data received from the rig is most likely subject to some kind of averaging anyway, since it seems to depend on the screen size (as that's what the manual says), so the rig outputs what the rig displays on the screen.

The spectrum data lengths are:

* IC-7300 / IC-9700 / IC-705 / IC-R8600: 475 bytes (bins)
* IC-7610 / IC-785x: 689 bytes (bins)

The data ranges (bin values) are:

* IC-7300 / IC-9700 / IC-705 / IC-R8600: 0-160
* IC-7610  / IC-785x: 0-200 (at least on newer firmware, older manual says 0-136 for IC-785x)

The scope span settings in center mode are for all rigs:

* 2.5k, 5k, 10k, 25k, 50k, 100k, 250k, 500k

* IC-R8600 supports also: 1000k and 2500k

In fixed mode, the scope span will vary, it can even be the whole band.

Kenwood TS-890S:
* Main scope: 32 * 20 bytes = 640 bytes (bins)
* Sub scope: 12 * 19 bytes = 228 bytes (bins)
* Kenwood also specifies the signal levels:
  * Main scope: 00h = 0dB, 8Ch = -100dB
  * Sub scope: 00h = 0dB, 32h = -50dB

Kenwood TS-990S:
* Main scope: 32 * 20 bytes = 640 bytes (bins)
* Sub scope: 15 * 19 bytes = 285 bytes (bins)
* Kenwood also specifies the signal levels:
  * Main scope: 00h = 0dB, 8Ch = -100dB
  * Sub scope: 00h = 0dB, 32h = -50dB

I'm not sure if it is possible to find at least some approximations on the signal levels for Icoms. There are some markings on the screen waterfall display regarding dB scale.

mdblack98, 2021-05-19:
You want to contact Icom and Kenwood to find out the way to compute the bin frequencies?To make the spectrum data really useful we need to know that.
Mike W9MDB

mikaelnousiainen, 2021-05-19:
@mdblack98 Not sure if there is anything to ask. The spec is clear. The bins are evenly spaced either around the center frequency/span or the low/high edges. I'm not sure if there is an issue here?

After all, if you can freely specify the spectrum span or the edges, there is no way the bins are going to represent some neat integer bandwidth in all cases.

We can also do some research on how wfview interprets this data: https://gitlab.com/eliggett/wfview -- I'm planning on looking at the source just to verify our approach is correct.

mdblack98, 2021-05-19:
I don't expect the bins to have integer bin widths.Would like to know the definitive answer since it isn't in the manual and assuming is always a bad thing.
I would guessbin[0] = LowFreqbin[Length-1] = HighFreq
And the intervals based on that.  But you never know what they come up with.
I'll ask them if you don't want to.  I've pointed out several errors in their manuals which they have fixed.
Mike W9DMB

g4wjs, 2021-05-19:
On 19/05/2021 16:47, Michael Black wrote:
> I don't expect the bins to have integer bin widths.Would like to know
> the definitive answer since it isn't in the manual and assuming is
> always a bad thing.
> I would guessbin[0] = [...]

Mike,

isn't the data simply a bitmap graphic. Frequency mapping would depend 
purely on the number of pixels on the rig's display.

73
Bill
G4WJS.

mikaelnousiainen, 2021-05-19:
@mdblack98 Ok, that's a fair point. Please do ask -- I assume you've already got some contact info for the manufacturers. For Icom, we can verify this quickly through wfview source code too, as they have a working open-source implementation of waterfall display that interprets this data.

@g4wjs That's what I assume too.

mdblack98, 2021-05-19:
Quite possibly....but they have to have some frequency reference and I'd prefer to hear it from the horse's mouth rather than assuming anything.
Mike 

 
 

Mike,

isn't the data simply a bitmap graphic. Frequency mapping would depend 
purely on the number of pixels on the rig's display.

73
Bill
G4WJS.

mikaelnousiainen, 2021-05-19:
@mdblack98 Wfview (Icom spectrum viewer) does the bin frequency calculation exactly like you described: https://gitlab.com/eliggett/wfview/-/blob/master/wfmain.cpp#L2413

mikaelnousiainen, 2021-05-20:
@mdblack98 I could already start adding some code to parse the spectrum data in the Icom backend, handling it in `icom_decode_event`. 

I think we could do the following changes:

1. Add a spectrum data callback to `rig_callbacks`: https://github.com/Hamlib/Hamlib/blob/master/include/hamlib/rig.h#L2304 -- so that direct Hamlib API usage of the data is possible. Spectrum data will need its own struct for metadata and a buffer for the actual FFT data.
2. I guess rigctld can the directly use this callback (and all the other callbacks too for transceive use) with the multicast server to cache and deliver data updates via the multicast UDP packets. The existing callbacks seem to be used only in `rigctl_parse.c` simply printing out the (change of) data.
3. One more challenge is the passing of spectrum data from rig backend to the callback, as we're talking about frequent (minimum of multiple times per second) delivery of ~500-1000 bytes of data. Allocating/freeing data dynamically doesn't sound like a good option. I'm thinking of creating some sort of simple pre-allocated buffer pool for this purpose. The buffer pool would be allocated by the rig backend and the spectrum data struct/buffer would be valid during the callback. I guess that should be enough as the callback should really push out the spectrum data as a UDP packet as soon as possible.

Comments?

Also, some comments about the multicast data format:

* There are some inconsistencies between the text token format and JSON: The format don't have the same fields, e.g. PTT and Status and ErrorMsg are missing in JSON. Also, the field Seq is only present in JSON.
* I'm not sure if we really need the Seq number, as there's not much the receiving end can do if a packet is missing. Regarding spectrum data usage, I could possibly imagine the receiver application could wait for a *very* short time (in the order of under 100ms) for any out-of-order packets to arrive. Though I do see the Seq number as an indicator for missing/out-of-order data, as even with localhost delivery the ordering of UDP packets is never guaranteed.
* Let's keep the field names and the naming convention consistent between the formats. If we go for the field names starting with a capital letter (to preserve the existing convention in other Hamlib field names), `vfos` should probably be `VFOs` or `Vfos`.

mikaelnousiainen, 2021-05-20:
@mdblack98 Regarding the multicast data content, should all the messages include all of the fields or simply the data that gets updated at that time? I'm thinking of the latter, otherwise we'd need to cache all the data, which doesn't work for the spectrum data.

mfncooper, 2021-05-20:
I would **_very_** strongly recommend the use of Semantic Versioning for this protocol, and its representation(s). This will give you a great deal of flexibility in defining compatibility levels and degrees of change, and allow implementations / clients to determine how much or how little they can support. Date-based versioning gives you **_no_** flexibility, since a client needs to treat every date change as if it were a major change, even if it was in fact a tiny change. See https://semver.org/ for details, if you're unfamiliar with Semantic Versioning.

mikaelnousiainen, 2021-05-21:
@mdblack98 Thanks for the changes. More comments here:

1. Why do we need rig "cookie" based control? Why does control need to be exclusive for a client?
2. I was confused earlier: there's need for buffer pools as we will need to have memory allocated for only one buffer of spectrum data at a time anyway -> aim mostly for synchronous processing of the 

I've done some hacking during the past few nights on parsing the spectrum data for Icoms (testing on my IC-7300) and while it's still not ready (not even for an initial PR yet), it's looking good and I'm getting steady data output. There is **no** need to turn transceive on in the radio settings, but only the scope output needs to be enabled to get this data. The routine that collects the data is the `icom_decode_event`, which simply parses the `27 00` command output. The rigctl transceive setting (command À ON`) must be turned on for this to work, of course.

I'll continue working on the code in the coming days, so we can get the parsing / callback part of spectrum data output ready. The other half of it is then to implement the multicast server and message protocol.

I even made a simple ASCII graphic output for the waterfall spectrum data to be able to debug it, see the screenshots below.

This ASCII graphic is upside-down compared to the radio screen, as the terminal scrolls up and rig waterfall scrolls down:

[image]

And the corresponding IC-7300 screen (some CW / FT-8 / FT-4 signals here on 20m):

[image]

Weirdly, there is some delay (around 1-2 seconds) in getting the spectrum data -- my IC-7300 is directly connected to the laptop USB port, after all.

@mfncooper Semantic versioning does indeed sound like a better approach for protocol versioning than date. @mdblack98 What do you think? What would the purpose of a date as version be?

mdblack98, 2021-05-21:
Nothing to do with the spectrum.
We're going to allow rig control also via UDP.    In order to best support this there are some calls where you want to ensure nobody else messes with rig while you're doing things.  
For example,  I have an app that auto-tunes the rig on band change...so it does a series of set VFOA freq/mode, set VFOB freq/mode, tune, set VFOB freq/mode.
Can take several seconds before the tuner is done.  So would like to lock out anybody else messing with the rig while this is going on.
Same goes for things like WSJT-X does when setting up transmit split on both VFOA and VFOB with a series of commands.
Even the internal rig_set_split needs it as it does several calls.  rigctld controls this but even that has a problem when commands aren't stacked up on one line and especially when doing commands with "rigctl" command line requests.
I'm coordinating with FLRig too which will also be implementing cookie control.

N0NB, 2021-05-21:
I gotta admit, that ASCII art version of the waterfall is rather cool!

mikaelnousiainen, 2021-05-22:
@mdblack98 Regarding rig control and error reporting, should we expose the standardized Hamlib error code (RPRT) in addition (or instead of) an error message? Will there be some additional info in the error messages?

Also, should there be a way to correlate responses to control messages with some transaction/correlation ID the client can supply within a rig control command? Otherwise it may be difficult to know which command produced an error, for example.

mdblack98, 2021-05-22:
The Error will contain a multi-line history (last 3 messages at least) as it currently does.
Maybe a LastCommand field and maybe putting the command along with the error msg?  I can see benefits to both.
Since the intent is to send a packet on every command LastCommand makes sense and would allow monitoring from an external app.  Could also include the port number of the client and make an ID mandatory on the command to identify itself.

mikaelnousiainen, 2021-05-22:
Ok, makes sense. Making some kind of correlation ID field mandatory would probably be a good idea, so that clients can track when the issued commands actually get executed and what the result was. I'm assuming rigctl would send responses (success/error code) for each command anyway, right? Including the command that was executed in the responses might help too (don't have any strong opinion this), though I do not immediately see the use cases here.

mdblack98, 2021-05-22:
Yes..each command should be broadcast too.   That would necessitate a flurry of packets for some command sequences but over the network that should not be a problem.

mdblack98, 2021-05-22:
The spectrum is a unique item...so I'd say the field is only included when a new spectrum is available.
My intent right now is to put out packets every 100ms and then immediately when any change is made.  For those rigs that are a bit slow it will be as fast as the rig can do it.  Might have to be a bit picky about how often we poll certain things like mode and width which don't change much and don't need immediate response. 
10 times per second seems fast enough to allow apps to track frequency changes in real time.  With broadcast that means nobody has to poll for it but will receive it constantly and very quickly when polling detects a change.
And change would trigger immediate broadcast and if there is spectrum data available it will be sent too during transmissions of any packet.
What's the maximum data rate of the spectrum data?
Mike W9MDB

mikaelnousiainen, 2021-05-22:
@mdblack98 I'd still really like to see the correlation ID between a request and a response in the protocl. It's very useful for a client to know when a command it sent has actually been executed -- this helps especially with UI development.

Otherwise, I agree with the rest of your plan!

However, I guess there must be a way to opt in or opt out from polling of the rig? This is important for backwards-compatibility (many rigctl clients still poll manually) and simply for allowing clients to choose which features to use (e.g. to save bandwidth if polling is not needed).

I can see the following use cases:

1. Rig poll output only (even if transceive is supported natively by rig)
2. Rig poll output + transceive (changes pushed by rig)
3. Transceive output only

And any of those could be combined with spectrum scope output for Icom/Kenwood rigs that support it.

The IC-7300 spectrum output pushes about 4-5 lines of spectrum scope data per second, so considering every line consists of 11 CI-V frames about 60 bytes long, the output is around 3 kilobytes per second. I can provide more accurate measurements later.

However, it's worth keeping in mind that this rate applies only for 115200 bps serial port connectivity (and is limited by that speed). Newer Icom rigs with network connectivity push the complete "line" of spectrum data in one CI-V frame and based on some videos of wfview it feels like the update rate is a significantly higher (maybe 3-4x?).

I'm not sure if anyone has tested network connections with the Hamlib Icom backend yet?

It'd be really nice to have an IC-705 or an IC-7610 to test this with...

mdblack98, 2021-05-22:
We should let the client provide an ID and return it in the response packet.  That should be a mandatory field.

  LastCommand {
      string Type; // Rig, Amp, Tuner, etc...
      string ID;
      string Command;
      string Status;
  }

A command would be this...or could be a different structure for each type.  Can't see an advantage either way right now.

       Command {
            string Type;  // Rig, Amp, Tuner, etc...
            string ID;
            string Command;
       }

I'm debating how much information to include in the packet.  Would like to keep the packet from fragmenting so needs to be under  the MTU size of 1500 bytes (minus headers).

No reason we couldn't have a thread gets the network spectrum too and inserts it into the mujlticast.

I do plan on supporting all 3 situations you talk about.  The multicast will be running it's own thread polling the rig.  Clients will still be able to connect and poll themselves.

Also plan on supporting transceive too built into the new multicast server which will update local cache and therefore feed the multicast too along with polling.

What's really good about that is for satellite ops the transceive will allow users to twiddle the knob and the

mikaelnousiainen, 2021-05-22:
@mdblack98 Forgot to add what I meant by correlation ID:

Client command would have a metadata field:

`CorrelationId=abc1234` where the client generates the correlation IDs and stores them in some cache

The server response for that command would then include a copy of the ID, simply:
`CorrelationId=abc1234`

This way the client knows its command was processed and the result of it -- whether it generated an error or not.

mdblack98, 2021-05-22:
That's what the ID will be -- whatever the client sends.

Could just be a 1-up counter to keep it simple.
Or better yet some client-unique thing to identify itself plus a 1-up counter.

So

Command.ID = "MyApp 1"

Mike

mikaelnousiainen, 2021-05-23:
@mdblack98 Cool -- I think this is the feature set we could get started with.

Regarding enabling of polling and/or transceive, I guess that could really be dynamically changeable per rig at run time (so not only a static command-line flag).

Some clients (like what my initial plans are) could be interested in the spectrum data only (-> no transceive or rig poll updates enabled). Though I'm seeing ta least I'll be migrating to use this protocol fully later on. Just thinking about the ways existing software can easily get started with this new protocol by enabling a subset of its features to avoid breaking existing functionality.

Spectrum data will be the same, there'll be a func switch to enable it (as all rigs have the command to explicitly subscribe to receiving spectrum data).

mdblack98, 2021-05-23:
Seems there should be 3 capabilities that could be or''d together.  Would be an argument to the multicast server startup.  By default all 3 would be enabled but could be overridden for custom clients and rigctld.
--set-conf=multicast=SPECTRUM=1,POLL=0,TRANSCEIVE=0

RIG_MULTICAST_POLL
RIG_MULTICAST_TRANSCEIVE
RIG_MULTICAST_SPECTRUM

mdblack98, 2021-05-23:
Since we can have multiple clients the client can just ask for what it wants but may get more than it wants.

One client may ask for just SPECTRUM but may get both POLL and TRANSCEIVE data too(from another client request) which it can just ignore in the json....one of the nice things about json deserialization.

To enable for the multicast server we'll need a set_multicast capability.  This will be able to operate from either rigctl or rigctld.

rigctl will have a new multicast client that will talk to an existing rigctld multicast server.

Could also give rigctl the -M option too and it would crank off it's own server.

Mike W9MDB

mdblack98, 2021-05-23:
Was thinking about the multicast client.  Specifically for multiple instances of WSJT-X.

The 1st instance of WSJT-X would crank off the multicast server -- but it would need a multicast config file or an option built-in to WSJTX/JTDX to enable it.  The config file option would work with existing WSJTX/JTDX code and perhaps we can get a checkbox to enable this for a rig.
All other instances would just select the multicast client -- no polling required.

Along with recent changes in FLRig where FLRig is now able to follow rigs in close to real-time (subject to rig speed in polling) we will also see multicast clients able follow the rig in close to real-time.

The multicast server will be polling pretty much as fast as it can.

Mike W9MDB

mikaelnousiainen, 2021-05-23:
@mdblack98 All this sounds very good and promising.

Would it be the `set_multicast` command that clients could use to toggle the multicast (`RIG_MULTICAST_*`) features on and off (in addition to the initial `set-conf` parameter)?

I'm curious about the "multiple instances of WSJT-X/JTDX" use case for a single rig. I do not fully understand why such setup is needed and what it provides?

I'm still working on an initial PR for the Icom spectrum output, though it'll be a couple of days until I'm able to push out anything useful. I'll be adding some standardized levels/funcs for controlling spectrum output. I'm basing these on a set of common features between Icom and Kenwood rigs. Some manufacturer-specific custom features will use extlevels.

Native transceive mode for rigs will have its standardized func too, as it'll need to be enabled separately.

mdblack98, 2021-05-23:
Given multicast is available to all apps I don't intend to give any app the ability to turn it off.
Once one asks for it they all get it.

Transceive mode will be fairly easy in the multicast mode since the multicast server will be doing all the polling and can easily read the transceive data interleaved.  Ultimately I want to enable the transceive in normal rig mode too but that's a bigger change since we always flush the rig right now before every command to be safe.  That really shouldn't be necessary though.

As for multi-instance WJST-X/JTDX that's for rigs like the Flex that can have multiple slices.  Though it would take a new Flex backend to handle the slices differently than it's done now.
The 1st WSJT-X/JTDX will have to specify the rig info plus enable multicast for all the other instances.  So as it is now that will take a hamlib_settings.json file to set it until the apps can add a multicast checkbox.  This 

Mike W9MDB

mikaelnousiainen, 2021-05-24:
@mdblack98 Ok, thanks for the explanations. The Flexradio slices indeed require that kind of setup.

There's one more thing I'm wondering about how this will work: my hope was to be able to use the existing rigctld TCP connection for rig control while still being able to receive spectrum data via multicast UDP.

Just making sure that your plans are how I've understood them: the multicast server would be an additional part of the normal rigctld server, so during transition time (or as an alternative) clients could use a combination of the TCP and UDP protocols. Though the Hamlib TCP protocol is so widely in use that I guess it won't really disappear -- and it has its use cases too.

mdblack98, 2021-05-24:
Yes...multicast is independent and will actually speed up normal tcp connections since the information will always be cached and not need any rig commands for non-set operations.

For example it will speed up like this example that asks for VFOA/VFOB frequencies back-to-back

34ms for the 1st pair as it requires asking the rig.
8ms for the 2nd example since it's cached now.

This caching is also what allows multiple programs to talk to the rig without loading down the serial I/O.  As long as you're just asking for information you'll get the cached response and generate no CAT commands so the response to commands will be very fast.

 Rig command: f VFOA f VFOB f VFOA f VFOB
2021-05-24:12:19:13.766733: rigctl_parse: cmd==0x66
2021-05-24:12:19:13.778444: rig.c(2090):rig_get_freq return(0)
Frequency: 10136000
2021-05-24:12:19:13.779421: rigctl_parse.c(2018):rigctl_get_freq return(0)
800
2021-05-24:12:19:13.781373: rigctl_parse: cmd==0x66
2021-05-24:12:19:13.799917: rig.c(2090):rig_get_freq return(0)
Frequency: 10134500
2021-05-24:12:19:13.800893: rigctl_parse.c(2018):rigctl_get_freq return(0)

Rig command: 2021-05-24:12:19:13.802845: rigctl_parse: cmd==0x66
Frequency: 10136000
2021-05-24:12:19:13.804797: rigctl_parse.c(2018):rigctl_get_freq return(0)
2021-05-24:12:19:13.804797: rigctl_parse.c(1743):rigctl_parse return(0)
2021-05-24:12:19:13.805773: rigctl_parse.c(648):rigctl_parse entered

Rig command: 2021-05-24:12:19:13.806750: rigctl_parse: cmd==0x66
Frequency: 10134500
2021-05-24:12:19:13.811629: rigctl_parse.c(2018):rigctl_get_freq return(0)

mikaelnousiainen, 2021-05-24:
@mdblack98 Great, that's how I more or less imagined it'd work -- and getting the cache updated is a great improvement.

One more thing I noticed: we need to allow multiple spectrum scope data streams identified by some number. Kenwood rigs allow output of Main and Sub scopes simultaneously (either via USB or LAN). Also, it looks like Icom rigs allow you to choose which scope to use with output, either Main or Sub, so the data stream needs the identifier anyway.

I'll include all of this in my code for spectrum support abstraction

mdblack98, 2021-05-24:
Then we need to change Spectrum into a Spectra array -- although this data example won't be accurate for Kenwood rigs.
Sound right?

  "__comment_spectra__": "Rigs that have spectrum output may include this array data",
  "Spectra": [
  {
    "Name": "Main",
    "Length": 475,
    "__comment_spectrum_data__": "2-char hex bytes so data len=2*Length",
    "Data": "00AAFF75BD2AAA...",
    "Type": "FIXED|CENTER",
    "MinLevel": 0,
    "MaxLevel": 140,
    "MinStrength": -100,
    "MaxStrength": 0,
    "__comment_spectrum_center__": "If Type=CENTER, the following fields will be present:",
    "CenterFreq": 14267000,
    "Span": 25000,
    "__comment_spectrum_fixed__": "If SpectrumType=FIXED, the following fields will be present:",
    "LowFreq": 14000000,
    "HighFreq": 14250000
  },
  {
    "Name": "Sub",
    "Length": 475,
    "__comment_spectrum_data__": "2-char hex bytes so data len=2*Length",
    "Data": "00AAFF75BD2AAA...",
    "Type": "FIXED|CENTER",
    "MinLevel": 0,
    "MaxLevel": 140,
    "MinStrength": -100,
    "MaxStrength": 0,
    "__comment_spectrum_center__": "If Type=CENTER, the following fields will be present:",
    "CenterFreq": 14267000,
    "Span": 25000,
    "__comment_spectrum_fixed__": "If SpectrumType=FIXED, the following fields will be present:",
    "LowFreq": 14000000,
    "HighFreq": 14250000
  }],

mikaelnousiainen, 2021-05-25:
@mdblack98 Yes, exactly. Though I would simply replace the Name field with a numeric ID, starting from zero. It's easier to use programmatically and that's what all the rigs supply. We can give a mapping from the numbers to a string name in rig caps.

That said, should we use numbers for the spectrum Type field too? Its just extra bytes and parsing to transfer those as strings and there's already an enum for the Type too defined as a part of Hamlib API.

mikaelnousiainen, 2021-05-25:
@mdblack98

So just as an suggestion:

* Replace `Name` in the `Spectra` array items with `Id` that will be numeric. Basically 0 will represent `Main` and 1 will represent `Sub`
* Replace value of `Type` in the `Spectra` array items with a numeric value (valid values 1-4) according to the following enum:

```
enum rig_spectrum_mode_e {
    RIG_SPECTRUM_MODE_NONE = 0,
    RIG_SPECTRUM_MODE_CENTER,            /*!< Spectrum scope centered around the VFO frequency */
    RIG_SPECTRUM_MODE_FIXED,             /*!< Spectrum scope edge frequencies are fixed */
    RIG_SPECTRUM_MODE_CENTER_SCROLL,     /*!< Spectrum scope edge frequencies are fixed, but identical to what the center mode would use. Scrolling is enabled. */
    RIG_SPECTRUM_MODE_FIXED_SCROLL,      /*!< Spectrum scope edge frequencies are fixed with scrolling enabled */
};
```

These spectrum modes represent all modes supported by Icom rigs. Kenwoods seem to offer 3 of these 4 modes (no explicit CENTER_SCROLL mode) so the mapping seems to be ok for a starting point.

mdblack98, 2021-05-25:
Internally we may want the enum for function calls about the spectra but I think the JSON should contain the text.  

The idea being the JSON should stand on it's own...a self-complete abstraction of the rig.

Mike W9MDB

mdblack98, 2021-05-25:
Answer from Kenwood on frequency computation
The following method is the same as the method used for the ARCP-990 bandscope.

 

1) Determine whether the current status is Center Mode or Fixed Mode by using the "BS3"command.

 

2)In case of Center Mode

Add or subtract the RIT/XIT frequency ("RF" command) to the MAIN Band frequency ("FA" command) or SUB Band frequency ("FB" command) to obtain the actual current center frequency.

Then add or subtract the SPAN frequency obtained by the "BS4" command to the center frequency to calculate the lower and upper frequencies of the bandscope.

 

3) In case of Fixed Mode

Directly obtain the lower and upper frequencies of the bandscope using the "BS5" command.

 

4)Calculate each spectral frequency by dividing the area between the lower and upper frequencies by 640.

[38 more replies]

closed 2023-11-10 by mdblack98