Hamlib/Hamlib #697: FT2000 roofing filter incorrect

mdblack98, 2021-05-12:
RF13 is not a valid roofing filter choice.

2021-05-12:12:51:19.406006: newcat_set_cmd: newcat_set_cmd_validate not implemented...continuing
2021-05-12:12:51:19.406030: write_block called
2021-05-12:12:51:19.406061: write_block(): TX 5 bytes
2021-05-12:12:51:19.406090: 0000    4e 41 31 30 3b                                      NA10;           
2021-05-12:12:51:19.406168: cmd_str = ID;
2021-05-12:12:51:19.406189: write_block called
2021-05-12:12:51:19.406240: write_block(): TX 3 bytes
2021-05-12:12:51:19.406269: 0000    49 44 3b                                            ID;             
2021-05-12:12:51:19.406292: read_string called, rxmax=129
2021-05-12:12:51:19.437336: read_string(): RX 7 characters
2021-05-12:12:51:19.437364: 0000    49 44 30 32 35 31 3b                                ID0251;         
2021-05-12:12:51:19.437386: newcat_set_cmd: read count = 7, ret_data = ID0251;
2021-05-12:12:51:19.437411: newcat.c(9990):newcat_set_cmd return(0)
2021-05-12:12:51:19.437435: newcat.c(7039):newcat_set_narrow return(0)
2021-05-12:12:51:19.437460: newcat.c(8051):set_roofing_filter_for_width entered
2021-05-12:12:51:19.437486: newcat.c(7988):set_roofing_filter entered
2021-05-12:12:51:19.437513: newcat_valid_command RF
2021-05-12:12:51:19.437538: newcat.c(6701):newcat_valid_command return(1)
2021-05-12:12:51:19.437561: newcat.c(9826):newcat_set_cmd entered
2021-05-12:12:51:19.437582: rig_flush: called for serial device
2021-05-12:12:51:19.437606: serial.c(629):serial_flush entered
2021-05-12:12:51:19.437632: tcflush
2021-05-12:12:51:19.437659: serial.c(661):serial_flush return(0)
2021-05-12:12:51:19.437680: cmd_str = RF13;
2021-05-12:12:51:19.437701: newcat.c(9709):newcat_set_cmd_validate entered
2021-05-12:12:51:19.437723: newcat_set_cmd_validate: priv->cmd_str=RF13;
2021-05-12:12:51:19.437745: newcat_set_cmd_validate: RF13; not implemented
2021-05-12:12:51:19.437767: newcat.c(9749):newcat_set_cmd_validate return(-4)
2021-05-12:12:51:19.437791: newcat_set_cmd: newcat_set_cmd_validate not implemented...continuing
2021-05-12:12:51:19.437814: write_block called
2021-05-12:12:51:19.437844: write_block(): TX 5 bytes
2021-05-12:12:51:19.437874: 0000    52 46 31 33 3b                                      RF13;           
2021-05-12:12:51:19.437897: cmd_str = ID;
2021-05-12:12:51:19.437918: write_block called
2021-05-12:12:51:19.437947: write_block(): TX 3 bytes
2021-05-12:12:51:19.437975: 0000    49 44 3b                                            ID;             
2021-05-12:12:51:19.437998: read_string called, rxmax=129
2021-05-12:12:51:19.453327: read_string(): RX 2 characters
2021-05-12:12:51:19.453354: 0000    3f 3b                                               ?;              
2021-05-12:12:51:19.453376: newcat_set_cmd: read count = 2, ret_data = ?;
2021-05-12:12:51:19.453397: newcat_set_cmd: Command rejected by the rig (set): 'RF13;'

mikaelnousiainen, 2021-05-12:
Answering here as I was implementing this logic. It seems there's some code trying to use targetable VFOs here. The first digit has to be `0` always for FT-2000 (but not necessarily for other rigs).

mdblack98, 2021-05-12:
Yes...the code in ft2000.c shows that -- so have to track down where it's getting the wrong pair from.
There is no 13 combo.
const struct newcat_priv_caps ft2000_priv_caps ={    .roofing_filter_count = 7,    .roofing_filters =    {        // The index must match ext level combo index        { .index = 0, .set_value = '0', .get_value = 0, .width = 15000, .optional = 0 },        { .index = 1, .set_value = '1', .get_value = '1', .width = 15000, .optional = 0 },        { .index = 2, .set_value = '2', .get_value = '2', .width = 6000, .optional = 0 },        { .index = 3, .set_value = '3', .get_value = '3', .width = 3000, .optional = 0 },        { .index = 4, .set_value = 0, .get_value = '4', .width = 15000, .optional = 0 },        { .index = 5, .set_value = 0, .get_value = '5', .width = 6000, .optional = 0 },        { .index = 6, .set_value = 0, .get_value = '6', .width = 3000, .optional = 0 },    }};

mikaelnousiainen, 2021-05-12:
It's this condition:

https://github.com/Hamlib/Hamlib/blob/870b9ca7481fc40255749194fe6215cbbc74df93/rigs/yaesu/newcat.c#L7997

Same code is in the `get_roofing_filter` function.

Leaves a question:

Should we have `RIG_TARGETABLE_MODE` for FT-2000 (not sure right now), should we use some other condition (from the `RIG_TARGETABLE_` set) or make an exception for this rig? Can't remember all of those capabilities right now.

mdblack98, 2021-05-12:
Looks like we need TARGETABLE_ROOFING
Here are the Yaesu ones that look like they have VFOB capabilities
ft1200.c:        { .index = 1, .set_value = '1', .get_value = '1', .width = 15000, .optional = 0 },ft2000.c:        { .index = 1, .set_value = '1', .get_value = '1', .width = 15000, .optional = 0 },ft3000.c:        { .index = 1, .set_value = '1', .get_value = '1', .width = 15000, .optional = 0 },ft5000.c:        { .index = 1, .set_value = '1', .get_value = '1', .width = 15000, .optional = 0 },ft950.c:        { .index = 1, .set_value = '1', .get_value = '1', .width = 15000, .optional = 0 },ftdx101.c:        { .index = 1, .set_value = '1', .get_value = '6', .width = 12000, .optional = 0 },ftdx101mp.c:        { .index = 1, .set_value = '1', .get_value = '6', .width = 12000, .optional = 0 },ftdx10.c:        { .index = 1, .set_value = '1', .get_value = '6', .width = 12000, .optional = 0 },

Mike

 
 

https://github.com/Hamlib/Hamlib/blob/870b9ca7481fc40255749194fe6215cbbc74df93/rigs/yaesu/newcat.c#L7997

mikaelnousiainen, 2021-05-12:
Only FTDX5000 and FTDX101 seem to have targetable roofing filter. FT-1200 or FT-3000 (or any other rig) needs to have `0` as the first digit.

mdblack98, 2021-05-12:
FT-1200 is wrong -- it was probably just copied from another rig.
Do you want to fix these?

const struct newcat_priv_caps ftdx1200_priv_caps ={    .roofing_filter_count = 7,    .roofing_filters =    {        // The index must match ext level combo index        { .index = 0, .set_value = '0', .get_value = 0, .width = 15000, .optional = 0 },        { .index = 1, .set_value = '1', .get_value = '1', .width = 15000, .optional = 0 },        { .index = 2, .set_value = '2', .get_value = '2', .width = 6000, .optional = 0 },        { .index = 3, .set_value = '3', .get_value = '3', .width = 3000, .optional = 0 },        { .index = 4, .set_value = 0, .get_value = '4', .width = 15000, .optional = 0 },        { .index = 5, .set_value = 0, .get_value = '5', .width = 6000, .optional = 0 },        { .index = 6, .set_value = 0, .get_value = '6', .width = 3000, .optional = 0 },    }};

mikaelnousiainen, 2021-05-12:
The values are correct for FTDX1200. The targeting of VFO is not specified in those roofing filter tables, but in the `set_roofing_filter` and `get_roofing_filter` functions. And like you correctly identified in your earlier message, we'd probably need `RIG_TARGETABLE_ROOFING` and it would be enabled only for FTDX5000 and FTDX101. -> The conditions in those two functions to determine VFO number need to use this new `RIG_TARGETABLE_ROOFING` flag then.

mdblack98, 2021-05-12:
Do you want to add it or do you want me to do it?
Mike

mikaelnousiainen, 2021-05-12:
I can have make the changes later today.

PR #699 (2021-05-12): Fix Yaesu roofing filter VFO targeting
* Add RIG_TARGETABLE_ROOFING to allow Yaesu roofing filter targeting by VFO only for FTDX5000 and FTDX101.
* Fix TS-480 VOXDELAY level definition.

Not tested, as I don't have access to a suitable Yaesu rig right now.

Fixes #697
closed 2021-05-12 by mdblack98