Hamlib/Hamlib #700: get_level & set_level not working with FLrig 1.3.54

NeoChen1024, 2021-05-16:
```
rigctl, Hamlib 3.3
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_init called
dummy: _init called
rig_register called
rig_register: rig_register (1)
rig_register called
rig_register: rig_register (2)
rig_register called
rig_register: rig_register (4)
rig_register called
rig_register: rig_register (5)
rig_open called
port_open called
network_open called
flrig_open version 1.4
write_block called
read_string called
read_transaction: string='HTTP/1.1 200 OK
'read_string called
read_transaction: string='Server: XMLRPC++ 0.8
'read_string called
read_transaction: string='Content-Type: text/xml
'read_string called
read_transaction: string='Content-length: 117
'read_string called
read_transaction: string='
'read_string called
read_transaction: string='<?xml version="1.0"?>
'read_string called
read_transaction: string='<methodResponse><params><param>
'read_string called
read_transaction: string='      <value>IC-7300</value>
'read_string called
read_transaction: string='</param></params></methodResponse>
'read_transaction: got </methodResponse>
xml_parse XML:
HTTP/1.1 200 OK
Server: XMLRPC++ 0.8
Content-Type: text/xml
Content-length: 117

<?xml version="1.0"?>
<methodResponse><params><param>
        <value>IC-7300</value>
</param></params></methodResponse>

Transceiver=IC-7300
write_block called
read_string called
read_transaction: string='HTTP/1.1 200 OK
'read_string called
read_transaction: string='Server: XMLRPC++ 0.8
'read_string called
read_transaction: string='Content-Type: text/xml
'read_string called
read_transaction: string='Content-length: 112
'read_string called
read_transaction: string='
'read_string called
read_transaction: string='<?xml version="1.0"?>
'read_string called
read_transaction: string='<methodResponse><params><param>
'read_string called
read_transaction: string='      <value>AM</value>
'read_string called
read_transaction: string='</param></params></methodResponse>
'read_transaction: got </methodResponse>
xml_parse XML:
HTTP/1.1 200 OK
Server: XMLRPC++ 0.8
Content-Type: text/xml
Content-length: 112

<?xml version="1.0"?>
<methodResponse><params><param>
        <value>AM</value>
</param></params></methodResponse>

flrig_open: getmodeA is available=AM
write_block called
read_string called
read_transaction: string='HTTP/1.1 200 OK
'read_string called
read_transaction: string='Server: XMLRPC++ 0.8
'read_string called
read_transaction: string='Content-Type: text/xml
'read_string called
read_transaction: string='Content-length: 173
'read_string called
read_transaction: string='
'read_string called
read_transaction: string='<?xml version="1.0"?>
'read_string called
read_transaction: string='<methodResponse><params><param>
'read_string called
read_transaction: string='      <value><array><data><value>10000</value><value></value></data></array></value>
'read_string called
read_transaction: string='</param></params></methodResponse>
'read_transaction: got </methodResponse>
xml_parse XML:
HTTP/1.1 200 OK
Server: XMLRPC++ 0.8
Content-Type: text/xml
Content-length: 173

<?xml version="1.0"?>
<methodResponse><params><param>
        <value><array><data><value>10000</value><value></value></data></array></value>
</param></params></methodResponse>

flrig_open: get_bwA is available=10000
write_block called
read_string called
read_transaction: string='HTTP/1.1 200 OK
'read_string called
read_transaction: string='Server: XMLRPC++ 0.8
'read_string called
read_transaction: string='Content-Type: text/xml
'read_string called
read_transaction: string='Content-length: 111
'read_string called
read_transaction: string='
'read_string called
read_transaction: string='<?xml version="1.0"?>
'read_string called
read_transaction: string='<methodResponse><params><param>
'read_string called
read_transaction: string='      <value>A</value>
'read_string called
read_transaction: string='</param></params></methodResponse>
'read_transaction: got </methodResponse>
xml_parse XML:
HTTP/1.1 200 OK
Server: XMLRPC++ 0.8
Content-Type: text/xml
Content-length: 111

<?xml version="1.0"?>
<methodResponse><params><param>
        <value>A</value>
</param></params></methodResponse>

rig_strvfo called
flrig_open: currvfo=VFOA value=A
write_block called
read_string called
read_transaction: string='HTTP/1.1 200 OK
'read_string called
read_transaction: string='Server: XMLRPC++ 0.8
'read_string called
read_transaction: string='Content-Type: text/xml
'read_string called
read_transaction: string='Content-length: 362
'read_string called
read_transaction: string='
'read_string called
read_transaction: string='<?xml version="1.0"?>
'read_string called
read_transaction: string='<methodResponse><params><param>
'read_string called
read_transaction: string='      <value><array><data><value>LSB</value><value>USB</value><value>AM</value><value>FM</value><value>CW</value><value>CW-R</value><value>RTTY</value><value>RTTY-R</value><value>LSB-D</value><value>USB-D</value><value>AM-D</value><value>FM-D</value></data></array></value>
'read_string called
read_transaction: string='</param></params></methodResponse>
'read_transaction: got </methodResponse>
xml_parse XML:
HTTP/1.1 200 OK
Server: XMLRPC++ 0.8
Content-Type: text/xml
Content-length: 362

<?xml version="1.0"?>
<methodResponse><params><param>
        <value><array><data><value>LSB</value><value>USB</value><value>AM</value><value>FM</value><value>CW</value><value>CW-R</value><value>RTTY</value><value>RTTY-R</value><value>LSB-D</value><value>USB-D</value><value>AM-D</value><value>FM-D</value></data></array></value>
</param></params></methodResponse>

flrig_open: modes=LSB|USB|AM|FM|CW|CW-R|RTTY|RTTY-R|LSB-D|USB-D|AM-D|FM-D
flrig_open: hamlib modes=0x00000dbf
rig_get_vfo called
write_block called
read_string called
read_transaction: string='HTTP/1.1 200 OK
'read_string called
read_transaction: string='Server: XMLRPC++ 0.8
'read_string called
read_transaction: string='Content-Type: text/xml
'read_string called
read_transaction: string='Content-length: 111
'read_string called
read_transaction: string='
'read_string called
read_transaction: string='<?xml version="1.0"?>
'read_string called
read_transaction: string='<methodResponse><params><param>
'read_string called
read_transaction: string='      <value>A</value>
'read_string called
read_transaction: string='</param></params></methodResponse>
'read_transaction: got </methodResponse>
xml_parse XML:
HTTP/1.1 200 OK
Server: XMLRPC++ 0.8
Content-Type: text/xml
Content-length: 111

<?xml version="1.0"?>
<methodResponse><params><param>
        <value>A</value>
</param></params></methodResponse>

flrig_get_vfo: vfo value=A
rig_strvfo called
flrig_get_vfo: vfo=VFOA
Opened rig model 4, 'FLRig'
rig_strstatus called
Backend version: 1.4, Status: Stable
rig_parse_level called
rig_has_get_level called
rig_ext_lookup called
get_level: error = Invalid parameter
rig_close called
port_close called
network_close called
rig_cleanup called
```

mdblack98, 2021-05-16:
You need to upgrade hamlib.  That should be fixed in 4.1 and 4.2 will be coming out shortly.
Mike W9MDB

NeoChen1024, 2021-05-17:
Okay, it seems it's the package in Arch Linux Repo. has been outdated for a while. I am sorry for not noticing it.

closed 2021-05-17 by NeoChen1024