Hamlib/Hamlib #710: Multicast feature breaks MinGW build

N0NB, 2021-05-22:
Hi Mike.

I guess this affected yesterday's run already but I just remembered to check this.

Cleaned out my build directory and ran my scripts manually.  Here is the start for 32 bit:

```
$ hamlib-daily-w32.sh
RELEASE = 4.3~git
GITSHA1 = 82c27e55
YMD = 20210521
Building Hamlib 4.3~git for W32
```
And it fails at this toward the end:

```
make[1]: Entering directory '/home/nate/builds/hamlib-4.3~git/src'
make  install-am
make[2]: Entering directory '/home/nate/builds/hamlib-4.3~git/src'
  CC       rig.lo
  CC       misc.lo
  CC       serial.lo
  CC       register.lo
rig.c: In function ‘rig_cookie’:
rig.c:6574:63: warning: implicit declaration of function ‘random’; did you mean ‘rand’? [-Wimplicit-function-declaration]
         snprintf(cookie, cookie_len, "%s %ld\n", cookie_save, random());
                                                               ^~~~~~
                                                               rand
rig.c:6574:44: warning: format ‘%ld’ expects argument of type ‘long int’, but argument 5 has type ‘int’ [-Wformat=]
         snprintf(cookie, cookie_len, "%s %ld\n", cookie_save, random());
                                          ~~^                  ~~~~~~~~
                                          %d
  CC       event.lo
  CC       cal.lo
  CC       conf.lo
  CC       tones.lo
  CC       rotator.lo
  CC       locator.lo
  CC       rot_reg.lo
  CC       rot_conf.lo
  CC       rot_settings.lo
  CC       rot_ext.lo
  CC       iofunc.lo
  CC       ext.lo
  CC       mem.lo
  CC       settings.lo
  CC       parallel.lo
  CC       usb_port.lo
  CC       debug.lo
  CC       network.lo
  CC       cm108.lo
  CC       gpio.lo
  CC       microham.lo
  CC       amplifier.lo
  CC       amp_reg.lo
  CC       amp_conf.lo
  CC       amp_settings.lo
  CC       extamp.lo
  CC       sleep.lo
  CC       sprintflst.lo
  CCLD     libhamlib.la
/usr/bin/i686-w64-mingw32-ld: .libs/rig.o: in function `rig_cookie':
/home/nate/builds/hamlib-4.3~git/src/rig.c:6574: undefined reference to `random'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:535: libhamlib.la] Error 1
make[2]: Leaving directory '/home/nate/builds/hamlib-4.3~git/src'
make[1]: *** [Makefile:709: install] Error 2
make[1]: Leaving directory '/home/nate/builds/hamlib-4.3~git/src'
make: *** [Makefile:603: install-recursive] Error 1
```

**Now 64 bit**

Looks to be about the same result:

```
$ hamlib-daily-w64.sh
RELEASE = 4.3~git
GITSHA1 = 82c27e55
YMD = 20210521
Building Hamlib 4.3~git for W64
```

And fails the same:

```
Making install in src
make[1]: Entering directory '/home/nate/builds/hamlib-4.3~git/src'
make  install-am
make[2]: Entering directory '/home/nate/builds/hamlib-4.3~git/src'
  CC       rig.lo
  CC       serial.lo
  CC       misc.lo
  CC       register.lo
rig.c: In function ‘rig_cookie’:
rig.c:6574:63: warning: implicit declaration of function ‘random’; did you mean ‘rand’? [-Wimplicit-function-declaration]
         snprintf(cookie, cookie_len, "%s %ld\n", cookie_save, random());
                                                               ^~~~~~
                                                               rand
rig.c:6574:44: warning: format ‘%ld’ expects argument of type ‘long int’, but argument 5 has type ‘int’ [-Wformat=]
         snprintf(cookie, cookie_len, "%s %ld\n", cookie_save, random());
                                          ~~^                  ~~~~~~~~
                                          %d
  CC       event.lo
  CC       cal.lo
  CC       conf.lo
  CC       tones.lo
  CC       rotator.lo
  CC       locator.lo
  CC       rot_reg.lo
  CC       rot_conf.lo
  CC       rot_settings.lo
  CC       rot_ext.lo
  CC       iofunc.lo
  CC       ext.lo
  CC       mem.lo
  CC       settings.lo
  CC       parallel.lo
  CC       usb_port.lo
  CC       debug.lo
  CC       network.lo
  CC       cm108.lo
  CC       gpio.lo
  CC       microham.lo
  CC       amplifier.lo
  CC       amp_reg.lo
  CC       amp_conf.lo
  CC       amp_settings.lo
  CC       extamp.lo
  CC       sleep.lo
  CC       sprintflst.lo
  CCLD     libhamlib.la
/usr/bin/x86_64-w64-mingw32-ld: .libs/rig.o: in function `rig_cookie':
/home/nate/builds/hamlib-4.3~git/src/rig.c:6574: undefined reference to `random'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:535: libhamlib.la] Error 1
make[2]: Leaving directory '/home/nate/builds/hamlib-4.3~git/src'
make[1]: *** [Makefile:709: install] Error 2
make[1]: Leaving directory '/home/nate/builds/hamlib-4.3~git/src'
make: *** [Makefile:603: install-recursive] Error 1
```

Hope that helps.

N0NB, 2021-05-22:
The rand() is fixed but now am seeing this from the linker:

```
Making install in tests
make[1]: Entering directory '/home/nate/builds/hamlib-4.3~git/tests'
  CC       rigctl-rigctl.o
  CC       rigctl-rigctl_parse.o
  CC       rigctl-dumpcaps.o
  CC       rigctld-rigctld.o
  CC       rigctld-rigctl_parse.o
  CC       rigctld-dumpcaps.o
  CC       rigmem-rigmem.o
  CC       rigmem-memsave.o
  CC       rigmem-memload.o
  CC       rigmem-memcsv.o
  CC       rigsmtr.o
  CC       rigswr.o
  CC       rotctl-rotctl.o
  CC       rotctl-rotctl_parse.o
  CC       rotctl-dumpcaps_rot.o
  CC       rotctld-rotctld.o
  CC       rotctld-rotctl_parse.o
  CC       rotctld-dumpcaps_rot.o
  CC       rigctlcom-rigctlcom.o
  CC       rigctlcom-rigctl_parse.o
  CC       rigctlcom-dumpcaps.o
  CC       ampctl-ampctl.o
  CC       ampctl-ampctl_parse.o
  CC       ampctl-dumpcaps_amp.o
  CC       ampctld-ampctld.o
  CC       ampctld-ampctl_parse.o
  CC       ampctld-dumpcaps_amp.o
  CCLD     rigctl.exe
  CCLD     rigctld.exe
  CCLD     rigmem.exe
/usr/bin/i686-w64-mingw32-ld: rigctld-rigctld.o: in function `main':
/home/nate/builds/hamlib-4.3~git/tests/rigctld.c:749: undefined reference to `network_multicast_server'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:1002: rigctld.exe] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/nate/builds/hamlib-4.3~git/tests'
```

mdblack98, 2021-05-22:
This would indicate it's linking an old library
The function is defined in network.o and it builds on Linux, Windows and github.
nm network.o | grep multi0000000000000f60 T network_multicast_server
Unless there's some limit on symbol length in your linker....it's an older system, right?  I know there was some limit many moons ago.MISRA C 2004 and the ISO standard said don't depend on more then 31 chars but we're not near that yet.
Check the object file for the definition and also check the library that's being linked....there's a disagreement there.
Mike W9MDB

N0NB, 2021-05-22:
No old library.  As a reminder this is in a virtual machine running Debian 10 where Hamlib has never been installed, only built.

mdblack98, 2021-05-22:
Did you check the library to see if it contains the reference?
nm network.o | grep multi0000000000000f60 T network_multicast_server
nm /usr/local/lib/libhamlib.so | grep multi00000000000a5d90 T network_multicast_server

N0NB, 2021-05-22:
The address is too specific in your example.  Here is what I get:

```
$ nm network.o | grep "network_multicast_server"
00001470 T _network_multicast_server
```

No `libhamlib.so` exists on this VM.  Hamlib was never installed.  However, the local installation directory from which the W32/W64 binaries are packaged shows:

```
nate@buster:~/builds/hamlib-4.3~git/mingw32-inst/lib 
$ ls -a
./  ../  libhamlib.dll.a*  libhamlib.la*
```

and:

```
$ nm libhamlib.dll.a | grep "network_multicast_server"
```

N0NB, 2021-05-26:
Another build failure in the VM.  Just checked with:

```
RELEASE = 4.3~git
GITSHA1 = d2f7957e
YMD = 20210526
Building Hamlib 4.3~git for W32
```

Here is the error output:

```
Making install in tests
make[1]: Entering directory '/home/nate/builds/hamlib-4.3~git/tests'
  CC       rigctl-rigctl.o
  CC       rigctl-rigctl_parse.o
  CC       rigctl-dumpcaps.o
  CC       rigctld-rigctld.o
  CC       rigctld-rigctl_parse.o
  CC       rigctld-dumpcaps.o
  CC       rigmem-rigmem.o
  CC       rigmem-memsave.o
  CC       rigmem-memload.o
  CC       rigmem-memcsv.o
  CC       rigsmtr.o
  CC       rigswr.o
  CC       rotctl-rotctl.o
  CC       rotctl-rotctl_parse.o
  CC       rotctl-dumpcaps_rot.o
  CC       rotctld-rotctld.o
  CC       rotctld-rotctl_parse.o
  CC       rotctld-dumpcaps_rot.o
  CC       rigctlcom-rigctlcom.o
  CC       rigctlcom-rigctl_parse.o
  CC       rigctlcom-dumpcaps.o
  CC       ampctl-ampctl.o
  CC       ampctl-ampctl_parse.o
  CC       ampctl-dumpcaps_amp.o
  CC       ampctld-ampctld.o
  CC       ampctld-ampctl_parse.o
  CC       ampctld-dumpcaps_amp.o
  CCLD     rigctl.exe
  CCLD     rigctld.exe
  CCLD     rigmem.exe
/usr/bin/i686-w64-mingw32-ld: rigctl-rigctl.o: in function `main':
/home/nate/builds/hamlib-4.3~git/tests/rigctl.c:177: undefined reference to `cookie_use'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:994: rigctl.exe] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/nate/builds/hamlib-4.3~git/tests'
make: *** [Makefile:603: install-recursive] Error 1
```

mdblack98, 2021-05-26:
This should be fixed now.
As for the random() that's not in the current code.
Line 6524 shows rand() on the github copy of rig.c
Mike W9MDB

N0NB, 2021-05-26:
That fixed that part of the build and the daily snapshots built and have been uploaded as of SHA1 8ded4461.

I so see this in both 32 and 64 bit MinGW builds:

```
  CC       register.lo
rig.c:93:1: warning: ‘cdecl’ attribute only applies to function types [-Wattributes]
 HAMLIB_EXPORT(int) cookie_use;
 ^~~~~~~~~~~~~
```

N0NB, 2021-05-26:
I understand the `random()` has been corrected as the builds were working for the past couple of days.  I just reopened this issue since this seemed related.

mdblack98, 2021-05-26:
Try again...
Mike

N0NB, 2021-05-26:
Clean as a hound's tooth!

Thanks, Mike.

closed 2021-05-26 by N0NB