Hamlib/Hamlib #714: FT857D

KA5TTZ, 2021-05-30:
I am loading Hamlib on my Raspberry Pi3 B+ I have loaded the 4.2 and still can now get it to recognize my FT857D. I have tried every combination and still no luck. Here is the reply i get. i even went as fat as pi@raspberrypi:~ $ rigctl -m 1022 and still have now luck. 

pi@raspberrypi:~ $ rigctl -m 1022 -r /dev/ttyUSB1 -s 38400 f
rig_open: error = IO error
port_open: serial_open status=-6
iofunc.c(80):port_open return(-6)
rig.c(773):rig_open return(-6) 
pi@raspberrypi:~ $ rigctl -m 1022 -r /dev/ttyUSB1 -s 19200 f
rig_open: error = IO error
port_open: serial_open status=-6
iofunc.c(80):port_open return(-6)
rig.c(773):rig_open return(-6) 
pi@raspberrypi:~ $ 9600
bash: 9600: command not found
pi@raspberrypi:~ $ rigctl -m 1022 -r /dev/ttyUSB1 -s 9600 f
rig_open: error = IO error
port_open: serial_open status=-6
iofunc.c(80):port_open return(-6)
rig.c(773):rig_open return(-6) 
pi@raspberrypi:~ $ rigctl -m 1022 -r /dev/ttyUSB1 -s 4800 f
rig_open: error = IO error
port_open: serial_open status=-6
iofunc.c(80):port_open return(-6)
rig.c(773):rig_open return(-6) 

I tried with two different cable I have on my rig. RTsystem USB 62C  and another cable i have. No luck what so ever. What I am trying to set up is a APRS station using this radio and PITNC_9K6 modem which use the Teensy 3.2.

mdblack98, 2021-05-30:
Are you sure ttyUSB1 is the right serial port?  And not ttyUSB0 or some other device?
If you increased the verbose a bit you'll probably see this
igctl -m 1022 -r /dev/ttyUSB0 -s 9600 -vvserial_open: Unable to open /dev/ttyUSB0 - No such file or directoryport_open: serial_open status=-6rig_open: error = IO errorport_open: serial_open status=-6iofunc.c(80):port_open return(-6)rig.c(776):rig_open return(-6) 

Mike W9MDB

KA5TTZ, 2021-05-30:
Thanks Mike. Yes the cable is USB1, because I used another cable before that one and it is USB0, but what I found out early this morning was that in the menu setting I had the port to TUNER only, instead to CAT. I am redoing it now and will be loading Hamlib 4.2 here soon. Thanks again for the help.
73's
David KC5DP

closed 2021-05-30 by KA5TTZ