Hamlib/Hamlib #716: Implement Icom network cat control

mdblack98, 2021-06-03:
e.g. IC-9700 has a network port for CAT control
Protocol is not CI-V and need manual for the network protocol.

mikaelnousiainen, 2021-06-03:
@mdblack98 It seems there are three UDP ports used by the network-capable Icom rigs: IC-7610, IC-785x, IC-9700, IC-R8600 and IC-705. First of them is some "control port" that is used (at least) for some form of authentication and enabling of remote control features. The second one is "serial port" that somehow uses the CI-V protocol frames inside UDP packets. The third one is audio port that is used to transfer (PCM?) audio.

It's weird that there is no mention of these additional protocols in the manuals, but it seems some hams have implemented the protocol already in an open source app: https://github.com/nonoo/kappanhang

Maybe there is some documentation available after all.

Now only missing suitable hardware for this :)

mikaelnousiainen, 2021-06-03:
@mdblack98 Is this something you could ask docs for from Icom? They must have something. Not really sure what it is called: UDP network protocol / RS-BA1 server protocol?

mdblack98, 2021-06-03:
I did ask...

mdblack98, 2021-06-04:
Icom's inadequate response 

Other than the CI-V command manual that is available on the ICOM America Knowledge Base there is no other information available for radio control. The link for this information is:  https://www.dropbox.com/s/fe7wf3icqzoueko/IC-9700_CI-V%20Reference%20Guide.pdf?dl=0
 
Engineering has not provided any information to us on the network interface.
 
 Regards,
 
Fred Varian
IA Technical Support

arodland, 2021-12-21:
Are you 100% sure it's not CI-V? I've used `socat` on a Linux machine to connect a TCP listen socket to a radio's CI-V port, then told RS-BA1 to connect to that port, and it worked.

Yes, there's some other stuff, including a discovery protocol over UDP, and a separate protocol for the remote audio, but I believe that the CAT aspect of it is 100% just CI-V over TCP.

mdblack98, 2021-12-21:
As I recall we tried a TCP connection but not UDP.
Hamlib does not have a UDP interface so perhaps could add that and give it a try.

mikaelnousiainen, 2021-12-21:
Like I've mentioned above, the Kappanhang project has a working implementation of the Icom network protocol.

The protocol seems to use separate UDP ports for different streams (control/auth, serial, audio) and the CI-V commands are wrapped inside UDP frames:

Control: https://github.com/nonoo/kappanhang/blob/main/controlstream.go#L41

Serial: https://github.com/nonoo/kappanhang/blob/main/serialstream.go#L33

In a way Kappanhang could used *with* (Linux only, of course) Hamlib as it seems to use loopback serial and audio devices to expose the radio connection.

This could be implemented in Hamlib too, but it requires quite a lot of logic -- and to be useful Hamlib should probably have a way to transfer audio to and from the device. Maybe through raw UDP audio streams?

mdblack98, 2021-12-21:
This does require the RS_BA software to be running, right?
Or does the rig itself does support this?
Mike W9MDB

arodland, 2021-12-21:
The rigs that have built-in Ethernet/WiFi have the equivalent of the RS-BA1
server onboard.

mikaelnousiainen, 2021-12-22:
Yeah, indeed, the Icom rigs with an Ethernet port are running the UDP networking internally (you can configure ports and all in the menus), so no 3rd party software is necessary.

I guess the RS-BA1 server is meant for enabling remote control for rigs with USB/serial port only (like my IC-7300).

guitarpicva, 2023-05-09:
kappenhang's github work is in golang which makes it a bit challenging for C/C++ folk to translate but you can derive the general sequence and format of the command and response frames to set up a connection to the server in the radio.

Thus far, I have gotten most of the way to creating the actual "login" to the radio.  To the point of initiating the UDP streams.  Had to set it aside for now.