Hamlib/Hamlib #718: Kenwood TH-D74 does not work with rigctld 4.2

giordyb, 2021-06-05:
Hi,

It seems that the support for the Kenwood TH-D74 broke with Hamlib 4.2, this is the error message that I get when I try to run rigctld:

rigctld -t 4533 -m 2042 -r /dev/cu.usbmodem14201 -vvvvv

rigctld Hamlib 4.2 Mon May 17 02:40:24 2021 +0000 SHA=cdad07
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_check_rig_caps: p1=0x105c09dd0, p2=0x105c10148, rig_model=0x105c09dd0, macro_name=0x105c10148
initrigs4_kenwood called
rig_register called
rig_register: rig_register (2012)
rig_register called
rig_register: rig_register (2013)
rig_register called
rig_register: rig_register (2001)
rig_register called
rig_register: rig_register (2025)
rig_register called
rig_register: rig_register (2003)
rig_register called
rig_register: rig_register (2004)
rig_register called
rig_register: rig_register (2016)
rig_register called
rig_register: rig_register (2024)
rig_register called
rig_register: rig_register (2005)
rig_register called
rig_register: rig_register (2007)
rig_register called
rig_register: rig_register (2009)
rig_register called
rig_register: rig_register (2010)
rig_register called
rig_register: rig_register (2022)
rig_register called
rig_register: rig_register (2014)
rig_register called
rig_register: rig_register (2030)
rig_register called
rig_register: rig_register (2021)
rig_register called
rig_register: rig_register (2029)
rig_register called
rig_register: rig_register (2043)
rig_register called
rig_register: rig_register (2044)
rig_register called
rig_register: rig_register (2045)
rig_register called
rig_register: rig_register (2047)
rig_register called
rig_register: rig_register (2038)
rig_register called
rig_register: rig_register (2002)
rig_register called
rig_register: rig_register (2011)
rig_register called
rig_register: rig_register (2006)
rig_register called
rig_register: rig_register (2008)
rig_register called
rig_register: rig_register (2015)
rig_register called
rig_register: rig_register (2026)
rig_register called
rig_register: rig_register (2017)
rig_register called
rig_register: rig_register (2033)
rig_register called
rig_register: rig_register (2042)
rig_register called
rig_register: rig_register (2020)
rig_register called
rig_register: rig_register (2023)
rig_register called
rig_register: rig_register (2027)
rig_register called
rig_register: rig_register (2034)
rig_register called
rig_register: rig_register (2031)
rig_register called
rig_register: rig_register (2039)
rig_register called
rig_register: rig_register (2037)
rig_register called
rig_register: rig_register (2028)
rig_register called
rig_register: rig_register (2019)
rig_register called
rig_register: rig_register (2032)
rig_register called
rig_register: rig_register (2036)
rig_register called
rig_register: rig_register (2048)
rig_register called
rig_register: rig_register (2040)
rig_register called
rig_register: rig_register (2041)
rig_register called
rig_register: rig_register (2046)
rig_register called
rig_register: rig_register (2049)
kenwood.c(5351):initrigs4_kenwood return(0)
rig_init: rx_range_list1 is empty, using rx_range_list2
rig_init: rig has VFO_A
rig_init: rig has VFO_B
kenwood_init called, version 20210427/20210427.0
kenwood_init: if_len = 37
kenwood.c(738):kenwood_init return(0)
rigctl_parse.c(1916):set_conf entered
rigctl_parse.c(1950):set_conf return(0)
main: twiddle=0, uplink=0, twiddle_rit=0
rig.c(669):rig_open entered
iofunc.c(68):port_open entered
serial.c(144):serial_open entered
serial_open: /dev/cu.usbmodem14201
serial.c(278):serial_setup entered
serial_setup: tcgetattr
serial_setup: cfmakeraw
serial_setup: cfsetispeed=9600,0x2580
serial_setup: cfsetospeed=9600,0x2580
serial_setup: data_bits=8
serial_setup: parity=0
serial_setup: tcsetattr TCSANOW
serial.c(618):serial_setup return(0)
serial.c(629):serial_flush entered
tcflush
serial.c(661):serial_flush return(0)
serial.c(253):serial_open return(0)
iofunc.c(185):port_open return(0)
rig.c(217):add_opened_rig return(0)
kenwood_transaction called
kenwood_transaction: cmdstr = 
rig_flush: called for serial device
serial.c(629):serial_flush entered
tcflush
serial.c(661):serial_flush return(0)
write_block called
write_block(): TX 1 bytes
0000    0d                                                  .               
write_block called
write_block(): TX 3 bytes
0000    49 44 0d                                            ID.             
read_string called, rxmax=35
read_string(): RX 2 characters
0000    3f 0d                                               ?.              
'enwood_transaction: read_string(len=2)='?
kenwood_transaction: Unknown command or rig busy ''
kenwood_transaction: Retrying shortly
kenwood_transaction: cmdstr = 
rig_flush: called for serial device
serial.c(629):serial_flush entered
tcflush
serial.c(661):serial_flush return(0)
write_block called
write_block(): TX 1 bytes
0000    0d                                                  .               
write_block called
write_block(): TX 3 bytes
0000    49 44 0d                                            ID.             
read_string called, rxmax=35
read_string(): RX 2 characters
0000    3f 0d                                               ?.              
'enwood_transaction: read_string(len=2)='?
kenwood_transaction: Unknown command or rig busy ''
kenwood_transaction: Retrying shortly
kenwood_transaction: cmdstr = 
rig_flush: called for serial device
serial.c(629):serial_flush entered
tcflush
serial.c(661):serial_flush return(0)
write_block called
write_block(): TX 1 bytes
0000    0d                                                  .               
write_block called
write_block(): TX 3 bytes
0000    49 44 0d                                            ID.             
read_string called, rxmax=35
read_string(): RX 2 characters
0000    3f 0d                                               ?.              
'enwood_transaction: read_string(len=2)='?
kenwood_transaction: Unknown command or rig busy ''
kenwood_transaction: Retrying shortly
kenwood_transaction: cmdstr = 
rig_flush: called for serial device
serial.c(629):serial_flush entered
tcflush
serial.c(661):serial_flush return(0)
write_block called
write_block(): TX 1 bytes
0000    0d                                                  .               
write_block called
write_block(): TX 3 bytes
0000    49 44 0d                                            ID.             
read_string called, rxmax=35
read_string(): RX 2 characters
0000    3f 0d                                               ?.              
'enwood_transaction: read_string(len=2)='?
kenwood_transaction: Unknown command or rig busy ''
kenwood_transaction: returning retval=-9
kenwood.c(577):kenwood_transaction return(-9)
rig.c(1000):rig_open return(-9)
rig_open: error = Command rejected by the rig
kenwood_transaction: returning retval=-9
kenwood.c(577):kenwood_transaction return(-9)
rig.c(1000):rig_open return(-9) 

If I run the same exact command but after compiling and installing Hamlib 4.1 it works fine.

This happens on both macOS Big Sur and Raspbian with Hamlib compiled from source.

Thank you

Giordano

mdblack98, 2021-06-05:
Can you give me a debug log from 4.1 too where it works?
Mike W9MDB

giordyb, 2021-06-05:
Hi Mike, 

sure thing, here's the log! let me know if you need anything else.
```
#hamlib-4.1 rigctld -t 4533 -m 2042 -r /dev/cu.usbmodem14201 -vvvvv
Recommend using --vfo switch for rigctld if client supports it
rigctl and netrigctl will automatically detect vfo mode
rigctld Hamlib 4.1
Last commit was Mon Feb 01 01:56:33 2021 +0000 SHA=645717
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig.c(361):rig_init entered
rig_check_rig_caps: p1=0x104779800, p2=0x10477fb70, rig_model=0x104779800, macro_name=0x10477fb70
initrigs4_kenwood called
rig_register called
rig_register: rig_register (2012)
rig_register called
rig_register: rig_register (2013)
rig_register called
rig_register: rig_register (2001)
rig_register called
rig_register: rig_register (2025)
rig_register called
rig_register: rig_register (2003)
rig_register called
rig_register: rig_register (2004)
rig_register called
rig_register: rig_register (2016)
rig_register called
rig_register: rig_register (2024)
rig_register called
rig_register: rig_register (2005)
rig_register called
rig_register: rig_register (2007)
rig_register called
rig_register: rig_register (2009)
rig_register called
rig_register: rig_register (2010)
rig_register called
rig_register: rig_register (2022)
rig_register called
rig_register: rig_register (2014)
rig_register called
rig_register: rig_register (2030)
rig_register called
rig_register: rig_register (2021)
rig_register called
rig_register: rig_register (2029)
rig_register called
rig_register: rig_register (2043)
rig_register called
rig_register: rig_register (2044)
rig_register called
rig_register: rig_register (2045)
rig_register called
rig_register: rig_register (2047)
rig_register called
rig_register: rig_register (2038)
rig_register called
rig_register: rig_register (2002)
rig_register called
rig_register: rig_register (2011)
rig_register called
rig_register: rig_register (2006)
rig_register called
rig_register: rig_register (2008)
rig_register called
rig_register: rig_register (2015)
rig_register called
rig_register: rig_register (2026)
rig_register called
rig_register: rig_register (2017)
rig_register called
rig_register: rig_register (2033)
rig_register called
rig_register: rig_register (2042)
rig_register called
rig_register: rig_register (2020)
rig_register called
rig_register: rig_register (2023)
rig_register called
rig_register: rig_register (2027)
rig_register called
rig_register: rig_register (2034)
rig_register called
rig_register: rig_register (2031)
rig_register called
rig_register: rig_register (2039)
rig_register called
rig_register: rig_register (2037)
rig_register called
rig_register: rig_register (2028)
rig_register called
rig_register: rig_register (2019)
rig_register called
rig_register: rig_register (2032)
rig_register called
rig_register: rig_register (2036)
rig_register called
rig_register: rig_register (2048)
rig_register called
rig_register: rig_register (2040)
rig_register called
rig_register: rig_register (2041)
rig_register called
rig_register: rig_register (2046)
rig_init: rx_range_list1 is empty, using rx_range_list2
rig_init: rig has VFO_A
rig_init: rig has VFO_B
kenwood_init called, version 20201231/20201231.0
kenwood_init: if_len = 37
rig.c(613):rig_init return
main: twiddle=0, uplink=0
rig.c(642):rig_open entered
port_open called
serial_open called
serial_setup called
serial_setup: tcgetattr
serial_setup: cfmakeraw
serial_setup: cfsetispeed=9600,0x2580
serial_setup: cfsetospeed=9600,0x2580
serial_setup: data_bits=8
serial_setup: parity=0
serial_setup: tcsetattr TCSANOW
serial_flush called
tcflush
rig.c(213):add_opened_rig return
kenwood_transaction called
kenwood_transaction: cmdstr =
rig_flush: called for serial device
serial_flush called
tcflush
write_block called
write_block(): TX 1 bytes
0000    0d                                                  .
write_block called
write_block(): TX 3 bytes
0000    49 44 0d                                            ID.
read_string called, rxmax=35
read_string(): RX 2 characters
0000    3f 0d                                               ?.
'enwood_transaction: read_string(len=2)='?
kenwood_transaction: Unknown command or rig busy ''
kenwood_transaction: Retrying shortly
read_string called, rxmax=35
read_string(): RX 10 characters
0000    49 44 20 54 48 2d 44 37 34 0d                       ID TH-D74.
'enwood_transaction: read_string(len=10)='ID TH-D74
 in ID TH-D74action: No data expected, checking ID
kenwood_transaction: returning RIG_OK, retval=0
kenwood_transaction: returning retval=0
rig.c(2308):rig_get_vfo entered
elapsed_ms: start = 0,0
rig_get_vfo: cache check age=1000000ms
rig_get_vfo: cache miss age=1000000ms
thd74_get_vfo: called
kenwood_transaction called
kenwood_transaction: cmdstr = BC
rig_flush: called for serial device
serial_flush called
tcflush
write_block called
write_block(): TX 3 bytes
0000    42 43 0d                                            BC.
read_string called, rxmax=11
read_string(): RX 5 characters
0000    42 43 20 30 0d                                      BC 0.
'enwood_transaction: read_string(len=5)='BC 0
kenwood_transaction: returning RIG_OK, retval=0
kenwood_transaction: returning retval=0
elapsed_ms: start = 0,0
elapsed_ms: after gettime, start = 1622920290,345128000
rig.c(2358):rig_get_vfo return
rig.c(1037):rig_open return
Opened rig model 2042, 'TH-D74'
Backend version: 20201231.0, Status: Alpha
main: Using IPV6
```

mdblack98, 2021-06-05:
Can you please clone the current master and see if it works.
Also...please add the -Z switch as this looks like it could be a timing issue too on both 4.1 and your testing of the current version.
Mike W9MDB

giordyb, 2021-06-06:
Hi Mike,

Master works, here are the logs with the timestamp added for 4.1 4.2 and master.

4.1
```
➜  Downloads rigctld -t 4533 -m 2042 -vvvvv -Z -r /dev/cu.usbmodem14101
Recommend using --vfo switch for rigctld if client supports it
rigctl and netrigctl will automatically detect vfo mode
2021-06-06:08:42:17.296901: rigctld Hamlib 4.1
Last commit was Mon Feb 01 01:56:33 2021 +0000 SHA=645717
2021-06-06:08:42:17.296947: Report bugs to <hamlib-developer@lists.sourceforge.net>

2021-06-06:08:42:17.296960: rig.c(361):rig_init entered
2021-06-06:08:42:17.296965: rig_check_rig_caps: p1=0x10fb0c800, p2=0x10fb12b70, rig_model=0x10fb0c800, macro_name=0x10fb12b70
2021-06-06:08:42:17.296983: initrigs4_kenwood called
2021-06-06:08:42:17.296987: rig_register called
2021-06-06:08:42:17.296992: rig_register: rig_register (2012)
2021-06-06:08:42:17.297003: rig_register called
2021-06-06:08:42:17.297006: rig_register: rig_register (2013)
2021-06-06:08:42:17.297010: rig_register called
2021-06-06:08:42:17.297015: rig_register: rig_register (2001)
2021-06-06:08:42:17.297019: rig_register called
2021-06-06:08:42:17.297022: rig_register: rig_register (2025)
2021-06-06:08:42:17.297026: rig_register called
2021-06-06:08:42:17.297030: rig_register: rig_register (2003)
2021-06-06:08:42:17.297033: rig_register called
2021-06-06:08:42:17.297038: rig_register: rig_register (2004)
2021-06-06:08:42:17.297041: rig_register called
2021-06-06:08:42:17.297045: rig_register: rig_register (2016)
2021-06-06:08:42:17.297049: rig_register called
2021-06-06:08:42:17.297052: rig_register: rig_register (2024)
2021-06-06:08:42:17.297056: rig_register called
2021-06-06:08:42:17.297060: rig_register: rig_register (2005)
2021-06-06:08:42:17.297064: rig_register called
2021-06-06:08:42:17.297067: rig_register: rig_register (2007)
2021-06-06:08:42:17.297071: rig_register called
2021-06-06:08:42:17.297075: rig_register: rig_register (2009)
2021-06-06:08:42:17.297078: rig_register called
2021-06-06:08:42:17.297082: rig_register: rig_register (2010)
2021-06-06:08:42:17.297085: rig_register called
2021-06-06:08:42:17.297089: rig_register: rig_register (2022)
2021-06-06:08:42:17.297093: rig_register called
2021-06-06:08:42:17.297096: rig_register: rig_register (2014)
2021-06-06:08:42:17.297100: rig_register called
2021-06-06:08:42:17.297104: rig_register: rig_register (2030)
2021-06-06:08:42:17.297107: rig_register called
2021-06-06:08:42:17.297111: rig_register: rig_register (2021)
2021-06-06:08:42:17.297114: rig_register called
2021-06-06:08:42:17.297118: rig_register: rig_register (2029)
2021-06-06:08:42:17.297121: rig_register called
2021-06-06:08:42:17.297125: rig_register: rig_register (2043)
2021-06-06:08:42:17.297129: rig_register called
2021-06-06:08:42:17.297132: rig_register: rig_register (2044)
2021-06-06:08:42:17.297136: rig_register called
2021-06-06:08:42:17.297139: rig_register: rig_register (2045)
2021-06-06:08:42:17.297143: rig_register called
2021-06-06:08:42:17.297146: rig_register: rig_register (2047)
2021-06-06:08:42:17.297150: rig_register called
2021-06-06:08:42:17.297155: rig_register: rig_register (2038)
2021-06-06:08:42:17.297158: rig_register called
2021-06-06:08:42:17.297172: rig_register: rig_register (2002)
2021-06-06:08:42:17.297176: rig_register called
2021-06-06:08:42:17.297179: rig_register: rig_register (2011)
2021-06-06:08:42:17.297183: rig_register called
2021-06-06:08:42:17.297187: rig_register: rig_register (2006)
2021-06-06:08:42:17.297190: rig_register called
2021-06-06:08:42:17.297194: rig_register: rig_register (2008)
2021-06-06:08:42:17.297197: rig_register called
2021-06-06:08:42:17.297202: rig_register: rig_register (2015)
[... 459 lines cut ...]
2021-06-06:08:44:36.878391: read_string(): RX 5 characters
2021-06-06:08:44:36.878415: 0000    42 43 20 30 0d                                      BC 0.
'021-06-06:08:44:36.878427: kenwood_transaction: read_string(len=5)='BC 0
2021-06-06:08:44:36.878458: kenwood_transaction: returning RIG_OK, retval=0
2021-06-06:08:44:36.878469: kenwood.c(576):kenwood_transaction return(0)
2021-06-06:08:44:36.878482: rig.c(2752):rig_get_vfo return(0)
2021-06-06:08:44:36.878494: rig.c(1086):rig_open return(0)
Opened rig model 2042, 'TH-D74'
2021-06-06:08:44:36.878543: Backend version: 20210427.1, Status: Beta
2021-06-06:08:44:36.878992: main: Using IPV6
2021-06-06:08:44:36.879024: network.c(431):network_multicast_server entered
2021-06-06:08:44:36.879035: network.c(432):network_multicast_server under development
2021-06-06:08:44:36.879047: network.c(433):ADDR=224.0.1.1, port=4532
2021-06-06:08:44:36.879058: network.c(440) MULTICAST_TRANSCEIVE enabled
2021-06-06:08:44:36.879068: network.c(444) MULTICAST_SPECTRUM enabled
2021-06-06:08:44:36.879155: network.c(458):network_multicast_server return(0)
2021-06-06:08:44:36.879207: network.c(407): Starting multicast server
2021-06-06:08:44:36.879230: network.c(409): Multicast server poll
2021-06-06:08:44:37.880884: network.c(409): Multicast server poll
2021-06-06:08:44:38.881045: network.c(409): Multicast server poll
2021-06-06:08:44:39.884887: network.c(409): Multicast server poll
2021-06-06:08:44:40.889380: network.c(409): Multicast server poll
2021-06-06:08:44:41.890101: network.c(409): Multicast server poll
2021-06-06:08:44:42.890243: network.c(409): Multicast server poll
```

mdblack98, 2021-06-06:
Try a pull on master and test again.I'm making the thd74 call kenwood_open to get the rig ID.
Mike W9MDB

giordyb, 2021-06-06:
Hi Mike,

I did a pull on master but I don't see any new commits.

mdblack98, 2021-06-06:
Oops...forgot to do the push.
Try again..

giordyb, 2021-06-06:
Hi Mike,

don't worry it happens to me all the time....

it's working, here is the log from the latest commit. 

```
2021-06-06:14:57:27.040167: rigctld Hamlib 4.3~git Dom Giu 06 12:57:33 2021 +0000 SHA=fcb252
2021-06-06:14:57:27.040219: Report bugs to <hamlib-developer@lists.sourceforge.net>

2021-06-06:14:57:27.040235: rig_check_rig_caps: p1=0x107d89bd0, p2=0x107d90140, rig_model=0x107d89bd0, macro_name=0x107d90140
2021-06-06:14:57:27.040250: initrigs4_kenwood called
2021-06-06:14:57:27.040255: rig_register called
2021-06-06:14:57:27.040260: rig_register: rig_register (2012)
2021-06-06:14:57:27.040271: rig_register called
2021-06-06:14:57:27.040276: rig_register: rig_register (2013)
2021-06-06:14:57:27.040281: rig_register called
2021-06-06:14:57:27.040286: rig_register: rig_register (2001)
2021-06-06:14:57:27.040291: rig_register called
2021-06-06:14:57:27.040295: rig_register: rig_register (2025)
2021-06-06:14:57:27.040300: rig_register called
2021-06-06:14:57:27.040305: rig_register: rig_register (2003)
2021-06-06:14:57:27.040310: rig_register called
2021-06-06:14:57:27.040314: rig_register: rig_register (2004)
2021-06-06:14:57:27.040319: rig_register called
2021-06-06:14:57:27.040324: rig_register: rig_register (2016)
2021-06-06:14:57:27.040329: rig_register called
2021-06-06:14:57:27.040333: rig_register: rig_register (2024)
2021-06-06:14:57:27.040356: rig_register called
2021-06-06:14:57:27.040362: rig_register: rig_register (2005)
2021-06-06:14:57:27.040367: rig_register called
2021-06-06:14:57:27.040371: rig_register: rig_register (2007)
2021-06-06:14:57:27.040376: rig_register called
2021-06-06:14:57:27.040381: rig_register: rig_register (2009)
2021-06-06:14:57:27.040386: rig_register called
2021-06-06:14:57:27.040390: rig_register: rig_register (2010)
2021-06-06:14:57:27.040395: rig_register called
2021-06-06:14:57:27.040400: rig_register: rig_register (2022)
2021-06-06:14:57:27.040405: rig_register called
2021-06-06:14:57:27.040410: rig_register: rig_register (2014)
2021-06-06:14:57:27.040414: rig_register called
2021-06-06:14:57:27.040419: rig_register: rig_register (2030)
2021-06-06:14:57:27.040424: rig_register called
2021-06-06:14:57:27.040428: rig_register: rig_register (2021)
2021-06-06:14:57:27.040433: rig_register called
2021-06-06:14:57:27.040438: rig_register: rig_register (2029)
2021-06-06:14:57:27.040443: rig_register called
2021-06-06:14:57:27.040448: rig_register: rig_register (2043)
2021-06-06:14:57:27.040452: rig_register called
2021-06-06:14:57:27.040457: rig_register: rig_register (2044)
2021-06-06:14:57:27.040462: rig_register called
2021-06-06:14:57:27.040467: rig_register: rig_register (2045)
2021-06-06:14:57:27.040472: rig_register called
2021-06-06:14:57:27.040477: rig_register: rig_register (2047)
2021-06-06:14:57:27.040482: rig_register called
2021-06-06:14:57:27.040486: rig_register: rig_register (2038)
2021-06-06:14:57:27.040491: rig_register called
2021-06-06:14:57:27.040496: rig_register: rig_register (2002)
2021-06-06:14:57:27.040501: rig_register called
2021-06-06:14:57:27.040506: rig_register: rig_register (2011)
2021-06-06:14:57:27.040511: rig_register called
2021-06-06:14:57:27.040517: rig_register: rig_register (2006)
2021-06-06:14:57:27.040524: rig_register called
2021-06-06:14:57:27.040531: rig_register: rig_register (2008)
2021-06-06:14:57:27.040538: rig_register called
2021-06-06:14:57:27.040546: rig_register: rig_register (2015)
2021-06-06:14:57:27.040554: rig_register called
2021-06-06:14:57:27.040561: rig_register: rig_register (2026)
2021-06-06:14:57:27.040568: rig_register called
2021-06-06:14:57:27.040575: rig_register: rig_register (2017)
[... 114 lines cut ...]
2021-06-06:14:57:27.307526: rig.c(2752):rig_get_vfo return(0)
2021-06-06:14:57:27.307544: rig.c(1086):rig_open return(0)
Opened rig model 2042, 'TH-D74'
2021-06-06:14:57:27.307614: Backend version: 20210427.2, Status: Beta
2021-06-06:14:57:27.308254: main: Using IPV6
2021-06-06:14:57:27.308288: network.c(431):network_multicast_server entered
2021-06-06:14:57:27.308323: network.c(432):network_multicast_server under development
2021-06-06:14:57:27.308340: network.c(433):ADDR=224.0.1.1, port=4532
2021-06-06:14:57:27.308351: network.c(440) MULTICAST_TRANSCEIVE enabled
2021-06-06:14:57:27.308361: network.c(444) MULTICAST_SPECTRUM enabled
2021-06-06:14:57:27.308537: network.c(407): Starting multicast server
2021-06-06:14:57:27.308569: network.c(409): Multicast server poll
2021-06-06:14:57:27.308722: network.c(458):network_multicast_server return(0)
2021-06-06:14:57:28.313715: network.c(409): Multicast server poll
2021-06-06:14:57:29.316203: network.c(409): Multicast server poll
2021-06-06:14:57:30.319199: network.c(409): Multicast server poll
2021-06-06:14:57:31.321077: network.c(409): Multicast server poll
2021-06-06:14:57:32.326234: network.c(409): Multicast server poll
2021-06-06:14:57:33.329365: network.c(409): Multicast server poll
2021-06-06:14:57:34.329441: network.c(409): Multicast server poll
2021-06-06:14:57:35.330320: network.c(409): Multicast server poll
2021-06-06:14:57:36.330632: network.c(409): Multicast server poll
2021-06-06:14:57:37.331908: network.c(409): Multicast server poll
2021-06-06:14:57:38.337082: network.c(409): Multicast server poll
```

closed 2021-06-07 by mdblack98