Hamlib/Hamlib #730: f command return unexpected freq. at IC-9700

jh4xsy, 2021-06-27:
I use IC-9700 and Hamlib 4.2.
In non-SAT mode, I could write the freq. to Sub VFO by using F command.  But f command returned the freq. of Main VFO.

```
% rigctl --version
rigctl Hamlib 4.2 Mon May 17 02:40:24 2021 +0000 SHA=cdad07
% rigctl -r /dev/ic9700a -s 19200 -m 3081

Rig command: V MainA

Rig command: f
Frequency: 433300000

Rig command: V SubA

Rig command: F 145900000

Rig command: f
Frequency: 433300000

mdblack98, 2021-06-27:
Grab the daily snapshot or clone the master repo -- that should be fixed already.
Daily
http://n0nb.users.sourceforge.net/

Clonegit clone https://github.com/Hamlib/Hamlib.git
Mike W9MDB

jh4xsy, 2021-06-27:
Hi Mike.

I got the daily snapshot, but same result. Hamlib 4.1 works fine.

```
% rigctl --version
rigctl Hamlib 4.3~git Fri Jun 25 12:31:15 2021 +0000 SHA=1f91bd
% rigctl -r /dev/ic9700a -s 19200 -m 3081

Rig command: V MainA

Rig command: f
Frequency: 145962710

Rig command: V SubA

Rig command: F 433000000

Rig command: f
Frequency: 145962710

mdblack98, 2021-06-27:
Could you please repeat that sequence with this

rigctl -r /dev/ic9700a -s 19200 -m 3081 -vvvvv V MainA f 145962710 V SubA F 433000000 f >log.txt 2>&1
And send me the log.txtMike W9MDB

jh4xsy, 2021-06-27:
This is my log file. 

[log.txt](https://github.com/Hamlib/Hamlib/files/6721102/log.txt)

At rig_get_freq, curr_vfo is MainA.

commit d5231ec0 (2021-07-13): Use loop to ensure RS-HFIQ hears first command. Closes #730
closed 2021-07-13 by mdblack98