Hamlib/Hamlib #747: Communication with Green Heron RT-21 Az/EL model

mitchem890, 2021-07-23:
Hello, we are trying to use rotctld with the RT-21 Az/El. This model communicates over 2 USB ports (one for Az and one for El). 
We are able to control each of the rotors independently with separate calls to rotctld

Azimuth 
`rotctld -m 405 -r /dev/ttyUSB0 -s 4800 -vvvvv
`
or
Elevation
`rotctld -m 405 -r /dev/ttyUSB1 -s 4800 -vvvvv
`
Then when using the prompt we are able to set the position of the rotors independently.

Are there future plans to implement control of dual USB rotor controllers? 
Or are there any known work-arounds for this issue?

Thanks

mdblack98, 2021-07-23:
What's the maximum elevation?
There's no workaround but I can make a change to allow both serial ports to be opened...perhaps something like this:
rotctld -m 405 -r /dev/ttyUSB0 -R /dev/ttyUSB1 -s 4800 -vvvvv

 The -R would be for the elevation controller.

mitchem890, 2021-07-23:
Awesome, thanks for the quick response,

The max elevation is 90 degrees on the RT-21 Az/El

closed 2021-08-29 by mdblack98