Hamlib/Hamlib #757: SEGV, handle_socket: fdopen(0x6) in: Bad file descriptor

cjritola, 2021-08-05:
rigctld (latest git) throws SEGV shortly after:
handle_socket: fdopen(0x6) in: Bad file descriptor
Making VFO, temperature, and power level queries from multiple processes on localhost and over LAN at a rate of 2/sec each, so it is getting hit by about 6 requests/sec. About 10-20 minutes rigctld throws SEGV and systemd restarts it. 

Unexpected error while long-testing my code before pull request. Could this be a race condition? Socket contention? Duplicate of #556 ? My own code at fault? Here's a snippet  :
```

Aug 05 01:56:12 host rigctld[29461]: RIG_LEVEL_TEMP_METER command=*T
Aug 05 01:56:12 host rigctld[29461]: write_block(): TX 3 bytes
Aug 05 01:56:12 host rigctld[29461]: 0000    2a 54 0d                                            *T.
Aug 05 01:56:12 host rigctld[29461]: read_string called, rxmax=9
Aug 05 01:56:12 host rigctld[29461]: read_string(): RX 7 characters
Aug 05 01:56:12 host rigctld[29461]: 0000    33 33 2e 35 33 43 0d                                33.53C.
Aug 05 01:56:12 host rigctld[29461]: RIG_LEVEL_TEMP_METER reply=33.53C
Aug 05 01:56:12 host rigctld[29461]: RIG_LEVEL_TEMP_METER val=33
Aug 05 01:56:12 host rigctld[29461]: rigctl_parse.c(3246):rigctl_get_level return(0)
Aug 05 01:56:12 host rigctld[29461]: sync_callback: client lock disengaged
Aug 05 01:56:12 host rigctld[29461]: rigctl_parse.c(1743):rigctl_parse return(0)
Aug 05 01:56:12 host rigctld[29461]: handle_socket: doing rigctl_parse vfo_mode=0
Aug 05 01:56:12 host rigctld[29461]: rigctl_parse.c(648):rigctl_parse entered
Aug 05 01:56:12 host rigctld[29461]: rigctl_parse: called, interactive=1
Aug 05 01:56:12 host rigctld[29461]: rigctl_parse: cmd=q(71)
Aug 05 01:56:12 host rigctld[29461]: rigctl_parse: quit returning NETRIGCTL_RET 0
Aug 05 01:56:12 host rigctld[29461]: rigctl_parse.c(798):rigctl_parse return(1)
Aug 05 01:56:12 host rigctld[29461]: handle_socket: rigctl_parse retcode=1
Aug 05 01:56:12 host rigctld[29461]: Connection closed from [lan address]:43266
Aug 05 01:56:12 host rigctld[29461]: Connection opened from [lan address]:43268
Aug 05 01:56:12 host rigctld[29461]: rigctld.c(1062):handle_socket entered
Aug 05 01:56:12 host rigctld[29461]: handle_socket: fdopen(0x6) in: Bad file descriptor
Aug 05 01:56:12 host systemd[1]: rigctld.service: Main process exited, code=killed, status=11/SEGV
Aug 05 01:56:12 host systemd[1]: rigctld.service: Failed with result 'signal'.
Aug 05 01:56:17 host systemd[1]: rigctld.service: Service RestartSec=5s expired, scheduling restart.
Aug 05 01:56:17 host systemd[1]: rigctld.service: Scheduled restart job, restart counter is at 1.
Aug 05 01:56:17 host systemd[1]: Stopped rigctl daemon.
Aug 05 01:56:17 host systemd[1]: Started rigctl daemon.
Aug 05 01:56:17 host rigctld[31805]: rigctld.c(272) Startup: /usr/local/bin/rigctld -vvvvvv -m 25019 -r /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
Aug 05 01:56:17 host rigctld[31805]: rigctld Hamlib 4.3~git Wed Aug 04 12:46:17 2021 +0000 SHA=407ba1
Aug 05 01:56:17 host rigctld[31805]: Report bugs to <hamlib-developer@lists.sourceforge.net>
Aug 05 01:56:17 host rigctld[31805]: rig_check_rig_caps: p1=0x76f12ea8, p2=0x76f185ec, rig_model=0x76f12ea8, macro_name=0x76f185ec
Aug 05 01:56:17 host rigctld[31805]: initrigs4_kit: _init called
```

mdblack98, 2021-08-05:
What version of hamlib?  
Can you show a normal successful command's behavior?  
I just did a commit that should fix that segfault hopefully but wondering what a normal sequence looks like with your code.
Are you opening/closing a connection for each command?

cjritola, 2021-08-05:
Yes, some commands are opening/closing a connection multiple times per second.

Local machine using rigctld and a rigctl client:
```
rigctld --version
rigctld.c(272) Startup: rigctld --version
rigctl Hamlib 4.3~git Wed Aug 04 12:46:17 2021 +0000 SHA=407ba1
```

Remote machine using rigctl only, from 'watch' command:
```
rigctl --version
rigctl.c(158) Startup: rigctl --version
rigctl Hamlib 4.3~git Wed Aug 04 14:48:07 2021 +0000 SHA=a44c3d
```

One remote client (quisk) is using a single consistent connection, one is a local python script which uses a single consistent connection to poll the VFO to tune a loop, and two commands running on remote machines:
```
watch -n .5 rigctl -m 2 -r [lan ip] get_level RFPOWER_METER
watch -n .5 rigctl -m 2 -r [lan ip] get_level TEMP_METER
```

Here's a normal cycle from the command for TEMP_METER:
```
Aug 05 17:56:15 host rigctld[6120]: Connection opened from [ LAN IP address ]:59462
Aug 05 17:56:15 host rigctld[6120]: rigctld.c(1062):handle_socket entered
Aug 05 17:56:15 host rigctld[6120]: sync_callback: client lock engaged
Aug 05 17:56:15 host rigctld[6120]: sync_callback: client lock disengaged
Aug 05 17:56:15 host rigctld[6120]: handle_socket: doing rigctl_parse vfo_mode=0
Aug 05 17:56:15 host rigctld[6120]: rigctl_parse.c(648):rigctl_parse entered
Aug 05 17:56:15 host rigctld[6120]: rigctl_parse: called, interactive=1
Aug 05 17:56:15 host rigctld[6120]: rigctl_parse: cmd=\(5c)
Aug 05 17:56:15 host rigctld[6120]: rigctl_parse: cmd=chk_vfo
Aug 05 17:56:15 host rigctld[6120]: sync_callback: client lock engaged
Aug 05 17:56:15 host rigctld[6120]: [31B blob data]
Aug 05 17:56:15 host rigctld[6120]: rigctl_parse: vfo_opt=0
Aug 05 17:56:15 host rigctld[6120]: rigctl_parse.c(5018):rigctl_chk_vfo entered
Aug 05 17:56:15 host rigctld[6120]: rigctl_parse.c(5029):rigctl_chk_vfo return(0)
Aug 05 17:56:15 host rigctld[6120]: sync_callback: client lock disengaged
Aug 05 17:56:15 host rigctld[6120]: rigctl_parse.c(1743):rigctl_parse return(0)
Aug 05 17:56:15 host rigctld[6120]: handle_socket: doing rigctl_parse vfo_mode=0
Aug 05 17:56:15 host rigctld[6120]: rigctl_parse.c(648):rigctl_parse entered
Aug 05 17:56:15 host rigctld[6120]: rigctl_parse: called, interactive=1
Aug 05 17:56:15 host rigctld[6120]: rigctl_parse: cmd=\(5c)
Aug 05 17:56:15 host rigctld[6120]: rigctl_parse: cmd=dump_state
Aug 05 17:56:15 host rigctld[6120]: sync_callback: client lock engaged
Aug 05 17:56:15 host rigctld[6120]: [31B blob data]
Aug 05 17:56:15 host rigctld[6120]: rigctl_parse: vfo_opt=0
Aug 05 17:56:15 host rigctld[6120]: rigctl_parse.c(4401):rigctl_dump_state entered
Aug 05 17:56:15 host rigctld[6120]: rigctl_parse.c(4560):rigctl_dump_state return(0)
Aug 05 17:56:15 host rigctld[6120]: sync_callback: client lock disengaged
Aug 05 17:56:15 host rigctld[6120]: rigctl_parse.c(1743):rigctl_parse return(0)
Aug 05 17:56:15 host rigctld[6120]: handle_socket: doing rigctl_parse vfo_mode=0
Aug 05 17:56:15 host rigctld[6120]: rigctl_parse.c(648):rigctl_parse entered
Aug 05 17:56:15 host rigctld[6120]: rigctl_parse: called, interactive=1
Aug 05 17:56:15 host rigctld[6120]: rigctl_parse: cmd=l(6c)
Aug 05 17:56:15 host rigctld[6120]: sync_callback: client lock engaged
Aug 05 17:56:15 host rigctld[6120]: rigctl(d): l 'currVFO' 'TEMP_METER' '' ''
Aug 05 17:56:15 host rigctld[6120]: rigctl_parse: vfo_opt=0
Aug 05 17:56:15 host rigctld[6120]: rigctl_parse.c(3155):rigctl_get_level entered
Aug 05 17:56:15 host rigctld[6120]: rig_parse_level called
Aug 05 17:56:15 host rigctld[6120]: rshfiq_get_level called. level type =281474976710656
Aug 05 17:56:15 host rigctld[6120]: rig_flush: called for serial device
Aug 05 17:56:15 host rigctld[6120]: serial.c(629):serial_flush entered
Aug 05 17:56:15 host rigctld[6120]: tcflush
Aug 05 17:56:15 host rigctld[6120]: serial.c(661):serial_flush return(0)
Aug 05 17:56:15 host rigctld[6120]: RIG_LEVEL_TEMP_METER command=*T
Aug 05 17:56:15 host rigctld[6120]: write_block(): TX 3 bytes
Aug 05 17:56:15 host rigctld[6120]: 0000    2a 54 0d                                            *T.
Aug 05 17:56:15 host rigctld[6120]: read_string called, rxmax=9
Aug 05 17:56:15 host rigctld[6120]: read_string(): RX 7 characters
Aug 05 17:56:15 host rigctld[6120]: 0000    32 37 2e 36 39 43 0d                                27.69C.
Aug 05 17:56:15 host rigctld[6120]: RIG_LEVEL_TEMP_METER reply=27.69C
Aug 05 17:56:15 host rigctld[6120]: RIG_LEVEL_TEMP_METER val=27
Aug 05 17:56:15 host rigctld[6120]: rigctl_parse.c(3246):rigctl_get_level return(0)
Aug 05 17:56:15 host rigctld[6120]: sync_callback: client lock disengaged
Aug 05 17:56:15 host rigctld[6120]: rigctl_parse.c(1743):rigctl_parse return(0)
Aug 05 17:56:15 host rigctld[6120]: handle_socket: doing rigctl_parse vfo_mode=0
Aug 05 17:56:15 host rigctld[6120]: rigctl_parse.c(648):rigctl_parse entered
Aug 05 17:56:15 host rigctld[6120]: rigctl_parse: called, interactive=1
Aug 05 17:56:15 host rigctld[6120]: rigctl_parse: cmd=q(71)
Aug 05 17:56:15 host rigctld[6120]: rigctl_parse: quit returning NETRIGCTL_RET 0
Aug 05 17:56:15 host rigctld[6120]: rigctl_parse.c(798):rigctl_parse return(1)
Aug 05 17:56:15 host rigctld[6120]: handle_socket: rigctl_parse retcode=1
Aug 05 17:56:15 host rigctld[6120]: Connection closed from [ LAN IP address ]:59462
```

cjritola, 2021-08-06:
5cab1ac fixes this

closed 2021-08-06 by cjritola

mdblack98, 2021-08-06:
As your program is runnning checknetstat -t|grep 4532 | wc -l
You'll get a count of 2 for every active connection.  I do see rigctl is leaving the socket in TIME_WAIT status which may be the problem if you're really banging on new socket connections.
I wonder if somebody isn't closing the socket politely and you're running out of ephemeral sockets.
Mike

cjritola, 2021-08-07:
When closing quisk or my python script, the connection disappears immediately from netstat, yet
with rigctl it lingers with TIME_WAIT for about a minute.

```
cat /proc/sys/net/ipv4/tcp_fin_timeout
60
```

The python script uses device 2 (network rig control) like with the test rigctl commands.

When running:
`watch -n .5 rigctl -m 2 -r [rigctld LAN ip address] get_level RFPOWER_METER`
The command watch "netstat -t | grep 4532 | wc -l" gives a number which gradually escalates and stabilizes to about 120.
When the 'watch rigctl' command above is terminated, the 'watch netstat' command's output number slowly dwindles to 0 within about a minute. 

On the rigctl side, no connections hanging.

Output from rigctl command side:
```
rigctl -vvvvvv -m 2 -r [rigctld lan IP] get_level TEMP_METER
rigctl.c(158) Startup: rigctl -vvvvvv -m 2 -r [rigctld LAN ip address] get_level TEMP_METER
rigctl Hamlib 4.3~git Thu Aug 05 13:38:58 2021 +0000 SHA=e57774
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_check_rig_caps: p1=0x7f850e861b20, p2=0x7f850e868090, rig_model=0x7f850e861b20, macro_name=0x7f850e868090
dummy.c(2568):initrigs4_dummy entered
initrigs4_dummy: _init called
rig_register called
rig_register: rig_register (1)
rig_register called
rig_register: rig_register (2)
rig_register called
rig_register: rig_register (4)
rig_register called
rig_register: rig_register (5)
rig_register called
rig_register: rig_register (6)
dummy.c(2577):initrigs4_dummy return(0)
rig_init: rx_range_list1 is empty, using rx_range_list2
rig_init: rig does not have tx_range!!
netrigctl_init version 20210712.0
rigctl_parse.c(1916):set_conf entered
rigctl_parse.c(1950):set_conf return(0)
rig.c(682):rig_open entered
rig_open: using network address [rigctld lan IP]
iofunc.c(68):port_open entered
network.c(171):network_open entered
network_open: hoststr=[rigctl lan IP], portstr=
network_open: client port=9403
network.c(308):network_open return(0)
iofunc.c(185):port_open return(0)
rig.c(220):add_opened_rig return(0)
netrigctl_open called
netrigctl_transaction: called len=9
rig_flush: called for network device
network_flush called
write_block(): TX 9 bytes
0000    5c 63 68 6b 5f 76 66 6f 0a                          \chk_vfo.       
read_string called, rxmax=1024
read_string(): RX 2 characters
0000    30 0a                                               0.              
netrigctl_open: vfo_mode=0
netrigctl_transaction: called len=12
rig_flush: called for network device
network_flush called
write_block(): TX 12 bytes
0000    5c 64 75 6d 70 5f 73 74 61 74 65 0a                 \dump_state.    
read_string called, rxmax=1024
read_string(): RX 2 characters
0000    31 0a                                               1.              
[... 568 lines cut ...]
Aug 07 01:08:28 rigctldhost rigctld[18084]: rshfiq_get_level called. level type =281474976710656
Aug 07 01:08:28 rigctldhost rigctld[18084]: rig_flush: called for serial device
Aug 07 01:08:28 rigctldhost rigctld[18084]: serial.c(629):serial_flush entered
Aug 07 01:08:28 rigctldhost rigctld[18084]: tcflush
Aug 07 01:08:28 rigctldhost rigctld[18084]: serial.c(661):serial_flush return(0)
Aug 07 01:08:28 rigctldhost rigctld[18084]: RIG_LEVEL_TEMP_METER command=*T
Aug 07 01:08:28 rigctldhost rigctld[18084]: write_block(): TX 3 bytes
Aug 07 01:08:28 rigctldhost rigctld[18084]: 0000    2a 54 0d                                            *T.
Aug 07 01:08:28 rigctldhost rigctld[18084]: read_string called, rxmax=9
Aug 07 01:08:28 rigctldhost rigctld[18084]: read_string(): RX 7 characters
Aug 07 01:08:28 rigctldhost rigctld[18084]: 0000    32 31 2e 37 35 43 0d                                21.75C.
Aug 07 01:08:28 rigctldhost rigctld[18084]: RIG_LEVEL_TEMP_METER reply=21.75C
Aug 07 01:08:28 rigctldhost rigctld[18084]: RIG_LEVEL_TEMP_METER val=21
Aug 07 01:08:28 rigctldhost rigctld[18084]: rigctl_parse.c(3246):rigctl_get_level return(0)
Aug 07 01:08:28 rigctldhost rigctld[18084]: sync_callback: client lock disengaged
Aug 07 01:08:28 rigctldhost rigctld[18084]: rigctl_parse.c(1743):rigctl_parse return(0)
Aug 07 01:08:28 rigctldhost rigctld[18084]: handle_socket: doing rigctl_parse vfo_mode=0
Aug 07 01:08:28 rigctldhost rigctld[18084]: rigctl_parse.c(648):rigctl_parse entered
Aug 07 01:08:28 rigctldhost rigctld[18084]: rigctl_parse: called, interactive=1
Aug 07 01:08:28 rigctldhost rigctld[18084]: rigctl_parse: cmd=q(71)
Aug 07 01:08:28 rigctldhost rigctld[18084]: rigctl_parse: quit returning NETRIGCTL_RET 0
Aug 07 01:08:28 rigctldhost rigctld[18084]: rigctl_parse.c(798):rigctl_parse return(1)
Aug 07 01:08:28 rigctldhost rigctld[18084]: handle_socket: rigctl_parse retcode=1
Aug 07 01:08:28 rigctldhost rigctld[18084]: Connection closed from localhost:4114
```

mdblack98, 2021-08-07:
Would you please send me your python script?
Mike W9MDB

cjritola, 2021-08-08:
```
import time
import Hamlib
import RPi.GPIO as GPIO
import signal
import sys

loop_gpio_pin=20
GPIO.setwarnings(False)
GPIO.setmode(GPIO.BCM)
GPIO.setup(loop_gpio_pin,GPIO.OUT)

Hamlib.rig_set_debug(Hamlib.RIG_DEBUG_NONE)
rig = Hamlib.Rig(2)
rig.set_conf("rig_pathname", "localhost")
rig.open()

running = True

def shutdown_handler(sig, frame):
            global running
            print ('Shutdown signal caught. Exiting...\n')
            running = False

signal.signal(signal.SIGTERM, shutdown_handler)
signal.signal(signal.SIGINT, shutdown_handler)

while (running):
            frequency = (rig.get_freq()/1.0e6)

            if( (frequency > 3.5 and frequency < 5) or (frequency > 10 and frequency < 14) or (frequen$
                #print ("gap")
                GPIO.output(loop_gpio_pin,1)
            else:
                #print ("no gap")
                GPIO.output(loop_gpio_pin,0)
            time.sleep(.5)

print ("Escaped main loop.\n")
rig.close()
sys.exit (0)
```