Hamlib/Hamlib #761: Problem when "make"

dl7ata, 2021-08-09:
I get the following error after  ./bootstrap and  ./configure when executing make:
`  CC       adat.lo
In file included from ../../src/register.h:28,
                 from adat.c:48:
../../include/config.h:514:8: error: redefinition of ‘struct addrinfo’
 struct addrinfo {
        ^~~~~~~~
In file included from adat.c:25:
../../include/config.h:514:8: note: originally defined here
 struct addrinfo {
        ^~~~~~~~
make[1]: *** [Makefile:483: adat.lo] Fehler 1
make[1]: Verzeichnis „/home/svxlink/Programme/Hamlib/rigs/adat“ wird verlassen
make: *** [Makefile:610: all-recursive] Fehler 1
`

mdblack98, 2021-08-10:
What operating system?
Is this build from the current master?

dl7ata, 2021-08-11:
Sorry, I missed the standards:
Raspi 3B+, Buster-lite, nothing special on the machine.

closed 2021-09-11 by mdblack98