Hamlib/Hamlib #766: grig shows empty dropdowns for ATT and PREAMP w/ FUNCube Dongle ProPlus (rigs/kit/funcube.c)

cjritola, 2021-08-15:
grig cannot control LNA or mixer gain due to empty dropdowns for ATT and PREAMP w/ FUNCube Dongle ProPlus (funcube.c)
Caused by funcubeplus_caps having no attenuator and preamp fields defined.

Currently, the LNA's on/off is mapped to PREAMP as 0 or 1 and the mixer gain's on/off is mapped to ATT as 0 or 1. The FCDPP's mixer gain is pretty consistently 20dB when enabled. The LNA's gain varies between 15dB to 5dB up the tuning range. There are a few ways I could go about making the FCDPP LNA and Mixer controllable over grig:

* Add funcubeplus_caps.preamp[] with a value of 1 in its array and funcubeplus_caps.attenuator[] with a value of 1 in its array for the mixer gain. This is the most backwards compatible to the original behavior, but grig shows the PREAMP option as 1dB and ATT option as -1dB.
* Add funcubeplus_caps.preamp[] with a value of 10 in its array to approximate the gain of the LNA and  funcubeplus_caps.attenuator[] with a value of 20 for the mixer gain, then change RIG_LEVEL_ATT switch case to disable Mixer Gain when set to 20, else enable Mixer Gain when set to 0. This would not be backwards compatible to the previous behavior.
* Merge both Mixer Gain and LNA as values in PREAMP by listing all permutations of the LNA and PREAMP on/off states such that:  0dB = (LNAoff, MixGainOff), 10dB = (LNAon, MixGainOff), 20dB= (LNAoff, MixGainOn), 30dB = (LNAon, MixGainOn), then discard the ATT level capability. This is not backwards compatible, but is logically the closest to actual behavior without losing any control.
* Leave it alone.

Which would you prefer?

mdblack98, 2021-08-16:
I show the mappping to be this for the funcube.
Can you point me to the documenation?

    .preamp =       { 5, 10, 15, 20, 25, 30, RIG_DBLST_END, },
    .attenuator =       { 2, 5, RIG_DBLST_END, },

mdblack98, 2021-08-16:
Oops...that was for just the funcube...not the plus.  I see where preamp and attenuator are commented out for the plus.

Would make the most sense to just map the preamp values as you mention in the 3rd option.

cjritola, 2021-08-17:
ok, will do

commit 118dcbb3 (2021-08-30): Merge pull request #779 from cjritola/fcdppAug2021

Map FCDPP LNA and Mixer Gain as PREAMP values. Closes #766
commit dc47740c (2021-08-30): Map FCDPP LNA and Mixer Gain as PREAMP values. Closes #766
closed 2021-08-30 by mdblack98