Hamlib/Hamlib #773: rigctl Option prompt displaying when not needed

mdblack98, 2021-08-22:
Rig command: Y 1 0
Option:

closed 2021-08-29 by mdblack98

mdblack98, 2021-09-13:
> Looks like all of these changes already exist in 4.3.

I may have accidentally put some old fixes in the 4.3.1 list.  For some reason I had them tagged for 4.4.
Mike

N0NB, 2021-09-13:
* On 2021 13 Sep 04:42 -0500, Michael Black wrote:
> I may have accidentally put some old fixes in the 4.3.1 list.  For
> some reason I had them tagged for 4.4.

That's fine.  Git lets me know when there is an issue and I'll just note
in each issue if any were applied.

73, Nate

-- 
"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."
Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819