Hamlib/Hamlib #778: Regression from recent commit (set_level: Feature not available)

cjritola, 2021-08-29:
```
rigctl -m 1 set_level PREAMP 0
set_level: error = Feature not available
read_string called, rxmax=1024
read_string(): RX 9 characters
rigctl_parse.c(3150):rigctl_set_level return(-11)
```

... issue arises in commit 
d16fa490d7aa7bfe1bb520fada04db3a3ad88cc1

Commenting parts of block starting at line 1000 in tests/rigctl_parse.c fixes it:
```
if (interactive)
            {
                rig_debug(RIG_DEBUG_TRACE, "%s: debug7\n", __func__);

                //arg1[0] = fgetc(fin);
                //arg1[1] = 0;

                if (prompt /* && arg1[0]  == 0x0a */)
                {
```
I was pulling my hair out trying to get my new code to work when I realized rig_caps.set_level wasn't even getting called. XD

Looks like the parser was passing an empty string to rig_parse_level()  in declare_proto_rig(set_level). I don't understand what is happening beyond that.

mdblack98, 2021-08-29:
I can't reproduce this problem.  Works fine here under Linux and Windows.
How are you getting error messages without using the verbose switches?
rigctl -vvv -m 1 set_level PREAMP 0
Opened rig model 1, 'Dummy'
Mike W9MDB

cjritola, 2021-08-29:
How about this:
```
me@host:~/Hamlib$ rigctld -m 1 &
[1] 20946
me@host:~/Hamlib$ rigctl -m 2 -r localhost set_level PREAMP 1
set_level: error = Feature not available
read_string called, rxmax=1024
read_string(): RX 9 characters
rigctl_parse.c(3150):rigctl_set_level return(-11)
```

I think I got confused 7 hrs in.

mdblack98, 2021-08-29:
Try again please...think I fixed it.
Mike W9MDB

 
 

How about this:
[1] 20946

closed 2021-08-29 by mdblack98

cjritola, 2021-08-30:
That fixed it, thanks