Hamlib/Hamlib #782: IC-910 sat mode turned on when split selected in WSJTX

mdblack98, 2021-09-01:
Bill Somerville <g4wjs@classdesign.com>
To:
hamlib-developer@lists.sourceforge.net

Wed, Sep 1 at 11:49 AM

Hi all,

subject line says what happens, here's a trace showing the unexpected 1a 07 01 command:

warning: [16:25:57.135286][RIGCTRL][debug] () : rig.c(4533):rig_set_split_vfo entered
warning: [16:25:57.135286][RIGCTRL][debug] () : rig_set_split_vfo: rx_vfo=currVFO, split=1, tx_vfo=VFOB
warning: [16:25:57.136038][RIGCTRL][trace] () : vfo_fixup: vfo=currVFO, vfo_curr=Main
warning: [16:25:57.136038][RIGCTRL][trace] () : vfo_fixup: Leaving currVFO alone
warning: [16:25:57.136038][RIGCTRL][trace] () : vfo_fixup: vfo=VFOB, vfo_curr=Main
warning: [16:25:57.136787][RIGCTRL][trace] () : vfo_fixup: final vfo=Sub
warning: [16:25:57.136787][RIGCTRL][trace] () : C:/Users/bill/src/hamlib/src/rig.c(4553) trace
warning: [16:25:57.136787][RIGCTRL][debug] () : rig.c(2616):rig_set_vfo entered
warning: [16:25:57.137537][RIGCTRL][debug] () : rig_set_vfo called vfo=VFOA
warning: [16:25:57.137537][RIGCTRL][trace] () : vfo_fixup: vfo=VFOA, vfo_curr=Main
warning: [16:25:57.137537][RIGCTRL][trace] () : vfo_fixup: vfo=Main, vfo_curr=Main
warning: [16:25:57.138288][RIGCTRL][trace] () : C:/Users/bill/src/hamlib/src/rig.c(2675) trace
warning: [16:25:57.138288][RIGCTRL][debug] () : icom_set_vfo called vfo=Main
warning: [16:25:57.138288][RIGCTRL][trace] () : icom_set_vfo: vfo line#2421 vfo=Main, split=0
warning: [16:25:57.139039][RIGCTRL][trace] () : icom_set_vfo: line#2473
warning: [16:25:57.139039][RIGCTRL][trace] () : icom_set_vfo: line#2630
warning: [16:25:57.139039][RIGCTRL][debug] () : frame.c(402):icom_transaction entered
warning: [16:25:57.139789][RIGCTRL][debug] () : icom_transaction: cmd=0x07, subcmd=0xd0, payload_len=0
warning: [16:25:57.139789][RIGCTRL][debug] () : frame.c(139):icom_one_transaction entered
warning: [16:25:57.140539][RIGCTRL][debug] () : frame.c(54):make_cmd_frame entered
warning: [16:25:57.312394][RIGCTRL][debug] () : frame.c(88):make_cmd_frame return(7)
warning: [16:25:57.312394][RIGCTRL][trace] () : rig_flush: called for serial device
warning: [16:25:57.313141][RIGCTRL][debug] () : serial.c(642):serial_flush entered
warning: [16:25:57.313141][RIGCTRL][debug] () : tcflush
warning: [16:25:57.313893][RIGCTRL][debug] () : serial.c(674):serial_flush return(0)
warning: [16:25:57.313893][RIGCTRL][trace] () : write_block(): TX 7 bytes
warning: [16:25:57.313893][RIGCTRL][trace] () : 0000    fe fe 60 e0 07 d0 fd                                ..`....
warning: [16:25:57.314642][RIGCTRL][debug] () : frame.c(452):read_icom_frame entered
warning: [16:25:57.314642][RIGCTRL][trace] () : read_string called, rxmax=80
warning: [16:25:57.755187][RIGCTRL][trace] () : read_string(): RX 7 characters
warning: [16:25:57.755187][RIGCTRL][trace] () : 0000    fe fe 60 e0 07 d0 fd                                ..`....
warning: [16:25:57.755938][RIGCTRL][debug] () : frame.c(486):read_icom_frame return(7)
warning: [16:25:57.755938][RIGCTRL][debug] () : frame.c(452):read_icom_frame entered
warning: [16:25:57.756688][RIGCTRL][trace] () : read_string called, rxmax=80
warning: [16:25:57.756688][RIGCTRL][trace] () : read_string(): RX 6 characters
warning: [16:25:57.757438][RIGCTRL][trace] () : 0000    fe fe e0 60 fb fd                                   ...`..
warning: [16:25:57.757438][RIGCTRL][debug] () : frame.c(486):read_icom_frame return(6)
warning: [16:25:57.757438][RIGCTRL][trace] () : icom_one_transaction: frm_len=6, frm_len-1=fd, frm_len-2=fb
warning: [16:25:57.758189][RIGCTRL][debug] () : frame.c(380):icom_one_transaction return(0)
warning: [16:25:57.758189][RIGCTRL][debug] () : frame.c(432):icom_transaction return(0)
warning: [16:25:57.758940][RIGCTRL][trace] () : icom_set_vfo: line#2633
warning: [16:25:57.758940][RIGCTRL][trace] () : icom_set_vfo: line#2656 curr_vfo=Main
warning: [16:25:57.758940][RIGCTRL][debug] () : icom.c(2657):icom_set_vfo return(0)
warning: [16:25:57.759690][RIGCTRL][trace] () : rig_set_vfo: rig->state.current_vfo=Main
warning: [16:25:57.759690][RIGCTRL][trace] () : C:/Users/bill/src/hamlib/src/rig.c(2701) trace
warning: [16:25:57.759690][RIGCTRL][debug] () : icom_get_freq called for Main, curr_vfo=Main
warning: [16:25:57.759690][RIGCTRL][debug] () : icom_get_freq: using vfo=Main
warning: [16:25:57.760440][RIGCTRL][trace] () : C:/Users/bill/src/hamlib/rigs/icom/icom.c(1359) trace
warning: [16:25:57.760440][RIGCTRL][trace] () : set_vfo_curr: vfo=Main, curr_vfo=Main
warning: [16:25:57.760440][RIGCTRL][trace] () : set_vfo_curr: curr_vfo now=Main
warning: [16:25:57.760440][RIGCTRL][debug] () : icom.c(8846):set_vfo_curr return(0)
warning: [16:25:57.761191][RIGCTRL][debug] () : frame.c(402):icom_transaction entered
warning: [16:25:57.761191][RIGCTRL][debug] () : icom_transaction: cmd=0x03, subcmd=0xffffffff, payload_len=0
warning: [16:25:57.761191][RIGCTRL][debug] () : frame.c(139):icom_one_transaction entered
warning: [16:25:57.761942][RIGCTRL][debug] () : frame.c(54):make_cmd_frame entered
warning: [16:25:57.761942][RIGCTRL][debug] () : frame.c(88):make_cmd_frame return(6)
warning: [16:25:57.761942][RIGCTRL][trace] () : rig_flush: called for serial device
warning: [16:25:57.761942][RIGCTRL][debug] () : serial.c(642):serial_flush entered
warning: [16:25:57.762693][RIGCTRL][debug] () : tcflush
warning: [16:25:57.762693][RIGCTRL][debug] () : serial.c(674):serial_flush return(0)
[... 78 lines cut ...]
warning: [16:25:58.035100][RIGCTRL][trace] () : read_string(): RX 7 characters
warning: [16:25:58.035100][RIGCTRL][trace] () : 0000    fe fe 60 e0 0f 01 fd                                ..`....
warning: [16:25:58.035848][RIGCTRL][debug] () : frame.c(486):read_icom_frame return(7)
warning: [16:25:58.035848][RIGCTRL][debug] () : frame.c(452):read_icom_frame entered
warning: [16:25:58.036599][RIGCTRL][trace] () : read_string called, rxmax=80
warning: [16:25:58.047106][RIGCTRL][trace] () : read_string(): RX 6 characters
warning: [16:25:58.047106][RIGCTRL][trace] () : 0000    fe fe e0 60 fa fd                                   ...`..
warning: [16:25:58.047857][RIGCTRL][debug] () : frame.c(486):read_icom_frame return(6)
warning: [16:25:58.047857][RIGCTRL][debug] () : frame.c(330):icom_one_transaction return(-9)
warning: [16:25:58.048609][RIGCTRL][debug] () : icom_transaction: failed: Command rejected by the rig
frame.c(486):read_icom_frame return(6)
frame.c(330):icom_one_transaction return(-9)
icom_transaction: failed: Command rejected by the rig
frame.c(486):read_icom_frame return(6)
frame.c(330):icom_one_transaction return(-9)
frame.c(330):icom_one_transaction return(-9)warning: [16:25:58.049359][RIGCTRL][debug] () : frame.c(432):icom_transaction return(-9)
warning: [16:25:58.049359][RIGCTRL][debug] () : icom.c(6097):icom_set_split_vfo return(-9)
warning: [16:25:58.049359][RIGCTRL][debug] () : rig.c(4578):rig_set_split_vfo return(-9)
73
Bill
G4WJS.
_______________________________________________
Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

closed 2021-09-11 by mdblack98

N0NB, 2021-09-13:
Cherry picked 6de45883 into the Hamlib-4.3 branch.  0e67bcd is apparently an exact duplicate so not cherry-picked.

mdblack98, 2021-09-13:
Yup...removed that one from the list too.  I failed to check the commit dates.