Hamlib/Hamlib #783: Add GPS reading

mdblack98, 2021-09-01:
New command perhaps?

Support NMEA, NOVATELX, and RTCMV3
Should be able to automatically tell the format

AuroraRAS, 2021-09-26:
mavlink2 is a good protocol for hamlib, It's include global position and attitude message.

see also:
[https://mavlink.io/en/messages/common.html#ATTITUDE](url)
[https://mavlink.io/en/messages/common.html#GLOBAL_POSITION_INT](url)

mdblack98, 2023-01-23:
Need to leave open the possibility of using Windows location ability...
Here's the Win32 API: [Location API - Win32 apps | Microsoft Learn ](https://learn.microsoft.com/en-us/windows/win32/locationapi/windows-location-api-portal)

And here's the latest UWP API: Windows.Devices.Geolocation Namespace - Windows UWP applications | Microsoft Learn

mdblack98, 2024-01-06:
Here's a maiden head location routine
#include <stdio.h>
#include <stdlib.h>

#define MAIDENHEAD_LEN 11 // Length including the null terminator

void toMaidenhead(double lat, double lon, char* locator, size_t bufferSize) {
    // Ensure buffer size is sufficient
    if (bufferSize < MAIDENHEAD_LEN) {
        fprintf(stderr, "Error: Buffer size is too small for Maidenhead locator.\n");
        return;
    }

    // Constants for Maidenhead conversion
    const char* letters = "ABCDEFGHIJKLMNOPQRSTUVWX";
    const int lonFieldSize = 20, latFieldSize = 10, lonSquareSize = 2, latSquareSize = 1;
    const double lonSubsquareSize = 5.0 / 60, latSubsquareSize = 2.5 / 60;
    const double lonExtendedSize = 5.0 / 3600, latExtendedSize = 2.5 / 3600;

    // Adjust for negative coordinates
    lon += 180;
    lat += 90;

    // Field
    int lonField = lon / lonFieldSize;
    int latField = lat / latFieldSize;
    locator[0] = letters[lonField];
    locator[1] = letters[latField];

    // Square
    lon -= lonField * lonFieldSize;
    lat -= latField * latFieldSize;
    locator[2] = '0' + (int)(lon / lonSquareSize);
    locator[3] = '0' + (int)(lat / latSquareSize);

    // Subsquare
    lon -= (locator[2] - '0') * lonSquareSize;
    lat -= (locator[3] - '0') * latSquareSize;
    locator[4] = letters[(int)(lon / lonSubsquareSize)];
    locator[5] = letters[(int)(lat / latSubsquareSize)];

    // Extended Subsquare
    lon -= (locator[4] - 'A') * lonSubsquareSize;
    lat -= (locator[5] - 'A') * latSubsquareSize;
    locator[6] = '0' + (int)(lon / lonExtendedSize);
    locator[7] = '0' + (int)(lat / lonExtendedSize);
    lon -= (locator[6] - '0') * lonExtendedSize;
    lat -= (locator[7] - '0') * latExtendedSize;
    locator[8] = '0' + (int)(lon / (lonExtendedSize / 24));
    locator[9] = '0' + (int)(lat / (latExtendedSize / 24));

    locator[10] = '\0'; // Null-terminate the string
}

int main() {
    double latitude, longitude;
    char locator[MAIDENHEAD_LEN];

    printf("Enter latitude: ");
    scanf("%lf", &latitude);
    printf("Enter longitude: ");
    scanf("%lf", &longitude);

    toMaidenhead(latitude, longitude, locator, MAIDENHEAD_LEN);
    printf("Maidenhead Locator: %s\n", locator);

    return 0;
}

hackery, 2024-01-19:
Many devices will probably offer location data without an NMEA message stream etc, and there'll be a need to represent a condition of `unknown` or `no lock`.
e.g. I've been looking at the Icom ID-52E in the last couple of days, the location data can be polled over CI-V (can't see a way to get continuous updates):

```
rigctl -m $id51 -r /dev/ttyACM0 -c 0xA6 -t -1 w '\0xFE\0xFE\0xA6\0xE0\0x23\0x00\0xFD'
\0xFE\0xFE\0xE0\0xA6\0x23\0x00\0xFF\0xFD 8 # no location on screen, FF looks like "not known" / no lock

\0xFE\0xFE\0xE0\0xA6\0x23\0x00\0x51\0x37\0x51\0x90\0x01\0x00\0x01\0x19\0x52\0x00\0x00\0x00\0x06\0x47\0x00\0x02\0x12\0x00\0x00\0x27\0x20\0x24\0x01\0x19\0x08\0x25\0x28\0xFD 34

* lat \0x51\0x37\0xMM\0xMM\0xMM             51d 37.MMMMM N
* lon \0x00\0x01\0x19\0xMM\0xMM\0xMM        1d 19.MMMMM W
* alt \0x00\0x06\0x47\0x00                  64.700m
* crs \0x02\0x12                            212 deg
* spd \0x00\0x00\0x27                       2.7km/h
* tim \0x20\0x24\0x01\0x19\0x08\0x25\0x28   2024-01-19T08:25:28
```
[image]

N0NB, 2025-07-30:
I'm not sure why Mike posted that code as Hamlib already includes Maidenhead calculation routines accessible via `rotctl[d]`.