Hamlib/Hamlib #785: What is the policy on checking for rig for the rig back end

wutje, 2021-09-03:
I was wondering about the assumption a rig back end can make on the rig pointer being `!null`.
I see some code that assume `rig != NULL`, while other code adds actual checks. Either not checking is a bug or checking is redundant.

Running a simple grep on check for rig pointer not being NULL:
`grep -R '!rig[^_-]' --include=*.[ch] rigs`
shows most older Yaesu rigs have checks for every possible function, most other rigs have no check or only one. Most check only in the close function.

I think all these check are (no longer?) required. I expect these should only be accessed via the API which could check this.

I am willing to code this cleanup, but I first wanted to know what the policy is.

rigs/kit/funcube.c:    if (!rig)
rigs/kit/elektor304.c:    if (!rig)
rigs/kit/fifisdr.c:    if (!rig)
rigs/kit/elektor507.c:    if (!rig)
rigs/kit/si570avrusb.c:    if (!rig)
rigs/kit/dds60.c:    if (!rig)
rigs/kit/drt1.c:    if (!rig)
rigs/kit/usrp_impl.cc:  if (!rig)
rigs/pcr/pcr.c:    if (!rig)
rigs/pcr/pcr.c:    if (!rig)
rigs/icom/icom.c:    if (!rig)
rigs/kenwood/tmd710.c:    if (!rig || !code)
rigs/kenwood/tmd710.c:    if (!rig || !ch)
rigs/kenwood/tmd710.c:    if (!rig)
rigs/kenwood/tmd710.c:    if (!rig || !chan)
rigs/kenwood/tmd710.c:    if (!rig || !chan)
rigs/kenwood/thf7.c:    if (!rig)
rigs/kenwood/kenwood.c:    if (!rig)
rigs/kenwood/thf6a.c:    if (!rig)
rigs/yaesu/ft600.c:    if (!rig)
rigs/yaesu/ft600.c:    if (!rig) { return -RIG_EINVAL; }
rigs/yaesu/ft600.c:    if (!rig) { return -RIG_EINVAL; }
rigs/yaesu/ft991.c:    if (!rig || !in_split)
rigs/yaesu/ft991.c:    if (!rig || !tx_mode || !tx_width)
rigs/yaesu/ft991.c:    if (!rig)
rigs/yaesu/ft847.c:    if (!rig)
rigs/yaesu/ft847.c:    if (!rig)
rigs/yaesu/ft847.c:    if (!rig)
rigs/yaesu/ft847.c:    if (!rig)
rigs/yaesu/ft1000d.c:    if (!rig)
rigs/yaesu/ft1000d.c:    if (!rig)
rigs/yaesu/ft1000d.c:    if (!rig)
rigs/yaesu/ft1000d.c:    if (!rig)
rigs/yaesu/ft1000d.c:    if (!rig)
rigs/yaesu/ft1000d.c:    if (!rig)
rigs/yaesu/ft1000d.c:    if (!rig)
rigs/yaesu/ft1000d.c:    if (!rig)
rigs/yaesu/ft1000d.c:    if (!rig)
rigs/yaesu/ft1000d.c:    if (!rig)
rigs/yaesu/ft1000d.c:    if (!rig)
rigs/yaesu/ft1000d.c:    if (!rig)
rigs/yaesu/ft1000d.c:    if (!rig)
rigs/yaesu/ft1000d.c:    if (!rig)
rigs/yaesu/ft1000d.c:    if (!rig)
rigs/yaesu/ft1000d.c:    if (!rig)
rigs/yaesu/ft1000d.c:    if (!rig)
rigs/yaesu/ft1000d.c:    if (!rig)
rigs/yaesu/ft1000d.c:    if (!rig)
rigs/yaesu/ft1000d.c:    if (!rig)
rigs/yaesu/ft1000d.c:    if (!rig)
rigs/yaesu/ft1000d.c:    if (!rig)
rigs/yaesu/ft1000d.c:    if (!rig)
rigs/yaesu/ft1000d.c:    if (!rig)
rigs/yaesu/ft1000d.c:    if (!rig)
rigs/yaesu/ft1000d.c:    if (!rig)
rigs/yaesu/ft1000d.c:    if (!rig)
rigs/yaesu/ft1000d.c:    if (!rig)
rigs/yaesu/ft1000d.c:    if (!rig)
rigs/yaesu/ft1000d.c:    if (!rig)
[... 160 lines cut ...]
rigs/yaesu/ft891.c:    if (!rig)
rigs/yaesu/ft891.c:    if (!rig)
rigs/yaesu/ft891.c:    if (!rig)
rigs/yaesu/ft767gx.c:    if (!rig)
rigs/yaesu/ft767gx.c:    if (!rig)
rigs/icmarine/icmarine.c:    if (!rig || !rig->caps)
rigs/icmarine/icmarine.c:    if (!rig)
rigs/icmarine/icm710.c:    if (!rig || !rig->caps)
rigs/icmarine/icm710.c:    if (!rig)
rigs/racal/ra37xx.c:    if (!rig || !rig->caps)
rigs/racal/ra37xx.c:    if (!rig)
rigs/racal/racal.c:    if (!rig || !rig->caps)
rigs/racal/racal.c:    if (!rig)
rigs/barrett/barrett.c:    if (!rig)
rigs/dummy/flrig.c:    if (!rig)
rigs/dummy/netrigctl.c:    if (!rig || !rig->caps)
rigs/dummy/trxmanager.c:    if (!rig)
rigs/winradio/g313-win.c:    if (!rig)
rigs/winradio/g313-posix.c:    if (!rig)
rigs/tentec/rx331.c:    if (!rig)
rigs/tentec/rx331.c:    if (!rig)
rigs/prm80/prm80.c:    if (!rig)
rigs/wj/wj.c:    if (!rig || !rig->caps)
rigs/wj/wj.c:    if (!rig)
`

mdblack98, 2021-09-03:
It's optional and would prevent a segfault if rig is null.  So it's just a sanity check.
What would be better is perhaps a macro and a function and expand the usage.The function could then be expanded then to use a list of rig ptrs  (obtained during rig_open and cleared during rig_close) to ensure they are really valid instead of just not null.  Would be nice to have a linked list of rig ptrs for this purpose so adding some linked list code to the library would be nice too.
#define RIGPTROK if (!rigPtrOk(rig)) {  return -RIG_EINTER; }
int rigPtrOk(RIG *rig){    if (rig) return 1;
    rig_debug(RIG_DEBUG_ERR, "%s: rig ptr %p is invalid\n", __func__, rig);    return 1;}

wutje, 2021-09-03:
> It's optional and would prevent a segfault if rig is null.  So it's just a sanity check. What would be better is perhaps a macro and a function and expand the usage.The function could then be [...]

But that should be done in the frontend, not the rig backend, right? 

Is it even possible to call a rig backend function with `rig==NULL` ? 
I took a quick look and all API function first check for rig.

mdblack98, 2021-09-03:
You never know when a pointer gets nulled or corrupted so good to check everywhere you can.  It's a bit pedantic but knowing you have a valid pointer instead of some buffer overwrite is a really good thing.
But yes...theoretically it SHOULD not be possible in the back ends....but it is possible...just highly unlikely.
That's why I like the idea of the macro/function combination so we can check for valid pointers.
Starting point would be in all functions in rig.c -- and could be added anywhere rig is referenced other than rig.c too....that pointer is critical for correct operations.
One thing we've never worried about is bit flips that can be caused by cosmic particles.  Checking the ptr is valid would detect that and doing a CRC on some of the caps data would really expand it.
Bit flips are responsible for software errors and computer crashes.https://science.slashdot.org/story/17/02/19/2330251/serious-computer-glitches-can-be-caused-by-cosmic-rays

Mike W9MDB

wutje, 2021-09-03:
> You never know when a pointer gets nulled or corrupted so good to check everywhere you can.  It's a bit pedantic but knowing you have a valid pointer instead of some buffer overwrite is a really [...]

Bit flips are real, but checking some pointer is not going to solve this. A bit on the return stack could be bit flipped and then the program is doomed as well. ECC hardware is the only proper solution against bit flips! Still hoping it gets cheaper. Besides this hobby software, not space grade stuff 👾 

Dropping the checks from the back end means it will expose front end bugs sooner instead of hiding behind 'random' checks. Since most backend rig code today has no checks I suggest to simply drop all checks. 
Simple code is better then 'half' code IMHO.

mdblack98, 2021-09-03:
Dropping back end checks will not expose sooner....it will expose later in a totally unrelated function.
e.g.
icom_get_freq(rig...) and inside something screws the pointer (buffer overflow, bad code)
rig.c will continue to next function with an invalid pointer where it will die...likely a segfault.
The more sanity checks the better.  So let's not remove any of these checks and instead make them more robust.
Mike W9MDB

wutje, 2021-09-04:
Dropping backend checks will expose missing front end checks sooner; a crash is easier spotted then an error return.
I think your example is flawed:
```
rig.c: 
 
rig_set_vfo(rig) //rig is local memory
 icom_set_vfo(rig)
     --do some funky code that corrupts the **memory pointed to** by rig. It cannot destroy rig 
 icom_set(rig) //rig is local still local
    --rig is still valid (not NULL). The **memory pointed to** is corrupted. So the null check is not helping.
```
Once again, I do not see how adding NULL checks on every call is even remotely helping. 
Let alone it justifies the effort to implement, maintain and test the huge amount of extra code.

There are very little checks currently so I think the current API is doing proper checks already.
Using tools like valgrind or AddressSanitizer are a way more efficient approach to find memory corruption.

mdblack98, 2021-09-04:
That was my point about improving the check to look for a valid pointer instead of just NULL.That would be useful.
So if you want to change anything follow my recommendation to make it a macro/function combo and we'll improve the function then which will apply it everywhere.

wutje, 2021-09-04:
Yes adding checks for the FRONT END is useful, not for the back end. I think that is a task for the front end and the back end should not bother with it.

mdblack98, 2021-09-05:
The problem is you never know when corruption occurs -- could be at any level.Plus...the backends call themselves so can also corrupt things.
At worst the checks are harmless.  At best they can catch a problem when they are improved to match real data.
The general rule is that there is no such thing as too much error checking (unless it causes performance issues).
Mike

closed 2021-09-22 by mdblack98