Hamlib/Hamlib #786: python3 bindings broken

mdblack98, 2021-09-03:
Broken in 4.3 too.

/py3test.py: Python 3.9.5; Hamlib 4.4~git

status(str):		Command completed successfully
rig_token_lookup called
rig_confparam_lookup called
rig_get_conf called
get_conf:		path = , retry = 5
Traceback (most recent call last):
  File "/home/mdblack98/src/Hamlib/bindings/./py3test.py", line 126, in <module>
    StartUp()
  File "/home/mdblack98/src/Hamlib/bindings/./py3test.py", line 36, in StartUp
    my_rig.set_freq(Hamlib.RIG_VFO_A, 5700000000)
AttributeError: module 'Hamlib' has no attribute 'RIG_VFO_A'

N0NB, 2021-09-03:
Mike,

If you want to have a bugfix 4.3.1 release, please cherry-pick the bug fixes into the Hamlib-4.3 and when ready I can do a 4.3.1 release from that branch.

mdblack98, 2021-09-03:
Probably a good idea....the python thing will be a needed fix.
I'll let you know....probably should let this one bake into the next WSJTX release candidate. 
Mike

az-av8r, 2021-09-10:
Hi Gents,

I'm seeing the same failure when running 4.2 using the dummy rig and my ic-7600 using the python bindings.  However, rigctl, rigctld seems to be working fine in 4.2 for both the dummy and my ic-7600.

if I comment out the failing calls (interaction with the VFO) py3test.py will pass some of the code like the Lat Lon static functions.

Not a professional coder and rather new to Hamlib, but let me know If I can help.

73,

-Andy

KB7ZUT

closed 2021-09-11 by mdblack98

mdblack98, 2021-09-12:
OK Nate...let's do 4.3.1.  No new bug reports for several days now and all the critical ones taken care of.
These are the commits needed
https://github.com/Hamlib/Hamlib/issues?q=milestone%3A4.3.1+

Mike W9MDB

N0NB, 2021-09-12:
* On 2021 12 Sep 17:09 -0500, Michael Black wrote:
> OK Nate...let's do 4.3.1.  No new bug reports for several days now and
> all the critical ones taken care of.  These are the commits needed
> https://github.com/Hamlib/Hamlib/issues?q=milestone%3A4.3.1+

Hmmm.

I thought maybe the Web interface would have a way to pull those into a
branch.  I'll have to check the site docs to see if there is, otherwise
some manual cherrypicking will be in order.

73, Nate

-- 
"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."
Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

mdblack98, 2021-09-13:
Would seem nice to be able to take a group of issues from milestone and merge them.  Easy enough to mark an issue with a milestone.
Let me create the 4.3.1 branch so you don't have to.
Mike

 
 

Hmmm.

I thought maybe the Web interface would have a way to pull those into a
branch. I'll have to check the site docs to see if there is, otherwise
some manual cherrypicking will be in order.

73, Nate

N0NB, 2021-09-13:
* On 2021 12 Sep 23:17 -0500, Michael Black wrote:
> Would seem nice to be able to take a group of issues from milestone and merge them.  Easy enough to mark an issue with a milestone.
> Let me create the 4.3.1 branch so you don't have to.

No, no new branches!  4.3.1 will be done in the Hamlib-4.3 branch.  Once
the commits are in place I'll tag it, etc.

I'll get it taken care of.

73, Nate

-- 
"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."
Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

N0NB, 2021-09-13:
Applied 004ddaf to the Hamlib-4.3 branch.  eb1a930 is stated as not belonging to a branch and 50ccb1f  is a "bad revision" but these two appear to be the exact same code as the first so safe to ignore.

mdblack98, 2021-09-13:
For some reason wutje's branch commits or merges are showing up.

N0NB, 2021-09-13:
* On 2021 13 Sep 07:45 -0500, Michael Black wrote:
> For some reason wutje's branch commits or merges are showing up.

So I noticed.  Oh well, I am checking things as I go and so far things
should be okay.

I'll push the commits into Hamlib-4.3 when I am done and will let you
look at them before I tag the branch.

73, Nate

-- 
"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."
Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819