Hamlib/Hamlib #789: ICOM 705 issues

km4ack, 2021-09-05:
Running on a Pi 4 with hamlib 4.3 installed. Everything worked ok on Icom 1.24 firmware. After upgrading the radio firmware to 1.26, I am having issues. Here is one example where I am simply trying to get the frequency from the radio:

rigctl -vvv -m 3085 -r /dev/serial/by-id/usb-Icom_Inc._IC-705_IC-705_12007996-if00 f
Opened rig model 3085, 'IC-705'
rigctl_parse: rig not open...trying to reopen
rigctl_parse: RIG_EIO?
main: i/o error
main: rig_close retcode=-1
main: rig_open retcode=0

If you need more info, just let me know. 

73, de KM4ACK

mdblack98, 2021-09-05:
Please run that again with -vvvvv
Mike W9MDB

km4ack, 2021-09-05:
rigctl -vvvvv -m 3085 -r /dev/serial/by-id/usb-Icom_Inc._IC-705_IC-705_12007996-if00 f
rigctl Hamlib 4.3 Wed Sep 01 02:14:39 2021 +0000 SHA=5a6134
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_check_rig_caps: p1=0xb6f99268, p2=0xb6f9e9ac, rig_model=0xb6f99268, macro_name=0xb6f9e9ac
initrigs4_icom: _init called
rig_register called
rig_register: rig_register (3055)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (3085)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (3009)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (3010)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (3011)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (3013)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (3014)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (3015)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (3019)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (3020)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (3021)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (3022)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (3067)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (3023)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (3046)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (3024)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (3028)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (3030)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (3026)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (3027)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (3047)
register.c(225):rig_register return(0)
rig_register called
[... 529 lines cut ...]
rig.c(2728):rig_set_vfo return(0)
icom_set_default_vfo: curr_vfo now VFOA
icom.c(970):icom_set_default_vfo return(0)
icom.c(854):icom_rig_open return(0)
rig.c(1033) trace
rig_open: default vfo = currVFO
rig.c(1102):rig_open return(0)
main: rig_open retcode=0
rigctl_parse.c(648):rigctl_parse entered
rigctl_parse: called, interactive=0
rigctl_parse.c(816):rigctl_parse return(1)
rig.c(1124):rig_close entered
icom_rig_close: called
icom.c(890):icom_rig_close return(0)
iofunc.c(200):port_close entered
ser_close called
ser_close: restoring options
serial.c(844):ser_close return(0)
iofunc.c(234):port_close return(0)
rig.c(243):remove_opened_rig return(0)
rig.c(1280):rig_close return(0)
rig.c(1299):rig_cleanup entered
icom_cleanup called
icom.c(717):icom_cleanup return(0)
rig.c(1324):rig_cleanup return(0)

mdblack98, 2021-09-05:
I assume you have a /dev/ttyUSB device for that.
Can you try that device instead?
Mike W9MDB

km4ack, 2021-09-05:
No ttyUSB. I have ttyACM0 and ttyACM1

pi@IC705:/dev/serial/by-id $ ls -l
total 0
lrwxrwxrwx 1 root root 13 Sep  5 21:08 usb-Icom_Inc._IC-705_IC-705_12007996-if00 -> ../../ttyACM0
lrwxrwxrwx 1 root root 13 Sep  5 21:08 usb-Icom_Inc._IC-705_IC-705_12007996-if02 -> ../../ttyACM1

km4ack, 2021-09-05:
same result using ttyACM0

mdblack98, 2021-09-06:
Can you try a firmware downgrade?  Since you said it was ok with a prior version...
If the prior version works I think you should raise a bug report with Icom.
According to the Icom site some serial #'s might have a problem with downgrading though.https://www.icomjapan.com/support/firmware_driver/3343/

Mike W9MDB

mdblack98, 2021-09-06:
And did you compile hamlib yourself?
Do you have the original hamlib still installed?
ls -l /usr/lib/libhamlib*ls -l /usr/local/lib/libhamlib*
Mike W9MDB

km4ack, 2021-09-06:
Both versions were compiled on my machine. The first (4.2) was compiled by [Build a Pi](https://github.com/km4ack/pi-build) which I am the developer. The specific code which compiled it can be seen [here](https://github.com/km4ack/pi-build/blob/master/functions/base.function#L111). 4.3 was compiled manually. Speaking of 4.3, it requires libusb1-0 (1.0.23) or later.  libusb1-0 (1.0.22) is the latest version available in the Pi repository. I had to compile libusb1.0 (1.0.24) before compiling hamlib 4.3. I wouldn't think this would make a difference but wanted to mention it anyway.

Using the command you supplied:

pi@IC705:/dev/serial/by-id $ ls -l /usr/lib/libhamlib*ls -l /usr/local/lib/libhamlib*
ls: cannot access '/usr/lib/libhamlib*ls': No such file or directory
-rw-r--r-- 1 root root   343800 Sep  5 20:43  /usr/local/lib/libhamlib++.a
-rw-r--r-- 1 root root 23987522 Sep  5 20:43  /usr/local/lib/libhamlib.a
-rwxr-xr-x 1 root root     1011 Sep  5 20:43  /usr/local/lib/libhamlib++.la
-rwxr-xr-x 1 root root      973 Sep  5 20:43  /usr/local/lib/libhamlib.la
lrwxrwxrwx 1 root root       20 Sep  5 20:43  /usr/local/lib/libhamlib++.so -> libhamlib++.so.4.0.3
lrwxrwxrwx 1 root root       18 Sep  5 20:43  /usr/local/lib/libhamlib.so -> libhamlib.so.4.0.3
lrwxrwxrwx 1 root root       20 Sep  5 20:43  /usr/local/lib/libhamlib++.so.4 -> libhamlib++.so.4.0.3
lrwxrwxrwx 1 root root       18 Sep  5 20:43  /usr/local/lib/libhamlib.so.4 -> libhamlib.so.4.0.3
-rwxr-xr-x 1 root root   231800 Sep  2 13:29  /usr/local/lib/libhamlib++.so.4.0.2
-rwxr-xr-x 1 root root 16652388 Sep  2 13:29  /usr/local/lib/libhamlib.so.4.0.2
-rwxr-xr-x 1 root root   232120 Sep  5 20:43  /usr/local/lib/libhamlib++.so.4.0.3
-rwxr-xr-x 1 root root 17140284 Sep  5 20:43  /usr/local/lib/libhamlib.so.4.0.3

I'll build a new Pi today and only install hamlib 4.3 to see the results. If that doesn't work, I'll see about downgrading the firmware for the 705. 

73, de KM4ACK

mdblack98, 2021-09-06:
ls /usr/lib/libhamlib*

km4ack, 2021-09-06:
Nothing listed for `ls /usr/lib/libhamlib*`

km4ack, 2021-09-06:
After loading 4.3 onto a clean Pi, everything works as expected. Radio is still on firmware 1.26. Didn't have to downgrade firmware. Not sure why this happened. Never had an issue upgrading hamlib before. Sorry for wasting your time. Thank you for the help!

73, de KM4ACK

closed 2021-09-06 by km4ack