Hamlib/Hamlib #802: What is the intended use case of rig_cookie?

wutje, 2021-09-19:
I think I understand what rig_cookie is trying to achieve: implement some sort of locking to allow multi-threaded applications.
But I think there are some serious issues;

1. Locking is only done when having pthread, so if that is not enabled there is no multi thread guarantee.
2. Even if pthread is available the lock should also be used when reading cookie to protect against writes, not just when setting.
3. The lock should also protect timestamp.
4. Fetching timestamp should be done AFTER getting the lock. otherwise significant time could have past waiting for the lock.

To me it really seems like a problem that should be solved at the application layer. Why is hamlib trying to provide this? Are there any known users?

mdblack98, 2021-09-19:
It's put in for future rigctld use.The problem is particularly bad for non-vfo mode where setting freq can be like this.
V VFOBF 14074000V VFOA
If another rigctld client wants VFOA freq it will do this which could be done in the middle of the 1st sequence without any control.  V VFOAf

This is really a mutex but I didn't want to call it that.  It's only meant for a short time -- 1 second right now which should cover most situations so doesn't quite qualify as a mutex.  A cookie implies expiring.

rigctld has no way of knowing that multiple commands are occurring or that another client is running so this is a solution to this potential problem.  In hamlib 5.0 this may be forced....haven't decided yet.  Nobody has reported a problem with this but the new mulitcast ability is going to aggravate this condition.  Hamlib will be polling the rig asynchronously soon which it's never done before. 
Mike W9MDB

wutje, 2021-09-19:
Yes the timeout thing make it somewhat different from a mutex. I already figured the timeout was the key thing.

Why can rigctld not simply have a lock it self? Why does it need be in rig.c? Surely it will be well behaved client :P

Do you agree the current implementation is flawed?

mdblack98, 2021-09-19:
rigctld does have a lock but it's only for one command.
The client may be sending multiple commands that are all needed to keep in sequence.
So far nobody has complained about it but it's a potential.
Mike

wutje, 2021-09-20:
Okay I implemented the fix; just lock over everything.

PR #804 (2021-09-21): Fix rig_cookie: use lock for reads and write.
Ensure we never print more then HAMLIB_COOKIE_SIZE otherwise we read out-of-bounds.
Drop stray printf.
Add tests for invalid input and overly large input.
Fix test2 to release the cookie.

This fixes #802
closed 2021-09-21 by mdblack98