Hamlib/Hamlib #803: error on make

Ian-G3VPX, 2021-09-20:
I downloaded the latest Hamlib-master.zip yesterday   - failed on make.
Rebuilt from 13th Aug version - ok

I downloaded the latest Hamlib-master.zip today   - failed on make.
Rebuilt from 13th Aug version - ok
see attached screen dump

Ian G3VPX

[image]

mdblack98, 2021-09-20:
Could you replace the top include section in rigtestlibusb.c with this and tell me if it works?
#include <stdio.h>#include <string.h>
#if defined(HAVE_LIBUSB_H)
#include "libusb.h"
#else
#include <libusb-1.0/libusb.h>
#endif

Mike W9MDB

Ian-G3VPX, 2021-09-20:
Hi Mike
Where is rigtestlibusb.c ???
I've searched all the folders in Hamlib-master (apart from the 
individual rotator fodlers)
and I can't find it
Ian
G3VPX

wutje, 2021-09-20:
In the tests folder?

Ian-G3VPX, 2021-09-20:
..
#include <stdio.h>#include <string.h>  threw an error

So   Using:
#include <stdio.h>
#include <string.h>
#if defined(HAVE_LIBUSB_H)
#include "libusb.h"
#else
#include <libusb-1.0/libusb.h>
#endif

I get the attached error message:
[image]

Ian
G3VPX

mdblack98, 2021-09-20:
I need to see your config.log and include/config.hPlease attach.

 
 

..

Ian-G3VPX, 2021-09-20:
config.h and config.log attached

[config.log](https://github.com/Hamlib/Hamlib/files/7198980/config.log)

[config.h.txt](https://github.com/Hamlib/Hamlib/files/7198983/config.h.txt)

Ian 
G3VPX

closed 2021-09-22 by mdblack98