Hamlib/Hamlib #811: IC-7300 bandwidth broken

mdblack98, 2021-09-27:
Howard Nurse <hlnurse@cmmsft.com>

Setting bandpass widths appears to be broken for the IC 7300.
 
--Howard
 
pi@rigpi3:~ $ rigctl -m 3073 -r /dev/ttyUSB0 -vvvvv
rigctl.c(444) Startup: rigctl -m 3073 -r /dev/ttyUSB0 -vvvvv
rigctl Hamlib 4.4~git from indeterminate source revision.
 
Rig command: m
rigctl_parse: input_line: m
rigctl_parse: vfo_opt=0
rigctl_parse.c(2144):rigctl_get_mode entered
rig.c(2353):rig_get_mode entered
rig_get_mode: currVFO cache check age=1967ms
rig_get_mode: cache miss age mode=1000000ms, width=1000000ms
rig.c(2400) trace
icom_get_mode_with_data called
icom_get_mode called vfo=currVFO
vfo_fixup: vfo=currVFO, vfo_curr=currVFO
vfo_fixup: Leaving currVFO alone
vfo_fixup: vfo=currVFO, vfo_curr=currVFO
vfo_fixup: Leaving currVFO alone
frame.c(408):icom_transaction entered
icom_transaction: cmd=0x26, subcmd=0x00, payload_len=0
frame.c(139):icom_one_transaction entered
frame.c(54):make_cmd_frame entered
frame.c(88):make_cmd_frame return(7)
rig_flush: called for serial device
serial.c(642):serial_flush entered
tcflush
serial.c(674):serial_flush return(0)
write_block(): TX 7 bytes
0000    fe fe 94 e0 26 00 fd                                ....&..        
frame.c(458):read_icom_frame entered
read_string called, rxmax=80
read_string(): RX 10 characters
0000    fe fe e0 94 26 00 01 00 01 fd                       ....&.....     
frame.c(492):read_icom_frame return(10)
icom_one_transaction: frm_len=10, frm_len-1=fd, frm_len-2=01
frame.c(386):icom_one_transaction return(0)
frame.c(438):icom_transaction return(0)
icom_get_mode: mode_len=5, modebuf=26 00 01 00 01
icom_get_mode: modebuf[0]=0x26, modebuf[1]=0x01, mode_len=1
frame.c(648):icom2rig_mode entered
icom2rig_mode: mode=0x01, pd=-1
icom_get_dsp_flt called, mode=USB
frame.c(408):icom_transaction entered
icom_transaction: cmd=0x1a, subcmd=0x03, payload_len=0
frame.c(139):icom_one_transaction entered
frame.c(54):make_cmd_frame entered
frame.c(88):make_cmd_frame return(7)
rig_flush: called for serial device
serial.c(642):serial_flush entered
tcflush
serial.c(674):serial_flush return(0)
write_block(): TX 7 bytes
0000    fe fe 94 e0 1a 03 fd                                .......        
frame.c(458):read_icom_frame entered
read_string called, rxmax=80
read_string(): RX 8 characters
0000    fe fe e0 94 1a 03 34 fd                             ......4.       
frame.c(492):read_icom_frame return(8)
icom_one_transaction: frm_len=8, frm_len-1=fd, frm_len-2=34
frame.c(386):icom_one_transaction return(0)
frame.c(438):icom_transaction return(0)
from_bcd called
icom_get_dsp_flt: i=34, [0]=1a, [1]=03, [2]=34, [3]=b6
icom_get_dsp_flt: using width=34
[... 179 lines cut ...]
read_string(): RX 8 characters
0000    fe fe e0 94 1a 03 34 fd                             ......4.       
frame.c(492):read_icom_frame return(8)
icom_one_transaction: frm_len=8, frm_len-1=fd, frm_len-2=34
frame.c(386):icom_one_transaction return(0)
frame.c(438):icom_transaction return(0)
from_bcd called
icom_get_dsp_flt: i=34, [0]=1a, [1]=03, [2]=34, [3]=b6
icom_get_dsp_flt: using width=34
icom.c(1643):icom_get_dsp_flt return(3000)
icom.c(2350):icom_get_mode return(0)
icom_get_mode_with_data mode=4
icom_get_mode_with_data databuf[2]=0, mode=4
icom.c(2167):icom_get_mode_with_data return(0)
rig_get_mode: retcode after get_mode=0
rig_get_mode(2444): debug
rig.c(1407):set_cache_mode entered
rig.c(1471):set_cache_mode return(0)
rig.c(2459):rig_get_mode return(0)
Mode: USB
Passband: 3000
rigctl_parse.c(2167):rigctl_get_mode return(0)
rigctl_parse.c(1758):rigctl_parse return(0)
rigctl_parse.c(651):rigctl_parse entered
rigctl_parse: called, interactive=1

hlnurse, 2021-10-02:
Mik,e, note this was working on Hamlib 4.2.

--Howard

mdblack98, 2021-10-02:
I just pushed a patch that should fix this.
I have an IC7300 user I will test with tomorrow too.
Mike W9MDB

hlnurse, 2021-10-02:
Hi Mike,

 

The power on function is working.

The f function is working.

The v function is still misbehaving, but I think that is the Icom limitation.

 

The filter set function worked at first, but has now stopped.  Go figure!

 

rig_register: rig_register (3054)

register.c(225):rig_register return(0)

rig_register called

rig_register: rig_register (3083)

register.c(225):rig_register return(0)

rig_register called

rig_register: rig_register (3084)

register.c(225):rig_register return(0)

rig_register called

rig_register: rig_register (3082)

register.c(225):rig_register return(0)

rig_register called

rig_register: rig_register (3071)

register.c(225):rig_register return(0)

rig_register called

rig_register: rig_register (3072)

register.c(225):rig_register return(0)

rig_register called

rig_register: rig_register (3074)

register.c(225):rig_register return(0)

rig_register called

rig_register: rig_register (3076)

register.c(225):rig_register return(0)

icom.c(9406):initrigs4_icom return(0)

rig_init: rig has VFO_A

rig_init: rig has VFO_B

rig_init: rig has VFO_MEM

[... 996 lines cut ...]
rig.c(1475):set_cache_mode return(0)

rig_get_mode: elapsed=6ms

rig.c(2486):rig_get_mode return(0)

Mode: CW

Passband: 1200

rigctl_parse.c(2167):rigctl_get_mode return(0)

rigctl_parse.c(1758):rigctl_parse return(0)

rigctl_parse.c(651):rigctl_parse entered

rigctl_parse: called, interactive=1

 

Rig command:

 

--Howard

mdblack98, 2021-10-03:
You could do me a favor in future and send the entire log as an attachment and always use the -Z argument for timing.  I'd like to see the whole power on sequence with timing as there was a complaint before on how lot it took to timeout when the rig was not on.   Please test that too when the rig is not on.  I'd like to see both timings.
Your emails are double-spaced and I also like to see the startup info as it can be relevant.
What do you mean "v is...misbehaving" -- there is no "v" command for Icoms.  I added a method for it recently but it interferes with VFO twiddling so disabled it.  You should be seeing "no get_vfo". 
I can see there's a problem with the width still....working on it.

closed 2021-10-14 by mdblack98